MichaIng/DietPi #4371: dkms: WARNING: Linux headers are missing

vinhtq115, 2021-05-13:
### Creating a bug report/issue

#### Required Information
- DietPi version | `7.1.2`
- Distro version | `buster`
- Kernel version | `Linux DietPi 5.10.17-v7l+ #1403 SMP Mon Feb 22 11:33:35 GMT 2021 armv7l GNU/Linux`
- SBC model | `RPi 4 Model B (armv7l)`
- Power supply used | 5V 3A Canakit
- SDcard used | Don't remember

#### Steps to reproduce
1. Run `sudo apt upgrade`. Apt says I need to update some packages which I don't remember their name (eeprom or something).

#### Actual behaviour
```
/etc/kernel/postinst.d/dkms:
dkms: WARNING: Linux headers are missing, which may explain the above failures.
      please install the linux-headers-5.10.17-v8+ package to fix this.
Setting up raspberrypi-kernel-headers (1:1.20210430-2) ...
run-parts: executing /etc/kernel/header_postinst.d/dkms 5.10.17+
run-parts: executing /etc/kernel/header_postinst.d/dkms 5.10.17-v7+
run-parts: executing /etc/kernel/header_postinst.d/dkms 5.10.17-v7l+
run-parts: executing /etc/kernel/header_postinst.d/dkms 5.10.17-v8+
dkms: WARNING: Linux headers are missing, which may explain the above failures.
      please install the linux-headers-5.10.17-v8+ package to fix this.
Setting up raspberrypi-bootloader (1:1.20210430-2) ...
Removing 'diversion of /boot/start.elf to /usr/share/rpikernelhack/start.elf by rpikernelhack'
Removing 'diversion of /boot/start_cd.elf to /usr/share/rpikernelhack/start_cd.elf by rpikernelhack'
Removing 'diversion of /boot/start_db.elf to /usr/share/rpikernelhack/start_db.elf by rpikernelhack'
Removing 'diversion of /boot/start_x.elf to /usr/share/rpikernelhack/start_x.elf by rpikernelhack'
Removing 'diversion of /boot/fixup.dat to /usr/share/rpikernelhack/fixup.dat by rpikernelhack'
Removing 'diversion of /boot/fixup_cd.dat to /usr/share/rpikernelhack/fixup_cd.dat by rpikernelhack'
Removing 'diversion of /boot/fixup_db.dat to /usr/share/rpikernelhack/fixup_db.dat by rpikernelhack'
Removing 'diversion of /boot/fixup_x.dat to /usr/share/rpikernelhack/fixup_x.dat by rpikernelhack'
Removing 'diversion of /boot/bootcode.bin to /usr/share/rpikernelhack/bootcode.bin by rpikernelhack'
Removing 'diversion of /boot/start4.elf to /usr/share/rpikernelhack/start4.elf by rpikernelhack'
Removing 'diversion of /boot/start4cd.elf to /usr/share/rpikernelhack/start4cd.elf by rpikernelhack'
Removing 'diversion of /boot/start4db.elf to /usr/share/rpikernelhack/start4db.elf by rpikernelhack'
Removing 'diversion of /boot/start4x.elf to /usr/share/rpikernelhack/start4x.elf by rpikernelhack'
Removing 'diversion of /boot/fixup4.dat to /usr/share/rpikernelhack/fixup4.dat by rpikernelhack'
Removing 'diversion of /boot/fixup4cd.dat to /usr/share/rpikernelhack/fixup4cd.dat by rpikernelhack'
Removing 'diversion of /boot/fixup4db.dat to /usr/share/rpikernelhack/fixup4db.dat by rpikernelhack'
Removing 'diversion of /boot/fixup4x.dat to /usr/share/rpikernelhack/fixup4x.dat by rpikernelhack'
Removing 'diversion of /boot/LICENCE.broadcom to /usr/share/rpikernelhack/LICENCE.broadcom by rpikernelhack'
Setting up libraspberrypi0 (1:1.20210430-2) ...
Setting up libraspberrypi-bin (1:1.20210430-2) ...
dietpi@DietPi:~$
```

MichaIng, 2021-05-13:
Many thanks for your report.

Do you have WireGuard installed, or another application which builds a kernel module via DKMS?
The RPi headers package can be installed via:
```
apt install raspberrypi-kernel-headers
```

Joulinar, 2021-05-13:
The warning is as expected because it's a 32bit system. Therefore the 64bit module will not be touched by `dkms`

MichaIng, 2021-05-13:
It could make sense to remove the 64-bit kernel from the 32-bit (armhf) kernel package. Running the OS in 64-bit mode while the installed userland packages are 32-bit has only downsides and confuses/breaks all installers which use `uname` to derive which binary format to download, which will fail when shared libraries are all 32-bit.

But to keep that ability, to install 32-bit and 64-bit kernels concurrently, the package content would need to be changed since some files are shipped by both 🤔.

MichaIng, 2021-05-13:
Issue open for discussion: https://github.com/RPi-Distro/repo/issues/246
Let's see if there are actual reasons for keeping both architectures mixed or if this is more for historical reasons.

Joulinar, 2021-05-13:
I guess historical reason. If I'm not mistaken this already changed on 64bit.

MichaIng, 2021-05-13:
Yes it did.

But this did not imply any regression as a 32-bit kernel cannot run 64-bit userland binaries, so switching to 32-bit kernel on a Raspberry Pi OS 64-bit image cannot work at all.

But the other way round it works and removing the 64-bit kernel breaks this ability. But not sure whether there is an actual case to run a 32-bit image in 64-bit mode.

vinhtq115, 2021-05-13:
> 
> 
> Many thanks for your report.
> 
> Do you have WireGuard installed, or another application which builds a kernel module via DKMS?
> The RPi headers package can be installed via:
> 
> ```
> apt install raspberrypi-kernel-headers
> ```
> 
> ...
> ah the reason is likely that the armhf kernel headers package (which you likely have installed) ships no 64-bit headers, which are instead only shipped with the arm64 kernel headers package.
> 
> The armhf kernel package itself however ships the 64-bit kernel was well. So basically kernel and kernel headers packages do not match, causing the warning.
> 
> However, you can ignore it since of course you run the 32-bit kernel when having the 32-bit package installed. I'll open an issue on the RPi repo, as it probably does not make sense to ship the 64-bit kernel with the 32-bit package, IMO.

I ran that command and it showed that:
```
Reading package lists... Done
Building dependency tree
Reading state information... Done
raspberrypi-kernel-headers is already the newest version (1:1.20210430-2).
0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.
```

WireGuard was not installed but prior to that, I had installed PiVPN few months ago and removed it few days ago.

Joulinar, 2021-05-13:
@vinhtq115 
Nothing to do from your side. All working as expected. You can safely ignore the warning 😃

vinhtq115, 2021-05-13:
> 
> 
> @vinhtq115
> Nothing to do from your side. All working as expected. You can safely ignore the warning 😃

Okay. Also, if I install the 64-bit version, will this warning go away? I saw that DietPi seemed to have official 64-bit version for Raspberry Pi now.

Joulinar, 2021-05-13:
Well there is no reason to run the 64bit version. As said, just ignore the message.

Yes DietPi support 64bit on RPi as well. But keep in mind DietPi is using Raspberry OS as base image and this is still not officially release by Raspberry Foundation.

MichaIng, 2021-05-13:
But to answer the message: The 64-bit image would indeed not show this warning as both, the kernel and the headers package contain only files for the 64-bit kernel.

closed 2021-05-14 by vinhtq115