MichaIng/DietPi #4372: Cant install HomeAssistant with DietPi-Software on RPi 2 Model B (armv7l)

QuaiGoner, 2021-05-13:
---------------------------------------------------------------------
- DietPi has encountered an error                                   -
- Please create a ticket: https://github.com/MichaIng/DietPi/issues -
- Copy and paste only the BLUE lines below into the ticket          -
---------------------------------------------------------------------
#### Details:
- Date           | Thu May 13 12:59:01 BST 2021
- DietPi version | v7.1.2 (MichaIng/master)
- Image creator  | DietPi Core Team
- Pre-image      | Raspberry Pi OS Lite
- Hardware       | RPi 2 Model B (armv7l) (ID=2)
- Kernel version | Linux DietPi 5.10.17-v7+ #1403 SMP Mon Feb 22 11:29:51 GMT 2021 armv7l GNU/Linux
- Distro         | buster (ID=5,RASPBIAN=1)
- Command        | sudo -u homeassistant dash -c
. /home/homeassistant/pyenv-activate.sh
pyenv install 3.8.8
pyenv local 3.8.8
pip3 install -U pip wheel
[ -z '' ] || pip3 install
exec pip3 install homeassistant
- Exit code      | 127
- Software title | DietPi-Software
#### Steps to reproduce:

1. ...
2. ...
#### Expected behaviour:

- ...
#### Actual behaviour:

- ...
#### Extra details:

- ...
#### Additional logs:
```
WARNING: `pyenv init -` no longer sets PATH.
Run `pyenv init` to see the necessary changes to make to your configuration.
Downloading Python-3.8.8.tgz...
-> https://www.python.org/ftp/python/3.8.8/Python-3.8.8.tgz
Installing Python-3.8.8...
Installed Python-3.8.8 to /home/homeassistant/.pyenv/versions/3.8.8

dash: 5: pip3: not found
dash: 7: exec: pip3: not found
```
---------------------------------------------------------------------

Joulinar, 2021-05-13:
Hi,

I guess this will be fixed on next release with this PR https://github.com/MichaIng/DietPi/pull/4364

QuaiGoner, 2021-05-13:
Hello. can i bypass the error manually?

Joulinar, 2021-05-13:
@MichaIng 
is there a way to overcome the situation until 7.2 has been released?

MichaIng, 2021-05-13:
Since it requires multiple lines to be added, at least there is no trivial one-liner. But you could apply the change to `/boot/dietpi/dietpi-software`: https://github.com/MichaIng/DietPi/pull/4364/files#diff-98445566b948cc76bb45181ccd463ee93a7a441dbc8b25763d1b88f05f4ef8c5
- Remove the red line and add the green lines.
- The Python version increment is not required.

BernieV1977, 2021-05-15:
Thx @MichaIng !

MichaIng, 2021-05-19:
I'll mark this as closed. Feel free to reopen if required.

closed 2021-05-19 by MichaIng

PrzemekSkw, 2021-10-06:
Hello, I try to install Home Assistant on Rpi2 but installation failed on Rust:
```
error: can't find Rust compiler
  
  If you are using an outdated pip version, it is possible a prebuilt wheel is available for this package but pip is not able to install from it. Installing from the wheel would avoid the need for a Rust compiler.
  
  To update pip, run:
  
      pip install --upgrade pip
  
  and then retry package installation.
  
  If you did intend to build this package from source, try installing a Rust compiler from your system package manager and ensure it is on the PATH during installation. Alternatively, rustup (available at https://rustup.rs) is the recommended way to download and update the Rust compiler toolchain.
  
  This package requires Rust >=1.41.0.
                                                                    
                                              │   ERROR: Failed building wheel for cryptography                                                                        
                                              │ Failed to build cryptography                                                                                           
                                              │ ERROR: Could not build wheels for cryptography which use PEP 517 and cannot be installed directly
```
Upgrade pip don't work, command not found.
Regards,
Przemek

PrzemekSkw, 2021-10-07:
Hi, @Joulinar I download last Rpi image Armv7-32bit and install it on Raspberry Pi 2 model B. I'm not home now so I cannot give You more details.
Regards.

Joulinar, 2021-10-07:
Ah yeah I remember that we had an issues once where `cryptography` requires rust to build now. Usually it was impacting ARMv8 only, as on 32bit we download pre-compiled packages from piwheels.org. #4104

Ok something we need to have a look how to bypass.

Joulinar, 2021-10-07:
@MichaIng actually it's ARMv7 and I could replicate on my RPi3B+

MichaIng, 2021-10-07:
Solved:
```
Collecting cryptography==3.4.8
  Downloading https://www.piwheels.org/simple/cryptography/cryptography-3.4.8-cp39-cp39-linux_armv7l.whl (2.0 MB)
```