MichaIng/DietPi #4373: NC broken after dietpi-update

Phil1988, 2021-05-13:
#### Required Information
- DietPi version
G_DIETPI_VERSION_CORE=7
G_DIETPI_VERSION_SUB=1
G_DIETPI_VERSION_RC=2
G_GITBRANCH='master'
G_GITOWNER='MichaIng'

- Distro version 
stretch 9.13

- Kernel version 
-Linux DietPi 4.9.0-15-amd64 #1 SMP Debian 4.9.258-1 (2021-03-08) x86_64 GNU/Linux

- SBC model 
Virtual Machine (x86_64)

#### Additional Information (if applicable)
NC bro

#### Steps to reproduce
I was on 6.34.3 and wanted to update to 6.35.0
I got this error
[image]

Then I did
`apt-key del '95BD4743' # delete the old key
curl -sSfL 'https://packages.sury.org/php/apt.gpg' -o /etc/apt/trusted.gpg.d/dietpi-php.gpg # download the new key
/boot/dietpi/func/dietpi-set_software apt-cache clean # clean the apt cache
apt update # update again the package cache`

after that I was able to run dietpi-software "successfully"... but NC is (not checked if anything else is also) down:
[image]

I rebooted the system -> still the same.
As I got the error that 15 packages can be updated, I also did 
`apt update && apt upgrade`

NC is still broken

If you need more infos, please let me know :)

Joulinar, 2021-05-13:
> As I got the error that 15 packages can be updated

This is not an error, this is working as designed and give you an information about available `apt` package updated.

pls can you share `dietpi-services status`

Phil1988, 2021-05-13:
I am already on it :) (and also changed the word error-> hint already)... sorry for this unprecise word

dietpi-services status:
 Mode: status

[  OK  ] DietPi-Services | redis-server         active (running) since Thu 2021-05-13 16:02:31 CEST; 12min ago
[  OK  ] DietPi-Services | mariadb              active (running) since Thu 2021-05-13 16:02:32 CEST; 12min ago
[  OK  ] DietPi-Services | apache2              active (running) since Thu 2021-05-13 16:02:32 CEST; 12min ago
[  OK  ] DietPi-Services | mosquitto            active (running) since Thu 2021-05-13 16:02:32 CEST; 12min ago
[  OK  ] DietPi-Services | node-red             active (running) since Thu 2021-05-13 16:02:32 CEST; 12min ago
[  OK  ] DietPi-Services | blynkserver          active (running) since Thu 2021-05-13 16:02:32 CEST; 12min ago
[  OK  ] DietPi-Services | home-assistant       active (running) since Thu 2021-05-13 16:02:32 CEST; 12min ago
[  OK  ] DietPi-Services | cron                 active (running) since Thu 2021-05-13 16:02:32 CEST; 12min ago
[  OK  ] DietPi-Services | ssh                  active (running) since Thu 2021-05-13 16:02:31 CEST; 12min ago
[  OK  ] DietPi-Services | pihole-FTL           active (running) since Thu 2021-05-13 16:02:29 CEST; 12min ago
[ INFO ] DietPi-Services | dietpi-vpn           inactive (dead)
[  OK  ] DietPi-Services | dietpi-ramlog        active (exited) since Thu 2021-05-13 16:02:29 CEST; 12min ago
[  OK  ] DietPi-Services | dietpi-preboot       active (exited) since Thu 2021-05-13 16:02:29 CEST; 12min ago
[  OK  ] DietPi-Services | dietpi-boot          active (exited) since Thu 2021-05-13 16:02:30 CEST; 12min ago
[  OK  ] DietPi-Services | dietpi-postboot      active (exited) since Thu 2021-05-13 16:02:30 CEST; 12min ago
[ INFO ] DietPi-Services | dietpi-wifi-monitor  inactive (dead)

looks like nextcloud isnt running

Joulinar, 2021-05-13:
NextCloud is not an onw service. It's a web application and available via web server `apache2`, database `mariadb` and `redis`. All these services seems to be running fine. I guess issue is with the installed PHP version. At least I see 2 PHP version installed on you screen above. PHP 7.3 and 7.4.

I guess it's same issue as this https://github.com/MichaIng/DietPi/issues/4258

what happen if you run `ncc maintenance:mode --off`

Phil1988, 2021-05-13:
Good finding...
[image]

So there is php 7.3 , 7.4 and 8.0 installed?

Joulinar, 2021-05-13:
yes because you use are running on old Debian Stretch and there Ondrej's PHP repository is used.

Can you check if the 7.3 config file is available 
`cat /etc/php/7.3/mods-available/dietpi-nextcloud.ini`

Phil1988, 2021-05-13:
sorry see my edit... I am not as rocket fast as you :D

MichaIng, 2021-05-13:
You have conflicting PHP versions installed. Please paste the output of the following command:
```
dpkg -l | grep 'php'
```

Phil1988, 2021-05-13:
Here it is:

```
root@DietPi:~# dpkg -l | grep 'php'
ii  libapache2-mod-php            2:8.0+82+0~20210313.35+debian9~1.gbpa8195f             all          server-side, HTML-embedded scripting language (Apache 2 module) (default)
ii  libapache2-mod-php7.3         7.3.28-1+0~20210503.84+debian9~1.gbp6819da             amd64        server-side, HTML-embedded scripting language (Apache 2 module)
ii  libapache2-mod-php7.4         7.4.18-2+0~20210503.44+debian9~1.gbp52d112             amd64        server-side, HTML-embedded scripting language (Apache 2 module)
ii  libapache2-mod-php8.0         8.0.5-2+0~20210503.19+debian9~1.gbp3e4ad0              amd64        server-side, HTML-embedded scripting language (Apache 2 module)
ii  php-apcu                      5.1.20+4.0.11-1+0~20210304.24+debian9~1.gbp1c3d4d      amd64        APC User Cache for PHP
ii  php-common                    2:82+0~20210313.35+debian9~1.gbpa8195f                 all          Common files for PHP packages
ii  php-curl                      2:8.0+82+0~20210313.35+debian9~1.gbpa8195f             all          CURL module for PHP [default]
ii  php-gd                        2:8.0+82+0~20210313.35+debian9~1.gbpa8195f             all          GD module for PHP [default]
ii  php-igbinary                  3.2.1+2.0.8-6+0~20210223.25+debian9~1.gbp4f4993        amd64        igbinary PHP serializer
ii  php-imagick                   3.4.4+php8.0+3.4.4-7+0~20210305.28+debian9~1.gbpdcb27b amd64        Provides a wrapper to the ImageMagick library
ii  php-intl                      2:8.0+82+0~20210313.35+debian9~1.gbpa8195f             all          Internationalisation module for PHP [default]
ii  php-mbstring                  2:8.0+82+0~20210313.35+debian9~1.gbpa8195f             all          MBSTRING module for PHP [default]
ii  php-mysql                     2:8.0+82+0~20210313.35+debian9~1.gbpa8195f             all          MySQL module for PHP [default]
ii  php-redis                     5.3.4+4.3.0-1+0~20210329.35+debian9~1.gbpdc7ead        amd64        PHP extension for interfacing with Redis
ii  php-sqlite3                   2:8.0+82+0~20210313.35+debian9~1.gbpa8195f             all          SQLite3 module for PHP [default]
ii  php-xml                       2:8.0+82+0~20210313.35+debian9~1.gbpa8195f             all          DOM, SimpleXML, WDDX, XML, and XSL module for PHP [default]
ii  php-zip                       2:8.0+82+0~20210313.35+debian9~1.gbpa8195f             all          Zip module for PHP [default]
ii  php7.3-bcmath                 7.3.28-1+0~20210503.84+debian9~1.gbp6819da             amd64        Bcmath module for PHP
ii  php7.3-cli                    7.3.28-1+0~20210503.84+debian9~1.gbp6819da             amd64        command-line interpreter for the PHP scripting language
ii  php7.3-common                 7.3.28-1+0~20210503.84+debian9~1.gbp6819da             amd64        documentation, examples and common module for PHP
ii  php7.3-curl                   7.3.28-1+0~20210503.84+debian9~1.gbp6819da             amd64        CURL module for PHP
ii  php7.3-gd                     7.3.28-1+0~20210503.84+debian9~1.gbp6819da             amd64        GD module for PHP
ii  php7.3-gmp                    7.3.28-1+0~20210503.84+debian9~1.gbp6819da             amd64        GMP module for PHP
ii  php7.3-igbinary               3.2.1+2.0.8-6+0~20210223.25+debian9~1.gbp4f4993        amd64        igbinary PHP serializer
ii  php7.3-intl                   7.3.28-1+0~20210503.84+debian9~1.gbp6819da             amd64        Internationalisation module for PHP
ii  php7.3-json                   7.3.28-1+0~20210503.84+debian9~1.gbp6819da             amd64        JSON module for PHP
ii  php7.3-mbstring               7.3.28-1+0~20210503.84+debian9~1.gbp6819da             amd64        MBSTRING module for PHP
ii  php7.3-mysql                  7.3.28-1+0~20210503.84+debian9~1.gbp6819da             amd64        MySQL module for PHP
ii  php7.3-opcache                7.3.28-1+0~20210503.84+debian9~1.gbp6819da             amd64        Zend OpCache module for PHP
ii  php7.3-readline               7.3.28-1+0~20210503.84+debian9~1.gbp6819da             amd64        readline module for PHP
ii  php7.3-sqlite3                7.3.28-1+0~20210503.84+debian9~1.gbp6819da             amd64        SQLite3 module for PHP
ii  php7.3-xml                    7.3.28-1+0~20210503.84+debian9~1.gbp6819da             amd64        DOM, SimpleXML, WDDX, XML, and XSL module for PHP
ii  php7.3-zip                    7.3.28-1+0~20210503.84+debian9~1.gbp6819da             amd64        Zip module for PHP
ii  php7.4-cli                    7.4.18-2+0~20210503.44+debian9~1.gbp52d112             amd64        command-line interpreter for the PHP scripting language
ii  php7.4-common                 7.4.18-2+0~20210503.44+debian9~1.gbp52d112             amd64        documentation, examples and common module for PHP
ii  php7.4-curl                   7.4.18-2+0~20210503.44+debian9~1.gbp52d112             amd64        CURL module for PHP
ii  php7.4-gd                     7.4.18-2+0~20210503.44+debian9~1.gbp52d112             amd64        GD module for PHP
ii  php7.4-igbinary               3.2.1+2.0.8-6+0~20210223.25+debian9~1.gbp4f4993        amd64        igbinary PHP serializer
ii  php7.4-intl                   7.4.18-2+0~20210503.44+debian9~1.gbp52d112             amd64        Internationalisation module for PHP
[... cut ...]
ii  php8.0-zip                    8.0.5-2+0~20210503.19+debian9~1.gbp3e4ad0              amd64        Zip module for PHP
```

/etc/php/7.3/apache2/conf.d/**98**-dietpi-nextcloud.ini is not there, but
/etc/php/7.3/apache2/conf.d/**99**-dietpi-nextcloud.ini does exist

```
root@DietPi:/etc/php/7.3/apache2/conf.d# ls -l /etc/php/7.3/apache2/conf.d/99-dietpi-nextcloud.ini
lrwxrwxrwx 1 root root 48 Mai 14  2019 /etc/php/7.3/apache2/conf.d/99-dietpi-nextcloud.ini -> /etc/php/7.3/mods-available/dietpi-nextcloud.ini

```

MichaIng, 2021-05-13:
Okay, obviously something pulled the PHP meta packages, which are kinda rolling-release packages, pullen the newest stable PHP versions. But for those to become effective, you'd need to alter webserver configs and copy/move custom PHP configs over to the new version.

I suggest to remove all new PHP packages:
```
apt purge 'php8.0*' 'php7.4*' libapache2-mod-php7.4 libapache2-mod-php8.0
apt install php7.3-apcu php7.3-redis
```

Phil1988, 2021-05-14:
That worked!

Thank both of you guys! 👍

closed 2021-05-14 by Phil1988