MichaIng/DietPi #4374: Iteration v7.2 (release plan)

fpetru, 2021-05-13:
## Preparation

* [ ]  Review / update [`CHANGELOG.md`](https://github.com/MichaIng/DietPi/blob/master/CHANGELOG.txt)

## Beta testing
* [ ]  Make sure the [appropriate pull request](github.com/michaIng/dietPi/pulls) is available for the [documentation](dietpi.com/docs/releases/) with all the necessary changes:
   * [ ] new software titles
   * [ ] configuration changes
   * [ ] other doc changes
* [ ] Launch the initial beta
  * [ ] Announce it on Twitter, asking for external review
* [ ]  Determine if one or more hotfix needs to be applied (so the minor version is incremented e.g. 7.2.1, 7.2.2)

## Release
* [ ] Make the release
* [ ] Publish the release notes ([Docs](https://dietpi.com/docs/releases/))
* [ ] Mention the release notes in [DietPi Forum](https://dietpi.com/phpbb/viewforum.php?f=10)
* [ ] Publish the main banner on the website, announcing new version
* [ ] Determine if a hotfix is needed

## Release announcements
* [ ] Publish on Twitter
* [ ] Publish on Facebook

## Prep for the _next_ release
* [ ] Select issues for the new iteration in the new milestone (v7.3)
* [ ] Create a new _release plan_
* [ ] Create a new section in [wiki](https://github.com/MichaIng/DietPi/wiki/Iteration-plans) associated to v7.3

fpetru, 2021-05-13:
@MichaIng - A draft release plan, with actions on which we could help - it's just the initial version and I'm sure it needs more updates.

MichaIng, 2021-05-13:
Good idea to make this an actual issue with each release cycle. Shall we add deadlines/dates here as well (see my basecamp notes)?

StephanStS, 2021-05-30:
Post in the Facebook DietPi group:
---
[image]

closed 2021-06-01 by MichaIng