MichaIng/DietPi #4375: Iteration plan 7.2 (May 2021)

fpetru, 2021-05-13:
This plan capture the initial planning of the work to be done in May 2021. The full list of items in progress is included in the [milestone v7.3](https://github.com/MichaIng/DietPi/milestone/102).

## Planned schedule
- 2021.05.22 - Beta
- 2021.05.29 - Release version 7.2

## Planned items

### DietPi

- [ ] :running_woman: Support new single board computers (SBC) NanoPi R4S, NanoPi R1 and Pinebook Pro.
- [ ] :running_woman: Desktop autologin as _non-root_ user will be enhanced :octicons-arrow-right-16: It will work with all SBCs/GPUs and without the need to add the user to the video and/or render system group.
- [ ] :running_woman: The browser, installed together with desktops, can be chosen via preference systems. As a start, the already implemented Firefox and Chromium can be chosen, but further browsers may be added soon.
- [ ] :running_woman: Support for further public VPN and DDNS providers will be added to [**DietPi-VPN**](https://dietpi.com/docs/dietpi_tools/#dietpi-vpn) and [**DietPi-DDNS**](https://dietpi.com/docs/dietpi_tools/#dietpi-ddns).
- [ ] :running_woman: [**Bitwarden_RS**](https://dietpi.com/docs/software/cloud/#bitwarden_rs) has been arenamed to "_vaultwarden_" by its author, to avoid confusing and possible legal conflicts with the original Bitwarden software. DietPi will apply to this change and (force or allow to) migrate existing installs. It's not a big change, but relevant to all existent Bitwarden_RS users.

The release will include more bugs fixes, and possibly more enhancements and features.  

### DietPi Website

- [ ] :butterfly: Create a new page with performance metrics

---
### Legend

Mark | Description
-- | --
:running_woman: | work in progress
:warning: | blocked task
:butterfly: | stretch goal for this iteration
:frog: | more investigation required
:point_up: | under discussion within the team
:black_large_square: | a large work item, larger than one iteration

fpetru, 2021-05-13:
@MichaIng @StephanStS - initial iteration plan

MichaIng, 2021-05-13:
Great. I'll add links to the issues/PRs of each task. Also there is a "Completed" item missing in the legend, isn't it?

fpetru, 2021-05-13:
@MichaIng - My initial idea was to have the checkmark as the sign as completed. But, if you see a need, I could still I add it

MichaIng, 2021-05-14:
Makes sense, so we remove the symbol when done 👍.

closed 2021-06-01 by MichaIng