MichaIng/DietPi #4376: DNS issues after trying to update to 7.1

HyperCriSiS, 2021-05-13:
### Creating a bug report/issue

#### Required Information
- DietPi version 
G_DIETPI_VERSION_CORE=7
G_DIETPI_VERSION_SUB=1
G_DIETPI_VERSION_RC=0
G_GITBRANCH='beta'
G_GITOWNER='MichaIng'

- Distro version |  buster
- Kernel version |  Linux DarkPi 5.10.17-v7l+ #1414 SMP Fri Apr 30 13:20:47 BST 2021 armv7l GNU/Linx
- SBC model |  RPi 4 Model B (armv7l)

- Power supply used | 5V 3.4A 17W 
- SDcard used | SanDisk class 10
#### Additional Information 
- Software title | Dietpi + Pi-Hole + Unbound
- I tried to update to latest version 7.1 final but it seems something went wrong, because it still says (beta).
   Afterwards I have heavy DNS issues. Most of the queries get "unknown" as answer.
   I can't start dietpi-software or dietpi-update again now, because the DNS queries do not resolve. 

I have no bug report as it can't be sent.

Where do I start now to get this fixed?

Thanks,
Hyper

Joulinar, 2021-05-13:
Hi,

to change to stable branch you would need to set the correct value inside `dietpi.txt`. It can be done like this.

`G_CONFIG_INJECT 'DEV_GITBRANCH=' 'DEV_GITBRANCH=master' /boot/dietpi.txt`

But your main issue seems to be DNS. Probably `unbound` is not running. Can you check if the service is active 

`systemctl status unbound`

As workaround you can set a public DNS provider for your DietPi system. This will bypass PiHole/Unbound. Wich is anyway recommended 😃

HyperCriSiS, 2021-05-13:
Thanks for the fast answer :)

I want to stay on the beta channel, I just thought it did not update to 7.1 final because in the SSH welcome message in the title it says (beta)

Yes the DNS seems to be the main issue. Some queries work some not it is strange. At my Windows PC most of the queries work, but Dietpi can't reach the resources needed for the setup tools.

Unbound seems to be OK:

`unbound.service - Unbound DNS server
   Loaded: loaded (/lib/systemd/system/unbound.service; enabled; vendor preset: enabled)
  Drop-In: /etc/systemd/system/unbound.service.d
           └─dietpi.conf
   Active: active (running) since Thu 2021-05-13 21:12:33 CEST; 38min ago
     Docs: man:unbound(8)
 Main PID: 576 (unbound)
    Tasks: 1 (limit: 4915)
   CGroup: /system.slice/unbound.service
           └─576 /usr/sbin/unbound -d

Mai 13 21:12:25 DarkPi systemd[1]: Starting Unbound DNS server...
Mai 13 21:12:33 DarkPi package-helper[513]: /var/lib/unbound/root.key has content
Mai 13 21:12:33 DarkPi package-helper[513]: success: the anchor is ok
Mai 13 21:12:33 DarkPi unbound[576]: [576:0] info: start of service (unbound 1.9.0).
Mai 13 21:12:33 DarkPi systemd[1]: Started Unbound DNS server.
`

Is it the right way to set the DNS in the "/etc/resolv.conf"?

Joulinar, 2021-05-13:
if using STATIC IP, you should be able to set DNS entry via `dietpi-config`. But yes, at the end it's a setting inside `/etc/resolv.conf`

But, as well you could try to switch PiHole from `custom` DNS (which I assume is `Unbound`) to a global public one. Maybe this help as well for the time being.

MichaIng, 2021-05-13:
Pi-hole and Unbound should not play any role for the systems DNS, as long as you do not added Pi-hole or Unbound manually as DNS server to `/etc/resolv.conf`, which I do not recommend (so that a service crash does not lock out the whole server).

Try to use 9.9.9.9 (Quad9, DietPi default) as system DNS server. When applying it to `/etc/resolv.conf`, first assure that it is a static file and not a symlink:
```
ls -l /etc/resolv.conf
```
Pi-hole itself does not pull `resolvconf` as direct dependency, but probably another package does, in which case the DNS server needs to be assigned via `/etc/network/interfaces`. In any case, `dietpi-config` can be used as well.

HyperCriSiS, 2021-05-13:
It gets more confusing now :)

In dietpi-config in the Adapter settings the title is:
│ Ethernet Details:                                                                                            │
│ Usage   : Sent = 9 MiB | Recieved = 9 MiB                                                                    │
│ Address : IP = 192.168.0.99 | Mask = 255.255.255.0 | Gateway = 192.168.0.1 | DNS = 127.0.0.1 

But below:                                                  │
Static DNS     : [1.1.1.1 1.0.0.1]       

In the file /etc/resolv.conf --> 127.0.0.1
Which points to /etc/resolvconf/run/resolv.conf

dietpi-update output:

` DietPi-Update
─────────────────────────────────────────────────────
 Phase: Checking for available DietPi update

[ INFO ] DietPi-Update | Getting repository version: https://raw.githubusercontent.com/MichaIng/DietPi/beta/.update/version
curl: (7) Failed to connect to raw.githubusercontent.com port 443: Connection timed out
[FAILED] DietPi-Update | Unable to get repository version. Please check your network connection, then rerun dietpi-update.
[FAILED] DietPi-Update | If this error persists, please report at: https://github.com/MichaIng/DietPi/issues
[ INFO ] DietPi-Update | APT update, please wait...
Err:1 http://raspbian.raspberrypi.org/raspbian buster InRelease
  Could not connect to raspbian.raspberrypi.org:80 (93.93.128.193), connection timed out
Err:2 https://archive.raspberrypi.org/debian buster InRelease
  Could not connect to archive.raspberrypi.org:443 (93.93.135.141), connection timed out Could not connect to archive.raspberrypi.org:443 (93.93.135.117), connection timed out Could not connect to archive.raspberrypi.org:443 (93.93.130.212), connection timed out Could not connect to archive.raspberrypi.org:443 (176.126.240.86), connection timed out Could not connect to archive.raspberrypi.org:443 (46.235.227.39), connection timed out Could not connect to archive.raspberrypi.org:443 (176.126.240.167), connection timed out Could not connect to archive.raspberrypi.org:443 (46.235.231.111), connection timed out Could not connect to archive.raspberrypi.org:443 (176.126.240.84), connection timed out Could not connect to archive.raspberrypi.org:443 (46.235.230.122), connection timed out Could not connect to archive.raspberrypi.org:443 (93.93.135.118), connection timed out Could not connect to archive.raspberrypi.org:443 (46.235.231.151), connection timed out Could not connect to archive.raspberrypi.org:443 (46.235.231.145), connection timed out
Reading package lists...
W: Failed to fetch http://raspbian.raspberrypi.org/raspbian/dists/buster/InRelease  Could not connect to raspbian.raspberrypi.org:80 (93.93.128.193), connection timed out
W: Failed to fetch https://archive.raspberrypi.org/debian/dists/buster/InRelease  Could not connect to archive.raspberrypi.org:443 (93.93.135.141), connection timed out Could not connect to archive.raspberrypi.org:443 (93.93.135.117), connection timed out Could not connect to archive.raspberrypi.org:443 (93.93.130.212), connection timed out Could not connect to archive.raspberrypi.org:443 (176.126.240.86), connection timed out Could not connect to archive.raspberrypi.org:443 (46.235.227.39), connection timed out Could not connect to archive.raspberrypi.org:443 (176.126.240.167), connection timed out Could not connect to archive.raspberrypi.org:443 (46.235.231.111), connection timed out Could not connect to archive.raspberrypi.org:443 (176.126.240.84), connection timed out Could not connect to archive.raspberrypi.org:443 (46.235.230.122), connection timed out Could not connect to archive.raspberrypi.org:443 (93.93.135.118), connection timed out Could not connect to archive.raspberrypi.org:443 (46.235.231.151), connection timed out Could not connect to archive.raspberrypi.org:443 (46.235.231.145), connection timed out
W: Some index files failed to download. They have been ignored, or old ones used instead.
[  OK  ] DietPi-Update | APT update
[ INFO ] DietPi-Update | No APT upgrades were found, not creating file: /run/dietpi/.apt_updates
`

dietpi-software output:

│ Checking network connectivity                                                                                │
│  - Command: ping -c 1 -W 5 9.9.9.9                                                                           │
│  - Exit code: 1                                                                                              │
│  - DietPi version: v7.1.0 (MichaIng/beta) | HW_MODEL: 4 | HW_ARCH: 2 | DISTRO: 5                             │
│  - Image creator: DietPi Core Team                                                                           │
│  - Pre-image: Raspbian Lite                                                                                  │
│  - Error log:                                                                                                │
│ PING 9.9.9.9 (9.9.9.9) 56(84) bytes of data.                                                                 │
│                                                                                                              │
│ --- 9.9.9.9 ping statistics ---                                                                              │
│ 1 packets transmitted, 0 received, 100% packet loss, time 0ms 

So I wonder which of the settings counts now. The one in the dietpi-settings "Static" or the one in the title which reflects /etc/resolvconf/run/resolv.conf.

Also it is strange that even the ping to 9.9.9.9 gets no answer :/

HyperCriSiS, 2021-05-13:
I tested some IPs now and I get no answer from any of them.
But pi-hole still answers query requests from my Android and my Windows PC.
I had such an issue month ago on Raspbian and I could not fix it. it happend after change of the hostname, lol. I had to reinstall. :(

Joulinar, 2021-05-13:
as well a simple `ping 9.9.9.9` doesn't give anything? It still errors out?

Can you share `ip a`. You can check as well to ping the gateway `ping 192.168.0.1`

HyperCriSiS, 2021-05-13:
Yes, even simple ping to 9.9.9.9 and other IPs.
Changed cable now, even switched to another Pi4 board and power cycled the switch.
WTF is this... :D

Joulinar, 2021-05-13:
what about connecting the RPi directly to your router? would that be possible? just to reduce the number of components. Are you able to ping the gateway?

HyperCriSiS, 2021-05-13:
Directly connected to the router there's no change.
I also get no answer from ping to gateway :(

Joulinar, 2021-05-13:
it might not do any difference but you have IPv6 enabled. If not needed you could disable it. Let's check default route as well `ip r`

MichaIng, 2021-05-13:
Please also try to ping the router (the IP should match the one which `ip r` shows), to see whether it is at least reaching the router and probably not handled/forwarded there:
```
ping 192.168.0.1
```

HyperCriSiS, 2021-05-13:
It ran well before with IP6 enabled. The problem is when I disable IPv6 blocking DNS queries will not work on devices where I can't disable IPv6 like on my Android.

ip r:
192.168.0.0/24 dev eth0 proto kernel scope link src 192.168.0.99

PING 192.168.0.1 (192.168.0.1) 56(84) bytes of data.
--- 192.168.0.1 ping statistics ---
16 packets transmitted, 0 received, 100% packet loss, time 599ms

Joulinar, 2021-05-13:
usually Android devices are able to work on IPv4 as well.

Does your router has a feature to block specific devices?

MichaIng, 2021-05-13:
> The problem is when I disable IPv6 blocking DNS queries will not work on devices where I can't disable IPv6 like on my Android.

> usually Android devices are able to work on IPv4 as well.

Not usually, but definitely. Most mobile networks are purely IPv4, large parts of the www are. When you disable IPv6 on the server, it has no IPv6 address anymore, which means that clients connect via IPv4 automatically. In the LAN network this should be the case anyway with a DHCP(v4) server. IPv4 DNS request can resolve IPv6 IPs as well, the protocol of the DNS request and the protocol of the contained IP address are two independent things. So it should not break anything.
I don't believe that disabling IPv6 solves anything, but it doesn't hurt to give it a try, just to rule it out.
__________________
```
ip r:
192.168.0.0/24 dev eth0 proto kernel scope link src 192.168.0.99
```
You have no default route assigned, so requests to any IP outside of the LAN are not sent anywhere. Please try to reconfigure the interface, **if you're not connected via SSH (as this disables network completely)**:
```
ifdown eth0
ifup eth0
```
If you're connected via SSH:
```
ip r add default via 192.168.0.1 dev eth0
```
And check:
```
cat /etc/network/interfaces
```

HyperCriSiS, 2021-05-13:
No change with IPv6 disabled.

sudo ip r add default via 192.168.0.1 dev eth0
RTNETLINK answers: File exists

If I want to edit the file manually what do I need to enter to specify the default route?

Joulinar, 2021-05-14:
do you have any packet filter installed on your DietPi device? At least it's quite strange that you can connect to the system but not from the system. Somehow outbound traffic seems to be blocked or routed incorrectly.

MichaIng, 2021-05-14:
> No change with IPv6 disabled.
> ```
> sudo ip r add default via 192.168.0.1 dev eth0
> RTNETLINK answers: File exists
> ```
> I manually added this to the file:
> default via 192.168.0.1 dev eth0

Was your [above](https://github.com/MichaIng/DietPi/issues/4376#issuecomment-840857948) `ip r` output complete? `RTNETLINK answers: File exists` means that the default route does already exist. This cannot be added to a file like that. You can remove and add routes only via the `ip r add/del` commands. Please remove `default via 192.168.0.1 dev eth0` from any file you added it manually.

HyperCriSiS, 2021-05-15:
It is getting even more strange now.
I disconnected the Raspi from network before I went to bed. At this time the routers DHPC was active. 
I saw a DNS issue in the system logs of the router, the phone numbers could not be registered.

Today the DNS issues in the router are not showed anymore.
I connected the Raspi (DHCP offline) and I had full connection, everything worked...
Then I ran the dietpi updater again and it updated to 7.1.2.
Thought everything is ok now and I installed PiVPN and remote.it, I know, bad idea at this stage :)

I disabled the routers DHCP and enabled it in pi-hole.

After reboot then, again, no connection...

As it is exactly the same issue as before I don't think it has something to do now with the PiVPN?
The issue persists if I disable the Pihole DHCP and enable the routers DHCP again.
All other devices have a proper connection.

Restarting network should be possbile with "sudo systemctl restart networking" is this right?
Because it did not work, I could not connect via SSH then anymore and had to restart.
Also while there's no ping possible to anything except the own IP, when I made a connection test in the dietpi-update tool, I got "SUCCESS".

Actual ip r output is:
`default via 192.168.0.1 dev eth0 onlink
10.6.0.0/24 dev wg0 proto kernel scope link src 10.6.0.1
192.168.0.0/24 dev eth0 proto kernel scope link src 192.168.0.99
`

MichaIng, 2021-05-16:
> Restarting network should be possbile with "sudo systemctl restart networking" is this right?

Not that has no effect for the "allow-hotplug" entries in `/etc/network/interfaces`. To restart the network (with temporary connection outage):
```
ifdown eth0
ifup eth0
```

The default route is there btw, so that looks correct now.

So do I get it right that network broke after you enabled the Pi-hole DHCP server? Was the VPN server already running before doing the reboot? Please try to check both separately, enabling only DHCP server with VPN disabled, or only VPN with DHCP disabled (using router DHCP instead).

If the DHCP server turns out to be the culprit, how exactly did you configure DNS settings in router and Pi-hole, e.g. the upstream DNS entries they use? Probably there is a DNS loop or other conflict between both.

Joulinar, 2021-05-16:
Well even if there would be a DNS loop, `ping` should be able to work using IP address. It seems more a routing issue or some strange `iptables` rules. 

Does the router has some package tracing function? At least this way we could check if packages are arriving at all

MichaIng, 2021-05-16:
Good point. Also to check iptables: `iptables -L`

HyperCriSiS, 2021-05-16:
is it problematic to deinstall VPN with apt instead of pihole-software?
Because it would be better to have less error sources.
Yes it seems to have stopped working when I enabled the pihole DHCP again.
But that makes just no sense that even pings to IPs are not possible.

Unfortunately I don't have such possibility in this router. Dumb Telekom device. (Speedport Pro Plus)

iptables -L
```
Chain INPUT (policy ACCEPT)
target     prot opt source               destination

Chain FORWARD (policy ACCEPT)
target     prot opt source               destination

Chain OUTPUT (policy ACCEPT)
target     prot opt source               destination

```

Joulinar, 2021-05-16:
Usually VPN and Pihole don't have a direct relationship. You should be fine to remove PiVPN

HyperCriSiS, 2021-05-16:
Removal of PiPN seems not possible via apt-get.
Should I better leave it for now?
The issue is the same anyway as it was without VPN.

Joulinar, 2021-05-16:
PiVPN is not an `apt` that can be removed. It's a set of scripts to give user a frontend manage OpenVPN or Wireguard Server 😃 

can you try to remove it using `dietpi-software uninstall 117`

Probably you will get IP/DNS check issues. If yes, we can tweak it for the time being. Just edit following 2 lines in `/boot/dietpi.txt` if needed

```
CONFIG_CHECK_CONNECTION_IP=127.0.0.1
CONFIG_CHECK_DNS_DOMAIN=localhost
```

HyperCriSiS, 2021-05-25:
Hey guys, sorry for the late answer.

Actually DHCP is enabled in the router and disabled in PiHole.

And the "magic" happened again, the connection is OK?!
Dietpi was running now since my last post and I did not do anything with it...

Now there's an update to 7.2 available. I have the feeling, if I do that now, it will be bricked again.
Can I maybe do anything before, to get some extra logs?

Joulinar, 2021-05-25:
7.2 has not been released yet and is in Beta phase. Before doing any update it's recommended to perform a backup. This way you can restore quickly if needed

MichaIng, 2021-05-25:
You also may upgrade APT packages first manually, step by step when those are many:
```
apt update
apt upgrade
# Check how much/which packages are offered for upgrade and in case to it in smaller steps via:
apt install package1 package2
# Verify that all is still in order, in case reboot to be sure, then upgrade the next batch
apt install package3 package4
...
# And finally
dietpi-update
```

MichaIng, 2021-06-02:
@HyperCriSiS 
Did you manage to update successfully this time?

MichaIng, 2021-06-27:
I mark this issue as closed. Feel free to reopen if required.

closed 2021-06-27 by MichaIng