MichaIng/DietPi #4377: dietpi-login:  Does not walk incorrectly with user dietpi

LexiconCode, 2021-05-14:
### Creating a bug report/issue

#### Required Information
- DietPi version | G_DIETPI_VERSION_CORE=7
G_DIETPI_VERSION_SUB=1
G_DIETPI_VERSION_RC=2
G_GITBRANCH='master'
G_GITOWNER='MichaIng'

- Distro version | buster
- Kernel version | `Linux DietPi 5.10.17+ #1414 BST 2021 armv6l GNU/Linux`
- SBC model | `RPi Zero W (armv6l)`
- Power supply used | 5V 2A RAVpower
- SDcard used | SanDisk ultra

#### Additional Information (if applicable)
bug report
- bfb085fb-d356-48f6-984b-f9ed7dd826a8

#### Steps to reproduce

1. flash the latest version of dietpi
2. Boot up

#### Expected behaviour
- 1st time set up should execute

#### Actual behaviour
Using username "dietpi".
sh: 1: /usr/bin/xauth: not found
/boot/dietpi/dietpi-login: line 226: ((: == 2 : syntax error: operand expected (error token is "== 2 ")

#### Extra details
 │ [FAILED] Root login required                                                                                         
 │ To finish DietPi first run setup, root permissions are required.                                                     
 │ Please login again as user "root" with password "dietpi", respectively the one you chose via "dietpi.txt".    

- /var/tmp/dietpi/logs/dietpi-firstrun-setup.log is empty
- `-rwxr-xr-x 1 root root 13491 May 13 22:53 /boot/dietpi.txt`
- `chown: changing group of '/boot/dietpi.txt': Operation not permitted`

At 1st I was able to login under root and do all the configuration but upon reboot this issue persisted.

Joulinar, 2021-05-14:
Hi,

did you tried to re-flash the SD card and perform a new first run/installation?

MichaIng, 2021-05-14:
Strange. As `xauth is tried to be executed (e.g. as part of `startx` desktop autostart?), did you install a desktop or Chromium? Would be just strange if `startx` did exist but `xauth` not.

> chown: changing group of '/boot/dietpi.txt': Operation not permitted

When did this happen? Did you run this chown manually or was it done automatically on login? It cannot work since the /boot partition has a FAT filesystem which does not support UNIX permissions. Only root can hence write it, or anyone with sudo permissions.
___________
I had a look into your bug report:
- There are some kernel error, not sure if those are related:
    ```
    [   18.790896] brcmfmac: brcmf_fw_alloc_request: using brcm/brcmfmac43430-sdio for chip BCM43430/1
    [   18.793006] usbcore: registered new interface driver brcmfmac
    [   18.826980] brcmfmac mmc1:0001:1: Direct firmware load for brcm/brcmfmac43430-sdio.raspberrypi,model-zero-w.txt failed with error -2
    [   19.098251] brcmfmac: brcmf_fw_alloc_request: using brcm/brcmfmac43430-sdio for chip BCM43430/1
    [   19.100355] brcmfmac: brcmf_c_process_clm_blob: no clm_blob available (err=-2), device may have limited channels available
    ```
    **EDIT: No issue: https://www.raspberrypi.org/forums/viewtopic.php?p=1723945**
- No X server is installed, hence xauth/startx should both not be available.
- Autostart index is 0 (regular console login), so I wonder where the xauth call was coming from.
- The dietpi-globals script and the install stage file are present, hence strange that dietpi-login has the related variable empty, leading to the syntax error.
- Install stage is "1", which means first run update went through (log is there and fine) but first run setup not (the initial dietpi-software call). That is strange since there is additional software marked as installed, so it must have been executed and run through. But also the related log is missing/empty, as you stated.

Unrelated:
- You tried to use `DietPi_Print` as hostname. Since the underscore is not valid character for hostnames, try to use `DietPi-Print`.
- `vmtouch.service` failed, which matches another recent report in the forum: https://dietpi.com/phpbb/viewtopic.php?t=9014
    This is completely optional, and does not cause issues, but strange that the unchanged package/binary suddenly causes a segmentation fault on ARMv6 RPi models. Probably the last kernel upgrade is responsible.

LexiconCode, 2021-05-15:
> 
> 
> Strange. As `xauth is tried to be executed (e.g. as part of `startx`desktop autostart?), did you install a desktop or Chromium? Would be just strange if`startx`did exist but`xauth` not.
> 
No desktop/Chromium
 was installed
> When did this happen? Did you run this chown manually or was it done automatically on login? It cannot work since the /boot partition has a FAT filesystem which does not support UNIX permissions. [...]

manually, make sense as FAT filesystem does not support UNIX permissions

> I had a look into your bug report:
> 
>     * There are some kernel error, not sure if those are related:
>       ```
>       [   18.790896] brcmfmac: brcmf_fw_alloc_request: using brcm/brcmfmac43430-sdio for chip BCM43430/1
>       [   18.793006] usbcore: registered new interface driver brcmfmac
>       [   18.826980] brcmfmac mmc1:0001:1: Direct firmware load for brcm/brcmfmac43430-sdio.raspberrypi,model-zero-w.txt failed with error -2
>       [   19.098251] brcmfmac: brcmf_fw_alloc_request: using brcm/brcmfmac43430-sdio for chip BCM43430/1
>       [   19.100355] brcmfmac: brcmf_c_process_clm_blob: no clm_blob available (err=-2), device may have limited channels available
>       ``` 
>       
>       **EDIT: No issue: [raspberrypi.org/forums/viewtopic.php?p=1723945](https://www.raspberrypi.org/forums/viewtopic.php?p=1723945)**
> 
>     * No X server is installed, hence xauth/startx should both not be available.

Yes no desktop environment was installed just Cups 

> * Autostart index is 0 (regular console login), so I wonder where the xauth call was coming from.
>
>     * The dietpi-globals script and the install stage file are present, hence strange that [...]

This only happened when I used `su -` to login when dietpi user failed to start the first run setup.

So I reflash and report back.

Joulinar, 2021-05-15:
Does the same happen if you login as user `root` directly?

LexiconCode, 2021-05-18:
> 
> 
> Does the same happen if you login as user `root` directly?

On a fresh install does not. I think I did not notice logging in as `dietpi` on 1st run because of my SSH client already having previous credentials/IP

MichaIng, 2021-05-19:
Okay, so the login as unprivileged user was the only issue it seems. This is expected on first login and the error message states it quite well, I think:
```
Please login again as user "root" with password "dietpi", respectively the one you chose via "dietpi.txt".
```

`su` is btw known to cause issues, as it pulls in the previous session's environment into the subshell, breaking e.g. PATH and other system variables, which is seen as a security/privacy issue as well as with variables malicious code executions can be injected into the root session. `sudo -i` has some benefits here, as it creates a fresh environment by default and also as it logs which user uses it when to gain root privileges. `su -` (with the dash) seems to create a new environment, in my tests, but not 100% sure, I'd always `logout` and freshly login, or open a new SSH session instead of creating nested shells, where possible.

LexiconCode, 2021-05-19:
This issue can be closed as I think we've have a great explanation.

closed 2021-05-19 by MichaIng