MichaIng/DietPi #4378: Error installing NextCloud

pechex, 2021-05-15:
#### Details:
- Date           | Fri May 14 21:40:09 -03 2021
- DietPi version | v7.1.2 (MichaIng/master)
- Image creator  | DietPi Core Team
- Pre-image      | Raspberry Pi OS (32-bit) Lite
- Hardware       | RPi 4 Model B (armv7l) (ID=4)
- Kernel version | Linux rpi4 5.10.17-v7l+ #1403 SMP Mon Feb 22 11:33:35 GMT 2021 armv7l GNU/Linux
- Distro         | buster (ID=5,RASPBIAN=1)
- Command        | mysql -e grant all privileges on *.* to tmp_root@localhost identified by 'tsdBM5Yoj6X7Hn92rkDq58RcAxiVGO' with grant option
- Exit code      | 1
- Software title | DietPi-Software
#### Steps to reproduce:

1. ... sudo dietpi-software install 114

#### Expected behaviour:

- ...
#### Actual behaviour:

- ...
#### Extra details:

- ...
#### Additional logs:
```
ERROR 1045 (28000): Access denied for user 'root'@'localhost' (using password: NO)
```

This error happens when installing NextCloud. I tried many times, allways got the same error. My Mysql root password IS changed from default.

Joulinar, 2021-05-15:
Hi,

Just a short question. Does it mean you changed the password inside the database manually?

MichaIng, 2021-05-15:
That would explain it. The MariaDB root user by default is using the unix_socket authentication plugin, which means that as UNIX root user one can login without entering a password, but not as any other user. This is the basis why we can do `mysql -e` calls without knowing or asking for any password.

When setting a MariaDB root user password, the authentication method is changed, so the `mysql -e` calls fail as no password is given.

You can revert to unix_socket authentication via:
```
mysql -u root -p -e 'alter user root@localhost identified via unix_socket;'
```

pechex, 2021-05-15:
I can confirm this fixed the error! Thanks!

And thanks for the explanation. I don't remember why I changed the password... maybe I tought it would be more secure.
Anyway, thanks.

closed 2021-05-15 by pechex