MichaIng/DietPi #4380: Key-based ssh authorization with dropbear fails due to wrong location of authorized_keys

mwip, 2021-05-16:
#### Required Information
- DietPi version 
```
G_DIETPI_VERSION_CORE=7
G_DIETPI_VERSION_SUB=1
G_DIETPI_VERSION_RC=2
G_GITBRANCH='master'
G_GITOWNER='MichaIng'
```
- Distro version 
```
buster
```
- Kernel version
```
Linux DietPi 5.10.17-v8+ #1414 SMP PREEMPT Fri Apr 30 13:23:25 BST 2021 aarch64 GNU/Linux
```
- SBC model
```
RPi 3 Model B (aarch64)
```
- Power supply used 
```5V 1.5A RAVpower```
- SDcard used 
```SanDisk of some sort```

#### Additional Information (if applicable)
- Software title | **dropbear**
- Was the software title installed freshly or updated/migrated? **Yes, default from the image**
- Can this issue be replicated on a fresh installation of DietPi? **It's a fresh install**

#### Steps to reproduce

1. create a ssh-rsa key on some machine using `ssh-keygen`
2. copy the ssh-key to your dietpi using `ssh-copy-id root@your-dietpi-hostname-or-ip`
3. enter the password for the `root` user
4. try to log in using `ssh root@your-dietpi-hostname-or-ip`

#### Expected behaviour

- I would expect, **NOT** being prompted for a password.

#### Actual behaviour

- I am being prompted for the root password

#### How I solved it

- The authorized key is placed in `/etc/dropbear/authorized_keys`
- However, it seems, for the root user at least, this file is not respected for key-based authorization. 
- I could login sucessfully after copying `/etc/dropbear/authorized_keys` to `/root/.ssh/authorized_keys`

*Note*: Full disclosure, I'm no expert on either dropbear or (ssh-)security. So unless this is verified, checked or even solved I do not encourage future visitors of this issue to reproduce the above solution. It is just meant to demonstrate *one possible* solution. 

#### Ideas for a solution
- either add `/etc/dropbear/authorized_keys` to the list of files checked for authorized keys, 
- or put the keys authorized by ssh-copy-id in correct file.

MichaIng, 2021-05-16:
Many thanks for your report.

Where is the info from that `/etc/dropbear/authorized_keys` should be the correct place? That AFAIK never was correct and it does not make sense either. You login with a specific user, hence the key must be located in a directory which is only accessible for this one user, which is its home director,. and not a system world-readable directory like in `/etc`. `~/.ssh/authorized_keys` is used by OpenSSH as well.

Or do you mean that `ssh-copy-id` is copying the local key automatically to that wrong location? It would be an external bug in the `openssh-client` package then, installed on the client system. In either way, we have no control over where Dropbear is reading the keys from, nor where `ssh-copy-id` is putting them.

mwip, 2021-05-16:
Indeed, `ssh-copy-id` put the keys in the wrong location. I don't know where this location is defined or even standardized and totally agree, that the key should be in any reference to the user it was authorized for. I found some mentions over at [StackExchange](https://serverfault.com/a/954257/495037) (to be taken with a grain of salt, o/c) but was not able to find any documentaion on dropbear (aside from `dropbear --help` resulting in an error and spitting out some options)

MichaIng, 2021-05-16:
Lol, the same StackExchange I was just reading 😄. That is about OpenWRT indeed (see my edit above).

The man page clearly states `~/.ssh/authorized_keys`: https://manpages.debian.org/dropbear

mwip, 2021-05-16:
Sorry for missing the debian man page... `man dropbear` was non-existant so I figured there might be no man page whatsoever... Maybe this would be indicative of that dietpi is using some minimal build of dropbear, that even strips out the man page?

I am using Arch on my clients with `OpenSSH` I tried it with.

mwip, 2021-05-16:
If you'd need the test, I'd have a Debian VM at the ready... just LMK

MichaIng, 2021-05-16:
Not the man page is missing, but the tool to view man pages from the command line: `apt install man-db`
The Dropbear build is the default package from Debian/Raspbian (non-RPi/RPi).

Then I'd open a bug report on the [Arch "openssh" package bug tracker](https://bugs.archlinux.org/?project=1&string=openssh) first, so that they check whether this is a build option their side or an upstream bug. Not sure how reactive Arch package maintainers are. Probably I find time to test it on Debian. It the issue is the same there, then we can be relatively sure that it's an upstream bug and can forward/report the issue to OpenSSH devs directly.

mwip, 2021-05-16:
Indeed. It seems to point to an arch issue. I just tried from a raspberrypi running stock RaspberrypiOS. This one works fine and puts the key to `~/.ssh/authorized_keys`.

I will report on the Arch side, link the issues and close when solved. If you like to close this issue earlier, feel free to do so.

mwip, 2021-05-16:
Here you can find the related issue: https://bugs.archlinux.org/task/70901

MichaIng, 2021-05-16:
Great. I'll leave the issue open since I like that method of applying keys for key authentication. I want to have that in our docs as step to enhance security and convenience both, key authentication in general, and to be true I didn't know that `ssh-copy-id` command from OpenSSH (client) before, which makes adding the key simple and safe from a GNU/Linux system 👍. That serves as reminder.

mwip, 2021-05-16:
Not to get sidetracked too far, but maybe it would be a neat option to have an entry in dietpi.txt where you can pre-insert ssh keys for the `root` user? Or even a switch to fully disable password-based authentication alltogether?

MichaIng, 2021-05-16:
Good idea, I like that. Jep let's do that in a new feature request.

Scimmia22, 2021-05-16:
There are no build options that affect this, ssh-copy-id is a simple, static script. Testing from Debian doesn't mean much, either, as even Sid is a few months out of date, and they patch the hell out of openssh.

MichaIng, 2021-05-16:
Then the Debian patches are quite reasonable it seems. And Arch does not apply any patches? Then its an upstream issue. I'll have a look into the source code, when its a hardcoded path based on the identified SSH server then it should be easy to fix and the reason is trivial. It would break applying the key to OpenWRT in a single step, but following a single single-purpose distribution which does not follow common defaults is hopefully not seen reasonable after all 😉.

Scimmia22, 2021-05-16:
Arch, as a general rule, patches as little as possible. Currently there's a single patch applied, and it's a fix for building sandbox-seccomp-filter.c with newer glibc. May be able to be dropped, though, it's old enough, I'll have to look into that.

Anyway, take a look at line 335 of the upstream script.

MichaIng, 2021-05-16:
PR up: https://github.com/openssh/openssh-portable/pull/250
I also found the OpenWRT patch, so the case is pretty clear now.