MichaIng/DietPi #4381: Add possibility to pre-share ssh-rsa keys in dietpi.txt and disable password-based authentication

mwip, 2021-05-16:
#### Is your feature request related to a problem? Please describe:
- ssh authorization through keys is both more convenient and secure than password-based authentication

#### Describe the solution you'd like:
- It would be cool to have two variables in `/boot/diepi.txt`

1. to share one or multiple public ssh-rsa key(s) of the client machine(s) one will want to access dietpi from
2. a toggle variable to disable password-based authentication alltogehter

#### Additional context
- this feature request is based on #4380

MichaIng, 2021-05-16:
Not to forget modern ECDSA and Ed25519 keys 😉.

mwip, 2021-05-17:
I tested Ed25519 on my dietpi install and it seems that the version of dropbear (v2018.76) does not yet support Ed25519. Reading through the [changelog](https://matt.ucc.asn.au/dropbear/CHANGES), it seems that only dropberar_2020.79 and higher support it. Thought I leave this here, in case someone runs into this.

MichaIng, 2021-05-17:
Jep that is true.

MichaIng, 2021-11-15:
I accidentally opened a duplicate with some more ideas on that: #4639

MichaIng, 2022-07-11:
Implemented with DietPi v8.5: #5509