MichaIng/DietPi #4382: WiringPi version Outdated for Pi 4B

C0D3-M4513R, 2021-05-16:
### Creating a bug report/issue

#### Required Information
- DietPi version | `cat /boot/dietpi/.version`
``` 
G_DIETPI_VERSION_CORE=7
G_DIETPI_VERSION_SUB=1
G_DIETPI_VERSION_RC=2
G_GITBRANCH='master'
G_GITOWNER='MichaIng'
```
- Distro version | `echo $G_DISTRO_NAME` or `cat /etc/debian_version`
`buster`
- Kernel version | `uname -a`
`Linux DietPi 5.10.17-v8+ #1414 SMP PREEMPT Fri Apr 30 13:23:25 BST 2021 aarch64 GNU/Linux`
- SBC model | `echo $G_HW_MODEL_NAME` or (EG: RPi3)
`RPi 4 Model B (aarch64)`
- Power supply used | (EG: 5V 1A RAVpower)
`5V3A `
- SDcard used | SanDisk?
- echo $G_HW_MODEL
`4`

#### Additional Information (if applicable)
- Software title | WiringPI
- Was the software title installed freshly or updated/migrated? fresh install
- Can this issue be replicated on a fresh installation of DietPi? yes

- Bug report ID | `echo $G_HW_UUID`

#### Steps to reproduce

1. Use a Pi 4B
2. Install WiringPi
3. run `gpio readall`
4. Overve the error: `Oops - unable to determine board type... model: 17`

#### Expected behaviour

- The command should execute without failing

#### Actual behaviour

- The command fails

#### Extra details

- http://wiringpi.com/wiringpi-updated-to-2-52-for-the-raspberry-pi-4b/
- https://github.com/MichaIng/DietPi/blob/86c5e377c607c2e1d8e97138c7cb9d377120a55d/dietpi/dietpi-software#L4862

Joulinar, 2021-05-16:
Hi,

I set I to know issue as there will be some updates on next release https://github.com/MichaIng/DietPi/pull/4324

C0D3-M4513R, 2021-05-16:
ok, great to see, that this was already worked on. Maybe either link to the pr #4324 or close, as duplicate of #4149?

Joulinar, 2021-05-16:
LoL @MichaIng

MichaIng, 2021-05-16:
Lol, concurrent actions. @C0D3-M4513R would be great if you could give it a try by migrating to the `dev` branch.

C0D3-M4513R, 2021-05-16:
sure. will post an update, when done

C0D3-M4513R, 2021-05-16:
Had a little trouble with an incorrect system time (despite having it setup in `dietpi-config`. resolved it, by running ` timedatectl set-ntp true`)
`dietpi-software reinstall 70` completed successfully 
Some Warnings occurred: 
```
[Compile] pseudoPins.c
pseudoPins.c: In function ‘myAnalogRead’:
pseudoPins.c:50:16: warning: cast to pointer from integer of different size [-Wint-to-pointer-cast]
   int *ptr   = (int *)node->data0 ;
                ^
pseudoPins.c: In function ‘myAnalogWrite’:
pseudoPins.c:59:16: warning: cast to pointer from integer of different size [-Wint-to-pointer-cast]
   int *ptr   = (int *)node->data0 ;
                ^
pseudoPins.c: In function ‘pseudoPinsSetup’:
pseudoPins.c:89:17: warning: cast from pointer to integer of different size [-Wpointer-to-int-cast]
   node->data0 = (unsigned int)ptr ;
                 ^
[Compile] wpiExtensions.c
drcNet.c: In function ‘authenticate’:
drcNet.c:112:24: warning: ‘%s’ directive writing up to 1023 bytes into a region of size 1021 [-Wformat-overflow=]
drcNet.c:89:14:
       return &buf [10] ;
              ~~~~~~~~~
drcNet.c:112:24:
   sprintf (salted, "$6$%s$", challenge) ;
                        ^~
drcNet.c:112:3: note: ‘sprintf’ output between 5 and 1028 bytes into a destination of size 1024
   sprintf (salted, "$6$%s$", challenge) ;
   ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
[Link (Dynamic)]
```
Otherwise `gpio readall` works as expected.

MichaIng, 2021-05-16:
Great, many thanks. The warnings could be reported to https://github.com/WiringPi/WiringPi/issues, both files have not been changed since that fork was created, so they seem to be very old.

MichaIng, 2021-05-18:
Marking this as closed in favour of #4149.

closed 2021-05-18 by MichaIng