MichaIng/DietPi #4383: Having problems after uninstalling XRDP

Flaam, 2021-05-16:
### Creating a bug report/issue

#### Required Information
- DietPi version | 7
- Distro version | buster
- Kernel version | Linux Server 5.10.17-v7l+ #1414 SMP Fri Apr 30 13:20:47 BST 2021 armv7l GNU/Linux
- SBC model | RPi 4 Model B (armv7l)
- Power supply used | Included in a case
- SDcard used | Using an external hdd as main drive

#### Additional Information (if applicable)
- Software title | XRDP / Every remote access software
- Was the software title installed freshly or updated/migrated? Uninstalled / Reinstalled
- Can this issue be replicated on a fresh installation of DietPi? I don't know

- Bug report ID | 2ff246f4-8c3f-4f75-bc9a-e7130eb083d0

#### Steps to reproduce

1. I uninstalled XRDP due to a problem I was having
2. When i tried to reinstall it, I started getting the error in the bug report
3. I tried uninstalling again but there wasn't the option
4. I tried to manually delete all XRDP related folders

#### Expected behaviour

- It should just reinstall fine

#### Actual behaviour

- I'm not able to reinstall X server related apps

Joulinar, 2021-05-16:
Hi,

Pls can share your error message?

Flaam, 2021-05-16:
Hello, here is the error!
[image]

Joulinar, 2021-05-16:
can you try following

```
G_AGP xrdp xorgxrdp
apt update
G_AGI xrdp xorgxrdp
```

Flaam, 2021-05-17:
Hello, thanks for the fix! Now it's working fine.

closed 2021-05-17 by Flaam