MichaIng/DietPi #4384: issue with the shairport - no sound

heapxor, 2021-05-17:
the latest dietpi version doesnt make any sound when using shairport-sync

last 3 comments here describe the issues.

https://github.com/mikebrady/shairport-sync

MichaIng, 2021-05-17:
Many thanks for your report.

> last 3 comments here describe the issues.

Please fix the link. Also a brief explanation here would not be bad, including which SBC/device is used etc 😉.

Joulinar, 2021-05-17:
I guess this is a related forum post https://dietpi.com/phpbb/viewtopic.php?f=11&t=9028

heapxor, 2021-05-17:
hello, 
what link do u mean? internet connection? SCB used is odroid xu4; https://www.amazon.com/Reiyin-WT-04-Bluetooth-Transmitter-Technology/dp/B077D4RKFD => bluetooth transmitter

MichaIng, 2021-05-17:
You said "last 3 comments here" above but gave a link to the repository root.

Okay there is a lot of multiple issues linked, with very different problems/questions, so I lost the overview of what your particular issue or question is now.

It seems you try to get it working with Bluetooth speakers? Does it work when playing an audio file from the DietPi system itself? Since setting up Bluetooth speakers with ALSA is not trivial: #3456

Joulinar, 2021-05-18:
@MichaIng 
The requestor posted some additional info on our forum. Probably you can have a look. I would recommend to continue the discussion on GitHub to avoid working on both Forum + GitHub.

heapxor, 2021-05-18:
@MichaIng  what i am trying to do is to stream (using roonlabs.com roon-bridge app) via airplay to dietpi and dietpi will forward it to the the bluetooth transmitter plugged into it ... so music will be playbacked on the bluetooth speaker (bose mini) connected to the dietpi via bluetooth transmitter. This doesnt work; sound is chopping or there is no sound.

When i tried more simple scenario to stream (using roonlabs.com) using roon-bridge to dietpi and let it forward sound bluetooth transmitter connected to bose bluetooth spekaer all works fine.

MichaIng, 2021-05-18:
Okay, so generally playing through the Bluetooth speakers works.

Is the regular USB disconnects in `dmesg` still a thing and does it also happen when going with the second scenario, skipping airplay?

And how did you set up the ALSA (if so) to use the Bluetooth transmitter/speakers?

heapxor, 2021-05-18:
Hi,
yes; you are correct, only once streaming via airplay doesnt work.

Yes it sometimes show up in dmesg. i dont know why / and how often, mostly if pc is idling.

I didnt do any setup of ALSA, I assume it happened somehow using roon-bridge.

MichaIng, 2021-05-18:
The USB disconnects is what I'd take care of first. That's on a level where audio is not involved at all but different players handle such disconnects differently. I saw some of the Shairport Sync errors were about device "in use"/"busy", probably Shairport Sync blocks on a disconnect while Roon alone releases immediately, or so.

Bad that the XU4 had a single USB port only. `dmesg` does not report any voltage/power issues, does it? Not sure if applicable, but aside of verifying that the BT transmitter sits tight and connector clean, probably you have another USB device that you can test on that port, just to see whether it get's disconnected regularly as well?

heapxor, 2021-05-18:
@michalng you are right, i also tried Banana Pi BPI-M1 and there are also disconnects. But at least on bpi-m1 i can get some sound via airplay - chopped one.  I just plugged usb BT to rpi4 to leave it on for 24h if there will be any disconnect in dmesg. 

What do you mean by single usb port only - it has 3 usb ports i think. No, no issue with voltage/power i have original adaptors - enought Ampers so by that way all is stable.

The device sits tight and all is clean; i can guarantee ... so maybe its issue of the vendor who created that BT - some incompatibility with drivers/kernel? Do I have to try other BT? I found https://avantree.com/dg60-bluetooth-usb-audio-transmitter#spec-tab and seller confirmed it can handle more then 44k sample rate and its compatible with linux.

Yes, so many "in use"/"busy" so i have no clue why? Disconnects happened lets say 4x in 24hours... so i dont believe the disconnects are connected with "in use"/"busy"

" Probably Roon Bridge accesses audio devices directly somehow. " -> i think so, or can get confirmed from roon folks.

Any idea what to do now?

heapxor, 2021-05-18:
@MichaIng so running rpi4 with ropiee 30min ... here u can see also disconnects

[    9.839397] bcmgenet fd580000.ethernet eth0: Link is Up - 1Gbps/Full - flow control off
[    9.839453] IPv6: ADDRCONF(NETDEV_CHANGE): eth0: link becomes ready
[   29.329144] sched: RT throttling activated
[  603.424978] usb 1-1.4: USB disconnect, device number 3
[  603.742443] usb 1-1.4: new full-speed USB device number 4 using xhci_hcd
[  603.886274] usb 1-1.4: New USB device found, idVendor=0a12, idProduct=1004, bcdDevice=36.06
[  603.886282] usb 1-1.4: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[  603.886290] usb 1-1.4: Product: REIYIN Bluetooth Audio WT04
[  603.886297] usb 1-1.4: SerialNumber: ABCDEF0123456789
[  603.967748] input: REIYIN Bluetooth Audio WT04 Consumer Control as /devices/platform/scb/fd500000.pcie/pci0000:00/0000:00:00.0/0000:01:00.0/usb1/1-1/1-1.4/1-1.4:1.3/0003:0A12:1004.0003/input/input2
[  604.032660] input: REIYIN Bluetooth Audio WT04 as /devices/platform/scb/fd500000.pcie/pci0000:00/0000:00:00.0/0000:01:00.0/usb1/1-1/1-1.4/1-1.4:1.3/0003:0A12:1004.0003/input/input3
[  604.032962] hid-generic 0003:0A12:1004.0003: input,hiddev96,hidraw0: USB HID v1.11 Device [REIYIN Bluetooth Audio WT04] on usb-0000:01:00.0-1.4/input3
[  604.040262] hid-generic 0003:0A12:1004.0004: hiddev97,hidraw1: USB HID v1.11 Device [REIYIN Bluetooth Audio WT04] on usb-0000:01:00.0-1.4/input4
[ 1205.373182] usb 1-1.4: USB disconnect, device number 4
[ 1205.825629] usb 1-1.4: new full-speed USB device number 5 using xhci_hcd
[ 1205.970782] usb 1-1.4: New USB device found, idVendor=0a12, idProduct=1004, bcdDevice=36.06
[ 1205.970790] usb 1-1.4: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[ 1205.970797] usb 1-1.4: Product: REIYIN Bluetooth Audio WT04
[ 1205.970804] usb 1-1.4: SerialNumber: ABCDEF0123456789
[ 1206.065249] input: REIYIN Bluetooth Audio WT04 Consumer Control as /devices/platform/scb/fd500000.pcie/pci0000:00/0000:00:00.0/0000:01:00.0/usb1/1-1/1-1.4/1-1.4:1.3/0003:0A12:1004.0005/input/input4
[ 1206.125843] input: REIYIN Bluetooth Audio WT04 as /devices/platform/scb/fd500000.pcie/pci0000:00/0000:00:00.0/0000:01:00.0/usb1/1-1/1-1.4/1-1.4:1.3/0003:0A12:1004.0005/input/input5
[ 1206.126130] hid-generic 0003:0A12:1004.0005: input,hiddev96,hidraw0: USB HID v1.11 Device [REIYIN Bluetooth Audio WT04] on usb-0000:01:00.0-1.4/input3
[ 1206.131516] hid-generic 0003:0A12:1004.0006: hiddev97,hidraw1: USB HID v1.11 Device [REIYIN Bluetooth Audio WT04] on usb-0000:01:00.0-1.4/input4
[ 1807.065202] usb 1-1.4: USB disconnect, device number 5
[ 1807.484074] usb 1-1.4: new full-speed USB device number 6 using xhci_hcd
[ 1807.627740] usb 1-1.4: New USB device found, idVendor=0a12, idProduct=1004, bcdDevice=36.06
[ 1807.627749] usb 1-1.4: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[ 1807.627756] usb 1-1.4: Product: REIYIN Bluetooth Audio WT04
[ 1807.627764] usb 1-1.4: SerialNumber: ABCDEF0123456789
[ 1807.711959] input: REIYIN Bluetooth Audio WT04 Consumer Control as /devices/platform/scb/fd500000.pcie/pci0000:00/0000:00:00.0/0000:01:00.0/usb1/1-1/1-1.4/1-1.4:1.3/0003:0A12:1004.0007/input/input6
[ 1807.774263] input: REIYIN Bluetooth Audio WT04 as /devices/platform/scb/fd500000.pcie/pci0000:00/0000:00:00.0/0000:01:00.0/usb1/1-1/1-1.4/1-1.4:1.3/0003:0A12:1004.0007/input/input7
[ 1807.774552] hid-generic 0003:0A12:1004.0007: input,hiddev96,hidraw0: USB HID v1.11 Device [REIYIN Bluetooth Audio WT04] on usb-0000:01:00.0-1.4/input3
[ 1807.780254] hid-generic 0003:0A12:1004.0008: hiddev97,hidraw1: USB HID v1.11 Device [REIYIN Bluetooth Audio WT04] on usb-0000:01:00.0-1.4/input4

heapxor, 2021-05-18:
these resets are increassing ... since the previous post its already on number

[24363.604074] usb 1-1.2: USB disconnect, device number 43
[24364.053061] usb 1-1.2: new full-speed USB device number 44 using xhci_hcd

MichaIng, 2021-05-19:
So it seems to be an issue with the BT transmitter itself then. Interesting is that the disconnects are quite precisely every 10 minutes (600 seconds), with a little shift that fits to the ~0.5 seconds disconnect time and ~1 second reconnect/setup time. Is it the same on the BPi?

heapxor, 2021-05-19:
hi,
it looks like its neverending even on bpi

[75223.758384] generic-usb 0003:0A12:1004.00FC: hiddev0,hidraw1: USB HID v1.11 Device [REIYIN Bluetooth Audio WT04] on usb-sw-ohci-1/input4
[75824.316641] usb 2-1: USB disconnect, device number 127
[75824.346878] ehci_irq: port change detect
[75824.795330] ehci_irq: port change detect
[75824.855367] The port change to OHCI now!
[75825.158279] usb 2-1: new full-speed USB device number 2 using sw-ohci
[75825.372311] usb 2-1: New USB device found, idVendor=0a12, idProduct=1004
[75825.392423] usb 2-1: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[75825.411146] usb 2-1: Product: REIYIN Bluetooth Audio WT04
[75825.427580] usb 2-1: SerialNumber: ABCDEF0123456789
[75825.585980] input: REIYIN Bluetooth Audio WT04 as /devices/platform/sw-ohci.1/usb2/2-1/2-1:1.3/input/input128
[75825.619420] generic-usb 0003:0A12:1004.00FD: input,hiddev0,hidraw0: USB HID v1.11 Device [REIYIN Bluetooth Audio WT04] on usb-sw-ohci-1/input3
[75825.663320] generic-usb 0003:0A12:1004.00FE: hiddev0,hidraw1: USB HID v1.11 Device [REIYIN Bluetooth Audio WT04] on usb-sw-ohci-1/input4
[76426.201288] usb 2-1: USB disconnect, device number 2
[76426.231625] ehci_irq: port change detect
[76426.671534] ehci_irq: port change detect
[76426.731583] The port change to OHCI now!
[76427.034385] usb 2-1: new full-speed USB device number 3 using sw-ohci
[76427.248320] usb 2-1: New USB device found, idVendor=0a12, idProduct=1004
[76427.284734] usb 2-1: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[76427.310353] usb 2-1: Product: REIYIN Bluetooth Audio WT04
[76427.327727] usb 2-1: SerialNumber: ABCDEF0123456789
[76427.498881] input: REIYIN Bluetooth Audio WT04 as /devices/platform/sw-ohci.1/usb2/2-1/2-1:1.3/input/input129
[76427.551923] generic-usb 0003:0A12:1004.00FF: input,hiddev0,hidraw0: USB HID v1.11 Device [REIYIN Bluetooth Audio WT04] on usb-sw-ohci-1/input3
[76427.612337] generic-usb 0003:0A12:1004.0100: hiddev0,hidraw1: USB HID v1.11 Device [REIYIN Bluetooth Audio WT04] on usb-sw-ohci-1/input4
[77028.085923] usb 2-1: USB disconnect, device number 3
[77028.116280] ehci_irq: port change detect
[77028.537520] ehci_irq: port change detect
[77028.598031] The port change to OHCI now!
[77028.900346] usb 2-1: new full-speed USB device number 4 using sw-ohci
[77029.115331] usb 2-1: New USB device found, idVendor=0a12, idProduct=1004
[77029.135697] usb 2-1: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[77029.154608] usb 2-1: Product: REIYIN Bluetooth Audio WT04
[77029.171104] usb 2-1: SerialNumber: ABCDEF0123456789
[77029.331694] input: REIYIN Bluetooth Audio WT04 as /devices/platform/sw-ohci.1/usb2/2-1/2-1:1.3/input/input130
[77029.369079] generic-usb 0003:0A12:1004.0101: input,hiddev0,hidraw0: USB HID v1.11 Device [REIYIN Bluetooth Audio WT04] on usb-sw-ohci-1/input3
[77029.422255] generic-usb 0003:0A12:1004.0102: hiddev0,hidraw1: USB HID v1.11 Device [REIYIN Bluetooth Audio WT04] on usb-sw-ohci-1/input4
[77629.970599] usb 2-1: USB disconnect, device number 4
[77630.000908] ehci_irq: port change detect
[77630.443422] ehci_irq: port change detect
[77630.503466] The port change to OHCI now!
[77630.806245] usb 2-1: new full-speed USB device number 5 using sw-ohci
[77631.020309] usb 2-1: New USB device found, idVendor=0a12, idProduct=1004
[77631.040640] usb 2-1: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[77631.059571] usb 2-1: Product: REIYIN Bluetooth Audio WT04
[77631.076153] usb 2-1: SerialNumber: ABCDEF0123456789
[77631.238660] input: REIYIN Bluetooth Audio WT04 as /devices/platform/sw-ohci.1/usb2/2-1/2-1:1.3/input/input131
[77631.273321] generic-usb 0003:0A12:1004.0103: input,hiddev0,hidraw0: USB HID v1.11 Device [REIYIN Bluetooth Audio WT04] on usb-sw-ohci-1/input3
[77631.320440] generic-usb 0003:0A12:1004.0104: hiddev0,hidraw1: USB HID v1.11 Device [REIYIN Bluetooth Audio WT04] on usb-sw-ohci-1/input4
[78231.855227] usb 2-1: USB disconnect, device number 5
[78231.885602] ehci_irq: port change detect
[78232.309639] ehci_irq: port change detect
[78232.369668] The port change to OHCI now!
[78232.672428] usb 2-1: new full-speed USB device number 6 using sw-ohci
[78232.886309] usb 2-1: New USB device found, idVendor=0a12, idProduct=1004
[78232.906736] usb 2-1: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[78232.927364] usb 2-1: Product: REIYIN Bluetooth Audio WT04
[78232.931319] usb 2-1: SerialNumber: ABCDEF0123456789
[78233.076838] input: REIYIN Bluetooth Audio WT04 as /devices/platform/sw-ohci.1/usb2/2-1/2-1:1.3/input/input132
[78233.106819] generic-usb 0003:0A12:1004.0105: input,hiddev0,hidraw0: USB HID v1.11 Device [REIYIN Bluetooth Audio WT04] on usb-sw-ohci-1/input3
[78233.150335] generic-usb 0003:0A12:1004.0106: hiddev0,hidraw1: USB HID v1.11 Device [REIYIN Bluetooth Audio WT04] on usb-sw-ohci-1/input4
[78833.739934] usb 2-1: USB disconnect, device number 6
[78833.770336] ehci_irq: port change detect
[78834.195830] ehci_irq: port change detect
[78834.255884] The port change to OHCI now!
[78834.558644] usb 2-1: new full-speed USB device number 7 using sw-ohci
[78834.772312] usb 2-1: New USB device found, idVendor=0a12, idProduct=1004
[78834.792469] usb 2-1: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[78834.811197] usb 2-1: Product: REIYIN Bluetooth Audio WT04
[78834.827639] usb 2-1: SerialNumber: ABCDEF0123456789
[78834.985535] input: REIYIN Bluetooth Audio WT04 as /devices/platform/sw-ohci.1/usb2/2-1/2-1:1.3/input/input133
[78835.021319] generic-usb 0003:0A12:1004.0107: input,hiddev0,hidraw0: USB HID v1.11 Device [REIYIN Bluetooth Audio WT04] on usb-sw-ohci-1/input3
[78835.074215] generic-usb 0003:0A12:1004.0108: hiddev0,hidraw1: USB HID v1.11 Device [REIYIN Bluetooth Audio WT04] on usb-sw-ohci-1/input4

MichaIng, 2021-05-20:
Even if those disconnects were not related to the choppy audio, it would break playback every 10 minutes for 1-2 seconds, which isn't great either. So before assuming a general issue with Shairport Sync, I'd definitely test a new BT transmitter first, that has no regular disconnects.

But I'll also rebuild our Shairport Sync packages as a new version is available: https://github.com/mikebrady/shairport-sync/releases/tag/3.3.8

heapxor, 2021-05-22:
@MichaIng do i have to compile it?

MichaIng, 2021-05-27:
New packages ready. Do you run the 32-bit or the 64-bit image?

For 32-bit (ARMv7/armhf):
```
cd /tmp
curl -sSfLO https://dietpi.com/downloads/binaries/buster/shairport-sync_armv7l.deb
apt install ./shairport-sync_armv7l.deb
rm shairport-sync_armv7l.deb
```

For 64-bit (ARMv8/arm64):
```
cd /tmp
curl -sSfLO https://dietpi.com/downloads/binaries/buster/shairport-sync_aarch64.deb
apt install ./shairport-sync_aarch64.deb
rm shairport-sync_aarch64.deb
```

heapxor, 2021-06-01:
@MichaIng  not working at all; no sound. Even i bought completely new BT avantree dg60.

I can get chopping sound on ropieee or banana but no sound from dietpi. 

How can i see / access logs from shairport running on dietpi?

heapxor, 2021-06-01:
root@DietPi:/tmp# shairport-sync -d "hw:DG60" -vvu
         0.000313542 "shairport.c:461" looking for configuration file at full path "/usr/local/etc/shairport-sync.conf"
shairport-sync was built without libdaemon, so does not support daemonisation using the -d, --daemon, -j or --justDaemoniseNoPIDFile options
         0.000685750 "shairport.c:1327" exit function called...
         0.000123083 "dbus-service.c:1089" stopping dbus service
         0.000023417 "dbus-service.c:1093" Zero OwnerID for "org.gnome.ShairportSync".
         0.000015458 "shairport.c:1355" Stopping DACP Monitor
         0.000027833 "shairport.c:1360" Stopping metadata hub
         0.000023917 "shairport.c:1421" exit
root@DietPi:/tmp# shairport-sync  "hw:DG60" -vvu
         0.000197708 "shairport.c:461" looking for configuration file at full path "/usr/local/etc/shairport-sync.conf"
         0.000790833 "shairport.c:1729" Started!
         0.000048417 "shairport.c:1756" software version: "3.3.8-OpenSSL-Avahi-ALSA-soxr-metadata-mqtt-dbus-mpris-sysconfdir:/usr/local/etc"
         0.000023625 "shairport.c:1762" log verbosity is 2.
         0.000124250 "audio_alsa.c:998" alsa: init() -- alsa_backend_state => abm_disconnected.
         0.000019750 "audio_alsa.c:1015" alsa: alsa_maximum_stall_time of 0.200000 sec.
         0.000039500 "audio_alsa.c:1302" alsa: disable_standby_mode is "never".
         0.000017833 "audio_alsa.c:1304" alsa: disable_standby_mode_silence_threshold is 0.040000 seconds.
         0.000020834 "audio_alsa.c:1306" alsa: disable_standby_mode_silence_scan_interval is 0.004000 seconds.
         0.000023750 "audio_alsa.c:1345" alsa: output device name is "default".
         0.000149875 "shairport.c:1777" The processor is running little-endian.
         0.000026791 "shairport.c:1810" disable resend requests is off.
         0.000036167 "shairport.c:1814" diagnostic_drop_packet_fraction is 0.000000. A value of 0.0 means no packets will be dropped deliberately.
         0.000003417 "audio_alsa.c:1968" keep_dac_busy is now "no"
         0.000035750 "shairport.c:1815" statistics_requester status is 0.
         0.000066750 "shairport.c:1820" rtsp listening port is 5000.
         0.000019625 "shairport.c:1821" udp base port is 6001.
         0.000016958 "shairport.c:1822" udp port range is 10.
         0.000019750 "shairport.c:1823" player name is "DietPi".
         0.000021125 "shairport.c:1824" backend is "alsa".
         0.000018125 "shairport.c:1825" run_this_before_play_begins action is "(null)".
         0.000018125 "shairport.c:1826" run_this_after_play_ends action is "(null)".
         0.000016708 "shairport.c:1827" wait-cmd status is 0.
         0.000020000 "shairport.c:1828" run_this_before_play_begins may return output is 0.
         0.000018875 "shairport.c:1829" run_this_if_an_unfixable_error_is_detected action is "(null)".
         0.000018209 "shairport.c:1830" run_this_before_entering_active_state action is  "(null)".
         0.000017916 "shairport.c:1831" run_this_after_exiting_active_state action is  "(null)".
         0.000019584 "shairport.c:1832" active_state_timeout is  10.000000 seconds.
         0.000020166 "shairport.c:1833" mdns backend "(null)".
         0.000016750 "shairport.c:1834" userSuppliedLatency is 0.
         0.000018417 "shairport.c:1837" interpolation setting is "auto".
         0.000016875 "shairport.c:1838" interpolation soxr_delay_threshold is 30.
         0.000016833 "shairport.c:1839" resync time is 0.050000 seconds.
         0.000017584 "shairport.c:1840" allow a session to be interrupted: 0.
         0.000016916 "shairport.c:1841" busy timeout time is 120.
         0.000017042 "shairport.c:1842" drift tolerance is 0.001995 seconds.
         0.000017542 "shairport.c:1843" password is "(null)".
         0.000016708 "shairport.c:1844" ignore_volume_control is 0.
         0.000016917 "shairport.c:1848" volume_max_db is not set
         0.000016833 "shairport.c:1850" volume range in dB (zero means use the range specified by the mixer): 0.
         0.000017750 "shairport.c:1854" volume_range_combined_hardware_priority (1 means hardware mixer attenuation is used first) is 0.
         0.000018583 "shairport.c:1856" playback_mode is 0 (0-stereo, 1-mono, 1-reverse_stereo, 2-both_left, 3-both_right).
         0.000019417 "shairport.c:1857" disable_synchronization is 0.
         0.000017125 "shairport.c:1858" use_mmap_if_available is 1.
         0.000017042 "shairport.c:1860" output_format automatic selection is enabled.
         0.000016875 "shairport.c:1864" output_rate automatic selection is enabled.
         0.000020708 "shairport.c:1868" audio backend desired buffer length is 0.200000 seconds.
         0.000018583 "shairport.c:1870" audio_backend_buffer_interpolation_threshold_in_seconds is 0.120000 seconds.
         0.000017917 "shairport.c:1871" audio backend latency offset is 0.000000 seconds.
         0.000019625 "shairport.c:1873" audio backend silence lead-in time is "auto".
         0.000018208 "shairport.c:1877" zeroconf regtype is "_raop._tcp".
         0.000017167 "shairport.c:1878" decoders_supported field is 1.
         0.000018500 "shairport.c:1879" use_apple_decoder is 0.
         0.000016375 "shairport.c:1880" alsa_use_hardware_mute is 0.
         0.000018958 "shairport.c:1884" no special mdns service interface was requested.
[... cut ...]
         0.002488834 "mpris-service.c:167" Build metadata
         0.002139041 "audio_alsa.c:1779" alsa: do_close() -- closing alsa handle
         0.000306709 "audio_alsa.c:1863" alsa: flush() -- alsa_backend_state => abm_disconnected.
         0.000135875 "player.c:945" flush request: flush output device.
         0.000092791 "player.c:997" flush request: flush frame 1988425493 pending -- buffer contains 85536 frames, from 1988339957 to 1988425492
         0.000099334 "player.c:1020" flush request: flush done.
^C
root@DietPi:/tmp#

MichaIng, 2021-06-01:
Logs can be reviewed via: `journalctl -u shairport-sync`

But there are no regular disconnects anymore with the new Bluetooth adapter? Does it work when playing a local audio file, e.g. a wav file via `aplay some.wav` or MP3 via `mpg123 some.mpg`. Ah or `speaker-test` to run the general ALSA speaker test.

heapxor, 2021-06-01:
everything plays fine while using roon-bridge  ie streaming music from roon to roon-bridge installed on dietpi and from BT transmitted to bose mini speaker. When I stream to airplay even from roon or directly from my iphone no sound at all. :(

No more disconnect in dmesg.

heapxor, 2021-06-01:
running on banana pi i have at least some sound; chopping but sound ... 
         0.431184292 "player.c:2364" Large negative sync error: -2323 with shoul                                                                                            d_be_frame_32 of 899954347, nt of 899874954 and current_delay of 6484.

but here nothing ;(

heapxor, 2021-06-01:
root@DietPi:/tmp# journalctl -u shairport-sync
-- Logs begin at Mon 2021-05-10 13:28:28 BST, end at Tue 2021-06-01 19:40:10 BST. --
Jun 01 19:03:04 DietPi systemd[1]: Started Shairport Sync - AirPlay Audio Receiver.
Jun 01 19:03:04 DietPi shairport-sync[5629]: warning: could not acquire a Shairport Sync native D-Bus interface "org.gnome.ShairportSync.i5629" on t
he system bus.
Jun 01 19:03:04 DietPi shairport-sync[5629]: warning: could not acquire an MPRIS interface named "org.mpris.MediaPlayer2.ShairportSync.i5629" on the
 system bus.
Jun 01 19:08:10 DietPi systemd[1]: Stopping Shairport Sync - AirPlay Audio Receiver...
Jun 01 19:08:10 DietPi systemd[1]: shairport-sync.service: Main process exited, code=killed, status=15/TERM
Jun 01 19:08:10 DietPi systemd[1]: shairport-sync.service: Succeeded.
Jun 01 19:08:10 DietPi systemd[1]: Stopped Shairport Sync - AirPlay Audio Receiver.
Jun 01 19:33:27 DietPi systemd[1]: Started Shairport Sync - AirPlay Audio Receiver.
Jun 01 19:34:23 DietPi systemd[1]: Stopping Shairport Sync - AirPlay Audio Receiver...
Jun 01 19:34:23 DietPi systemd[1]: shairport-sync.service: Main process exited, code=killed, status=15/TERM
Jun 01 19:34:23 DietPi systemd[1]: shairport-sync.service: Succeeded.
Jun 01 19:34:23 DietPi systemd[1]: Stopped Shairport Sync - AirPlay Audio Receiver.
Jun 01 19:34:23 DietPi systemd[1]: Started Shairport Sync - AirPlay Audio Receiver.
root@DietPi:/tmp#

heapxor, 2021-06-01:
dunno how to run that test

root@DietPi:/tmp# speaker-test  -D "hw:DG60"

speaker-test 1.1.8

Playback device is hw:DG60
Stream parameters are 48000Hz, S16_LE, 1 channels
Using 16 octaves of pink noise
Channels count (1) not available for playbacks: Invalid argument
Setting of hwparams failed: Invalid argument

even BT is there 
aplay -L

hw:CARD=DG60,DEV=0
    Avantree DG60, USB Audio
    Direct hardware device without any conversions

MichaIng, 2021-06-01:
How did you actually configure ALSA? Did you use `dietpi-config`, or manually set it up? If so please share your `cat /etc/asound.conf`.

It seems you don't have the automatic conversion plugin enabled, which means best sound quality, but if the sound card/driver does not support the format of the audio source, it won't play something. If you used `dietpi-config` to setup ALSA, please try to enabled `Auto-conversion`.

heapxor, 2021-06-01:
i didnt configure it; so there is not etc/asound.conf.

heapxor, 2021-06-01:
i executed now dietpi-config for the conf of audio end enabled Auto-conversion.

root@DietPi:/tmp# cat /etc/asound.conf
pcm.!default {
        type hw
        card 1
        device 0
}

ctl.!default {
        type hw
        card 1
}

heapxor, 2021-06-01:
but still no sound from the speaker while connected via iphone to airplay device :(

heapxor, 2021-06-01:
hows! it works now from iphone wou

heapxor, 2021-06-01:
okay i can confirm that all works flawlessly! thank you so much

MichaIng, 2021-06-01:
Can you check `alsamixer` whether there are controls muted or on low level. If `speaker-test` works this shouldn't be the case, but just to rule it out. Also you should be able to determine the main volume control name, shown at the bottom of the mixer GUI. In `/usr/local/etc/shairport-sync.conf` you can then uncomment and set `mixer_control_name` to have this control name, so it can adjust volume natively via ALSA instead of doing it software-wise. The config file contains many other options to look through. `output_device` and `mixer_device` should be fine, when commented, as the default is "default", which is the correct name of both, as named in `asound.conf`, but it could be tested to set it explicitly.

... ah okay, what was the final step? 😃

heapxor, 2021-06-01:
i think that execution of > dietpi-config for the conf of audio end enabled Auto-conversion then restart of shairport helped.

heapxor, 2021-06-01:
it also works fine with previous BT.

The issue was missing / off 
Automatic software conversions

MichaIng, 2021-06-01:
Okay great. We should add a note to audio related software docs, that, if audio does not work, that software conversion plugin is always a good thing to test. Probably we should even enable it by default 🤔. If I understand it correctly, the conversion (and hence CPU usage + theoretical audio quality loss) is anyway only done when the card/driver does not support the format if the audio source, otherwise it is skipped.

heapxor, 2021-06-01:
and what exactly is that software conversion plugin? how can i install it on other distribution? thanks

heapxor, 2021-06-01:
seems like config param in alsa config file 
https://www.alsa-project.org/alsa-doc/alsa-lib/pcm_plugins.html

MichaIng, 2021-06-01:
I'm everything else than an audio expert, but that suites quite well some experience I had on Windows, when changing the sampling rate or bit rate in audio control panel, sometimes audio was still playing, but horribly distorted.

That plugin is natively integrated in ALSA, at least it should be, on Debian it is. Check out the `asound.conf`:
```
pcm.!default {
        type plug
        slave.pcm {
                type hw
                card 1
                device 0
        }
}
```
This means the PCM called "default" is of type "plug", which is that software conversion plugin: https://www.alsa-project.org/alsa-doc/alsa-lib/pcm_plugins.html#pcm_plugins_plug
It has the actual sound card driver (type "hw" with card and device index defined) as child/slave PCM, to pass the converted audio stream to. Without that plugin, the audio stream is directly passed to the driver:
```
pcm.!default {
        type hw
        card 1
        device 0
}
```
The link above lists many other plugins that can be used to do things to the audio stream, interesting also `dmix`, which mixes multiple input stream and hence allows to run audio from multiple players concurrently, which is not natively supported by the ALSA driver. Crazy is it when enabling the equalizer:
```
pcm.!default {
	type plug
	slave.pcm {
		type equal
		slave.pcm {
			type plug
			slave.pcm {
				type hw
				card 1
				device 0
			}
		}
		controls "/var/lib/dietpi/dietpi-config/.alsaequal.bin"
	}
}
```
"type equal" is the equalizer plugin (which needs to be installed, as it is not part of the core ALSA package). But it requires the audio stream as integer stream or so (some format that is never used in the audio source and nearly never supported by the sound card), so the auto conversion plugin is used first, which passes the converted stream to the equalizer, which again passes the back-converted stream to the final audio driver 😄. Awesome is that "auto"-conversion really means auto, so that plugin determines quite reliable which format it needs to convert the input stream into, to satisfy its child/slave PCM.

So much about ALSA configuration which I also learned quite recently 😄.

heapxor, 2021-06-01:
even i replicated /etc/asound.conf on bananian is still chopping ::( any idea where to take a look? thanks!

2 root@bananapi ~/shairport-sync (git)-[master] # cat /etc/asound.conf       :(
pcm.!default {
        type plug
        slave.pcm {
                type hw
                card 1 # for headphone, turn 1 to 0
                device 0
        }
}

ctl.!default {
        type hw
        card 1 # for headphone, turn 1 to 0
}

heapxor, 2021-06-01:
or maybe can you share me your shairport-sync source and what ./configuration parameters did u use to create package? so i can try to compile it on bananian? thanks

MichaIng, 2021-06-01:
I use:
```
./configure --with-alsa --with-avahi --with-ssl=openssl --with-soxr --with-metadata --with-systemd --with-dbus-interface --with-mpris-interface --with-mqtt-client
```
Build dependencies accordingly:
```
automake pkg-config make g++ libpopt-dev libconfig-dev libssl-dev libsoxr-dev libavahi-client-dev libasound2-dev libglib2.0-dev libmosquitto-dev
```
DBus, MPRIS and MQTT however should not play a role.

heapxor, 2021-06-01:
and from which source do you compile?

MichaIng, 2021-06-01:
From official Git [master](https://github.com/mikebrady/shairport-sync/tree/master) branch as well, which equals the latest [release](https://github.com/mikebrady/shairport-sync/releases/tag/3.3.8).

heapxor, 2021-06-01:
hm thats strange... so i did 
git clone https://github.com/mikebrady/shairport-sync.git
cd sh...
autoreconf -fi
 ./configure --with-alsa --with-avahi --with-ssl=openssl --with-soxr --with-metadata  --with-dbus-interface --with-mpris-interface --with-mqtt-client   --with-libdaemon

But the issue is that i am not getting any sound from speaker-test ;(

heapxor, 2021-06-01:
okay i make it works also on banana! pow! the mixer volume was completely off. 

thanks for all your help!

heapxor, 2021-06-01:
the only thing i do not understand is ... why there are constant disconnects of BT in dmesg... weird :(

MichaIng, 2021-06-01:
Probably it is simply a hardware issue of that adapter, a broken cable, loose connection internally, or so.

heapxor, 2021-06-01:
well its not a cable ;/ its something like usb stick; so its directly plugged into pc ;-(

MichaIng, 2021-06-01:
I mean a cable or connection or broken solder inside of the sticks case.

MichaIng, 2021-06-02:
Ah true, I forgot the 10 minutes interval. Yeah that is doesn't fit to an internal connection issue, but more a firmware or driver bug. If you have a chance, I'd contact the manufacturer about it, in case of warranty, making use of it. Especially for audio applications, it's a no-go to have regular connection losses, of course, even if it's just for a very short time.

heapxor, 2021-06-14:
hi @MichaIng, i reinstalled pc with armbian; and i am gettin lets say almost same issue as we started that thread... i am not getting disconnects in dmesg; but sound is chopping even via bluetooth or via airplay :( 

thats the shairport-sync log 
https://paste.debian.net/1201142/

alsa-info 
https://paste.debian.net/1201142/

The distro is based on the latest ubuntu; so thats why i am confused why its not working as expected ... as i replicated all the configuration, as we discussed here :(

if u mind to take a look...
thanks so much

MichaIng, 2021-06-15:
Please check your `/usr/local/etc/shairport-sync.conf`. Is `output_rate` set (uncommented)? Not sure if Shairport Sync "sees" that the sound card supports/wants 48000 Hz, but as this is (for whatever reason) not valid value for Shairport Sync, is reverts to auto, effectively 44100 Hz, which then fails. Indeed the "plug" plugin should exactly solve such incompatibilities between input format and sound card/driver. And is `output_device` set?

And can you check which sampling rate is supported by that sound card?

heapxor, 2021-06-15:
hi,
 i tried output_rate which is as u said not supported, i tried only auto, but card supports up to 48k.

no clue then.

MichaIng, 2021-06-15:
And these "Speed requested: 44100. Speed available: 48000" stay even with the plug auto conversion plugin enabled?

> 0.000517586 "audio_alsa.c:1345" alsa: output device name is "hw:WT04".

It seems not to be used, as `hw:WT04` is the final hardware PCM. Can you try to set `output_device = "default";` explicitly in `shairport-sync.conf`, as defined in your `asound.conf` [above](https://github.com/MichaIng/DietPi/issues/4384#issuecomment-852406458).

heapxor, 2021-06-17:
@MichaIng hi,
i installed armbian debian instead of armbian ubuntu and it works... seems some issue in ubuntu? 

I can try to do that and git it last shot as its crazily time consuming :( i will need to restore ubuntu edit conf as u suggested.
So using the plug hw:WT04 should not be used as the last device? or what will be achieved using output_device = "default"? thanks!

MichaIng, 2021-06-17:
If this is your `/etc/asound.conf`
```
pcm.!default {
        type plug
        slave.pcm {
                type hw
                card 1
                device 0
        }
}
ctl.!default {
        type hw
        card 1
}
```
then `output_device = "default";` will force Shairport Sync to use the intended parent PCM, which we actively called "default" with this `!default` syntax, more precisely this syntax overrides any possible existing "default" PCM. It is `type plug`, hence does the auto-conversion, when required, and passes the converted stream to it's slave PCM, which is the final `type hw` audio driver. Your logs seem to indicated that `hw:WT04` was used as device name by Shairport Sync, which would be a `type hw` driver already, without any parent conversion plugin. `plughw:WT04` would be a pre-defined PC name/shortcut, which is a parent plug PCM with a slave hw PCM. What I'm not sure about it the `WT04` name. This needs to be defined somewhere in the ALSA config to be effective. Usually I'd expect `hw:1,0`, `plughw:1,0` or similar, are known pre-defined PCM names with a final card "1" + device "0", hence would match the manual declaration above.

heapxor, 2021-06-17:
Hi  @MichaIng 
you were exactly right
i was set as 

alsa =
{

  output_device = "hw:WT04";

When i changed shairport-sync conf to 

alsa =
{

  output_device = "default";

All started to work flawlessly! 

Great, so much appreciate all your help!

heapxor, 2021-06-17:
But on ubuntu there has to be one more problem... error... 
i do use roonbridge; and shairport-sync... shairport-sync works OK now, but if i stop its playback... the issue is that when i execute playback via roonbridge the chooping is coming from roon bridge. When I dont do playback via shairport-sync before doing playback viac roonbridge; all the sound produced via roonbridge works fine.

MichaIng, 2021-06-18:
Generally, running multiple players does not work OOTB. I'm not sure how fast/consequently audio devices are released when you stop playback in one player.

It could be tested if the dmix plugin makes a difference, which allows to mix multiple playback streams:
```
pcm.!default {
        type plug
        slave.pcm "dmix:1,0"
}
ctl.!default {
        type hw
        card 1
}
```

heapxor, 2021-06-19:
maybe almost same behavior? but it looks like if i do playback via roonbridge or shairport; if i stop one of these and way lets say 5 sec ; start the other one... all seems to be fine.

MichaIng, 2021-06-20:
The dmix plugin should actually solve such transition issues, although the kind where one player fails with a "device is busy"-like error as long as the other didn't release it. With the dmix plugin you should be possible to play audio streams via Stairport Sync and Roon Bridge concurrently and they should be mixed together.

[3 more replies]

closed 2021-11-11 by MichaIng