MichaIng/DietPi #4385: Update from DietPi version 6.2 to 6.35 generate issue

orotarobas, 2021-05-17:
### Creating a bug report/issue

#### Required Information

I'm using NanPi Neo Core for our industrial embedded application, I can succesfuly install the dietpi image provided on the manufacturer repo here https://drive.google.com/file/d/1FTJpsptgWkUFHprD1RLUBXOSCXyVaLWH/view?usp=sharing.

If I perform the installaton without any network connection, and the system boots properly. 
Since at boot the tool dietpi-software gets automatically invoked, if i connect the network interface the dietpi gets updated to the newer version 6.35, but after that after login i start getting this message:

"DietPi-Login | Waiting for DietPi-RAMDISK to finish mounting DietPi to RAM..." 

this keeps going in an infinite loop till i press control-C.

Since the application needs to be autolaunched after board boot this behaviour is not acceptable, so i would like to know how to prevent dietpi-software to be automatically invoked. 

I tried also to use the new dietpi version built for Nanopi Neo, but that version has issues with audio drivers that the 6.2/6.35 doesn't have.

Joulinar, 2021-05-17:
Hi,

by default DietPi will perform an update to latest version always. For this a valid internet connection is required on first boot to complete first initial setup. Once initial setup completes, the SBC can be operated without network connection as well. 

But if I understood correctly,. initial setup did not complete. Correct?

MichaIng, 2021-05-17:
Many thanks for your report. Is there a reboot done after/during the automated update?

orotarobas, 2021-05-17:
Initial Setup without network connection completes, softwar update do 
not complete and i can operate with the board without any further issue.

After reboot automatic software updates get launched again so i need to 
skip it every time and this is not feasible for our applicaiton.

If i let the automatic update to run it gets completed successfuly and 
dietpi version gets upgraded from 6.2 to 6.35 but this

DietPi-Login | Waiting for DietPi-RAMDISK to finish mounting DietPi to 
RAM...

error gets generated after login.

I tested this installation serveral times, and it's fully repetitive.
The issue gets generated either i let the board update immediately at 
first installation time or if i let the update run after first 
installation without network.

Bests

------ Messaggio originale ------

generate issue (#4385)

Joulinar, 2021-05-17:
You need a valid internet connection to complete initial setup, otherwise it will re-run on every reboot trying to complete the setup

orotarobas, 2021-05-17:
yes, that's clear

the problem is that once i provide the internet connection and the 
initial setup is completed then

DietPi-Login | Waiting for DietPi-RAMDISK to finish mounting DietPi to 
RAM...

gets generated.

So the question is, how to perform initial update without having this 
issue generated??

------ Messaggio originale ------

generate issue (#4385)

MichaIng, 2021-05-17:
The image is ancient, so not sure whether the update path is still intact. Let me explain what "should" happen:
- On first login, `DietPi-Software` should start and in case after a few initial prompts about password and serial console, `DietPi-Update` should be invoked.
- The update should update the DietPi scripts/core files, then "incremental patches" should start, DietPi-RAMdisk should be disabled as first step, then you should see a dialogue saying "The system will now reboot to finish this update.".
- After the reboot and login, DietPi-Update should run again and this time finish the update to DietPi v7.1.2.

In your case it didn't seem to happen like that, so please tell us how if differs, at best with the related console output, copy&pasted here, or as screenshot, or the log file: `/var/tmp/dietpi/logs/dietpi-firstrun-setup.log`

MichaIng, 2021-05-17:
Actually, your NanoPi NEO image should support the NEO Core as well. Not 100% sure if the internal eMMC as system partition works, but worth to give it a try, have much newer Linux and Debian version and avoid the ancient update path: https://dietpi.com/downloads/images/DietPi_NanoPiNEO-ARMv7-Buster.7z

orotarobas, 2021-05-17:
As i stated in first email i already try that

the problem with that image is that not sound card are detected, even if 
i configured by dietpi-config tool.

So which is better to debug audio driver not working or old version 
update not working??

------ Messaggio originale ------

generate issue (#4385)

MichaIng, 2021-05-17:
I'd go with the new image and debugging the sound card issue. Does it have an onboard sound chip or is it an attached USB DAC? I see it has audio I/O pins, but not sure whether this means onboard chip or not.

The other issue with the ancient image is that it's Debian Stretch, where we drop support for this year.

orotarobas, 2021-05-17:
Allwinner H3 has integrated audio codec, we need to use the integrated 
mic input.

what log files do you need to check it out?

Bests

------ Messaggio originale ------

generate issue (#4385)

ravenclaw900, 2021-05-17:
Could you post the logfile at `/var/tmp/dietpi/logs/dietpi-firstrun-setup.log`?

orotarobas, 2021-05-17:
There you go.

Bests

------ Messaggio originale ------

generate issue (#4385)

Joulinar, 2021-05-17:
Pls can you attach the log on GitHub directly as it is missing if you send it via email. Thx

orotarobas, 2021-05-17:
[dietpi-firstrun-setup.log](https://github.com/MichaIng/DietPi/files/6495823/dietpi-firstrun-setup.log)

Hope this works

MichaIng, 2021-05-17:
Okay and when you login now, you still see the "DietPi-Login | Waiting for DietPi-RAMDISK to finish mounting DietPi to RAM..." message?
Because the log file indicates that the code is up-to-date, which means that there should be no check the RAMdisk anymore 🤔.

orotarobas, 2021-05-17:
No, this version of OS has troubles with the audio driver do not detect any
audio peripheral.

The problem with RAM disk arise with another version.

But previously you suggested to check the audio driver bug instead of RAM
bug.

Do you want me to install the old version that generates the RAM issue??

MichaIng, 2021-05-17:
Ah okay, no all fine then, lets check the audio issue. As it's the H3 onboard chip, the kernel should contain the driver. Can you please the output of:
```
aplay -l
dmesg | grep -E '(audio|snd|sound)'
```

MichaIng, 2021-05-18:
Fixed the issue with old images. Basically all DietPi versions prior to v6.17 had v7 applied as core version so that no new update was found on restart: https://github.com/MichaIng/DietPi/commit/b1de551a356943a83a3470e94367cfcb5149327c

orotarobas, 2021-05-18:
Here is consolle output:

dmesg | grep -E '(audio|snd|sound)'
[    0.000000] Kernel command line: 
root=UUID=b57cc6e3-8ac3-425b-acd0-eb5dec85d0f3 rootwait rootfstype=ext4 
console=ttyS0,115200 hdmi.audio=EDID:0 
disp.screen0_output_mode=1920x1080p60 consoleblank=0 loglevel=4 
ubootpart=2d99deeb-01 ubootsource=mmc 
usb-storage.quirks=0x2537:0x1066:u,0x2537:0x1068:u   
sunxi_ve_mem_reserve=0 sunxi_g2d_mem_reserve=0 sunxi_fb_mem_reserve=16
[    3.658155]   No soundcards found.

aplay -l
aplay: device_list:272: no soundcards found...

------ Messaggio originale ------

generate issue (#4385)

orotarobas, 2021-05-18:
okay i'll try to install the old version then to check if everything is 
smooth.

------ Messaggio originale ------

generate issue (#4385)

MichaIng, 2021-05-18:
> okay i'll try to install the old version then to check if everything is
smooth.

No wait this won't work yet. It will work when v7.2 has been released.

orotarobas, 2021-05-18:
okay, waiting for your good to go then.

I posted the log output of sound issue with newer image meanwhile.

------ Messaggio originale ------

generate issue (#4385)

MichaIng, 2021-05-18:
Okay, please edit `/boot/armbianEnv.txt` and append ` analog-codec` to the `overlays=` line, then reboot. Onboard audio seems to be disabled by default on H3 boards with this kernel/device tree setup, the related device tree overlay enables it.

You find all device tree overlays in `/boot/dtb/overlay/`, the ones with `sun8i-h3-` prefix, enabling further features. Add the name without prefix and file extension to the `/boot/armbianEnv.txt` overlays line, so e.g. "uart1" to enable the 1st UART device for serial console or such.

You seem to know the board already, but for others, here is the GPIO/pin layout: http://wiki.friendlyarm.com/wiki/index.php/NanoPi_NEO_Core#Layout

orotarobas, 2021-05-18:
Did the modifications and i can confirm everything works smooth.

i have my last three questions then we're all set

I need to setup the board so our application gets launched automatically 
at board boot, reading the docs i tried a couple of config but something 
odd:

1) if i select option 14 i setup the script in 
/var/lib/dietpi/dietpi-autostart/custom.sh, it works properly but only 
after login (which i have to perform manually)
2) I run dietpi-config and selected option 7 but it still requires me to 
enter login credentials after boot and do not perform the script i 
configured at point 1

Please let me know how to sort this out

Last question is about mac address:
if i perform cat /sys/class/net/eth0/address i notice mac address 
changes after every reboot.
I was assuming it shoud be fixed and unique by hardware transceiver, can 
you explain why?

Bests

------ Messaggio originale ------

generate issue (#4385)

MichaIng, 2021-05-19:
`/var/lib/dietpi/dietpi-autostart/custom.sh` should actually run interdependently of the login. It might be a coincidence that it ran afterwards, but when you wait a little longer it should run by itself. It is an own systemd service (`systemctl status dietpi-autostart_custom`) that runs independently from the login.

Option 7 still requires you to manually login? Or do you speak of SSH? Note that this means login at local terminal console TTY1, so the one you reach when connecting a screen and keyboard. SSH autologin would need to be configured with the SSH client, the server has no charge on this. When choosing option 7, away from option 14, the custom script service is removed, so that is expected, but as said, autologin is not required for it to start.

If the task you need to run is not a long-term background job, you can have it executed as well by adding a script to `/var/lib/dietpi/postboot.d`. This can be also the custom.sh you used before:
```
mv /var/lib/dietpi/dietpi-autostart/custom.sh /var/lib/dietpi/postboot.d/custom.sh
```

The changing MAC address is indeed something common on those Allwinner SoCs: #3618
First please try to set it via boot config by adding the following two lines to `/boot/armbianEnv.txt`:
```
ethaddr <mac_address>
eth1addr <mac_address>
```
Replacing `<mac_address>` with the actual one you want to have, at best the currently used one so that it's assured to be a valid mac. Then reboot.

If those have no effect, then there is a second solution:
```
echo 'SUBSYSTEM=="net", KERNEL=="eth0", ACTION=="add", RUN+="ip l set dev eth0 address <mac_address>"' > /etc/udev/rules.d/99-eth0-mac.rules
```
Same here, replacing `<mac_address>` with the actual one you want to have.

orotarobas, 2021-05-22:
Sorry for my late answer

1) i can confirm /var/lib/dietpi/dietpi-autostart/custom.sh works 
autonomously, it was just a matter of initial delay
2)i'm using debug terminal consolle on TTYS0, the default one, and 
option 7 doesn't work, do not allow me to autologin
3)i checked mac setup as you suggested and i can confirm it works

so 1+3 solves our need, thanks for your precious support we can move on 
to production with our board NanoPI Dietpi powered.

Bests

------ Messaggio originale ------

generate issue (#4385)

MichaIng, 2021-05-22:
> 2)i'm using debug terminal consolle on TTYS0, the default one, and
option 7 doesn't work, do not allow me to autologin

Okay that makes sense. The console autologin sets this up on TTY1 only, not TTYS0. But you could manually setup autologin there:
```
mkdir -p /etc/systemd/system/getty@ttyS0.service.d
cat << '_EOF_' > /etc/systemd/system/getty@ttyS0.service.d/autologin.conf
[Service]
ExecStart=
ExecStart=-/sbin/agetty -a root -J -s %I 115200,57600,38400,9600 $TERM
_EOF_
```
But this has no effect on any `dietpi-autostart` option.

> 3)i checked mac setup as you suggested and i can confirm it works

Did editing `/boot/armbianEnv.txt` alone fix the MAC address already or was the udev rule required?

MichaIng, 2021-06-02:
The upgrade issue for old images has been solved. I'm marking this hence as closed. Feel free to reopen if required.

orotarobas, 2021-06-11:
Hello

i tried again the installation beginning from old image but it still 
fails.

------ Messaggio originale ------

generate issue (#4385)

MichaIng, 2021-06-11:
Can you share the output of the install process up to where it failed?

orotarobas, 2021-06-29:
Here are all the logs files.

[dietpi-firstrun-setup.log](https://github.com/MichaIng/DietPi/files/6734038/dietpi-firstrun-setup.log)
[dietpi-ramdisk.log](https://github.com/MichaIng/DietPi/files/6734039/dietpi-ramdisk.log)
[dietpi-ramlog.log](https://github.com/MichaIng/DietPi/files/6734040/dietpi-ramlog.log)
[dietpi-software_apt.log](https://github.com/MichaIng/DietPi/files/6734041/dietpi-software_apt.log)
[dietpi-update.log](https://github.com/MichaIng/DietPi/files/6734042/dietpi-update.log)
[fs_expand.log](https://github.com/MichaIng/DietPi/files/6734043/fs_expand.log)
[fs_partition_resize.log](https://github.com/MichaIng/DietPi/files/6734044/fs_partition_resize.log)

MichaIng, 2021-06-30:
```
[ INFO ] DietPi-Update | Getting repository version: https://raw.githubusercontent.com/MichaIng/DietPi/master/.update/version
[FAILED] DietPi-Update | Unable to get repository version. Please check your network connection, then rerun dietpi-update.
```
The download if the version file failed, but there is no related curl error message. From the error prompt, you should be able to exit to the shell. Would be interesting to try manually downloading this file from there:
```sh
cd /tmp
curl -sSfL https://raw.githubusercontent.com/MichaIng/DietPi/master/.update/version -o version
echo $? # check exit code
```