MichaIng/DietPi #4387: WIFI Realtek RTL8811AU seen by OS, not configurable in dietpi-config

MDAR, 2021-05-18:
### Creating a bug report/issue

#### Required Information
- DietPi version |  cat /boot/dietpi/.version
G_DIETPI_VERSION_CORE=7
G_DIETPI_VERSION_SUB=1
G_DIETPI_VERSION_RC=2
G_GITBRANCH='master'
G_GITOWNER='MichaIng'

- Distro version | `echo $G_DISTRO_NAME` or `cat /etc/debian_version`

10.9

- Kernel version | `uname -a`
Linux openHAB2-C4 4.9.241-arm64 #1 SMP PREEMPT Thu Feb 25 17:57:15 CET 2021 aarch64 GNU/Linux

- SBC model | `echo $G_HW_MODEL_NAME` or (EG: RPi3)
Odroid C4/HC4 (aarch64)

- Power supply used | (EG: 5V 1A RAVpower)
5V 2Amp
- SDcard used | (EG: SanDisk ultra)
Emmc 16GB

#### Additional Information (if applicable)
Purchased a WiFi USB dongle from Odroid to go with this C4.

From these options, [Odroid Connectivity options](https://www.odroid.co.uk/odroid-accessories/odroid-connectivity)

I chose this one, which is detected as a Realtek RTL8811AU
[https://www.odroid.co.uk/odroid-accessories/odroid-connectivity?product_id=870](https://www.odroid.co.uk/odroid-accessories/odroid-connectivity?product_id=870)

#### Steps to reproduce
LSUSB shows the USB device connected

```
root@openHAB2-C4:~# lsusb
Bus 002 Device 002: ID 2109:0817 VIA Labs, Inc.
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 001 Device 004: ID 0bda:a811 Realtek Semiconductor Corp. RTL8811AU 802.11a/b/g/n/ac WLAN Adapter
Bus 001 Device 003: ID 10cf:0b1b Velleman Components, Inc.
Bus 001 Device 002: ID 2109:2817 VIA Labs, Inc.
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
```
But `dietpi-config > 7 > WiFi`
reports that "No supported WiFi hardware was found"

#### Expected behaviour
I'm not entirely sure, as I've never setup a WiFi device on an SBC, I've only ever added second Ethernet dongles

#### Extra details

```
Linux openHAB2-C4 4.9.241-arm64 #1 SMP PREEMPT Thu Feb 25 17:57:15 CET 2021 aarc                                                            h64

The programs included with the Debian GNU/Linux system are free software;
the exact distribution terms for each program are described in the
individual files in /usr/share/doc/*/copyright.

Debian GNU/Linux comes with ABSOLUTELY NO WARRANTY, to the extent
permitted by applicable law.
 ─────────────────────────────────────────────────────
 DietPi v7.1.2 : 10:40 - Tue 18/05/21
 ─────────────────────────────────────────────────────
 - Device model : Odroid C4/HC4 (aarch64)
 - CPU temp : 32'C : 89'F (Cool runnings)
 - LAN IP : 192.168.178.72 (eth0)
 - MOTD : Did you know that you can run "dietpi-banner" to change its content?
 ─────────────────────────────────────────────────────
[... 133 lines cut ...]
           Device-3: g12a-sound-card driver: asoc_aml_card v: N/A bus ID: N/A
           Display: tty server: X.org 1.20.4 driver: none tty: 140x24
           Message: Advanced graphics data unavailable in console for root.
Audio:     Device-1: audio- driver: aml_audio_controller bus ID: N/A
           Device-2: snd-iomap driver: auge_snd_iomap bus ID: N/A
           Device-3: sm1-audio-clocks driver: audio_clocks bus ID: N/A
           Device-4: sm1-audio-ddr-manager driver: audio_ddr_manager bus ID: N/A
           Device-5: sm1-snd-spdif-a driver: snd_spdif bus ID: N/A
           Device-6: amhdmitx driver: amhdmitx bus ID: N/A
           Device-7: audio_data driver: audio_data_debug bus ID: N/A
           Device-8: audiolocker driver: audiolocker bus ID: N/A
           Device-9: g12a-sound-card driver: asoc_aml_card bus ID: N/A
           Device-10: vdac-sm1 driver: aml_vdac bus ID: N/A
           Sound Server: ALSA v: k4.9.241-arm64
Network:   Device-1: g12a-eth-dwmac driver: meson6_dwmac v: N/A port: N/A bus ID: N/A
           IF: eth0 state: up speed: 1000 Mbps duplex: full mac: <filter>
           Device-2: Realtek RTL8811AU 802.11a/b/g/n/ac WLAN Adapter type: USB driver: usb-network bus ID: 1-1.3:4
Drives:    Local Storage: total: 14.56 GiB used: 3.18 GiB (21.8%)
           ID-1: /dev/mmcblk0 model: AJTD4R size: 14.56 GiB
Partition: ID-1: / size: 13.96 GiB used: 3.11 GiB (22.3%) fs: ext4 dev: /dev/mmcblk0p2
           ID-2: /boot size: 255.7 MiB used: 71.1 MiB (27.8%) fs: vfat dev: /dev/mmcblk0p1
Sensors:   Missing: Required tool sensors not installed. Check --recommends
Info:      Processes: 132 Uptime: 12m Memory: 3.63 GiB used: 748.9 MiB (20.2%) Init: systemd runlevel: 5 Compilers: gcc: 8.3.0
           Shell: bash v: 5.0.3 inxi: 3.0.32
```

MDAR, 2021-05-18:
I just tried loading a new driver, as explained here, https://blog.abysm.org/2020/03/realtek-802-11ac-usb-wi-fi-linux-driver-installation/

I don't understand what the last two lines of this mean ---

```
root@openHAB2-C4:~# apt install ./realtek-rtl88xxau-dkms_5.6.4.2~git20210327.c0ce817-0kali1_all.deb
Reading package lists... Done
Building dependency tree
Reading state information... Done
Note, selecting 'realtek-rtl88xxau-dkms' instead of './realtek-rtl88xxau-dkms_5.6.4.2~git20210327.c0ce817-0kali1_all.deb'
The following packages will be upgraded:
  realtek-rtl88xxau-dkms
1 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.
Need to get 0 B/1,795 kB of archives.
After this operation, 4,096 B of additional disk space will be used.
Get:1 /root/realtek-rtl88xxau-dkms_5.6.4.2~git20210327.c0ce817-0kali1_all.deb realtek-rtl88xxau-dkms all 5.6.4.2~git20210327.c0ce817-0kali1 [1,795 kB]
(Reading database ... 50981 files and directories currently installed.)
Preparing to unpack .../realtek-rtl88xxau-dkms_5.6.4.2~git20210327.c0ce817-0kali1_all.deb ...

------------------------------
Deleting module version: 5.6.4.2~git20210118
completely from the DKMS tree.
------------------------------
Done.
Unpacking realtek-rtl88xxau-dkms (5.6.4.2~git20210327.c0ce817-0kali1) over (5.6.4.2~git20210118-0kali1) ...
Setting up realtek-rtl88xxau-dkms (5.6.4.2~git20210327.c0ce817-0kali1) ...
Loading new realtek-rtl88xxau-5.6.4.2~git20210327.c0ce817 DKMS files...
Building for 4.9.241-arm64
Module build for kernel 4.9.241-arm64 was skipped since the
kernel headers for this kernel does not seem to be installed.
```

 However on the previous attempt to load the driver, it did say that the header were missing, so I tried this 

`apt install linux-headers-4.9.241-arm64=4.9.241-20210225-C4`

then --reinstall the kali driver

`apt install ./realtek-rtl88xxau-dkms_5.6.4.2~git20210327.c0ce817-0kali1_all.deb  --reinstall`

Which eventually returned this :

```
root@openHAB2-C4:~# apt install ./realtek-rtl88xxau-dkms_5.6.4.2~git20210327.c0ce817-0kali1_all.deb  --reinstall
Reading package lists... Done
Building dependency tree
Reading state information... Done
Note, selecting 'realtek-rtl88xxau-dkms' instead of './realtek-rtl88xxau-dkms_5.6.4.2~git20210327.c0ce817-0kali1_all.deb'
apt install linux-headers-4.9.241-arm64=4.9.241-20210225-C40 upgraded, 0 newly installed, 1 reinstalled, 0 to remove and 0 not upgraded.
After this operation, 0 B of additional disk space will be used.
Get:1 /root/realtek-rtl88xxau-dkms_5.6.4.2~git20210327.c0ce817-0kali1_all.deb realtek-rtl88xxau-dkms all 5.6.4.2~git20210327.c0ce817-0kali1 [1,795 kB]
(Reading database ... 73518 files and directories currently installed.)
Preparing to unpack .../realtek-rtl88xxau-dkms_5.6.4.2~git20210327.c0ce817-0kali1_all.deb ...

------------------------------
Deleting module version: 5.6.4.2~git20210327.c0ce817
completely from the DKMS tree.
------------------------------
Done.
Unpacking realtek-rtl88xxau-dkms (5.6.4.2~git20210327.c0ce817-0kali1) over (5.6.4.2~git20210327.c0ce817-0kali1) ...
Setting up realtek-rtl88xxau-dkms (5.6.4.2~git20210327.c0ce817-0kali1) ...
Loading new realtek-rtl88xxau-5.6.4.2~git20210327.c0ce817 DKMS files...
Building for 4.9.241-arm64
Building initial module for 4.9.241-arm64
Done.

88XXau.ko:
Running module version sanity check.
Error! Module version v5.6.4.2_35491.20191025 for 88XXau.ko
is not newer than what is already found in kernel 4.9.241-arm64 (v5.6.4.2_35491.20191025).
You may override by specifying --force.

depmod...

DKMS: install completed.
```

But still dietpi-config > 7 > network doesn't show it, however `inxi -Fxz` continues to show the network device

MDAR, 2021-05-18:
I've just found an old USB Wifi dongle in the bottom of my laptop bag and that does seem to be discovered and configurable

`Realtek Semiconductor Corp. RTL8192CU 802.11n WLAN Adapter`

MichaIng, 2021-05-18:
Many thanks for your report.

First of all `dietpi-config` detects network adapters based on the kernel devices found in `/sys/class/net/` respectively the output of `ip l`, which lists all those as well. `lsusb` shows USB devices and their names, but the system/kernel cannot derive from that whether it's a network adapter or something else. For this a kernel driver is required. When it is invoked correctly, the USB device is detected as network adapter and can be used then.

Since you did some driver install tests, I'm not 100% sure now, but it seems the kernel ships this driver already?
Can you run:
```
find /lib/modules -name '88XXau.ko' -exec dpkg -S {} \;
```

MDAR, 2021-05-18:
> find /lib/modules -name '88XXau.ko' -exec dpkg -S {} \;

Hey Michal

Thanks for taking the time to respond. 

I ran the command and got this...

`linux-image-4.9.241-arm64: /lib/modules/4.9.241-arm64/kernel/drivers/net/wireless/rtl8812au/88XXau.ko`

MDAR, 2021-05-18:
Based on your reply, I'm wondering if this Realtek RTL8811AU chipset isn't listed in the set of headers I have loaded?
(Not that I really understand what the sentence means)

So I've tried the following command

`apt install linux-headers-arm64`

Which appears to want to install headers with the version `4.19`, rather than `4.9`

MichaIng, 2021-05-18:
Okay, so it's indeed part of the kernel already. Let's Verify the driver version and load it manually and see if it has an effect:
```
modinfo 88XXau
modprobe 88XXau
sleep 1
ls -l /sys/class/net/
ip l
dmesg | tail -10
```

MDAR, 2021-05-18:
> modinfo 88XXau
> modprobe 88XXau
> sleep 1
> ls -l /sys/class/net/
> ip l
> dmesg | tail -10

So I added that to a script and ran it.

Here is the screen dump of everything I've tried in the last 10 minutes

```
Linux openHAB2-C4 4.9.241-arm64 #1 SMP PREEMPT Thu Feb 25 17:57:15 CET 2021 aarc                                                                                                                                                             h64

The programs included with the Debian GNU/Linux system are free software;
the exact distribution terms for each program are described in the
individual files in /usr/share/doc/*/copyright.

Debian GNU/Linux comes with ABSOLUTELY NO WARRANTY, to the extent
permitted by applicable law.
 ─────────────────────────────────────────────────────
 DietPi v7.1.2 : 15:40 - Tue 18/05/21
 ─────────────────────────────────────────────────────
 - Device model : Odroid C4/HC4 (aarch64)
 - CPU temp : 34'C : 93'F (Cool runnings)
 - LAN IP : 192.168.178.72 (eth0)
 - MOTD : Did you know that you can run "dietpi-banner" to change its content?
 ─────────────────────────────────────────────────────

 DietPi Team     : MichaIng (lead), Daniel Knight (founder), Joulinar (support)
 Image by        : DietPi Core Team (pre-image: Meveric)
 Web             : https://dietpi.com | https://twitter.com/DietPi_
 Patreon Legends : Camry2731
 Contribute      : https://dietpi.com/contribute.html
 DietPi Hosting  : Powered by https://myvirtualserver.com

 dietpi-launcher : All the DietPi programs in one place.
 dietpi-config   : Feature rich configuration tool for your device.
 dietpi-software : Select optimized software for installation.
 htop            : Resource monitor.
 cpu             : Shows CPU information and stats.

root@openHAB2-C4:~# find /lib/modules -name '88XXau.ko' -exec dpkg -S {} \;
linux-image-4.9.241-arm64: /lib/modules/4.9.241-arm64/kernel/drivers/net/wireles                                                                                                                                                             s/rtl8812au/88XXau.ko
root@openHAB2-C4:~# find /lib/modules -name '88XXau.ko' -exec dpkg -S {} \;
linux-image-4.9.241-arm64: /lib/modules/4.9.241-arm64/kernel/drivers/net/wireless/rtl8812au/88XXau.ko
root@openHAB2-C4:~# dietpi-config
root@openHAB2-C4:~# apt install linux-header
Reading package lists... Done
Building dependency tree
Reading state information... Done
E: Unable to locate package linux-header
root@openHAB2-C4:~# apt install linux-header* -s
Reading package lists... Done
Building dependency tree
Reading state information... Done
Note, selecting 'linux-headers-rt-arm64' for glob 'linux-header*'
Note, selecting 'linux-headers-4.19.0-16-common' for glob 'linux-header*'
Note, selecting 'linux-headers-5.9.0-0.bpo.5-common-rt' for glob 'linux-header*'
Note, selecting 'linux-headers-4.19.0-14-all-arm64' for glob 'linux-header*'
Note, selecting 'linux-headers-5.10.0-0.bpo.4-common-rt' for glob 'linux-header*'
Note, selecting 'linux-headers-generic' for glob 'linux-header*'
Note, selecting 'linux-headers-686-pae' for glob 'linux-header*'
Note, selecting 'linux-headers-s390x' for glob 'linux-header*'
Note, selecting 'linux-headers-5.10.0-0.bpo.4-rt-arm64' for glob 'linux-header*'
Note, selecting 'linux-headers-4.9.236-arm64' for glob 'linux-header*'
Note, selecting 'linux-headers-5.9.0-0.bpo.5-rt-arm64' for glob 'linux-header*'
Note, selecting 'linux-headers-5.10.0-0.bpo.4-cloud-arm64' for glob 'linux-header*'
Note, selecting 'linux-headers-2.6' for glob 'linux-header*'
Note, selecting 'linux-headers-4.19.0-14-common' for glob 'linux-header*'
[... 370 lines cut ...]
parm:           rtw_decrypt_phy_file:Enable Decrypt PHY File (int)
parm:           rtw_en_napi:int
parm:           rtw_en_gro:int
parm:           rtw_iqk_fw_offload:int
parm:           rtw_ch_switch_offload:int
modprobe: ERROR: could not insert '88XXau': Exec format error
total 0
lrwxrwxrwx 1 root root 0 Jan  1  1970 eth0 -> ../../devices/platform/ff3f0000.ethernet/net/eth0
lrwxrwxrwx 1 root root 0 Jan  1  1970 lo -> ../../devices/virtual/net/lo
1: lo: <LOOPBACK,UP,LOWER_UP> mtu 65536 qdisc noqueue state UNKNOWN mode DEFAULT group default qlen 1
    link/loopback 00:00:00:00:00:00 brd 00:00:00:00:00:00
2: eth0: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc pfifo_fast state UP mode DEFAULT group default qlen 1000
    link/ether 00:1e:06:48:30:a6 brd ff:ff:ff:ff:ff:ff
[    8.503403] cpufreq_interactive: cpufreq_hmp_boost_start()
[    8.506595] cpufreq_interactive: no need to active hmp boost!
[    8.807390] meson6-dwmac ff3f0000.ethernet eth0: fail to init PTP.
[    8.808446] IPv6: ADDRCONF(NETDEV_UP): eth0: link is not ready
[   11.786831] vout: aml_tvout_mode_work: monitor_timeout
[   12.906807] meson6-dwmac ff3f0000.ethernet eth0: Link is Up - 1Gbps/Full - flow control rx/tx
[   12.909907] IPv6: ADDRCONF(NETDEV_CHANGE): eth0: link becomes ready
[   64.522618] fb: mem_free_work, free memory: addr:800000
[  611.338531] fb: osd[0] enable: 0 (kworker/0:0)
[  820.749844] module 88XXau: unsupported RELA relocation: 275
root@openHAB2-C4:~#
```

MDAR, 2021-05-18:
> apt install linux-headers-arm64-odroid-c4

Okay....

I'm running that right now......

MDAR, 2021-05-18:
> > apt install linux-headers-arm64-odroid-c4
> 
> Okay....
> 
> I'm running that right now......

And this is what I get

```
root@openHAB2-C4:~# apt install linux-headers-arm64-odroid-c4
Reading package lists... Done
Building dependency tree
Reading state information... Done
The following NEW packages will be installed:
  linux-headers-arm64-odroid-c4
0 upgraded, 1 newly installed, 0 to remove and 0 not upgraded.
Need to get 694 B of archives.
After this operation, 8,192 B of additional disk space will be used.
Get:1 https://dietpi.com/meveric all/c4 arm64 linux-headers-arm64-odroid-c4 arm64 4.9.241-arm64-1 [694 B]
Fetched 694 B in 0s (4,381 B/s)
Selecting previously unselected package linux-headers-arm64-odroid-c4.
(Reading database ... 90064 files and directories currently installed.)
Preparing to unpack .../linux-headers-arm64-odroid-c4_4.9.241-arm64-1_arm64.deb ...
Unpacking linux-headers-arm64-odroid-c4 (4.9.241-arm64-1) ...
Setting up linux-headers-arm64-odroid-c4 (4.9.241-arm64-1) ...
```

MDAR, 2021-05-18:
> find /lib/modules -name '88XXau.ko' -exec dpkg -S {} \;

(After a reboot)
I do see that this command reports the presence of **RTL8812au**, rather than **RTL8811au**

MichaIng, 2021-05-18:
> modprobe: ERROR: could not insert '88XXau': Exec format error

That is strange. Please try to reinstall the kernel and let's see whether loading the module works then. Also remove the wrong headers and kernel:
```
apt purge --autoremove linux-headers-arm64
apt install --reinstall linux-image-4.9.241-arm64
modprobe 88XXau
```

The package you downloaded installs the same kernel module `88XXau.ko`, the directory name it is located in is not relevant. Either it does work for 8811au as well or not. But first we need to be able to load it into the kernel.

MDAR, 2021-05-18:
> apt purge --autoremove linux-headers-arm64

if it helps, I just found this forum thread, that is suggesting a driver needs to be compiled?

https://forum.mxlinux.org/viewtopic.php?t=49102

https://github.com/aircrack-ng/rtl8812au

MDAR, 2021-05-18:
> apt purge --autoremove linux-headers-arm64
> apt install --reinstall linux-image-4.9.241-arm64
> modprobe 88XXau

Okay, so I ran those commands and apart from waiting a long time at 60%, it seems to have finished.

```
root@openHAB2-C4:~# lsusb
Bus 002 Device 002: ID 2109:0817 VIA Labs, Inc.
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 001 Device 004: ID 0bda:a811 Realtek Semiconductor Corp. RTL8811AU 802.11a/b/g/n/ac WLAN Adapter
Bus 001 Device 003: ID 10cf:0b1b Velleman Components, Inc.
Bus 001 Device 002: ID 2109:2817 VIA Labs, Inc.
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
root@openHAB2-C4:~# apt purge --autoremove linux-headers-arm64
Reading package lists... Done
Building dependency tree
Reading state information... Done
The following packages will be REMOVED:
  linux-headers-4.19.0-16-arm64* linux-headers-4.19.0-16-common* linux-headers-arm64* linux-kbuild-4.19*
0 upgraded, 0 newly installed, 4 to remove and 0 not upgraded.
After this operation, 57.2 MB disk space will be freed.
Do you want to continue? [Y/n] y
(Reading database ... 90064 files and directories currently installed.)
Removing linux-headers-arm64 (4.19+105+deb10u11) ...
Removing linux-headers-4.19.0-16-arm64 (4.19.181-1) ...
dpkg: warning: while removing linux-headers-4.19.0-16-arm64, directory '/lib/modules/4.19.0-16-arm64' not empty so not removed
Removing linux-headers-4.19.0-16-common (4.19.181-1) ...
Removing linux-kbuild-4.19 (4.19.181-1) ...
root@openHAB2-C4:~# apt install --reinstall linux-image-4.9.241-arm64
Reading package lists... Done
Building dependency tree
Reading state information... Done
0 upgraded, 0 newly installed, 1 reinstalled, 0 to remove and 0 not upgraded.
Need to get 20.8 MB of archives.
After this operation, 0 B of additional disk space will be used.
Get:1 https://dietpi.com/meveric all/c4 arm64 linux-image-4.9.241-arm64 arm64 4.9.241-20210225-C4 [20.8 MB]
Fetched 20.8 MB in 3s (7,685 kB/s)
(Reading database ... 73518 files and directories currently installed.)
Preparing to unpack .../linux-image-4.9.241-arm64_4.9.241-20210225-C4_arm64.deb ...
dkms: removing: realtek-rtl88xxau 5.6.4.2~git20210327.c0ce817 (4.9.241-arm64) (aarch64)

-------- Uninstall Beginning --------
Module:  realtek-rtl88xxau
Version: 5.6.4.2~git20210327.c0ce817
Kernel:  4.9.241-arm64 (aarch64)
-------------------------------------

Status: Before uninstall, this module version was ACTIVE on this kernel.

88XXau.ko:
 - Uninstallation
   - Deleting from: /lib/modules/4.9.241-arm64/
rmdir: failed to remove '': No such file or directory
 - Original module
   - No original module was found for this module on this kernel.
   - Use the dkms install command to reinstall any previous module version.

depmod...

DKMS: uninstall completed.
Unpacking linux-image-4.9.241-arm64 (4.9.241-20210225-C4) over (4.9.241-20210225-C4) ...
Setting up linux-image-4.9.241-arm64 (4.9.241-20210225-C4) ...
Error! Module version v5.6.4.2_35491.20191025 for 88XXau.ko
is not newer than what is already found in kernel 4.9.241-arm64 (v5.6.4.2_35491.20191025).
You may override by specifying --force.
update-initramfs: Generating /boot/initrd.img-4.9.241-arm64
Image Name:   uInitrd
Created:      Tue May 18 16:11:39 2021
Image Type:   AArch64 Linux RAMDisk Image (uncompressed)
Data Size:    8005899 Bytes = 7818.26 KiB = 7.64 MiB
Load Address: 00000000
Entry Point:  00000000
root@openHAB2-C4:~#

```

MichaIng, 2021-05-18:
Hmm, that is the very same kernel module `88XXau`. Generally it should work.

MDAR, 2021-05-18:
> Generally it should work.

Yeah, that old classic.

**Should** and **Does** aren't always good bedfellows

MichaIng, 2021-05-18:
DKMS seems to interfere, please remove it as well:
```
apt purge --autoremove dkms realtek-rtl88xxau-dkms
apt install --reinstall linux-image-4.9.241-arm64
modprobe 88XXau
```

MDAR, 2021-05-18:
It all went well... until...

```
modprobe 88XXau
modprobe: ERROR: could not insert '88XXau': Exec format error
```

MDAR, 2021-05-18:
Does that fact that the dongle is seen in this list mean anything???

MDAR, 2021-05-18:
Purely out of curiousity, I threw the older dongle in to see if there is any difference in the inxi query

Does it mean anything that one says the USB-Driver is `rtl88xxxu` and the other is `usb-network`

```
root@openHAB2-C4:~# inxi -Fxz
System:    Host: openHAB2-C4 Kernel: 4.9.241-arm64 aarch64 bits: 64 compiler: gcc v: 6.3.0 Console: tty 0
           Distro: Debian GNU/Linux 10 (buster)

Network:   Device-1: g12a-eth-dwmac driver: meson6_dwmac v: N/A port: N/A bus ID: N/A
           IF: eth0 state: up speed: 1000 Mbps duplex: full mac: <filter>
           Device-2: Realtek RTL8192CU 802.11n WLAN Adapter type: USB driver: rtl8xxxu bus ID: 1-1.2:6
           IF: wlan0 state: up mac: <filter>
           Device-3: Realtek RTL8811AU 802.11a/b/g/n/ac WLAN Adapter type: USB driver: usb-network bus ID: 1-1.3:5
```

MichaIng, 2021-05-18:
Not sure how `inxi` derives which hardware is of which type, but as long as `ip r` (and `/sys/class/net/`) does not show the interface, it cannot be used.

Does loading any other kernel module work, e.g.:
```
modprobe 8188eu
```

Ah and there is a force option, just to try:
```
modprobe -f 88XXau
```

The driver package you installed first btw states that its for x86, x86_64 platforms. Not sure whether it plays a role when it's compiled via DKMS but following the manual build via https://github.com/aircrack-ng/rtl8812au looks more promising to me.

MDAR, 2021-05-18:
Because I can... I plugged this "Odroid" WiFi dongle in to my Windows 10 machine, just to see if it works.

It does.

Win10 happily loads a RealTek RTL8811AU driver for it

MDAR, 2021-05-18:
> ```
> modprobe 8188eu
> ```
> 
> 
>     
>   
> 
> Ah and there is a force option, just to try:
> 
> 
> ```
> modprobe -f 88XXau
> ```

Oh dear

trying either of those commands gives me

`modprobe: ERROR: could not insert '88XXau': Exec format error`

MichaIng, 2021-05-18:
Did you reboot once? Which kernel is actually loaded? `uname -a`

MDAR, 2021-05-18:
> following the manual build via https://github.com/aircrack-ng/rtl8812au looks more promising to me.

I tried that and nothing much changed.

Apart from it suggested that the driver was no newer than the one already loaded.

******

`uname -a` returns

`Linux openHAB2-C4 4.9.241-arm64 #1 SMP PREEMPT Thu Feb 25 17:57:15 CET 2021 aarch64 GNU/Linux`

**** I do appreciate you spending so much time on this.... I would have given up hours ago

MichaIng, 2021-05-18:
I'm puzzled how no kernel module can be loaded. Is there any other module loaded, actually? Some, 10 or so, should be loaded by default.
```
lsmod
```

MDAR, 2021-05-18:
Its very strange

> lsmod

root@openHAB2-C4:/# lsmod
Module                  Size  Used by
rtl8xxxu              114688  0
amvdec_ports          200704  0
stream_input          180224  1 amvdec_ports
decoder_common        188416  2 amvdec_ports,stream_input
firmware               28672  2 decoder_common,stream_input
media_clock            45056  4 decoder_common,amvdec_ports,firmware,stream_input
ip_tables              32768  0
x_tables               49152  1 ip_tables
ipv6                  462848  46

MichaIng, 2021-05-18:
`rtl8xxxu` is loaded, probably they conflict? Try:
```
modprobe -r rtl8xxxu
modprobe 88XXau
```

Also let's try to load a very different basic module:
```
modprobe tun # that is for OpenVPN interfaces
```

And let's see whether the timestamps are as expected:
```
find /lib/modules ( -name 88XXau.ko -o -name rtl8xxxu.ko ) -exec ls -l {} \;
```

MDAR, 2021-05-18:
Something strange is happening...

This command returns ...

> modprobe 88XXau

`modprobe: ERROR: could not insert '88XXau': Exec format error`

and this command returns --

> find /lib/modules ( -name 88XXau.ko -o -name rtl8xxxu.ko ) -exec ls -l {} \;

`-bash: syntax error near unexpected token `('`

MDAR, 2021-05-18:
Thanks for your efforts Michal

I'm going to give up for the day now.

Thanks again for all your effort.

If you have any wild ideas, I'm happy to give them a try later

MichaIng, 2021-05-18:
Ah sorry, I forgot to escape the parenthesis:
```
find /lib/modules \( -name 88XXau.ko -o -name rtl8xxxu.ko \) -exec ls -l {} \;
```

MDAR, 2021-05-18:
> find /lib/modules \( -name 88XXau.ko -o -name rtl8xxxu.ko \) -exec ls -l {} \;

Arrr yes, that makes a difference

```
root@openHAB2-C4:~# find /lib/modules \( -name 88XXau.ko -o -name rtl8xxxu.ko \) -exec ls -l {} \;
-rw-r--r-- 1 root root 179200 Feb 25 16:59 /lib/modules/4.9.241-arm64/kernel/drivers/net/wireless/realtek/rtl8xxxu/rtl8xxxu.ko
-rw-r--r-- 1 root root 4304096 Feb 25 16:59 /lib/modules/4.9.241-arm64/kernel/drivers/net/wireless/rtl8812au/88XXau.ko
```

MichaIng, 2021-05-18:
Okay thanks. Looks like a bug then. I'll forward it.

Custom compiling the module from the GitHub source might work, when you're in mood to go with this 🙂.

MDAR, 2021-05-21:
> Custom compiling the module from the GitHub source might work

Nice thought, but I think that is WAY beyond my skill set

MichaIng, 2021-06-03:
Meveric by chance has that adapter as well, verified that the driver module shipped with the Hardkernel kernel does not work and packed an own DEB package for that module:
```
cd /tmp
curl -sSfLO https://oph.mdrjr.net/meveric/other/C4/rtl8812au-dkms_4.3.14_all.deb
apt install ./rtl8812au-dkms_4.3.14_all.deb
rm rtl8812au-dkms_4.3.14_all.deb
```
Best would be if Hardkernel got that fixed, but this should serve as workaround.

MDAR, 2021-06-03:
Ummm

Thanks for the hard work.

What I now get it that I can scan for networks and it correctly sees everything in my area, but it just can't connect.

I have multiple access points on the same SSID.

Whatever I have tried brings me back to the same situation.

Slot 0 gets the correct SSID and password combination, but it just can't get a lease.

******

Using Ubuntu it works perfectly.

What can I do next to test this for you?

Or, as you hint, should be just flag it to Hardkernel? (Or will they just say.... "Use Ubuntu, it works")

Thanks a million

MichaIng, 2021-06-03:
> but it just can't get a lease.

Oh, that is bad. Yeah it needs to be reported to Hardkernel. I already asked Meveric about it, else will open an issue: https://github.com/hardkernel/linux/issues

MichaIng, 2021-06-04:
Issue reported: https://github.com/hardkernel/linux/issues/418

MDAR, 2021-06-05:
I thought it might be a config issue, so I tried to follow these instructions, (which is what I had to do in the Ubuntu 20 command line)

https://bhattigurjot.wordpress.com/2013/10/26/connect-to-wi-fi-network-through-ubuntu-terminal/comment-page-1/

Ok, so ifconfig isn't available, but iwconfig seemed to work

But it still refused to connect to the SSID

MDAR, 2021-06-05:
But saying that.

I have restarted the C4 and your script prompted me to update 1 app

Which turned out to be

rt18812au from version 4.3.14 to 5.9.3.2

BUT

That has removed the WiFi card from the config again.

Oh well, it was worth a try 😢

MichaIng, 2021-06-05:
Don't use 8 years old guides, you'll easily destroy the setup when it contains configuring OS-specific and/or outdated stacks and tools 😄. `ifconfig` is from the deprecated `net-tools` package, while nowadays you'd want to use the `ip` command from the `iproute2` package, e.g. `ip a` to get an overview of attached network adapters and IP addresses.

`iwconfig` is deprecated the same way (superseded by `iw`), but the `wireless-tools` package contains a few other nice implementations, which is why we still have it pre-installed. It does not support WPA at all, so simply cannot work with modern WiFi networks, this is what `wpasupplicant` is for, controlled with the `dietpi-wifidb`/`dietpi-config` script, or via `/etc/wpa_supplicant/wpa_supplicant.conf`. `dhclient` is already started by `ifupdown` at boot based on `/etc/network/interfaces` config, so running it again would lead to two DHCP clients running concurrently.

Lol I see there is already such a package in the repo: https://dietpi.com/meveric/pool/main/r/rtl8812au-dkms/
That would need to be replaced, of course. However, as both do not fully work, it currently doesn't matter.

MDAR, 2021-06-05:
Some how I don't feel that "sorry" is enough of an apology.

I will unplug my keyboard and leave it alone.

MichaIng, 2021-06-05:
> Some how I don't feel that "sorry" is enough of an apology.

😄, if you mean a "sorry" from your side, that is definitely not required and I didn't meant this as blame. The info on ifconfig, iwconfig, iw, ip, dhclient etc are really meant as info, as this is definitely confusing, and I ran into the same issues with the same questions and hours of searching through the web to find out how things work.

Generally nothing bad about trying out things, as long as the system is not in urgent production state 👍.

Joulinar, 2021-08-19:
not sure but this might be a similar issue. https://dietpi.com/phpbb/viewtopic.php?f=11&t=9319

MichaIng, 2022-08-01:
With the new Odroid N2 image, this should do it:
```sh
sudo apt install git dkms bc linux-headers-current-meson64
git clone -b v5.6.4.2 --depth=1 https://github.com/aircrack-ng/rtl8812au.git
cd rtl8812au*
sudo make dkms_install
sudo modprobe 88XXau # or reboot
```

MichaIng, 2022-08-01:
Ah sorry two important steps forgotten above but edited them inside now:
```sh
sed -i 's/CONFIG_PLATFORM_I386_PC = y/CONFIG_PLATFORM_I386_PC = n/g' Makefile
sed -i 's/CONFIG_PLATFORM_ARM64_RPI = n/CONFIG_PLATFORM_ARM64_RPI = y/g' Makefile
```
Would have been neat if it would find out the architecture automatically, but it obviously doesn't.

wirelessjimp, 2022-08-01:
When I tried to run those commands on the Odroid C4 everything worked until I got to the make dkms_install step. It churned for a while but then errored out with this output: 

root@DietPi:~/rtl8812au# sudo make dkms_install
mkdir: created directory '/usr/src/8812au-5.6.4.2_35491.20191025'
cp -r * /usr/src/8812au-5.6.4.2_35491.20191025
dkms add -m 8812au -v 5.6.4.2_35491.20191025

Creating symlink /var/lib/dkms/8812au/5.6.4.2_35491.20191025/source ->
                 /usr/src/8812au-5.6.4.2_35491.20191025

DKMS: add completed.
dkms build -m 8812au -v 5.6.4.2_35491.20191025

Kernel preparation unnecessary for this kernel.  Skipping...

Building module:
cleaning build area...
'make' -j4 KVER=5.10.123-meson64 KSRC=/lib/modules/5.10.123-meson64/build...................................................................................(bad exit status: 2)
Error! Bad return status for module build on kernel: 5.10.123-meson64 (aarch64)
Consult /var/lib/dkms/8812au/5.6.4.2_35491.20191025/build/make.log for more information.
make: *** [Makefile:2311: dkms_install] Error 10

MichaIng, 2022-08-01:
Please check:
```sh
cat /var/lib/dkms/8812au/5.6.4.2_35491.20191025/build/make.log
```
for the error. Probably some dev header is missing.

wirelessjimp, 2022-08-01:
Was just in there looking.
DKMS make.log for 8812au-5.6.4.2_35491.20191025 for kernel 5.10.123-meson64 (aarch64)
Mon 01 Aug 2022 11:25:09 AM MDT
make[1]: Entering directory '/var/lib/dkms/8812au/5.6.4.2_35491.20191025/build'
make ARCH=arm64 CROSS_COMPILE= -C /lib/modules/5.10.123-meson64/build M=/var/lib/dkms/8812au/5.6.4.2_35491.20191025/build  modules
make[2]: Entering directory '/usr/src/linux-headers-5.10.123-meson64'
  CC [M]  /var/lib/dkms/8812au/5.6.4.2_35491.20191025/build/core/rtw_cmd.o
  CC [M]  /var/lib/dkms/8812au/5.6.4.2_35491.20191025/build/core/rtw_security.o
  CC [M]  /var/lib/dkms/8812au/5.6.4.2_35491.20191025/build/core/rtw_debug.o
  CC [M]  /var/lib/dkms/8812au/5.6.4.2_35491.20191025/build/core/rtw_io.o
  CC [M]  /var/lib/dkms/8812au/5.6.4.2_35491.20191025/build/core/rtw_ioctl_query.o
  CC [M]  /var/lib/dkms/8812au/5.6.4.2_35491.20191025/build/core/rtw_ioctl_set.o
  CC [M]  /var/lib/dkms/8812au/5.6.4.2_35491.20191025/build/core/rtw_ieee80211.o
  CC [M]  /var/lib/dkms/8812au/5.6.4.2_35491.20191025/build/core/rtw_mlme.o
  CC [M]  /var/lib/dkms/8812au/5.6.4.2_35491.20191025/build/core/rtw_mlme_ext.o
  CC [M]  /var/lib/dkms/8812au/5.6.4.2_35491.20191025/build/core/rtw_mi.o
  CC [M]  /var/lib/dkms/8812au/5.6.4.2_35491.20191025/build/core/rtw_wlan_util.o
  CC [M]  /var/lib/dkms/8812au/5.6.4.2_35491.20191025/build/core/rtw_vht.o
  CC [M]  /var/lib/dkms/8812au/5.6.4.2_35491.20191025/build/core/rtw_pwrctrl.o
  CC [M]  /var/lib/dkms/8812au/5.6.4.2_35491.20191025/build/core/rtw_rf.o
  CC [M]  /var/lib/dkms/8812au/5.6.4.2_35491.20191025/build/core/rtw_chplan.o
  CC [M]  /var/lib/dkms/8812au/5.6.4.2_35491.20191025/build/core/rtw_recv.o
  CC [M]  /var/lib/dkms/8812au/5.6.4.2_35491.20191025/build/core/rtw_sta_mgt.o
  CC [M]  /var/lib/dkms/8812au/5.6.4.2_35491.20191025/build/core/rtw_ap.o
  CC [M]  /var/lib/dkms/8812au/5.6.4.2_35491.20191025/build/core/mesh/rtw_mesh.o
  CC [M]  /var/lib/dkms/8812au/5.6.4.2_35491.20191025/build/core/mesh/rtw_mesh_pathtbl.o
  CC [M]  /var/lib/dkms/8812au/5.6.4.2_35491.20191025/build/core/mesh/rtw_mesh_hwmp.o
  CC [M]  /var/lib/dkms/8812au/5.6.4.2_35491.20191025/build/core/rtw_xmit.o
  CC [M]  /var/lib/dkms/8812au/5.6.4.2_35491.20191025/build/core/rtw_p2p.o
  CC [M]  /var/lib/dkms/8812au/5.6.4.2_35491.20191025/build/core/rtw_rson.o
  CC [M]  /var/lib/dkms/8812au/5.6.4.2_35491.20191025/build/core/rtw_tdls.o
  CC [M]  /var/lib/dkms/8812au/5.6.4.2_35491.20191025/build/core/rtw_br_ext.o
  CC [M]  /var/lib/dkms/8812au/5.6.4.2_35491.20191025/build/core/rtw_iol.o
  CC [M]  /var/lib/dkms/8812au/5.6.4.2_35491.20191025/build/core/rtw_sreset.o
  CC [M]  /var/lib/dkms/8812au/5.6.4.2_35491.20191025/build/core/rtw_btcoex_wifionly.o
  CC [M]  /var/lib/dkms/8812au/5.6.4.2_35491.20191025/build/core/rtw_btcoex.o
  CC [M]  /var/lib/dkms/8812au/5.6.4.2_35491.20191025/build/core/rtw_beamforming.o
  CC [M]  /var/lib/dkms/8812au/5.6.4.2_35491.20191025/build/core/rtw_odm.o
  CC [M]  /var/lib/dkms/8812au/5.6.4.2_35491.20191025/build/core/rtw_rm.o
  CC [M]  /var/lib/dkms/8812au/5.6.4.2_35491.20191025/build/core/rtw_rm_fsm.o
  CC [M]  /var/lib/dkms/8812au/5.6.4.2_35491.20191025/build/core/efuse/rtw_efuse.o
  CC [M]  /var/lib/dkms/8812au/5.6.4.2_35491.20191025/build/os_dep/osdep_service.o
  CC [M]  /var/lib/dkms/8812au/5.6.4.2_35491.20191025/build/os_dep/linux/os_intfs.o
  CC [M]  /var/lib/dkms/8812au/5.6.4.2_35491.20191025/build/os_dep/linux/usb_intf.o
  CC [M]  /var/lib/dkms/8812au/5.6.4.2_35491.20191025/build/os_dep/linux/usb_ops_linux.o
  CC [M]  /var/lib/dkms/8812au/5.6.4.2_35491.20191025/build/os_dep/linux/ioctl_linux.o
  CC [M]  /var/lib/dkms/8812au/5.6.4.2_35491.20191025/build/os_dep/linux/xmit_linux.o
  CC [M]  /var/lib/dkms/8812au/5.6.4.2_35491.20191025/build/os_dep/linux/mlme_linux.o
  CC [M]  /var/lib/dkms/8812au/5.6.4.2_35491.20191025/build/os_dep/linux/recv_linux.o
  CC [M]  /var/lib/dkms/8812au/5.6.4.2_35491.20191025/build/os_dep/linux/ioctl_cfg80211.o
  CC [M]  /var/lib/dkms/8812au/5.6.4.2_35491.20191025/build/os_dep/linux/wifi_regd.o
  CC [M]  /var/lib/dkms/8812au/5.6.4.2_35491.20191025/build/os_dep/linux/rtw_android.o
  CC [M]  /var/lib/dkms/8812au/5.6.4.2_35491.20191025/build/os_dep/linux/rtw_proc.o
  CC [M]  /var/lib/dkms/8812au/5.6.4.2_35491.20191025/build/os_dep/linux/rtw_rhashtable.o
  CC [M]  /var/lib/dkms/8812au/5.6.4.2_35491.20191025/build/os_dep/linux/ioctl_mp.o
  CC [M]  /var/lib/dkms/8812au/5.6.4.2_35491.20191025/build/hal/hal_intf.o
  CC [M]  /var/lib/dkms/8812au/5.6.4.2_35491.20191025/build/hal/hal_com.o
  CC [M]  /var/lib/dkms/8812au/5.6.4.2_35491.20191025/build/hal/hal_com_phycfg.o
  CC [M]  /var/lib/dkms/8812au/5.6.4.2_35491.20191025/build/hal/hal_phy.o
  CC [M]  /var/lib/dkms/8812au/5.6.4.2_35491.20191025/build/hal/hal_dm.o
  CC [M]  /var/lib/dkms/8812au/5.6.4.2_35491.20191025/build/hal/hal_dm_acs.o
  CC [M]  /var/lib/dkms/8812au/5.6.4.2_35491.20191025/build/hal/hal_btcoex_wifionly.o
  CC [M]  /var/lib/dkms/8812au/5.6.4.2_35491.20191025/build/hal/hal_btcoex.o
  CC [M]  /var/lib/dkms/8812au/5.6.4.2_35491.20191025/build/hal/hal_mp.o
  CC [M]  /var/lib/dkms/8812au/5.6.4.2_35491.20191025/build/hal/hal_mcc.o
  CC [M]  /var/lib/dkms/8812au/5.6.4.2_35491.20191025/build/hal/hal_hci/hal_usb.o
  CC [M]  /var/lib/dkms/8812au/5.6.4.2_35491.20191025/build/hal/led/hal_led.o
  CC [M]  /var/lib/dkms/8812au/5.6.4.2_35491.20191025/build/hal/led/hal_usb_led.o
  CC [M]  /var/lib/dkms/8812au/5.6.4.2_35491.20191025/build/hal/HalPwrSeqCmd.o
[... cut ...]
make[4]: *** No rule to make target 'scripts/module.lds', needed by '/var/lib/dkms/8812au/5.6.4.2_35491.20191025/build/88XXau.ko'.  Stop.
make[4]: *** Waiting for unfinished jobs....
  CC [M]  /var/lib/dkms/8812au/5.6.4.2_35491.20191025/build/88XXau.mod.o
make[3]: *** [scripts/Makefile.modpost:130: __modpost] Error 2
make[2]: *** [Makefile:1726: modules] Error 2
make[2]: Leaving directory '/usr/src/linux-headers-5.10.123-meson64'
make[1]: *** [Makefile:2251: modules] Error 2
make[1]: Leaving directory '/var/lib/dkms/8812au/5.6.4.2_35491.20191025/build'

MichaIng, 2022-08-01:
Nasty, seems to be an issue with Linux v5.10 headers: https://github.com/aircrack-ng/rtl8812au/issues/762

Ah, I mixed it up and built it on NanoPi R4S with "next" kernel v5.18. I'll try to apply the mentioned fix/workaround on Odroid N2 now.

MichaIng, 2022-08-01:
This did it:
```sh
sudo ln -s module.lds.S /usr/src/linux-headers-5.10.123-meson64/scripts/module.lds
sudo make dkms_remove
sudo make dkms_install
sudo modprobe 88XXau
```
But it says:
```
Good news! Module version v5.6.4.2_35491.20191025 for 88XXau.ko
exactly matches what is already found in kernel 5.10.123-meson64.
DKMS will not replace this module.
You may override by specifying --force.
```
So the kernel module is there already. Quite confusing, present here but missing on rockchip64 kernel next branch 🤔. So again:
```sh
sed -i 's/dkms install/dkms install --force/' Makefile
sudo make dkms_remove
sudo make dkms_install
sudo modprobe 88XXau
```
Now it was actually installed but:
```
modprobe: ERROR: could not insert '88XXau': Exec format error
```
What a hassle, trying another possible fix from the readme:
```sh
export ARCH=arm64
sed -i 's/^MAKE="/MAKE="ARCH=arm64\ /' dkms.conf
sudo make dkms_remove
sudo make dkms_install
sudo modprobe 88XXau
```
Same error ...

Trying with "edge" Linux packages version now.

wirelessjimp, 2022-08-01:
Wow! Thanks for the hard work. I will be busy the rest of the afternoon, I will check in this evening.

MichaIng, 2022-08-01:
```sh
sudo make dkms_remove
sudo apt -y install linux-dtb-next-meson64 linux-headers-next-meson64 linux-image-next-meson64 linux-u-boot-odroidn2-edge
. /usr/lib/u-boot/platform_install.sh
sudo write_uboot_platform "$DIR" "$(lsblk -no PKNAME "$(findmnt -no SOURCE /)")"
sudo reboot
# after reboot
sudo apt -y purge linux-dtb-current-meson64 linux-headers-current-meson64 linux-image-current-meson64
```
You could check now whether the driver shipped by the new kernel works, else again:
```sh
cd rtl8812au*
sudo make dkms_install
modprobe 88XXau
```
Now it works. So it looks like the recent version of this driver is not compatible with Linux v5.10 anymore. Additional hassle causes, that even if the internal version string suffix has been raised to [20220715](https://github.com/aircrack-ng/rtl8812au/commit/4ab079f), the driver still reports `v5.6.4.2_35491.20191025`, matching the one shipped with the kernel, so that DKMS is not installing the new one without passing the `--force` option.

wirelessjimp, 2022-08-02:
Unfortunately when I was able to try it this morning, I came up with this error when trying to install the packages:

```
root@DietPi:~/rtl8812au# sudo apt -y install linux-dtb-next-meson64 linux-headers-next-meson64 linux-image-next-meson64 linux-u-boot-odroidn2-edge. /usr/lib/u-boot/platform_install.sh
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
E: Unable to locate package linux-dtb-next-meson64
E: Unable to locate package linux-headers-next-meson64
E: Unable to locate package linux-image-next-meson64
```

MichaIng, 2022-08-02:
Ah sorry, it's "edge", not "next" in the package names. I edited it above.

wirelessjimp, 2022-08-02:
I have somehow managed to mangle the build so I need to wipe the install and start over fresh. Do I need to start from your comment from yesterday with the "fresh Odroid N2" steps or from the steps from today?

wirelessjimp, 2022-08-02:
Latest update. After getting stuck I went ahead and rebooted, and the RTL8812au card was working! The desktop, ssh, and xrdp were broken and didn't work, but the card was working. So, in an effort to start over fresh, I wiped the card and reinstalled. Went through the normal start-up and initial config stuff, and the card didn't work. So, I started over with the install part of the packages that were renamed to edge. That worked fine. Then, this happened:

```
root@DietPi:~# sudo write_uboot_platform "$DIR" "$(lsblk -no PKNAME "$(findmnt -no SOURCE /)")"
sudo: write_uboot_platform: command not found
```
I went ahead and rebooted the board as that was the next step. I had the rtl8812au card plugged in and upon reboot, it got a blue light on it which tells me it is working. Almost too well. I had to unplug it to gain access to the device through the Ethernet. Once back in I was able to run the last command to purge the packages, and now I can see this:

```
root@DietPi:~# iw wlan0 info
Interface wlan0
	ifindex 3
	wdev 0x1
	addr 00:c0:ca:ac:42:c3
	type managed
	wiphy 0
	channel 1 (2412 MHz), width: 20 MHz, center1: 2412 MHz
	txpower 20.00 dBm

```
I still need to do some more work to make sure the adapter is working as intended, but this is a very promising step in the right direction! I'll update here after I do more testing.

MichaIng, 2022-08-04:
`write_uboot_platform` is part of the sourced script the line before on the instructions. However/obviously, likely it is not needed. Older U-Boot usually works with newer kernel as well.

So the WiFi adapter is working now without compiling it via dkms? Or did you do that as well? Probably it got updated on edge kernel, fixing the issue.

wirelessjimp, 2022-08-04:
The dkms compile didn't work, but I also didn't need to do it. After I purged the packages I started to investigate the operation. When I went to `dietpi-config` and looked at the network adapters, it showed an active interface so I never went back to do those final three commands. 

I have also been able to confirm that the adapter works as intended and can scan both bands. I can also put it into monitor mode (my primary intention) and it works well in scanning all the spectrum the chip can support. I haven't tried packet or frame injection but I will try that eventually. For most people, I think this is all they would need to make the rtl8812au chip work, at least for the time being.

```
sudo apt -y install linux-dtb-edge-meson64 linux-headers-edge-meson64 linux-image-edge-meson64 linux-u-boot-odroidn2-edge . /usr/lib/u-boot/platform_install.sh
sudo reboot
# after reboot
sudo apt -y purge linux-dtb-current-meson64 linux-headers-current-meson64 linux-image-current-meson64
```
Just to put a final bow on this instance, this is what I am running currently:

```
DietPi v8.7.1
Odroid C4/HC4 (aarch64)
$ uname -a
Linux DietPi 5.17.5-meson64 #22.05.1 SMP PREEMPT Sat May 28 08:20:29 UTC 2022 aarch64 GNU/Linux
```

MichaIng, 2022-08-07:
Okay great, so it has been resolved with edge kernel already 🙂. It may still take a while until this gets merged into "current" kernel branch, so I leave this open. On Linux 5.10 an older version of the driver sources should build successfully, but we can find this out when someone else is landing here and moving to "edge" is not an option.

MichaIng, 2022-11-05:
I'm going to close this. There was quite some back and forth with the `88XXau`/`88XXcu` kernel driver in Armbian edge kernels, but now they reliable work with my RTL8811CU adapters.

I only hope it does not break again with Linux 6.0: https://github.com/aircrack-ng/rtl8812au/issues/1006, https://github.com/aircrack-ng/rtl8812au/pull/1008

Better maintained and on newer branch seems to be: https://github.com/morrownr/8821au-20210708
Probably we can suggest to Armbian to switch.

A pain that these drivers are not maintained upstream as part of Linux kernel itself, so that it does not rely on reports and fixes from the community when running into compiling issues with newer Linux versions...

closed 2022-11-05 by MichaIng