MichaIng/DietPi #4392: DietPi-Software | Gerbera

ravenclaw900, 2021-05-19:
### Creating a software request

#### Formal software information
- Software title | 
Gerbera
- Short description | 
UPnP media streamer
- Official URL (if available) | 
https://gerbera.io
- GitHub URL (if available) | 
https://github.com/gerbera/gerbera
- (Official) install documentation | 
https://docs.gerbera.io/en/stable/install.html

#### Are there similar/alternative software titles available with DietPi-Software?

- GMediaRender

#### What makes your requested software better than the above solutions, if available?

- UPnP media server, not control point

#### How can DietPi make the installation easier or compatible, than following the install instructions or do APT installation, if available?

- Automatically adding streaming directories, setting password for the web interface

#### Can you provide the installation steps that you would suggest DietPi-Software to do?

1. APT install gerbera
2. Edit `/etc/gerbera/config.xml`

#### Are you willing to help maintaining the software installation, e.g. in case of needed setup changes due to updates etc.? This is not needed, but could speed up our decision to implement it, as man power is always a topic :wink:.

- Probably

ravenclaw900, 2021-05-19:
I just found ReadyMedia, and the 2 servers are actually not that different.  I'm going to close this in favor of not adding another unneeded media server.

closed 2021-05-19 by ravenclaw900

MichaIng, 2021-05-20:
Not bad to have alternatives, but as long as ReadyMedia or GMediaRender servers your needs, probably other new software has larger benefit.