MichaIng/DietPi #4393: dietpi-banner:  WAN IP : curl: (22) The requested URL returned error: 503

maartenlangeveld, 2021-05-19:
Recently dietpi-banner gives an Http 503 error for WAN IP.

```
root@M-DNS:~# dietpi-banner
 ─────────────────────────────────────────────────────
 DietPi v7.1.2 : 13:19 - Wed 05/19/21
 ─────────────────────────────────────────────────────
 - Device model : NanoPi NEO2 (aarch64)
 - Uptime : up 43 minutes
 - CPU temp : 34'C : 93'F (Cool runnings)
 - FQDN/hostname : M-DNS
 - LAN IP : 192.xxx.xxx.xxx (eth0)
 - WAN IP : curl: (22) The requested URL returned error: 503
 - Freespace (RootFS) : 23G
 - MOTD : Did you know that you can run "dietpi-banner" to change its content?
 ─────────────────────────────────────────────────────
```

Joulinar, 2021-05-19:
Hi,

Might be a temporary issue with the provider we get the WAN IP from. Try later again.

ravenclaw900, 2021-05-19:
Indeed, it looks like it is a problem on their side.  See https://ifconfig.co/.

Joulinar, 2021-05-19:
Yep quite unstable service the last time.

MichaIng, 2021-05-19:
Indeed. Also it has some rate limiting now, I think 5 call in not too short time range, e.g. causing issues when playing a bit with DietPi-VPN, which obtains and prints the WAN IP as part of the main menu.

Probably we should revert to our own IP service: https://dietpi.com/myip.php
But it lacks country/state info, so that would require another service/API to be deployed.

Joulinar, 2021-05-19:
Probably more important to have a valid WAN IP in first place. But yeah country information would be good for people using VPN to ensure being connected to correct side. Not sure if there are other providers offering something like this.

Joulinar, 2021-05-20:
Service should be working again

maartenlangeveld, 2021-05-20:
Thanks, but still does not work or does not work again ;-)
Not that import for me so close this issue if you prefer.

Joulinar, 2021-05-20:
Well the service seems to be down again or overloaded

MichaIng, 2021-05-20:
Probably we should switch, e.g.: https://ipapi.co/json/
Open source API as well: https://github.com/ipapi-co/ipapi-python

One that suites us even better, allowing to return just the info we need separated by newline only (no JSON): https://ipwhois.app/line/?objects=ip,country,region

Joulinar, 2021-05-20:
The first one is something to be installed right? So we would need to obtain the IP first before geo details can be requested. Am I correct?

MichaIng, 2021-05-20:
Nope, all those return the IP of the client automatically. It should all be done in a single request, otherwise it can be too slow.

MichaIng, 2021-05-20:
Sadly ipapi.co is the slowest service:
```
2021-05-20 18:00:09 root@micha:/tmp# time curl -s https://ipapi.co/json > /dev/null

real    0m0.523s
user    0m0.181s
sys     0m0.038s
2021-05-20 18:00:14 root@micha:/tmp# time curl -s https://ipapi.co/json > /dev/null

real    0m0.560s
user    0m0.192s
sys     0m0.045s
2021-05-20 18:00:15 root@micha:/tmp# time curl -s https://dietpi.com/myip.php > /dev/null

real    0m0.466s
user    0m0.201s
sys     0m0.037s
2021-05-20 18:00:25 root@micha:/tmp# time curl -s https://dietpi.com/myip.php > /dev/null

real    0m0.347s
user    0m0.171s
sys     0m0.046s
2021-05-20 18:00:26 root@micha:/tmp# time curl -s https://dietpi.com/myip.php > /dev/null

real    0m0.387s
user    0m0.175s
sys     0m0.041s
2021-05-20 18:00:27 root@micha:/tmp# time curl -s https://dietpi.com/myip.php > /dev/null

real    0m0.349s
user    0m0.161s
sys     0m0.059s
2021-05-20 18:00:29 root@micha:/tmp# time curl -s https://ipapi.co/json > /dev/null

real    0m0.524s
user    0m0.155s
sys     0m0.074s
2021-05-20 18:00:33 root@micha:/tmp# time curl -s https://ipwhois.app/line/?objects=ip,country,region > /dev/null

real    0m0.410s
user    0m0.176s
sys     0m0.074s
2021-05-20 18:00:52 root@micha:/tmp# time curl -s https://ipwhois.app/line/?objects=ip,country,region > /dev/null

real    0m0.345s
user    0m0.158s
sys     0m0.055s
2021-05-20 18:00:53 root@micha:/tmp# time curl -s https://ipwhois.app/line/?objects=ip,country,region > /dev/null

real    0m0.388s
user    0m0.205s
sys     0m0.032s
```
But probably that depends on the location of the client as well. Tested on dietpi.com server, and ipwhois.app is much faster there,   less than 0.1 seconds, ipapi.co ~0.4 seconds.

Compromise: https://freegeoip.app/csv/
- It's quite fast
- It supports IPv6
- Rate limit more than sufficient
- But its location database is not as complete as https://ipapi.co/json, e.g. on dietpi.com, it finds the country, but not the city or even region. On my home service it finds both, at least.

Enough searching and testing for this, I'll go with the last above.

Joulinar, 2021-05-20:
Well it seems location dependent. For me it's quite fast

```
root@DietPi4:~#  time curl -s https://ipwhois.app/line/?objects=ip,country,region > /dev/null

real0m0.147s
user0m0.033s
sys0m0.012s
root@DietPi4:~#
```

MichaIng, 2021-05-20:
Yes that one is fast as well, but IPv4 only. How about those in your case?
- https://ipapi.co/json
- https://freegeoip.app/csv/
- https://reallyfreegeoip.org/csv/

MichaIng, 2021-05-20:
Done: https://github.com/MichaIng/DietPi/pull/4398
Little bonus of CSV vs JSON output is that it's much easier to parse it with awk 😃.

closed 2021-05-23 by MichaIng