MichaIng/DietPi #4395: DietPi-Software | Filtron

ravenclaw900, 2021-05-19:
### Creating a software request

#### Formal software information
- Software title | 
Filtron
- Short description | 
Reverse proxy
- Official URL (if available) | 
N/A
- GitHub URL (if available) | 
https://github.com/asciimoo/filtron
- (Official) install documentation | 
https://searx.github.io/searx/utils/filtron.sh.html

#### Are there similar/alternative software titles available with DietPi-Software?

- Nginx

#### What makes your requested software better than the above solutions, if available?

- Can use any webserver, webUI

#### How can DietPi make the installation easier or compatible, than following the install instructions or do APT installation, if available?

- Make it non-interactive, don't double-install Go

#### Can you provide the installation steps that you would suggest DietPi-Software to do?

1. Install go
2. Create service user
3. Create config files

#### Are you willing to help maintaining the software installation, e.g. in case of needed setup changes due to updates etc.? This is not needed, but could speed up our decision to implement it, as man power is always a topic :wink:.

- Probably

MichaIng, 2021-05-20:
Seems to be comparable to HAProxy, a standalone proxy with specifically filtering and rate limiting in mind, so a much lighter alternative to e.g. Nginx or any other fully-fledged webserver, when only a proxy is required.

Pre-compiled binaries are available as well: https://github.com/asciimoo/filtron/releases
Only arm64 binaries are missing, although the arm binary can be used on those as well:
```
2021-05-20 19:12:17 root@micha:/tmp# curl -sSfLo filtron https://github.com/asciimoo/filtron/releases/download/v0.2.0/filtron_linux_arm
2021-05-20 19:12:31 root@micha:/tmp# chmod +x filtron
2021-05-20 19:12:34 root@micha:/tmp# ./filtron --help
Usage of ./filtron:
  -api string
        API listen address (default "127.0.0.1:4005")
  -listen string
        Proxy listen address (default "127.0.0.1:4004")
  -read-buffer-size int
        Read buffer size (default 16384)
  -rules string
        JSON rule list (default "rules.json")
  -target string
        Target address for reverse proxy (default "127.0.0.1:8888")
  -version
        Version information

2021-05-20 19:14:53 root@micha:/tmp# curl -sSfLo rules.json https://raw.githubusercontent.com/asciimoo/filtron/v0.2.0/example_rules.json
2021-05-20 19:15:31 root@micha:/tmp# ./filtron &
[1] 4031
2021/05/20 19:15:34 7 rules loaded from rules.json
2021/05/20 19:15:34 Proxy listens on 127.0.0.1:4004
2021/05/20 19:15:34 API listens on 127.0.0.1:4005

2021-05-20 19:15:37 root@micha:/tmp# curl 127.0.0.1:4004
[log rule] 2021-05-20 19:15:46.136  GET 127.0.0.1:4004/ "" "curl/7.74.0"

2021-05-20 19:19:03 root@micha:/tmp# ps aux | grep filtron
root      4031  0.1  0.5 800076  5240 pts/1    Sl   19:15   0:00 ./filtron
```
- ~5 MiB memory usage (see above 5240 KiB).

So I suggest we skip the Go install and keep it simple, if it works as expected.

Joulinar, 2021-05-20:
Question is if it is something still being developed? It's in a version 0.2, no real PR over last half a year. Not that much activities on GitHub.

Joulinar, 2021-05-20:
interesting, pre-compiled binaries are available for all platforms.

but hard to understand the wiki 🤣 https://github.com/fatedier/frp/wiki

ravenclaw900, 2021-05-20:
Luckily, all the needed information is in the README.

ravenclaw900, 2021-05-22:
Did some testing with `frp`, works well, but slightly annoying to have 2 binaries running.

closed 2021-06-26 by MichaIng