MichaIng/DietPi #4396: Kodi video playback stutter

rainfallsevensamurai, 2021-05-20:
### Creating a bug report/issue

#### Required Information
- DietPi version | 7.1.2
- Distro version | Buster
- Kernel version | 4.9.241-arm64 #1 SMP PREEMPT Thu Feb 25 17:57:15 CET 2021 aarch64 GNU/Linux
- SBC model | Odroid C4/HC4 (aarch64)
- Power supply used | 12V 2A
- SDcard used | 32GB eMMC Module

#### Additional Information (if applicable)
- Software title | Kodi
- Was the software title installed freshly or updated/migrated? Freshly installed using dietpi-software
- Can this issue be replicated on a fresh installation of DietPi?

- Bug report ID | `echo $G_HW_UUID`

#### Steps to reproduce

1. Obtain Odroid C4
2. Flash image from dietpi website
3. Mount NAS drive using dietpi-config
4. Install Kodi using dietpi-software
5. Set Kodi to autostart
6. Reboot
7. Connect Odroid to TV
8. Open file from NAS to play in Kodi

#### Expected behaviour

- The video and sound should play smoothly 

#### Actual behaviour

- The video stutters and the sound is barely audible.
- If you pause the video it continues a while until it 'catches up' 

#### Extra details

- Installed software: PiHole, Unbound, Deluge (which is also not working: service is running but cant be reach via browser or application)
- I'm not sure how fast they should be, but there's a wired connection from the Odroid to the router and a wired connection from the router to the NAS. If I run the provided speed check, I got 

DietPi-Benchmark 
Filesystem Benchmark Results:  

- Filepath  = /mnt/nas218video/benchmark.file           
- Test size = 100 MiB    
- WRITE     = 4 MiB/s     
- READ      = 6 MiB/s  

I found a related thread here: https://github.com/MichaIng/DietPi/issues/2584

At the moment I'm really unsure of what's happening. The Odroid seems to be under no stress at all playing the video. Yesterday I killed the kodi process from ssh and started it myself using kodi --standalone. When I played a video I got messages in my ssh session like: 

codec_init amstream version 2.0
[codec_pause]p->has_audio=0
video_pause!
...etc

I tried playing videos in different formats, h264 and h265, etc.

Anything I get check to help investigating? Thanks so much for your time!

MichaIng, 2021-05-20:
Many thanks for your report.

You read he file from a network drive?
6 MiB/s read speed is very slow. Did you test it from a local hard drive or with an SD quality video?

rainfallsevensamurai, 2021-05-20:
I'll try to test with a 'local' file which I'll upload with FileZilla to `/mnt/dietpi_userdata`. I'll get back to you.

rainfallsevensamurai, 2021-05-20:
Uploaded the file and added it to Kodi. Plays just as bad. Exactly the same behavior. Info from Kodi (uses `o` to display information on screen while playing):

`Video decoder: am-h265 (HW)`

MichaIng, 2021-05-21:
Btw, you did install the Odroid build of Kodi, right (done when using `dietpi-software`)?
```
dpkg -l | grep -E '(kodi|aml|mali)'
```
And required kernel modules are loaded?
```
lsmod
```
And you did try to start Kodi as root user, right?

rainfallsevensamurai, 2021-05-22:
Yes, I did install using dietpi-software, autostart as root.

`dpkg -l | grep -E '(kodi|aml|mali)'`
`ii  aml-libs-odroid                    20190328-1+deb9               arm64        AmLogic libraries for video decoding`
`ii  gpg                                2.2.12-1+deb10u1              arm64        GNU Privacy Guard -- minimalist public key operations`
`ii  kodi-aml-fbdev-odroid              18.6-1+deb10                  arm64        Kodi - award-winning free and open source (GPL) software media player and entertainment hub (ODROID build)`
`ii  libhiredis0.14:arm64               0.14.0-3                      arm64        minimalistic C client library for Redis`
`ii  libmnl0:arm64                      1.0.4-2                       arm64        minimalistic Netlink communication library`
`ii  malig31-fbdev-opencl-odroid        20200503-r12p0-1              arm64        Mali binary blob and development headers (ODROID build)`

`lsmod`
`Module                  Size  Used by`
`md4                    16384  0`
`nls_utf8               16384  2`
`cifs                  475136  4`
`fscache                86016  1 cifs`
`nf_conntrack_ipv4      20480  2`
`nf_defrag_ipv4         16384  1 nf_conntrack_ipv4`
`nft_chain_route_ipv4    16384  1`
`nf_tables_ipv4         16384  8`
`nf_conntrack_ipv6      20480  2`
`nf_defrag_ipv6         20480  1 nf_conntrack_ipv6`
`nft_chain_route_ipv6    16384  1`
`xt_connmark            16384  4`
`nf_conntrack          143360  3 nf_conntrack_ipv6,nf_conntrack_ipv4,xt_connmark`
`xt_mark                16384  2`
`nft_counter            16384  6`
`nft_meta               16384  6`
`nf_tables_ipv6         16384  8`
`xt_comment             16384  0`
`xt_addrtype            16384  3`
`nft_compat             20480  8`
`nf_tables              86016  19 nft_compat,nf_tables_ipv6,nf_tables_ipv4,nft_chain_route_ipv6,nft_meta,nft_chain_route_ipv4,nft_counter`
`wireguard             122880  0`
`ip6_udp_tunnel         16384  1 wireguard`
`udp_tunnel             16384  1 wireguard`
`amvdec_ports          200704  0`
`i2c_meson_master       20480  0`
`amvdec_vp9            122880  0`
`amvdec_vc1             65536  0`
`amvdec_real            45056  0`
`amvdec_mmpeg4          40960  0`
`amvdec_mpeg4           65536  0`
`amvdec_mpeg12         106496  0`
`amvdec_mmjpeg          28672  0`
`amvdec_mjpeg           40960  0`
`amvdec_h265           147456  0`
`amvdec_h264mvc         57344  0`
`amvdec_mh264          159744  0`
`amvdec_h264           147456  0`
`amvdec_avs             69632  0`
`stream_input          180224  9 amvdec_h265,amvdec_mh264,amvdec_h264mvc,amvdec_ports,amvdec_real,amvdec_vp9,amvdec_h264,amvdec_mpeg12,amvdec_avs`
`decoder_common        188416  15 amvdec_h265,amvdec_mjpeg,amvdec_mh264,amvdec_mmpeg4,amvdec_h264mvc,amvdec_mmjpeg,amvdec_ports,amvdec_real,stream_input,amvdec_vp9,amvdec_h264,amvdec_mpeg12,amvdec_avs,amvdec_vc1,amvdec_mpeg4`
`firmware               28672  15 amvdec_h265,amvdec_mjpeg,amvdec_mh264,amvdec_mmpeg4,amvdec_h264mvc,amvdec_mmjpeg,decoder_common,amvdec_real,stream_input,amvdec_vp9,amvdec_h264,amvdec_mpeg12,amvdec_avs,amvdec_vc1,amvdec_mpeg4`
`media_clock            45056  10 amvdec_h265,amvdec_mh264,decoder_common,amvdec_ports,firmware,stream_input,amvdec_vp9,amvdec_h264,amvdec_mpeg12,amvdec_avs`
`ip_tables              32768  0`
`x_tables               49152  6 xt_comment,ip_tables,nft_compat,xt_mark,xt_connmark,xt_addrtype`
`ipv6                  462848  86 nf_conntrack_ipv6,nft_chain_route_ipv6,wireguard`
`spidev                 20480  0`
`spi_meson_spicc        20480  0`

MichaIng, 2021-05-22:
That all looks pretty correct. I'll have a look into the Kodi executable script...

Can you try to start it (as root) without `--standalone` flag?
```
kodi
```

rainfallsevensamurai, 2021-05-22:
Did that. Played a local file from /mnt/dietpi-userdata/. Still the same stutter.

```
[codec_reset:1220]ret=0
[codec_pause]p->has_audio=0
video_pause!
[codec_reset:1220]ret=0
[codec_pause]p->has_audio=0
video_pause!
[codec_reset:1220]ret=0
[codec_pause]p->has_audio=0
video_pause!
[codec_reset:1220]ret=0
[codec_pause]p->has_audio=0
video_pause!
[codec_reset:1220]ret=0
[codec_pause]p->has_audio=0
video_pause!
[codec_reset:1220]ret=0
```

This goes on and on and on.

rainfallsevensamurai, 2021-05-23:
Ok, I did a clean install, flashed the mmc module again I mean. Installed Kodi, used dietpi-drive_manager to mount my NAS. Opened Kodi and added some files from the NAS to the library. The same problem, fps would not go over 10 when playing a video. Then I started tinkering with Kodi settings. Found this:

`Settings > Player > Videos > Processing > Allow hardware acceleration - amcodes: True`

When I set it to `False` all videos started playing great with fps > 29. I'm still confused however :)

MichaIng, 2021-05-24:
Hmm, these should be the Amlogic hardware acceleration which we explicitly try to have enabled here 🤔.

Another test, starting Kodi via:
```
/usr/local/lib/kodi/kodi-aml
```

And can you check the following:
```
ls -lL /sys/class/video/axis /sys/class/video/screen_mode /sys/class/video/disable_video /sys/class/tsync/pts_pcrscr /sys/class/audiodsp/digital_raw /sys/class/audiodsp/digital_raw /sys/kernel/debug/aml_reg/paddr /sys/module/ionvideo/parameters/freerun_mode /sys/class/video/freerun_mode /sys/class/amhdmitx/amhdmitx0/config /sys/class/vfm/map /sys/class/tsync/enable /sys/class/amhdmitx/amhdmitx0/frac_rate_policy /sys/class/display/mode
find /sys/class/ -maxdepth 2 -name 'am*' -exec ls -lL {} +
find /sys/class/mics/ -maxdepth 1 -name 'mali*' -exec ls -lL {} +
```

rainfallsevensamurai, 2021-05-25:
I started Kodi again using the code you provided. I re-enabled the hardware acceleration in settings and tried to play a video. None would even start playing. Memory and CPU go way up, but nothing happens.

```
# ls -lL /sys/class/video/axis /sys/class/video/screen_mode /sys/class/video/disable_video /sys/class/tsync/pts_pcrscr /sys/class/audiodsp/digital_raw /sys/class/audiodsp/digital_raw /sys/kernel/debug/aml_reg/paddr /sys/module/ionvideo/parameters/freerun_mode /sys/class/video/freerun_mode /sys/class/amhdmitx/amhdmitx0/config /sys/class/vfm/map /sys/class/tsync/enable /sys/class/amhdmitx/amhdmitx0/frac_rate_policy /sys/class/display/mode
-rw-rw-r-- 1 root root 4096 May 23 23:27 /sys/class/amhdmitx/amhdmitx0/config
-rw-rw-r-- 1 root root 4096 May 23 23:27 /sys/class/amhdmitx/amhdmitx0/frac_rate_policy
-rw-rw-r-- 1 root root 4096 Jan  1  1970 /sys/class/audiodsp/digital_raw
-rw-rw-r-- 1 root root 4096 May 25 07:47 /sys/class/audiodsp/digital_raw
-rw-r--r-- 1 root root 4096 Jan  1  1970 /sys/class/display/mode
-rw-rw-r-- 1 root root 4096 Jan  1  1970 /sys/class/tsync/enable
-rw-rw-r-- 1 root root 4096 Jan  1  1970 /sys/class/tsync/pts_pcrscr
-rw-rw-r-- 1 root root 4096 Jan  1  1970 /sys/class/vfm/map
-rw-rw-r-- 1 root root 4096 Jan  1  1970 /sys/class/video/axis
-rw-rw-r-- 1 root root 4096 Jan  1  1970 /sys/class/video/disable_video
-rw-r--r-- 1 root root 4096 Jan  1  1970 /sys/class/video/freerun_mode
-rw-rw-r-- 1 root root 4096 Jan  1  1970 /sys/class/video/screen_mode
-r--r----- 1 root root    0 Jan  1  1970 /sys/kernel/debug/aml_reg/paddr
-rw-rw-r-- 1 root root 4096 May 23 23:27 /sys/module/ionvideo/parameters/freerun_mode
root@DietPi:~# find /sys/class/ -maxdepth 2 -name 'am*' -exec ls -lL {} +
--w-------  1 root root 4096 Jan  1  1970 /sys/class/efuse/amlogic_set
-r--r--r--  1 root root 4096 Jan  1  1970 /sys/class/media-configs/amports
-r--r--r--  1 root root 4096 May 23 23:27 /sys/class/vdec/amrisc_regs

/sys/class/amaudio:
total 0
drwxr-xr-x 3 root root    0 May 25 07:47 amaudio
-rw-rw-r-- 1 root root 4096 Jan  1  1970 audio_channels_mask
-rw-r--r-- 1 root root 4096 Jan  1  1970 dac_mute_const
-rw-rw-r-- 1 root root 4096 Jan  1  1970 debug
-r--r--r-- 1 root root 4096 Jan  1  1970 dolby_enable
-r--r--r-- 1 root root 4096 Jan  1  1970 dts_enable
-rw-r--r-- 1 root root 4096 Jan  1  1970 mute_left_right
-rw-r--r-- 1 root root 4096 Jan  1  1970 mute_unmute
-r--r--r-- 1 root root 4096 Jan  1  1970 output_enable
-rw-r--r-- 1 root root 4096 Jan  1  1970 record_type

/sys/class/amaudio-dev:
total 0
drwxr-xr-x 3 root root 0 May 25 07:47 amaudio_ctl
drwxr-xr-x 3 root root 0 May 25 07:47 amaudio_utils

/sys/class/amaudio-dev/amaudio_ctl:
total 0
-r--r--r-- 1 root root 4096 May 25 07:47 dev
drwxr-xr-x 2 root root    0 May 25 07:47 power
drwxr-xr-x 2 root root    0 May 25 07:47 subsystem
-rw-r--r-- 1 root root 4096 Jan  1  1970 uevent

/sys/class/amaudio-dev/amaudio_utils:
total 0
-r--r--r-- 1 root root 4096 May 25 07:47 dev
drwxr-xr-x 2 root root    0 May 25 07:47 power
drwxr-xr-x 2 root root    0 May 25 07:47 subsystem
-rw-r--r-- 1 root root 4096 Jan  1  1970 uevent

/sys/class/amaudio/amaudio:
total 0
-r--r--r-- 1 root root 4096 May 25 07:47 dev
drwxr-xr-x 2 root root    0 May 25 07:47 power
drwxr-xr-x 2 root root    0 May 25 07:47 subsystem
-rw-r--r-- 1 root root 4096 Jan  1  1970 uevent

/sys/class/amcanvas:
total 0
-rw-rw-r-- 1 root root 4096 Jan  1  1970 canvas_pool_debug
-r--r--r-- 1 root root 4096 Jan  1  1970 canvas_pool_map
-r--r--r-- 1 root root 4096 Jan  1  1970 canvas_pool_states

/sys/class/amdolby_vision:
total 0
[... 325 lines cut ...]
total 0
-r--r--r-- 1 root root 4096 May 25 07:47 dev
drwxr-xr-x 2 root root    0 May 25 07:47 power
drwxr-xr-x 2 root root    0 May 23 23:27 subsystem
-rw-r--r-- 1 root root 4096 May 23 23:27 uevent

/sys/class/video/amvideo:
total 0
-r--r--r-- 1 root root 4096 May 25 07:47 dev
drwxr-xr-x 2 root root    0 May 25 07:47 power
drwxr-xr-x 2 root root    0 Jan  1  1970 subsystem
-rw-r--r-- 1 root root 4096 Jan  1  1970 uevent

/sys/class/video_poll/amvideo_poll:
total 0
-r--r--r-- 1 root root 4096 May 25 07:47 dev
drwxr-xr-x 2 root root    0 May 25 07:47 power
drwxr-xr-x 2 root root    0 Jan  1  1970 subsystem
-rw-r--r-- 1 root root 4096 Jan  1  1970 uevent
```

```
root@DietPi:~# find /sys/class/mics/ -maxdepth 1 -name 'mali*' -exec ls -lL {} +
find: ‘/sys/class/mics/’: No such file or directory 
```

Joulinar, 2021-05-25:
> root@DietPi:~# find /sys/class/mics/ -maxdepth 1 -name 'mali*' -exec ls -lL {} +
find: ‘/sys/class/mics/’: No such file or directory 

There is a typo. It should be following
`find /sys/class/misc/ -maxdepth 1 -name 'mali*' -exec ls -lL {} +`

I will correct it above as well

rainfallsevensamurai, 2021-05-25:
```
find /sys/class/misc/ -maxdepth 1 -name 'mali*' -exec ls -lL {} +
total 0
-r--r--r-- 1 root root 4096 May 25 09:51 dev
drwxr-xr-x 4 root root    0 Jan  1  1970 device
drwxr-xr-x 2 root root    0 May 25 09:51 power
drwxr-xr-x 2 root root    0 Jan  1  1970 subsystem
-rw-r--r-- 1 root root 4096 Jan  1  1970 uevent
```

rainfallsevensamurai, 2021-05-28:
Sorry for the delay. Above suggestion had no effect, whenever I enable HA the videos won’t load or play extremely slow.

MichaIng, 2021-05-28:
Okay. I'll check this back with Meveric. Good to know that disabling hardware acceleration leads to acceptable frame rates, although CPU usage will be high I guess (?).

rainfallsevensamurai, 2021-06-01:
Yes it is. Thanks for checking...

MichaIng, 2021-09-09:
@meveric
Another one where I'm not sure what the issue is and if we miss something:
- Mali driver is installed
- Amlogic libraries are installed
- The related Kodi package is installed
- Kodi is started outside of X11
- Permissions are not the issue, Kodi runs as root user anyway
- `amcodes` for hardware acceleration is found and enabled, but it seems to work even better when disabled

meveric, 2021-09-12:
Hardware accelerated video decoding on C4 is always a bit of a hassle for me.
Overall the basics all seem to be correct, but I also experienced issues with hardware decoding on the C4 via Kodi.
My guess is that maybe the aml-libs-odroid could cause some issues, and I have to try if the once from HardKernel work anyway different.
Problem with those is, that they are not packaged correctly and I have to rebuild the package with the correct placement of files first before I can try this.

It's a long time standing issue for me, with very little priority up to now, as most people normally use OpenELEC or something instead.

I need time to investigate this, please don't expect any results in the next couple of days.

MichaIng, 2021-12-08:
I'll mark this as closed. New Bullseye image uses the Debian Kodi build and GBM when not started from desktop, so things may be pretty different. I'll do tests on Odroid N2 currently, and result should be pretty similar on C4.

rainfallsevensamurai, 2021-12-13:
Today I tried dietpi again, because I was notified by github about this issue being closed. Did a clean install using a fresh download. Used dietpi-software to install Kodi. It doesn't even start. 
```

Failed to open zone.tab
ERROR: Unable to create GUI. Exiting
```

I know you're working very hard on dietpi and I respect the huge amount of support you're giving, but I can't understand why this issue got closed while the problem got worse.

MichaIng, 2021-12-13:
Many thanks for your feedback. I have an Odroid N2 here since a few days an will do Kodi testing on it, which should behave very similar to the C4.

How did you try to execute Kodi? It should be the package from Debian now which has X11 support as well as GBM support included, so it should work in any way. Important only to not start it from desktop via GBM, as DRM allows only one exclusive device access.

rainfallsevensamurai, 2021-12-14:
This from the kodi.log (/root/.kodi/temp/)

```
2021-12-14 18:06:22.016 T:3893    ERROR <general>: GetDirectory - Error getting /usr/lib/aarch64-linux-gnu/kodi/addons
2021-12-14 18:06:22.016 T:3893    ERROR <general>: GetDirectory - Error getting special://xbmcbin/addons
...
2021-12-14 18:06:22.078 T:3893    ERROR <general>: DBus error: org.freedesktop.DBus.Error.ServiceUnknown - The name org.freedesktop.UPower was not provide>
...
2021-12-14 18:06:22.403 T:3893    ERROR <general>: CDRMUtils::OpenDrm - no drm devices found: (No such file or directory)
2021-12-14 18:06:22.403 T:3893    ERROR <general>: CWinSystemGbm::InitWindowSystem - failed to initialize Atomic DRM
2021-12-14 18:06:22.403 T:3893    ERROR <general>: CDRMUtils::OpenDrm - no drm devices found: (No such file or directory)
2021-12-14 18:06:22.403 T:3893    ERROR <general>: CWinSystemGbm::InitWindowSystem - failed to initialize Legacy DRM
2021-12-14 18:06:22.403 T:3893    ERROR <general>: CDRMUtils::OpenDrm - no drm devices found: (No such file or directory)
2021-12-14 18:06:22.404 T:3893    ERROR <general>: CWinSystemGbm::InitWindowSystem - failed to initialize off screen DRM
2021-12-14 18:06:22.930 T:3893    FATAL <general>: CApplication::CreateGUI - unable to init windowing system
2021-12-14 18:06:22.934 T:3893     INFO <general>: Stopping player

```

MichaIng, 2021-12-14:
> GetDirectory - Error getting /usr/lib/aarch64-linux-gnu/kodi/addons

Hmm, the Kodi package itself does not ship this directory or any addons. Probably you have/had `kodi-addons-odroid` installed, or manually installed and enabled addons in Kodi, and the new version now looks for them at a different place?

Please check the following:
```sh
dpkg -l | grep kodi
```
Although you did that already above, and mentioned package was not installed.

Please check whether starting Kodi with a clean config works:
```sh
mv ~/.kodi ~/.kodi_bak
kodi
```

And does this exist?
```sh
ls -l /usr/local/lib/kodi/addons
ls -l /usr/lib/kodi/addons
```

About the DRM errors, I hope GBM is generally supported by this C4 kernel. Can you check:
```sh
ls -l /dev/dri
```

Joulinar, 2021-12-14:
The missing addon directory I faced as well on my test system today. Looks like it has been introduced with 19.3

rainfallsevensamurai, 2021-12-14:
I did a complete clean install of dietpi, so no legacy settings.

```

root@DietPi:~/.kodi/temp# dpkg -l | grep kodi
ii  kodi                               2:19.1+dfsg2-2                    arm64        Open Source Home Theatre (executable binaries)
ii  kodi-bin                           2:19.1+dfsg2-2                    arm64        Open Source Home Theatre (architecture-dependent files)
ii  kodi-data                          2:19.1+dfsg2-2                    all          Open Source Home Theatre (arch-independent data package)
ii  kodi-repository-kodi               2:19.1+dfsg2-2                    all          Open Source Home Theatre (official addons repository feed)

```
...
```

root@DietPi:~/.kodi/temp# ls -l /usr/local/lib/kodi/addons
ls -l /usr/lib/kodi/addons
ls: cannot access '/usr/local/lib/kodi/addons': No such file or directory
ls: cannot access '/usr/lib/kodi/addons': No such file or directory
root@DietPi:~/.kodi/temp# ls -l /dev/dri
ls: cannot access '/dev/dri': No such file or directory

```
Clean config for Kodi gives the same error.

rainfallsevensamurai, 2021-12-14:
That didn’t really go as planned I guess:

```
dpkg --force-depends -P malig31-fbdev-opencl-odroid aml-libs-odroid
(Reading database ... 30652 files and directories currently installed.)
Removing aml-libs-odroid (20190328-1+deb9) ...
Purging configuration files for aml-libs-odroid (20190328-1+deb9) ...
dpkg: warning: while removing aml-libs-odroid, directory '/usr/local/lib' not empty so not removed
dpkg: malig31-fbdev-opencl-odroid: dependency problems, but removing anyway as you requested:
 kodi-bin depends on libegl1; however:
  Package libegl1 is not installed.
  Package malig31-fbdev-opencl-odroid which provides libegl1 is to be removed.

Removing malig31-fbdev-opencl-odroid (20200503-r12p0-1) ...
Purging configuration files for malig31-fbdev-opencl-odroid (20200503-r12p0-1) ...
root@DietPi:~# apt install --reinstall libegl1 libgles2 libdrm2 libgbm1
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
You might want to run 'apt --fix-broken install' to correct these.
The following packages have unmet dependencies:
 libegl1 : Depends: libegl-mesa0 but it is not going to be installed
E: Unmet dependencies. Try 'apt --fix-broken install' with no packages (or specify a solution).
root@DietPi:~# kodi
/usr/lib/aarch64-linux-gnu/kodi/kodi.bin: error while loading shared libraries: libEGL.so.1: cannot open shared object file: No such file or directory
root@DietPi:~#
```

rainfallsevensamurai, 2021-12-14:
```
root@DietPi:~# apt install libegl-mesa0
Reading package lists... 0%
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
You might want to run 'apt --fix-broken install' to correct these.
The following packages have unmet dependencies:
 kodi-bin : Depends: libegl1
E: Unmet dependencies. Try 'apt --fix-broken install' with no packages (or specify a solution).
root@DietPi:~# 
```

MichaIng, 2021-12-14:
APT hassle 😄:
```sh
apt install libegl-mesa0 libegl1
```

rainfallsevensamurai, 2021-12-15:
That worked, but the problem persists. Kodi doesn't start. Here are the errors from the kodi.log file again:

```
2021-12-15 08:49:44.491 T:33134   ERROR <general>: GetDirectory - Error getting /usr/lib/aarch64-linux-gnu/kodi/addons
2021-12-15 08:49:44.491 T:33134   ERROR <general>: GetDirectory - Error getting special://xbmcbin/addons
2021-12-15 08:49:44.552 T:33134   ERROR <general>: DBus error: org.freedesktop.DBus.Error.ServiceUnknown - The name org.freedesktop.UP>
2021-12-15 08:49:44.882 T:33134   ERROR <general>: CDRMUtils::OpenDrm - no drm devices found: (No such file or directory)
2021-12-15 08:49:44.882 T:33134   ERROR <general>: CWinSystemGbm::InitWindowSystem - failed to initialize Atomic DRM
2021-12-15 08:49:44.882 T:33134   ERROR <general>: CDRMUtils::OpenDrm - no drm devices found: (No such file or directory)
2021-12-15 08:49:44.882 T:33134   ERROR <general>: CWinSystemGbm::InitWindowSystem - failed to initialize Legacy DRM
2021-12-15 08:49:44.883 T:33134   ERROR <general>: CDRMUtils::OpenDrm - no drm devices found: (No such file or directory)
2021-12-15 08:49:44.883 T:33134   ERROR <general>: CWinSystemGbm::InitWindowSystem - failed to initialize off screen DRM
2021-12-15 08:49:45.405 T:33134   FATAL <general>: CApplication::CreateGUI - unable to init windowing system
```

I really appreciate the work you're doing, but maybe I would like to take a step back and remind myself why I'm doing this. I just wanted to run Kodi and some other applications on the same machine. This shouldn't be that hard. Before you closed the ticket I used CoreELEC for a while. It really just worked great. I gave dietpi another change because I really want to run pihole and deluge on the same machine. I don't want to sound entitled, but maybe dietpi should inform potential new users of its supported applications on the website. Sometimes I feel like I'm the only one that uses dietpi on a Odroid C4 and have this strange request I want to use this little, obscure piece of software called Kodi :D 

Again, I'm willing and able to debug and test for a certain period, but I would also like to see a movie once in a while. If dietpi doesn't support kodi on a C4, that's ok. But then I would like to know, so I can move on.

MichaIng, 2021-12-15:
Have you tried what I suggested above, running it via X server?
```sh
xinit "$(which kodi)"
```
If the X server (`xinit`) not installed yet:
```
dietpi-software install 6
```
> I don't want to sound entitled, but maybe dietpi should inform potential new users of its supported applications on the website.

That would be great, however, as long as we do not have individual "maintainers" for every software title which then also do test it on every SBC, or the other way round, this currently cannot be achieved. Very most software works fine on every SBC and implementations break rarely, but GUI/GPU related software is really a special difficult case, since it highly depends on the exact kernel, GPU drivers, libraries, GUI software builds etc and all of these need to work together. And I have not the SBCs and not the time to test each of them regularly on every SBC. And then e.g. on RPi different available sound drivers and configs come on top, which make it again more complicated. So it depends on user reports when something does not work.

In this particular case it was the upgrade from Buster to Bullseye which broke Kodi since on Odroids it is a dedicated Build (which utilises the framebuffer driver and not KMS/GBM/DRM), while there is no such Kodi build for Bullseye yet and I wasn't aware that those Odroids don't have a DRM interface at all, though it makes sense this being the reason for the fbdev/framebuffer drivers and builds 😉.  Such a Debian upgrade happens every two years, so naturally implementations can then break. It will be much more reliable when we switch to Debian's own packages, away from those special SBC-dependant builds, but obviously this requires a more generic kernel then as well. As said, I'll test the Armbian mainline kernel for Odroids N2 and C4, which hopefully provides the DRM interface so that we can use regular Debian GUI libraries and packages which are assured to be maintained and tested by Debian itself 🙂.

rainfallsevensamurai, 2021-12-16:
Thank you for your explanation!

I installed X server without problems and then tried your suggestion:

```
root@DietPi:~/.kodi/temp# xinit "$(which kodi)"

X.Org X Server 1.20.11
X Protocol Version 11, Revision 0
Build Operating System: linux Debian
Current Operating System: Linux DietPi 4.9.241-arm64 #1 SMP PREEMPT Thu Feb 25 17:57:15 CET 2021 aarch64
Kernel command line: root=UUID=e139ce78-9841-40fe-8823-96a304a09859 rootwait ro console=ttyS0,115200n8  no_console_suspend fsck.repair=yes net.ifnames=0 elevator=noop hdmimode=2160p30hz cvbsmode=576cvbs max_freq_a55= maxcpus=4 voutmode=hdmi  disablehpd=false cvbscable= overscan=100  monitor_onoff=false logo=osd0,loaded  sdrmode=auto enable_wol=0
Build Date: 13 April 2021  04:07:31PM
xorg-server 2:1.20.11-1 (https://www.debian.org/support)
Current version of pixman: 0.40.0
        Before reporting problems, check http://wiki.x.org
        to make sure that you have the latest version.
Markers: (--) probed, (**) from config file, (==) default setting,
        (++) from command line, (!!) notice, (II) informational,
        (WW) warning, (EE) error, (NI) not implemented, (??) unknown.
(==) Log file: "/var/log/Xorg.0.log", Time: Wed Dec 15 16:19:49 2021
(==) Using config directory: "/etc/X11/xorg.conf.d"
(==) Using system config directory "/usr/share/X11/xorg.conf.d"
failed to open zone.tab
/usr/lib/aarch64-linux-gnu/kodi/kodi-xrandr: Failed to get size of gamma for output default
libEGL warning: DRI2: failed to authenticate
libva info: VA-API version 1.10.0
libva error: vaGetDriverNameByIndex() failed with unknown libva error, driver_name = (null)
Failed to open VDPAU backend libvdpau_nvidia.so: cannot open shared object file: No such file or directory
sh: 1: nmblookup: not found
sh: 1: nmblookup: not found
sh: 1: nmblookup: not found
sh: 1: nmblookup: not found
sh: 1: nmblookup: not found
sh: 1: nmblookup: not found
sh: 1: nmblookup: not found
sh: 1: nmblookup: not found
```

This did in fact start Kodi 19.1 (!). I have not been able to test it yet, to see if it works, plays without stutter and utilizes the hardware acceleration. I'll keep you informed.

rainfallsevensamurai, 2021-12-17:
Ok that’s incredibly slow. I couldn’t even start a movie, the interface takes 2 seconds to respond to each input.

MichaIng, 2022-02-20:
We updated our image to use mainline Linux with generic DRM/KMS support. The issue shouldn't apply anymore, I hence mark this as closed.

closed 2022-02-20 by MichaIng