MichaIng/DietPi #4400: transmission install failed and now no software can be installed

pioniere, 2021-05-20:
### Creating a bug report/issue
#### Details:
- Date           | Thu May 20 23:04:08 BST 2021
- DietPi version | v7.0.2 (MichaIng/master)
- Image creator  | DietPi Core Team
- Pre-image      | Raspberry Pi OS Lite
- Hardware       | RPi 3 Model B+ (armv7l) (ID=3)
- Kernel version | Linux DietPi 5.10.17-v7+ #1414 SMP Fri Apr 30 13:18:35 BST 2021 armv7l GNU/Linux
- Distro         | buster (ID=5,RASPBIAN=1)
- Command        | apt-get -qq --allow-change-held-packages install deluged deluge-web deluge-console
- Exit code      | 100
- Software title | DietPi-Software
#### Steps to reproduce:

1. Try to install any software.
2. This error occurs and the install terminates
#### Expected behaviour:

- The software install should complete without error
#### Actual behaviour:

- The software install fails.
#### Extra details:

- This also occurred when trying to upgrade to version 7.1.2
#### Additional logs:
```
Setting up transmission-daemon (2.94-2+deb10u2) ...
chown: invalid group: ‘debian-transmission:debian-transmission’
dpkg: error processing package transmission-daemon (--configure):
 installed transmission-daemon package post-installation script subprocess returned error exit status 1
Errors were encountered while processing:
 transmission-daemon
E: Sub-process /usr/bin/dpkg returned an error code (1)
```

Joulinar, 2021-05-20:
Hi,

I guess you opened already another issue we closed because of missing feedback from your side https://github.com/MichaIng/DietPi/issues/4244

I would recommend to continue on the old one and have it reopen

Joulinar, 2021-05-29:
Going to close this one as duplicate. If needed feel free to reopen the original issue.

closed 2021-05-29 by Joulinar