MichaIng/DietPi #4401: error while running dietpi-update from v7.0.2 to v7.1.2.

k1r0s, 2021-05-21:
#### Details:
- Date           | Fri May 21 10:51:10 CEST 2021
- Bug report     | f9f80af8-1a20-449c-bebb-c941d9a2bc30
- DietPi version | v7.0.2 (MichaIng/master)
- Image creator  | DietPi Core Team
- Pre-image      | Raspberry Pi OS 64-bit
- Hardware       | RPi 4 Model B (aarch64) (ID=4)
- Kernel version | Linux DietPi 5.10.17-v8+ #1403 SMP PREEMPT Mon Feb 22 11:37:54 GMT 2021 aarch64 GNU/Linux
- Distro         | buster (ID=5,RASPBIAN=0)
- Command        | apt-get -qq upgrade
- Exit code      | 100
- Software title | DietPi-Update
#### Steps to reproduce:

1. ...
2. ...
#### Expected behaviour:

- ...
#### Actual behaviour:

- ...
#### Extra details:

- ...
#### Additional logs:
```
E: Packages were downgraded and -y was used without --allow-downgrades.
```

k1r0s, 2021-05-21:
TL;DR

closed 2021-05-21 by k1r0s

MichaIng, 2021-05-21:
I guess this issue? https://github.com/MichaIng/DietPi/issues/4358#issuecomment-835786793

k1r0s, 2021-06-24:
sorry for the late asnwer. In my case I had to --allow-downgrades

MichaIng, 2021-06-24:
Which package has been downgraded? If it was `cmake`, then the next APT upgrade will upgrade it again, closing the loop 😉.