MichaIng/DietPi #4402: [node-RED] - "Depreciated" error message

MDAR, 2021-05-21:
Required Information
DietPi version | cat /boot/dietpi/.version
G_DIETPI_VERSION_CORE=7
G_DIETPI_VERSION_SUB=1
G_DIETPI_VERSION_RC=2
G_GITBRANCH='master'
G_GITOWNER='MichaIng'

Distro version | echo $G_DISTRO_NAME or cat /etc/debian_version

10.9

Kernel version | uname -a
Linux openHAB2-C4 4.9.241-arm64 #1 SMP PREEMPT Thu Feb 25 17:57:15 CET 2021 aarch64 GNU/Linux

SBC model | echo $G_HW_MODEL_NAME or (EG: RPi3)
Odroid C4/HC4 (aarch64)

Power supply used | (EG: 5V 1A RAVpower)
5V 2Amp

SDcard used | (EG: SanDisk ultra)
Emmc 16GB

#### Steps to reproduce
Install Node-RED from `diet-pi-software` [112]

Looks to go ahead correctly and seems to run happily.

These warnings are shown while installing.
```
npm WARN deprecated har-validator@5.1.5: this library is no longer supported
npm WARN deprecated request@2.88.0: request has been deprecated, see https://github.com/request/request/issues/3142
npm WARN deprecated bcrypt@3.0.8: versions < v5.0.0 do not handle NUL in passwords properly
npm WARN deprecated node-pre-gyp@0.12.0: Please upgrade to @mapbox/node-pre-gyp: the non-scoped node-pre-gyp package is deprecated and only the @mapbox scoped package will recieve updates in the future
npm WARN deprecated node-pre-gyp@0.14.0: Please upgrade to @mapbox/node-pre-gyp: the non-scoped node-pre-gyp package is deprecated and only the @mapbox scoped package will recieve updates in the future
npm WARN deprecated bcrypt@3.0.6: versions < v5.0.0 do not handle NUL in passwords properly
```

I'm having an issue with a specific set of nodes and wondered if the NUL password issue above could be my problem?

I have tried the commands listed here, but I am still having the Node-RED issue, which I'll take up with their authors

[https://github.com/mapbox/node-pre-gyp](https://github.com/mapbox/node-pre-gyp)

```
npm uninstall node-pre-gyp --save
npm install @mapbox/node-pre-gyp --save
```

MichaIng, 2021-05-21:
Many thanks for your reports.

These are no errors but warnings, quite common ones when Node modules have a lot of dependencies, like Node-RED. It defines it's dependencies with specific versions to assure that all is compatible, but those versions might get deprecated before being updated by the upstream developer.

Do not update those manually, as this may break Node-RED. The specific versions are used as dependencies for a reason, usually. Furthermore Node-RED is installed as local module into a service users home directory (`/mnt/dietpi_userdata/node-red`). Running npm uninstall/install just like that from console manages either the global system modules (when running as root) or another local Node environment based on the current login user. Both does not what you want and does not affect the local Node-RED Node environment.

If you have specific issues with Node-RED, please report then individually, either as individual issue here (so we can try to replicate and verify whether its an issue with our implementation or an upstream bug) or to the module's authors directly.

MDAR, 2021-05-21:
Thanks Michal

> If you have specific issues with Node-RED, please report then individually, either as individual issue here

Yes, that was the plan, but as you say there is nothing to be concerned by right now

closed 2021-05-21 by MDAR