MichaIng/DietPi #4408: Using perf anyhow?

danergo, 2021-05-22:
Guys, please help me out.

I was using an outdated DietPi, and wanted to use `perf` on it.

It was not working, so I updated my system by `dietpi-update`. Now I'm on DietPi v7.1.2.

Kernel is: 4.19.66+ which honestly seems a bit outdated still (but dietpi-update wasn't updated it).

When I try running perf, it writes me:

```
# perf
/usr/bin/perf: line 13: exec: perf_4.19: not found
E: linux-perf-4.19 is not installed.
```
Searching for linux-tools:

```
# apt-cache search linux-tools
linux-perf-4.4 - Performance analysis tools for Linux 4.4
linux-perf-4.9 - Performance analysis tools for Linux 4.9
linux-tools - Performance analysis tools for Linux (dummy package)
linux-tools-3.10 - Performance analysis tools for Linux 3.10
linux-tools-3.16 - Performance analysis tools for Linux 3.16
linux-tools-3.18 - Performance analysis tools for Linux 3.18
linux-tools-3.6 - Performance analysis tools for Linux 3.6
```
It seems I can install only very old perf versions.

Can you recommend me any way to:
a.) update to latest recommended stable kernel? (in case it's not 4.19.66+),
b.) install perf matching with that kernel? (if there is an option without completely compiling it from scratch)?

Thank you so much,
Cheers,
Daniel

Joulinar, 2021-05-22:
Hi,

DietPi is not creating any kernel. They are installed according the base image used. Maybe you like to share the SBC used?

danergo, 2021-05-22:
SBC is an RPI CM, based on the old BCM2835 chip.

danergo, 2021-05-22:
What is the kernel version for the very latest base image for RPi CM1?

Joulinar, 2021-05-22:
Can you share `echo $G_DISTRO_NAME`

danergo, 2021-05-22:
Sure thing, it's "stretch". Kindof outdated, but I thought dietpi-update will also do dist-upgrade. Am I mistaken?

MichaIng, 2021-05-22:
DietPi-Update does no major Debian/Raspbian distribution upgrades (`dist-upgrade` alone doesn't do that, but you first need to change the `/etc/apt/sources.list(.d/*.list)` package sources to use the next Debian version) as these easily break installed software and may require manual adjustments, fixes or software reinstalls. This needs to be done manually by each admin so that you have full package upgrade output and can take care potentially arising errors manually. We can tell you how to do that and what to take care of afterwards, but this cannot be automated. Generally we recommend to flash a new image in such case which is cleaner and less error-prone, but of course may create more work.

Raspberry Pi Stretch is stuck on Linux 4.19 indeed, Debian Stretch however uses Linux 4.9, which is the reason you see the Linux 4.9 perf package only, which is for the Debian kernel package, not for the Raspberry Pi kernel. Not sure if it can work with it, even if the version did match. The Raspberry Pi foundation does not ship any perf package for their kernel.

Joulinar, 2021-05-22:
regarding the update Stretch > Buster we have a small summary https://dietpi.com/phpbb/viewtopic.php?p=18988#p18988

danergo, 2021-05-23:
Guys, thank you so much, it's always a pleasure to get these nice detailed and professional responses :)

closed 2021-05-23 by danergo