MichaIng/DietPi #4409: Open Beta v7.2 | Please help testing and hardening the upcoming release

MichaIng, 2021-05-22:
RC version | v7.2.0
---------- | -------
v7.2.0 PR | #4406
Changelog | https://github.com/MichaIng/DietPi/blob/beta/CHANGELOG.txt
Code changes | https://github.com/MichaIng/DietPi/compare/master...beta
Release planned | **2021-05-29**

### How to apply: https://github.com/MichaIng/DietPi/blob/master/BRANCH_SYSTEM.md

#### Related/solved issues: https://github.com/MichaIng/DietPi/issues?q=is%3Aissue+milestone%3Av7.2
____
### Beta v7.2.0
_(2021-05-22)_

#### New SBC support
- Pinebook Pro | Initial hardware identifier (ID: 46) and support for this device has been added to DietPi.
- NanoPi R4S | Initial hardware identifier (ID: 47) and support for this device has been added to DietPi.
- NanoPi R1 | Initial hardware identifier (ID: 48) and support for this device has been added to DietPi.

#### Changes
- DietPi-DDNS | Added native support for FreeDNS: https://freedns.afraid.org/
- DietPi-VPN | Added native support for IPVanish and Private Internet access (PIA)
- DietPi-VPN | When "Reset" is selected, you're now offered to purge the OpenVPN package, as long as neither the OpenVPN (server) install option nor PiVPN is installed. Many thanks to @maartenlangeveld for doing this suggestion: https://github.com/MichaIng/DietPi/issues/4346
- DietPi-AutoStart | Selecting desktop autologin with non-root user has been made much more reliabe. Previously the X server itself was started by the login user, which often requires additional permissions and sometimes, based on GPU and driver, is not possible at all without further customisation. The method has been changed so that LightDM is used for non-root autologins, just like it is for manual desktop logins, but logging into the desktop automatically with the chosen user.
- DietPi-Software | A desktop preference choice has been added. Before first boot via new dietpi.txt AUTO_SETUP_DESKTOP_INDEX setting or with the new menu entry it can be now chosen which desktop to install when another software selection requires one, especially remote desktop solutions. It works like the webserver preference, so it only play a role when a desktop is installed as dependency and is overridden by manual desktop selections.
- DietPi-Software | A browser preference choice has been added. Before first boot via new dietpi.txt AUTO_SETUP_BROWSER_INDEX setting or with the new menu entry it can be now chosen whether and which browser to install automatically together with desktops. Previously Firefox was installed with every desktop. It only plays a role when a desktop is installed and is overridden by manual browser selections. With this change, Firefox has become an independent software option (see new software below).
- DietPi-Software | WiringPi: On RPi, a new updated fork of the deprecated original project is now used, which enables support for RPi 4/400 and CM4: https://github.com/WiringPi/WiringPi
- DietPi-Software | WiringPi: On new installs and reinstalls, the source/examples directory is now installed to /mnt/dietpi_userdata/WiringPi instead of /root/wiringPi, to enable general access to non-root users.
- DietPi-Software | Node.js: On ARMv6, new Node.js versions are now installed via unofficial builds. Official builds are provided up to Node v11 only. Many thanks to @ollliegits for adding support for this builds to our Node.js installer fork: https://github.com/MichaIng/nodejs-linux-installer/pull/2
- DietPi-Software | EmonPi: This software option has been renamed to "emonHub", the name of the data collector for the "emonPi" RPi energy monitor addon board. A much newer Python 3 compatible version form the official OpenEnergyMonitor repository is installed from now on, providing additional features and fixes.
- DietPi-Software | RPi Cam Control: This install option has been disabled for 64-bit systems. It uses a 32-bit/armhf raspimjpeg binary that depends in 32-bit/armhf C library. It will be re-enabled once the project maintainer or we provide a native 64-bit/arm64 raspimjpeg binary.
- DietPi-Software | Roon Extension Manager: After a major upgrade to v1.0, it is now implemented as Docker container, rather than as Node.js module. The upgrade can be applied manually via "dietpi-software reinstall 86". Many thanks to @JanKoudijs for developing Roon Extension Manager and implementing the required changes into DietPi-Software: https://github.com/MichaIng/DietPi/pull/4399

#### New Software
- Firefox | The Mozilla browser has now become an independent software option with the ID 67 (see changes above). 

#### Removed Software
- LibSSL1.0.0 | This old library was kept for backwards-compatibility with old binaries but is not required anymore for any binary installed by DietPi-Software. It has hence been removed from the software list.

#### Fixes
- Odroid XU4 | Resolved an issue where installs and possibly other tasks hang, because the device ran out of entropy. All Odroid XU4 system will have the unsupported hardware random generator daemon removed and the software HAVEGE daemon installed instead for entropy generation. Many thanks to @Speeedfire for reporting this issue: https://github.com/MichaIng/DietPi/issues/4318
- DietPi-Banner | Resolved an issue where the MOTD was not updated via daily cron job, if the banner settings have not been changed yet, hence no config file exists. Since the MOTD is enabled by default, it needs to be updated as well if the config file does not exist. Many thanks to @gorby-pranata for helping us discovering this issue: https://github.com/MichaIng/DietPi/pull/4292#issuecomment-830787256
- DietPi-Banner | Resolved an issue where DietPi-VPN stats could not be shown due to a false file path. Many thanks to @sl002 for reporting this issue: https://github.com/MichaIng/DietPi/issues/4354
- DietPi-Banner and DietPi-VPN | Resolved an issue where the WAN IP could not be derived, as the external API service we used has server issues by times. We switched to a (hopefully) more reliable one with higher rate limit as well, until we find time to host an own GeoIP API. Many thanks to @maartenlangeveld for reporting this issue: https://github.com/MichaIng/DietPi/issues/4393
- DietPi-Update | Resolved an issue where updating from DietPi v6.16 or older did not complete as intended. Many thanks to @orotarobas for reporting this issue: https://github.com/MichaIng/DietPi/issues/4385
- DietPi-Drive_Manager | Resolved an issue where NTFS (and potentially other) filesystems were not detected correctly, leading to failing mount attempts at boot. Many thanks to `@NoLifer` for reporting this issue: https://dietpi.com/phpbb/viewtopic.php?t=8971
- DietPi-Drive_Manager | Resolved an issue where info for drives with a systemd.automount unit was not detected correctly, leading to invalid fstab entries. Many thanks to @cocoflan and @BillyCorgan1 for reporting this issue: https://github.com/MichaIng/DietPi/issues/4339
- DietPi-JustBoom | Resolved an issue where the equalizer had no effect in some cases. Many thanks to `@zackdvd` for reporting this issue: https://dietpi.com/phpbb/viewtopic.php?t=8992
- DietPi-LetsEncrypt | Resolved an issue with Lighttpd, where lighty-enable-mod or lighty-disable-mod failed, if the related config was already enabled or disabled, respectively. Many thanks to @staxfax for reporting this issue: https://github.com/MichaIng/DietPi/issues/4336
- DietPi-Config | Resolved an issue where changing the locale generated it but did not apply it as new system default. Many thanks to @bamyasi for reporting the issue: https://github.com/MichaIng/DietPi/issues/3515#issuecomment-840751875
- DietPi-Software | Python 3: Resolved an issue where installing pip on Stretch systems failed, due to a changed download URL. Many thanks to `@tfmeier` for reporting this issue: https://dietpi.com/phpbb/viewtopic.php?t=8968
- DietPi-Software | Webmin: Resolved an issue where restarts from the web interface only stopped the service. Many thanks to @Burgess85 and `@Keridos` for reporting this issue: https://dietpi.com/phpbb/viewtopic.php?t=8839, https://github.com/MichaIng/DietPi/pull/4331
- DietPi-Software | Docker Compose: Resolved an issue on ARMv8 Debian Stretch systems, where the install failed because of missing development headers. Many thanks to `@tfmeier` for reporting this issue: https://dietpi.com/phpbb/viewtopic.php?p=34293#p34293
[... cut ...] uring DietPi update. As compiling can take up to several hours, users are informed at the beginning of the DietPi update, with the option to cancel and apply it at a later time. All data and configs will be preserved during the reinstall. Many thanks to @math-gout for informing us about this change: https://github.com/MichaIng/DietPi/issues/4325
- DietPi-Software | Home Assistant: Resolved an issue where the install failed, as running "pyenv init -" does not complement the PATH variable anymore.

Joulinar, 2021-05-29:
Ok I just had a brief look. Sorry did not had that much time. Seems to be ok so far.

MichaIng, 2021-05-29:
DietPi v7.2 has been released: #4422

Many thanks to all testers! ❤️

closed 2021-05-29 by MichaIng