MichaIng/DietPi #4410: I updated my DietPi2 from 7.1.2 to 7.2 and become an error:

omavoss, 2021-05-22:
#### Details:
- Date           | Sat May 22 22:47:04 CEST 2021
- DietPi version | v7.1.2 (MichaIng/beta)
- Image creator  | DietPi Core Team
- Pre-image      | Raspbian Lite
- Hardware       | RPi 2 Model B (armv7l) (ID=2)
- Kernel version | Linux DietPi 5.10.17-v7+ #1403 SMP Mon Feb 22 11:29:51 GMT 2021 armv7l GNU/Linux
- Distro         | buster (ID=5,RASPBIAN=1)
- Command        | cargo build -j 2 --features sqlite --release
- Exit code      | 101
- Software title | DietPi-Software
#### Steps to reproduce:

1. ...
2. ...
#### Expected behaviour:

- ...
#### Actual behaviour:

- ...
#### Extra details:

- ...
#### Additional logs:
```
   Compiling libc v0.2.93
   Compiling proc-macro2 v1.0.26
thread 'rustc' panicked at 'Missing SyntaxContext 70 for crate "std"', compiler/rustc_metadata/src/rmeta/decoder.rs:364:36
note: run with `RUST_BACKTRACE=1` environment variable to display a backtrace

error: internal compiler error: unexpected panic

note: the compiler unexpectedly panicked. this is a bug.

note: we would appreciate a bug report: https://github.com/rust-lang/rust/issues/new?labels=C-bug%2C+I-ICE%2C+T-compiler&template=ice.md

note: rustc 1.53.0-nightly (132b4e5d1 2021-04-13) running on armv7-unknown-linux-gnueabihf

note: compiler flags: -C embed-bitcode=no -C debug-assertions=off --crate-type bin

note: some of the compiler flags provided by cargo are hidden

query stack during panic:
end of query stack
error: could not compile `libc`

To learn more, run the command again with --verbose.
warning: build failed, waiting for other jobs to finish...
thread 'rustc' panicked at 'Missing SyntaxContext 70 for crate "std"', compiler/rustc_metadata/src/rmeta/decoder.rs:364:36
note: run with `RUST_BACKTRACE=1` environment variable to display a backtrace

error: internal compiler error: unexpected panic

note: the compiler unexpectedly panicked. this is a bug.

note: we would appreciate a bug report: https://github.com/rust-lang/rust/issues/new?labels=C-bug%2C+I-ICE%2C+T-compiler&template=ice.md

note: rustc 1.53.0-nightly (132b4e5d1 2021-04-13) running on armv7-unknown-linux-gnueabihf

note: compiler flags: -C embed-bitcode=no -C debug-assertions=off --crate-type bin

note: some of the compiler flags provided by cargo are hidden

query stack during panic:
end of query stack
error: build failed
```

Joulinar, 2021-05-22:
You have installed Bitwarden_RS??

omavoss, 2021-05-22:
Yes.

Joulinar, 2021-05-22:
looks like Rust installation is failing

```
note: the compiler unexpectedly panicked. this is a bug.

note: we would appreciate a bug report: https://github.com/rust-lang/rust/issues/new?labels=C-bug%2C+I-ICE%2C+T-compiler&template=ice.md
```

let me do a test installation

omavoss, 2021-05-22:
I would like to migrate to vaultwarden.
What should I do? There is no hurry.
Thank you very much.

Joulinar, 2021-05-22:
This is still 7.2 BETA and probably an issue outside DietPi as indicated by the error message. Something we need to check first before we can give an advise.

omavoss, 2021-05-22:
As I said: it only concerns a test system, nothing used productively, please do not rush.

Thanks for any advice.

omavoss, 2021-05-22:
I need to reboot the system, it is no longer responding. Also an attempt to access directly on the DietPi via monitor fails. There is no image output.
I have now the configuration destroyed by the failed update on (so the micro SD card) and set the 7.2 BETA again. Nextcloud and Vaultwarden are installed. I will report back.

If the destroyed configuration is still needed to possibly read information on it, please let me know.

Many kind regards.

Joulinar, 2021-05-23:
I did test installations of Bitwarden_RS on 32bit as well as 64bit. Once done, I triggered the 7.2 update and migration to vaultwarden went fine without issues. I'm not able to replicate the behaviour you describe. 🤔

MichaIng, 2021-05-23:
I'll also run a test, I have the same RPi here. Probably also the memory was full. We forgot to raise the RAM requirements a little (so a larger swap file is created, when required), as planned. I only observed possible full RAM on x86_64 machines so far, but it is definitely close with 1.9 GiB total memory, too close IMO.

omavoss, 2021-05-23:
Hello, thanks for the message.

I have now rebuilt DietPi 7.2 BETA with OpenSSH, Nextcloud and Vaultwarden on the RasPi2. It took several hours to compile, now the DietPi runs smoothly as far as I can tell so far. I used a 32 GB micro SD card for this, the "old" card was also 32 GB.

 DietPi v7.2.0 (beta) : 18:27 - So 23.05.2021
 ─────────────────────────────────────────────────────
 - Device model : RPi 2 Model B (armv7l)
 - CPU temp : 39'C : 102'F (Cool runnings)
 - LAN IP : 192.168.115.146 (eth0)
 - Freespace (RootFS) : 23G
 - Freespace (userdata) : 23G

I can't see that the memory is not enough.
Kind regards.

MichaIng, 2021-05-23:
Great that it worked now. I did a test reinstall as well, and on RPi did memory usage is indeed quite lower, 2 GiB max but with four build jobs, which are only used when 3 GiB memory (including swap files) are available. So only on x86_64 PCs/VMs we need to raise that. I'll do a little tweak before final v7.2 release.

Joulinar, 2021-05-23:
@MichaIng 
do you like to keep it open?

closed 2021-05-23 by MichaIng

omavoss, 2021-05-23:
No, please close this ticket, thank you.