MichaIng/DietPi #4413: dietpi-update from v7.0.2 to v7.1.2 fails - services hanging

rhpsFan65, 2021-05-23:
#### Details:
- Date           | Sun May 23 15:52:15 CEST 2021
- Bug report     | 95a7d99e-684f-4195-a0a7-fadaa92b659a
- DietPi version | v7.0.2 (MichaIng/master)
- Image creator  | DietPi Core Team
- Pre-image      | Raspberry Pi OS 64-bit
- Hardware       | RPi 4 Model B (aarch64) (ID=4)
- Kernel version | Linux Raspi07 5.10.17-v8+ #1414 SMP PREEMPT Fri Apr 30 13:23:25 BST 2021 aarch64 GNU/Linux
- Distro         | buster (ID=5,RASPBIAN=0)
- Command        | dpkg --remove-architecture armhf
- Exit code      | 2
- Software title | DietPi-Patch
#### Steps to reproduce:

1. Login with ssh
2. call sudo Dietpi-update

-> happened the second time now after a restore
#### Expected behaviour:
finish update without error
#### Actual behaviour:
Error Message
Some Services are not running any more after the error message
#### Extra details:

Would be great to get an hint on how to recover without restoring a backup.

- ...
#### Additional logs:
```
dpkg: error: cannot remove architecture 'armhf' currently in use by the database
```
logs attached
[dietpi-update.log](https://github.com/MichaIng/DietPi/files/6528221/dietpi-update.log)
[DietpiUpdate210523.txt](https://github.com/MichaIng/DietPi/files/6528222/DietpiUpdate210523.txt)

MichaIng, 2021-05-23:
Many thanks for your report.

Here the underlying issue and fix has been applied for v7.2 already: #4338

When you face the error prompt, select "Change command" and replace it with "`true`" or "`:`".

Out of interest, which armhf packages have you installed?
```
dpkg --get-selections '*:armhf' | mawk '{print $1}'
```

Joulinar, 2021-05-23:
Still I wonder how many people have installed `armhf` binaries on an `aarch64` system 🤔

MichaIng, 2021-05-23:
Obviously it was possible to install e.g. `chromium-browser` in some cases. When I tested it last time, I ran into a conflict, but it doesn't seem to be the case always. With v7.1 we switched it to install `chromium` from the Debian repo instead, i.e. an arm64 package, but here it's v7.0.

rhpsFan65, 2021-05-23:
Dear Micha,

thanks for your quick response. Here is the information requested: 

$ dpkg --get-selections '*:armhf' | mawk '{print $1}'
gcc-8-base:armhf
libc6:armhf
libgcc1:armhf
shairport-sync:armhf
wiringpi:armhf

MichaIng, 2021-05-23:
So it's WiringPi in your case, and ehm Shairport Sync? How did you install the letter one?

I recommend, after DietPi v7.2 has been released and you updated, to install WiringPi via `dietpi-software` and remove the DEB package. It's an old version while `dietpi-software` will use a new repository for it with better RPi4 support, compiling it on your system natively as aarch64 library and binary: https://github.com/MichaIng/DietPi/pull/4324

For Shairport Sync we have aarch64 packages, actually: https://dietpi.com/downloads/binaries/buster/shairport-sync_aarch64.deb
But I'll rebuild them later today, as a new version is available.

With the two changes, you would be able to remove the armhf architecture.

rhpsFan65, 2021-05-24:
Thanks for the analysis. I'll modify the installation as you've recommended.
Acutally I did not install this packages explicitly. They must have come along with other packages.

And - your hint solved the problem - Thanks for your help.

closed 2021-05-24 by rhpsFan65