MichaIng/DietPi #4414: DietPi-Software | MPD: Re-add deb.kaliko.me repo for Buster

MichaIng, 2021-05-23:
Provided by the a Debian contributor, so we can assume it's very similar to official Debian build: https://kaliko.me/debian/
Promoted by MPD: https://www.musicpd.org/download-unoff-debian/

We used it in the past already, but it was available for x86_64 only. Now Raspbian armhf (means ARMv6+ARMv7 compatible) has been added as well.

Currently Debian Buster + Stretch and Raspbian Buster are supported, shipping always the MPD version of the next Debian release, like the official backports suite would do.

hrast01, 2022-01-30:
There's additional functionality exposed in myMPD with MPD 0.23 as the backend.

hrast01, 2022-08-02:
Any idea when this might get implemented?

MichaIng, 2022-08-02:
No ETA. I'm still hesitating since aarch64 isn't supported (yet) so that we complicate the install and configuration based an distro and arch. However, the instructions on the website are simple to follow manually.