MichaIng/DietPi #4416: Issue updating to DietPi v7.1.2 from v7.0.2

Cov-fefe, 2021-05-24:
Having an issue updating DietPi, when attempting I get an error 

`#### Details:
- Date           | Mon May 24 16:36:57 BST 2021
- Bug report     | 484f6290-e517-410a-9283-e0278610e325
- DietPi version | v7.0.2 (MichaIng/master)
- Image creator  | DietPi Core Team
- Pre-image      | Raspberry Pi OS Lite
- Hardware       | RPi 3 Model B (armv7l) (ID=3)
- Kernel version | Linux DietPi 5.10.17-v7+ #1403 SMP Mon Feb 22 11:29:51 GMT 2021 armv7l GNU/Linux
- Distro         | buster (ID=5,RASPBIAN=1)
- Command        | apt-get -qq upgrade
- Exit code      | 100
- Software title | DietPi-Update
#### Steps to reproduce:

1. ...
2. ...
#### Expected behaviour:

- ...
#### Actual behaviour:

- ...
#### Extra details:

- ...
#### Additional logs:
```
debconf: delaying package configuration, since apt-utils is not installed
(Reading database ... 25765 files and directories currently installed.)
Preparing to unpack .../archives/apt_1.8.2.3_armhf.deb ...
Unpacking apt (1.8.2.3) over (1.8.2.2) ...
dpkg: error processing archive /var/cache/apt/archives/apt_1.8.2.3_armhf.deb (--unpack):
 unable to create '/usr/share/bug/apt/script.dpkg-new' (while processing './usr/share/bug/apt/script'): Structure needs cleaning
dpkg-deb: error: paste subprocess was killed by signal (Broken pipe)
apt-daily-upgrade.timer is a disabled or a static unit not running, not starting it.
apt-daily.timer is a disabled or a static unit not running, not starting it.
Errors were encountered while processing:
 /var/cache/apt/archives/apt_1.8.2.3_armhf.deb
E: Sub-process /usr/bin/dpkg returned an error code (1)
```
`

Joulinar, 2021-05-24:
Hi,

Can you try following and post the output if there are errors

```
/boot/dietpi/func/dietpi-set_software apt-cache clean
apt update
apt upgrade

```

MichaIng, 2021-05-24:
Many thanks for your report.
```
Structure needs cleaning
```
This indicates a filesystem corruption.

Please try to have it repaired on reboot and check back fsck log:
```
> /forcefsck
reboot
# after reboot
journalctl -t systemd-fsck
```

Cov-fefe, 2021-05-24:
Thanks for the quick responses to both of you.  Tried a forcefsck w/ reboot and the system never came back up.  On screen I got an message stating: [end Kernel panic - not syncing: Attempted to kill init! exitcode=0x00007f00]

I think it is a corrupt file system but let me know if I should try anything before starting from scratch and installing DietPi all over again.  Thanks

Joulinar, 2021-05-24:
Do you have another Linux box where you can connect your SD card on and would be able to perform some file system checks?

Cov-fefe, 2021-05-24:
Yes I do, I have my Macbook that I can hook it upto.

MichaIng, 2021-05-24:
That is good, macOS as `fsck` utility as well, so you can run something like this: `fsck /dev/mmcblk0p2`
Or it has even a GUI for filesystem checks, I guess.

Cov-fefe, 2021-05-24:
I keep getting Permission Denied when using fsck /dev/disk3s1

I tried using the Mac disk utility and it doesn't find any errors.  I think this SD card may be toast

MichaIng, 2021-05-24:
Can you use `sudo fsck /dev/disk3s1` on macOS?

Joulinar, 2021-05-24:
Or credit a Linux VM?

Cov-fefe, 2021-05-24:
> Can you use `sudo fsck /dev/disk3s1` on macOS?

sudo fsck_msdos -y /dev/disk3s1
** /dev/rdisk3s1 (NO WRITE)
** Phase 1 - Preparing FAT
** Phase 2 - Checking Directories
** Phase 3 - Checking for Orphan Clusters
420 files, 206134 KiB free (412268 clusters)
MARK FILE SYSTEM CLEAN? no

***** FILE SYSTEM IS LEFT MARKED AS DIRTY *****

MichaIng, 2021-05-24:
Good. That was the boot partition, now the root partition:
```
sudo fsck /dev/disk3s2
```
Or probably `fsck_ext4` or so.

Cov-fefe, 2021-05-24:
> Good. That was the boot partition, now the root partition:
> 
> ```
> sudo fsck /dev/disk3s2
> ```
> 
> Or probably `fsck_ext4` or so.

sudo fsck_msdos /dev/disk3s2
** /dev/rdisk3s2
Invalid BS_jmpBoot in boot block: 000000

MichaIng, 2021-05-24:
That is not a FAT filesystem, but ext4, so if there is no generic fsck command, then it needs to have _ext or _ext4 suffix or similar.

Cov-fefe, 2021-05-24:
> That is not a FAT filesystem, but ext4, so if there is no generic fsck command, then it needs to have _ext or _ext4 suffix or similar.

 sudo fsck_ext /dev/disk3s2
sudo: fsck_ext: command not found

sudo fsck_ext4 /dev/disk3s2
sudo: fsck_ext4: command not found

Joulinar, 2021-05-24:
If I'm not mistaken MacOS did not support ext4. Isn't it?

ravenclaw900, 2021-05-24:
If you have Homebrew installed, you can run `brew install e2fsprogs` and the binary will then be located at `/usr/local/opt/e2fsprogs/sbin/e2fsck`.  You could also build it from source: https://e2fsprogs.sourceforge.net.

Cov-fefe, 2021-05-24:
> If you have Homebrew installed, you can run `brew install e2fsprogs` and the binary will then be located at `/usr/local/opt/e2fsprogs/sbin/e2fsck`. You could also build it from source: https://e2fsprogs.sourceforge.net.

I don't have Homebrew installed.  For some reason the file system gets corrupt each time I restart the Pi which has been very seldom because I've had this issue.  So it could be the Pi (3B) itself or the SD card.  I think I may just purchase a 4 and new SD card and start from scratch.

Joulinar, 2021-05-24:
Or you go with a USB stick and boot directly from it. This is what I'm doing on all my RPi's 😄

Cov-fefe, 2021-05-24:
> Or you go with a USB stick and boot directly from it. This is what I'm doing on all my RPi's 😄

Didn't realize you could use USB to boot.  I currently have a 3, does it work with Rpi3 or only Rpi4?  Any specific USB drive necessary?

Joulinar, 2021-05-24:
3B+ will work ootb. On a 3B it would need to be activated first. See RPi docs   https://www.raspberrypi.org/documentation/hardware/raspberrypi/bootmodes/msd.md

Personally I use a SanDisk 16GB Ultra Fit USB 3.1 Flash Drive. It's quite good for my purposes

MichaIng, 2021-05-25:
`dietpi-config` > Advanced Options can be also used to set the USB boot flag on RPi 3 non-plus. You need to reboot once, the bit is programmed into the firmware then. Afterwards you can attach a USB stick with DietPi (or any other image) flashed onto, remove the SD card and it will boot from the stick. As long as a bootable SD card is attached, it will still boot from it, so assure to either remove the SD card, or reformat it as pure data storage, without RPi boot partition on it.

Cov-fefe, 2021-05-26:
> 3B+ will work ootb. On a 3B it would need to be activated first. See RPi docs https://www.raspberrypi.org/documentation/hardware/raspberrypi/bootmodes/msd.md
> 
> Personally I use a SanDisk 16GB Ultra Fit USB 3.1 Flash Drive on my 3b+ and 4B. It's quite good for my purposes. But you could use a SSD as well 😄

> `dietpi-config` > Advanced Options can be also used to set the USB boot flag on RPi 3 non-plus. You need to reboot once, the bit is programmed into the firmware then. Afterwards you can attach a USB [...]

Thanks to both of you.  I am going to reflash my microSD card then change the USB boot flag and hook up a USB to boot from.  Thanks for all the info!

Joulinar, 2021-05-31:
Guys,

can this be closed or do we need to keep it open?

closed 2021-05-31 by Cov-fefe