MichaIng/DietPi #4418: armbian-bsp-cli-nanopineo2 upgrade is being withheld

maartenlangeveld, 2021-05-25:
Upgrade is being withheld. Probably for a good reason.
Just to let you know.
```
 ─────────────────────────────────────────────────────
 DietPi v7.1.2 : 1 APT updates available
 ─────────────────────────────────────────────────────
 - Device model : NanoPi NEO2 (aarch64)
 - Uptime : up 10 hours, 1 minute
 - CPU temp : 36'C : 96'F (Cool runnings)
 - FQDN/hostname : M-DNS
 - LAN IP : 192.xxx.xxx.xxx (eth0)
 - WAN IP : xxx.xxx.xxx.xxx ----- -----------------
 - Freespace (RootFS) : 23G
 - MOTD : Open Beta v7.2 | Please help testing the upcoming release:
          https://github.com/MichaIng/DietPi/issues/4409
 ─────────────────────────────────────────────────────

 DietPi Team     : MichaIng (lead), Daniel Knight (founder), Joulinar (support)
 Image by        : DietPi Core Team (pre-image: Armbian)
 Web             : https://dietpi.com | https://twitter.com/DietPi_
 Patreon Legends : Camry2731
 Contribute      : https://dietpi.com/contribute.html
 DietPi Hosting  : Powered by https://myvirtualserver.com

 apt upgrade     : Run now to apply 1 available APT package upgrades.

 dietpi-launcher : All the DietPi programs in one place.
 dietpi-config   : Feature rich configuration tool for your device.
 dietpi-software : Select optimized software for installation.
 htop            : Resource monitor.
 cpu             : Shows CPU information and stats.

root@M-DNS:~# apt upgrade
Reading package lists... Done
Building dependency tree
Reading state information... Done
Calculating upgrade... Done
The following packages have been kept back:
  armbian-bsp-cli-nanopineo2
0 upgraded, 0 newly installed, 0 to remove and 1 not upgraded.
```

MichaIng, 2021-05-25:
Many thanks for your report.

Armbian changed their root overlay/tweaks/tools package naming. Previously it was like `linux-buster-root-current-<sbc_name>`, and instead of "current" another "legacy" existed and before that an "edge" one as well, which was a bit confusing. The changed this now into `armbian-bsp-cli-<sbc_name>` for all SBCs and moved it into the `buster-utils` suite as well, which is a reasonable change to have things unified. I'll however remote those packages completely from new Armbian-based images, so that only kernel, device tree and bootloader packages are included.

The old packages are now transitional dummy packages and in your case the new one has been installed already, as intended. Now an upgrade seems to be available which implies dependency changes, so it is not applies via regular "apt upgrade". To see what would change and in case apply the upgrade, run: `apt full-upgrade`

maartenlangeveld, 2021-05-25:
```
root@M-DNS:~# apt full-upgrade
Reading package lists... Done
Building dependency tree
Reading state information... Done
Calculating upgrade... Done
The following packages were automatically installed and are no longer required:
  armbian-bsp-cli-nanopineo2 distro-info-data fping lsb-release u-boot-tools
Use 'apt autoremove' to remove them.
The following packages will be REMOVED:
  linux-buster-root-current-nanopineo2
The following packages will be upgraded:
  armbian-bsp-cli-nanopineo2
1 upgraded, 0 newly installed, 1 to remove and 0 not upgraded.
Need to get 415 kB of archives.
After this operation, 0 B of additional disk space will be used.
Do you want to continue? [Y/n]
```

MichaIng, 2021-05-25:
Nice, so the transitional dummy package even got removed by this, i.e. a clean migration. Should be all fine now. The `armbian-ramlog.service` mask is of course intended, as it conflicts with `dietpi-ramlog.service`, one of the reasons I want to not pre-install those packages anymore at all. I think only the `update-initramfs: Converting to u-boot format` part is taken care of by these packages, and are required, but we can place an own little script for that instead of installing the whole Armbian package.

MichaIng, 2021-05-28:
For now I added those new packages to DietPi-PREP, to keep things as they are until I find time to test an own replacement for it: https://github.com/MichaIng/DietPi/commit/15d2da75256ccfed1fd105295734585f4d1bbbeb

closed 2021-05-28 by MichaIng