MichaIng/DietPi #4424: Mosquitto installation from dietpi-software

fra87, 2021-05-26:
### Creating a bug report/issue

#### Required Information
- DietPi version | `7.1.2`
- Distro version | `buster 10.9`
- Kernel version | `5.10.35-rockchip64 #21.05.1 SMP PREEMPT Fri May 7 13:53:11 UTC 2021 aarch64 GNU/Linux`
- SBC model | `ROCK64 (aarch64)`
- Power supply used | N/A
- SDcard used | N/A

#### Additional Information (if applicable)
- Software title | Mosquitto
- Was the software title installed freshly or updated/migrated? New installation of Mosquitto
- Can this issue be replicated on a fresh installation of DietPi? Not tried, but I think so
- Bug report ID | N/A

#### Steps to reproduce
1. Start installing Mosquitto from the dietpi-software menu

#### Expected behaviour
- The installation completes without issues

#### Actual behaviour
- The installation halts when generating the password

#### Extra details
The dietpi-software halts with the following error

```
 - Command: mosquitto_passwd -c -b /etc/mosquitto/passwd mosquitto XXX
 - Exit code: 1
 - DietPi version: v7.1.2 (MichaIng/master) | HW_MODEL: 43 | HW_ARCH: 3 | DISTRO: 5
 - Image creator: DietPi Core Team
 - Pre-image: Armbian
 - Error log:
Error: -c argument given but password file or username missing.
```

It seems that -c and -b options are not supported at the same time. I managed to complete the installation opening a subshell, writing `touch /etc/mosquitto/passwd && exit`, and finally executing the command without the `-c`

Joulinar, 2021-05-27:
Hi,

thx for sharing. Hmm looks like both option can't be used together anymore. Not sure if this is an intended change or an issue within the `mosquitto_passwd` tool

MichaIng, 2021-05-27:
Great find. Here is the related commit, which fixed/allowed this since v1.6.10: https://github.com/eclipse/mosquitto/commit/3758e0c3285c9155f281e71c26435cb1c8117205

Let's pre-create the file for now. I'll open a PR.

fra87, 2021-05-27:
Thank you both for the quick feedback and for the fix 😃

MichaIng, 2021-05-28:
Great. I'll mark this as closed.

closed 2021-05-28 by MichaIng