MichaIng/DietPi #4426: dietpi-drive_manager - Add network drive error

BillyCorgan1, 2021-05-27:
**Required Information:**

- DietPi version | v7.2.1 (beta)
- Distro version | Buster
- Kernel version | Linux 5.10.17-v8+
- SBC model | Raspberry Pi 4GB
- Power supply used | Raspberry Pi Official Power Supply
- SDcard used | Samsung Pro Endurance 64GB

**Additional Information (if applicable):**

- Software title | dietpi-drive_manager
- Was the software title installed freshly or updated/migrated? Fresh Install and Updated
- Can this issue be replicated on a fresh installation of DietPi? Yes
- Bug report ID | echo $G_HW_UUID Not Applicable

**Steps to reproduce:**

1. Insert Micro SD Card into Odroid N2 Micro SD Slot.
2. Open dietpi-drive_manager.
3. Select "Add network drive" in the menu.
4. Select "samba".
5. Enter the fileservers IP address, I am using the same IP as the IP that is given to the DietPi installation.
6. Enter the fileservers shared folder name, I am using Plex.
7. Enter the fileservers username, I am using root.
8. Enter the fileservers password.
9. Enter a unique folder name for the mount location (eg: samba), I am using samba.

**Expected behaviour:**

The Samba share should be created.

**Actual behaviour:**

The Samba mount fails with the following error:

```
┌────────────────────────────────────────┤ DietPi-Drive_Manager | File viewer ├────────────────────────────────────────┐
│                                                                                                                      │
│ [FAILED] Samba mount failed with the following error output:                                                         │
│                                                                                                                      │
│ Unable to find suitable address.                                                                                     │
│ Unable to find suitable address.                                                                                     │
│ Unable to find suitable address.                                                                                     │
│ Unable to find suitable address.                                                                                     │
│ Unable to find suitable address.                                                                                     │
│                                                                                                                      │
│                                                        <Ok>                                                          │
│                                                                                                                      │
└──────────────────────────────────────────────────────────────────────────────────────────────────────────────────────┘

```

**Extra details:**

I am fully aware this could be a user error, as in I am stupid, but I just can't get samba to work. I have tried on Odroid N2 and a Raspberry Pi 4 4GB and the error is the same. As I said this could be a user error, but any help will be gratefully received. Thanks for your time.

Joulinar, 2021-05-27:
Hi,

Can you let me know what you are trying to do? What is your goal?

MichaIng, 2021-05-27:
> Can you let me know what you are trying to do? What is your goal?

Jep, especially if you're trying to mount an SMB/Samba share, hosted elsewhere on your network, on that RPi (that is what the `dietpi-drive_manager` option is for), or if you want to host a Samba server/share on the RPi (for that you'd need to install the Samba server from `dietpi-software` instead).

BillyCorgan1, 2021-05-27:
Hello,

Thank you for your quick reply. I am trying to setup a Samba share so that I can copy files to and from it. I wish to run Plex and want to be able to drag and drop media files to the samba share. I had successfully done this in the past with DietPi with no problems on an Odroid N2. I am unable to create a samba share and the config file is just the default samba one, there are no DietPi settings. I don't know if that has changed. It has been a while since I set it up and I may have forgotten how to do it.

Joulinar, 2021-05-27:
well than you do it wrongly. You need to install SAMBA server and activate file sharing on `dietpi-software`

MichaIng, 2021-05-27:
> setup a Samba share

Sounds like the Samba server shall run on the RPi. See above, for this install "Samba Server" via `dietpi-software`. `dietpi-drive_manager` is for "mounting" drives, respectively elsewhere running file servers.

BillyCorgan1, 2021-05-27:
Thanks. I thought it might be a case of the stupids. I have already installed Plex and Samba before I tried to get it working, so there is some hope. I am just trying to get the smb.conf working.

Joulinar, 2021-05-27:
you can test from an external client to test smb connection.

BillyCorgan1, 2021-05-27:
> 
> 
> you can test from an external client to test smb connection.

Thanks I have got a samba share working on another computer, so I will try to mirror that .

I will close this bug as I am sure this is a User Error and therefore not an actual bug. I thank you for your input and if I can't get it working I will post for help in a more appropriate place like the forum. Thanks again for your time and for helping me. I do appreciate your hard work.

closed 2021-05-27 by BillyCorgan1

Joulinar, 2021-05-27:
for some basic instructions, have a look to our online docs https://dietpi.com/docs/software/file_servers/#samba