MichaIng/DietPi #4427: Issue updating DietPi from v7.0.2 to v7.1.2.

musgen, 2021-05-27:
### Creating a bug report/issue

#### Required Information
- DietPi version | 
G_DIETPI_VERSION_CORE=7
G_DIETPI_VERSION_SUB=0
G_DIETPI_VERSION_RC=2
G_GITBRANCH='master'
G_GITOWNER='MichaIng'
- Distro version | buster
- Kernel version | Linux DietPi 4.19.0-16-amd64 #1 SMP Debian 4.19.181-1 (2021-03-19) x86_64 GNU/Linux
- SBC model | Native PC (x86_64)
- Power supply used | 220V
- SDcard used | HDD

#### Steps to reproduce

1. Run dietpi-upgrade

#### Expected behaviour

- Update complete

#### Actual behaviour

- Error during update.

#### Extra details

The following packages have unmet dependencies:
 bind9-host : Depends: libdns1104 (= 1:9.11.5.P4+dfsg-5.1+deb10u5) but 1:9.11.5.P4+dfsg-5.1+deb10u3 is installed
              Depends: libisc1100 (= 1:9.11.5.P4+dfsg-5.1+deb10u5) but 1:9.11.5.P4+dfsg-5.1+deb10u3 is installed
 dnsutils : Depends: libdns1104 (= 1:9.11.5.P4+dfsg-5.1+deb10u5) but 1:9.11.5.P4+dfsg-5.1+deb10u3 is installed
            Depends: libisc1100 (= 1:9.11.5.P4+dfsg-5.1+deb10u5) but 1:9.11.5.P4+dfsg-5.1+deb10u3 is installed
 libbind9-161 : Depends: libdns1104 (= 1:9.11.5.P4+dfsg-5.1+deb10u5) but 1:9.11.5.P4+dfsg-5.1+deb10u3 is installed
                Depends: libisc1100 (= 1:9.11.5.P4+dfsg-5.1+deb10u5) but 1:9.11.5.P4+dfsg-5.1+deb10u3 is installed
 libirs161 : Depends: libisc1100 (= 1:9.11.5.P4+dfsg-5.1+deb10u5) but 1:9.11.5.P4+dfsg-5.1+deb10u3 is installed
 libisccfg163 : Depends: libdns1104 (= 1:9.11.5.P4+dfsg-5.1+deb10u5) but 1:9.11.5.P4+dfsg-5.1+deb10u3 is installed
                Depends: libisc1100 (= 1:9.11.5.P4+dfsg-5.1+deb10u5) but 1:9.11.5.P4+dfsg-5.1+deb10u3 is installed
                Depends: libisccc161 (= 1:9.11.5.P4+dfsg-5.1+deb10u5) but 1:9.11.5.P4+dfsg-5.1+deb10u3 is installed
E: Unmet dependencies. Try 'apt --fix-broken install' with no packages (or specify a solution).

Joulinar, 2021-05-27:
Hi,

Pls can you run `apt --fix-broken instal` as stated in the error message

musgen, 2021-05-27:
Hi, thanks for taking a look.

sudo apt --fix-broken install

Reading package lists... Done
Building dependency tree       
Reading state information... Done
Correcting dependencies... Done
The following additional packages will be installed:
  libdns1104 libisc1100 libisccc161
The following packages will be upgraded:
  libdns1104 libisc1100 libisccc161
3 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.
10 not fully installed or removed.
Need to get 1910 kB of archives.
After this operation, 41.0 kB disk space will be freed.
Do you want to continue? [Y/n] 
Get:1 https://deb.debian.org/debian-security buster/updates/main amd64 libisccc161 amd64 1:9.11.5.P4+dfsg-5.1+deb10u5 [237 kB]
Get:2 https://deb.debian.org/debian-security buster/updates/main amd64 libdns1104 amd64 1:9.11.5.P4+dfsg-5.1+deb10u5 [1215 kB]
Get:3 https://deb.debian.org/debian-security buster/updates/main amd64 libisc1100 amd64 1:9.11.5.P4+dfsg-5.1+deb10u5 [458 kB]               
Fetched 1910 kB in 7s (268 kB/s)                                                                                                            
debconf: delaying package configuration, since apt-utils is not installed
(Reading database ... 27447 files and directories currently installed.)
Preparing to unpack .../libisccc161_1%3a9.11.5.P4+dfsg-5.1+deb10u5_amd64.deb ...
dpkg: error processing archive /var/cache/apt/archives/libisccc161_1%3a9.11.5.P4+dfsg-5.1+deb10u5_amd64.deb (--unpack):
 unable to open triggers ci file '/var/lib/dpkg/info/libisccc161:amd64.triggers': Bad message
Preparing to unpack .../libdns1104_1%3a9.11.5.P4+dfsg-5.1+deb10u5_amd64.deb ...
dpkg: error processing archive /var/cache/apt/archives/libdns1104_1%3a9.11.5.P4+dfsg-5.1+deb10u5_amd64.deb (--unpack):
 unable to open triggers ci file '/var/lib/dpkg/info/libdns1104:amd64.triggers': Bad message
Preparing to unpack .../libisc1100_1%3a9.11.5.P4+dfsg-5.1+deb10u5_amd64.deb ...
dpkg: error processing archive /var/cache/apt/archives/libisc1100_1%3a9.11.5.P4+dfsg-5.1+deb10u5_amd64.deb (--unpack):
 unable to open triggers ci file '/var/lib/dpkg/info/libisc1100:amd64.triggers': Bad message
Errors were encountered while processing:
 /var/cache/apt/archives/libisccc161_1%3a9.11.5.P4+dfsg-5.1+deb10u5_amd64.deb
 /var/cache/apt/archives/libdns1104_1%3a9.11.5.P4+dfsg-5.1+deb10u5_amd64.deb
 /var/cache/apt/archives/libisc1100_1%3a9.11.5.P4+dfsg-5.1+deb10u5_amd64.deb
E: Sub-process /usr/bin/dpkg returned an error code (1)

Joulinar, 2021-05-27:
let's see if we can install the packages manually

```
cd /tmp
wget http://security.debian.org/debian-security/pool/updates/main/b/bind9/libisccc161_9.11.5.P4+dfsg-5.1+deb10u5_amd64.deb
wget http://security.debian.org/debian-security/pool/updates/main/b/bind9/libdns1104_9.11.5.P4+dfsg-5.1+deb10u5_amd64.deb
wget http://security.debian.org/debian-security/pool/updates/main/b/bind9/libisc1100_9.11.5.P4+dfsg-5.1+deb10u5_amd64.deb
dpkg -i *.deb
```

Joulinar, 2021-06-11:
closed due to missing user feedback. Feel free to reopen if needed.

closed 2021-06-11 by MichaIng