MichaIng/DietPi #4428: SqueezeLite version update & -W option support

scan80269, 2021-05-28:
### Creating a feature request

#### Is your feature request related to a problem? Please describe:

SqueezeLite has a confirmed issue with 24/384K & 24/352.8K FLAC file playback to USB/I2S DACs (no sound) that was recently resolved.  1.9.9-1386 is the first SqueezeLite version with the proper fix.  Also, the -W option is needed to support 24/768K & 24/705.6K WAV file playback.

#### Describe the solution you'd like:

Update SqueezeLite version to 1.9.9-1386 or later for DietPi.  Add -W as user-configurable option for SqueezeLite settings.

#### Describe alternatives you've considered:

Alternative is to avoid playing 24/384K & 24/352.8K FLAC files and 24/768K & 24/705.6K WAV files with SqueezeLite.

#### Additional context

- ...

#### Vote for this feature on FeatHub: https://feathub.com/MichaIng/DietPi/

MichaIng, 2021-05-28:
Perfect timing, just finished Amiberry an Shairport Sync, Squeezelite was the next I wanted to look at.

MichaIng, 2021-05-28:
@scan80269 
Little question about the `-W` option: That should be always supported, so do you mean that we should set it by default in our systemd unit?
Actually I aim to reduce the settings, e.g. after some testing also the output parameters (`-a 4096:1024:16:0`) and hence not adding `-W`. The defaults should work in most cases, and otherwise users can change parameters as stated in our docs: https://dietpi.com/docs/software/media/#squeezelite ("Change Squeezelite command line options")

scan80269, 2021-05-29:
Thanks for the SqueezeLite update to 1.9.9-1386!

The (-a 4096 1024 16 0) parameters didn't work for 352.8K/384K FLAC & WAV and 768K WAV playback which were all badly distorted with constant buzzing & wrong tempo.  Replacing this with -W restores the playback of these formats to normal: /usr/bin/squeezelite -C 5 -W -n DietPi-SqueezeLite.

Tested with Allo Sparky (USBridge) driving Oppo Sonica DAC over USB.

MichaIng, 2021-05-29:
Thanks for testing. For now I'll leave it as it is, but probably we can collect some more opinions about the best/most compatible defaults, also whether we remove the `-a` as well. It also controls the buffers, and AFAIK also `-a 4096:1024` would be possible, skipping the trailing parameters, to set this buffer size only. But I lack experience about what makes most sense here. These parameters were set long ago by the DietPi founder and I simply never touched them 🙂.

What we could definitely do is provide some common options required for this or that types or audio sources in our docs: https://dietpi.com/docs/software/media/#squeezelite

scan80269, 2021-05-29:
Just tested -a 4096 1024 and it didn't work either, same problem as with -a 4096 1024 16 0.  Playback was normal up to 192K but not higher.  Going back to -W works.  So it looks like -a is not an effective substitute for -W for the setups I tested.

352.8K/384K FLAC & WAV and 768K WAV may be pushing the envelope, as little to no content is available in such formats.  These sample rates are related to external upsampling experiments my friends & I are conducting, so consider them as corner cases if you like.  Also, there were very few DACs capable of playing at 768K until recently, so probably very little test history at such high sample rates in the past.

I also tested SqueezeLite 1.9.9-1386 running in Arch Linux, and also with different SBCs (Raspberry Pi 4, Tinker Board S & Cubox i4x4) with virtually identical results: the -W is required for sample rates above 192K to play properly.  I'll be sticking with -W and leaving out -a for my setups.

MichaIng, 2021-05-29:
Just to be sure, you added the `-a` values separated by colon, not by space, right? So `-a 4096:1024:16:0`, not `-a 4096 1024 16 0`.

Does the `-W` option leads to any output (`journalctl -u squeezelite`) about which format was read from the the header and used? Would be interesting to know.

scan80269, 2021-05-30:
Yes, tried both -a 4096:1024:16:0 and -a 4096:1024 in /lib/systemd/system/squeezelite.service. Only -W (without -a) yields correct playback above 192K.

With -W running, nothing is seen in output of journalctl -u squeezelite, other than "DietPi systemd[1]: Started Squeezelite (DietPi)."

MichaIng, 2022-10-13:
Finally, new packages up with suggested argument changes applied. Also the arguments can now be easier adjusted via `/etc/default/squeezelite`. To test:
```sh
cd /tmp
curl -sSfLO https://dietpi.com/downloads/binaries/$G_DISTRO_NAME/testing/shairport-sync_$G_HW_ARCH_NAME.deb
apt install ./shairport-sync_$G_HW_ARCH_NAME.deb
rm shairport-sync_$G_HW_ARCH_NAME.deb
```