MichaIng/DietPi #4433: dietpi-vpn service has state "failed"

woelfli, 2021-05-29:
### Creating a bug report/issue

#### Required Information
- DietPi version: v7.1.2, v7.2.1 (beta), v7.2.2 (beta)  // latter currently installed
- Distro version: buster
- Kernel version: Linux dietpi-gitea 5.10.17-v7+ #1414 SMP Fri Apr 30 13:18:35 BST 2021 armv7l GNU/Linux
- SBC model: RPi 3 Model B (armv7l)
- Power supply used: Standard RPi 3 Model B power adapter
- SDcard used: SanDisk 8GB

#### Additional Information (if applicable)
Software installed:
- 0 - OpenSSH Client
- 3 - Midnight Commander
- 17 - Git
- 64 - phpSysInfo
- 73 - Fail2Ban
- 82 - LLMP
- 84 - Lighttpd
- 88 - MariaDB
- 89 - PHP
- 92 - Cerbot
- 103 - DietPi-RAMlog
- 105 - OpenSSH Server
- 109 - NFS Server
- 130 - Python 3
- 165 - Gitea

#### Steps to reproduce
- Taking a look in dialog "DietPi-Services" in DietPi Version v7.1.2 I realized, that there was a "openvpn" service (state = "active") as well as a "dietpi-vpn" service (state = "?"), even though I never  installed OpenVPN (97), PiVPN (117), WireGuard (172) or any other VPN software.
- Trying to change the state of "dietpi-vpn" service results in state "failed". This state is persistent and you can't set it back to state "active" or "inactive". This is true for DietPi version v7.1.2, v7.2.1 (beta) and v7.2.2 (beta), which is currently running on my machine.

#### Expected behaviour
- Maybe I'm wrong, but, as no VPN software is installed on my machine, I don't expect a VPN service on it. Somehow it seems to me, "openvpn" and "dietpi-vpn", whereupon at least "openvpn" was "active, might have been installed after the update to version "v7.1.2" or even perhaps to the release before (I don't know).
- A service should be either "active", "inactive", masked, ... or the service is simply not installed, but it should not keep staying in state "failed".
- After installing OpenVPN (index 97), there is a "/boot/DietPi_OpenVPN_Client.ovpn" file as well as a "/etc/openvpn/DietPi_OpenVPN_Client.ovpn" file, which are both equal (see "Extra details" below). After uninstalling OpenVPN, the "/etc/openvpn/DietPi_OpenVPN_Client.ovpn" file gets removed but not the other file. Here I expect either both files to get removed, or, why is there also a "/boot/DietPi_OpenVPN_Client.ovpn" file?

#### Extra details
1. Changing the state of "openvpn" in dialog "DietPi-Services" was possible. It could be changed from "active" to "inactive" and vice-versa. Even changing the mode ("DietPi controlled", "Systemd controlled", "mask" ,"unmask) was possible.
2. But: Trying to change the state of "dietpi-vpn" results in state "failed". This state persists, you can't set it back to "inactive" or "active" anymore. Also it's not possible to change the mode to "mask", e.g. This is true for version v7.1.2, v7.2.1 (beta) and v7.2.2 (beta).
3. Calling "systemctl status dietpi-vpn" in a shell showed among other things:
```
dietpi-vpn.service - VPN Client (DietPi)
Loaded: loaded (/etc/systemd/system/dietpi-vpn.service; disabled; vendor preset: enabled)
Active: failed (Result: exit-code) since Sat 2021-05-29 10:34:58 CEST; 40s ago
Process: 12357 ExecStart=/usr/sbin/openvpn --suppress-timestamps --nobind --config /etc/openvpn/client.ovpn (code=exited, status=203/EXEC)
Main PID: 12357 (code=exited, status=203/EXEC)
```
4. The output "/usr/sbin/openvpn --suppress-timestamps --nobind --config /etc/openvpn/client.ovpn" seemed to be interesting for me.
5. As "/usr/sbin/openvpn" and "/etc/openvpn/client.ovpn" both did not exist, I installed OpenVPN (index 97).
6. After installing OpenVPN, there was a "/boot/DietPi_OpenVPN_Client.ovpn" as well as a "/etc/openvpn/DietPi_OpenVPN_Client.ovpn" file on my machine, which were both equal.
7. After copying one of them to "/etc/openvpn/client.ovpn" and going back to dialog "DietPi-Services", it was possible to set the state of "dietpi-vpn" service back to either "active" or "inactive" (I kept it "inactive").
8. After uninstalling OpenVPN, the "openvpn" service was removed, but the "dietpi-vpn" service is still present in state "inactive".
9. Is the "dietpi-vpn" service necessary? If not, how can I uninstall it?

Joulinar, 2021-05-29:
`dietpi-vpn` is based on OpenVPN and shown always even if not used. `dietpi-vpn` is a script that work around OpenVPN and not a software package.

ravenclaw900, 2021-05-29:
However, should we mask the DietPi-VPN service when it's not being used?

MichaIng, 2021-05-29:
The `dietpi-vpn.service` cannot be masked, as it is located at `/etc/system/system/dietpi-vpn.service`, where the mask itself would be placed. But it also doesn't need to. It is disabled by default (not starting), same with `openvpn.service`, as long as none was explicitly enabled.

It should also not conflict with the OpenVPN server install option, or it's service:
- Our install option, same as the `openvpn.service` (the way it works), checks for and controls an auto-generated `openvpn@<config>.service` based on any existing `/etc/openvpn/*.conf` file.
- DietPi-VPN client config files however have an `*.ovpn` ending, hence won't lead to any auto-generated actual `openvpn@<config>.service` file. So as long as their content does not conflict. Both should start and be handled independently from the other.

The OpenVPN install option however needs to be reworked indeed. It creates an `/etc/openvpn/DietPi_OpenVPN_Client.ovpn` config, meant to be moved/installed on the client that shall connect to the server, which is fine. It however is copied as well to `/boot` (on RPi FAT partition), to be easily accessible from a Windows system, and even to `/mnt/dietpi_userdata`, to be easily accessible via file servers. But having three such files, with the private client key included + possibly a fourth on the client, is bad security-wise. It's better to put the client config at a single secure location and make it pretty clear to users, that, once installed on the client, for security reasons it should be removed from the server. In theory we could also merge this into DietPi-VPN, to make it a client + server tool. Setup as a server, it could then make client config generation easier, as well as revocation of client keys etc, and store all configs+keys at a secure place with strict access permissions.

@woelfli 
Do I understand it correctly, that you installed the OpenVPN server option after DietPi-VPN did not start? As the `client.ovpn` did not exist, did you previously set it up via the `dietpi-vpn` command? That would automatically install openvpn as well, so that the `/usr/sbin/openvpn` command is then available. Interesting is indeed that `openvpn.service` is shown as active. I'd have expected that this is only the case when a `/etc/openvpn/*.conf` file exists, otherwise the `openvpn.service` does nothing. Probably it is simply always shown as active, even when it's practically only a dummy, like a systemd target, then.

woelfli, 2021-06-02:
Hello Micha,
thank you for your quick response. Here are my answers:

To "Do I understand it correctly, that you installed the OpenVPN server option after DietPi-VPN did not start?": Yes. I installed the OpenVPN server after DietPi-VPN showed state = "failed". See also "Steps to reproduce" below.

To "As the `client.ovpn` did not exist, did you previously set it up via the `dietpi-vpn` command?": No. I installed OpenVPN server and then simply copied `/etc/openvpn/DietPi_OpenVPN_Client.ovpn` to `/etc/openvpn/client.ovpn`.

Steps to reproduce: Take a current DietPi environment (also v7.2.3, e.g.) without OpenVPN server installed, start `dietpi-services`, select service "dietpi-vpn", press return, select option "State" (should be in state "inactive"), press return. Result: Service "dietpi-vpn" is in state "failed". I could not set it back to "inactive" or anything else anymore, at least not inside `dietpi-services`. After this I installed OpenVPN server. Then I could set "dietpi-vpn"'s state back to "inactive". After this I uninstalled the OpenVPN server and deleted `/boot/DietPi_OpenVPN_Client.ovpn` manually afterwards.

MichaIng, 2021-06-02:
Okay, then it is based on a misunderstanding indeed 😃:
- Our OpenVPN install option is a "server" option. It creates a `server.conf` and a first client configuration `DietPi_OpenVPN_Client.ovpn` which is to be copied to another system, so that this other system can connect to the server where you installed "OpenVPN", running with the generated `server.conf`. Using `DietPi_OpenVPN_Client.ovpn` in the same system cannot work, as then no server runs with the matching `server.conf`. The server is controlled with the `openvpn.service`.
- DietPi-VPN is a VPN "client" script to connect to other (e.g. public) VPN servers. It does not run OOTB but is needs to be configured first. Obviously you need to tell it to which server to connect to, input the connection credentials etc. For this use the `dietpi-vpn`. It will then create the `client.ovpn` that is used by `dietpi-vpn.service`.

Both uses OpenVPN, but whether it runs in client or in server mode is the major difference. It would be indeed possible to run an OpenVPN server on one DietPi system and connect to it via DietPi-VPN from another DietPi system.

Joulinar, 2021-07-14:
Guys can this issue be closed or are there open questions left?

MichaIng, 2021-11-07:
I'll mark this as closed. Feel free to reopen when required.

closed 2021-11-07 by MichaIng