MichaIng/DietPi #4435: Dietpi-Software: Samba installation unexpected behavior.

BillyCorgan1, 2021-05-29:
**Required Information**

- DietPi version | v7.2.2 (beta)
- Distro version | Buster
- Kernel version | Linux 5.10.17-v8+
- SBC model | Raspberry Pi 4GB
- Power supply used | Raspberry Pi Official Power Supply
- SDcard used | Samsung Pro Endurance 64GB

**Additional Information (if applicable)**

- Software title | dietpi-software
- Was the software title installed freshly or updated/migrated? Fresh Install and Updated
- Can this issue be replicated on a fresh installation of DietPi? Yes
- Bug report ID | echo $G_HW_UUID Not Applicable

**Steps to reproduce**

1. Open dietpi-software.
2. Select "Add network drive" in the menu.
3. Select "Browse Software".
4. Select "Samba".
5. Select "OK"
6. Select "Install: Go >> Start installation for selected software"
7. Installation will proceed with out error.
8. Restart Raspberry Pi
9. Enter "sudo nano /etc/samba/smb.conf" after login.
10. Default Samba config file will load, not the DietPi version.
11. Samba shares will not be accessible despite being entered correctly.

**Expected behaviour**

The installation should be the same as File Server setting in the "dietpi-software". The Samba DietPi share should be created in "sudo nano /etc/samba/smb.conf".

**Actual behaviour**

The Samba mount fails and is not accessible.

**Extra details**

1. Open dietpi-software.
2. Select "Add network drive" in the menu.
3. Select "File Server".
4. Select "Samba".
5. Select "OK"
6. Select "Install: Go >> Start installation for selected software"
7. Installation will proceed with out error.
8. Restart Raspberry Pi
9. Enter "sudo nano /etc/samba/smb.conf" after login.
10. DietPi version Samba config file will load.
11. Samba shares will be accessible.

I hope this makes sense. This has been tripping me up as I use browse software to install Plex and Samba. I ignore the file server setting and samba installs but does not work correctly. It should work the same as when you use the file server menu I think. Let me know if this is not a bug and is working as intended or if you need any more information or logs etc. Thanks

Joulinar, 2021-05-29:
You are trying to set-up a samba server? Because `Select "Add network drive" in the menu.` is not an option inside `dietpi-software`. This is an option in drive manager

BillyCorgan1, 2021-05-29:
No, this is a separate issue to my stupidity. I am not trying to "Add network drive", I am not using drive manager. This is bug(?) within Dietpi-Software. I am successfully able to access shares when samba is installed when selected via "File server". When installed via the "Browse software" menu I am not able to access shares even when they are the same as the file server setting ones entered into the samba config file. There are two ways to install Samba as noted above. I would have expected that the end result would be the same but it is not installation one way, to me, is broken. I hope this makes a bit more sense. Thanks for your reply, let me know if you need more information or if I have not explained something. Thanks.

Joulinar, 2021-05-29:
There is just a single way how SAMBA is being installed. There is no difference on the way you are going to select it.

BillyCorgan1, 2021-05-29:
Then this is definitely a bug. This is what I get when entering "sudo nano /etc/samba/smb.conf" after using the browse menu.

`#
# Sample configuration file for the Samba suite for Debian GNU/Linux.
#
#
# This is the main Samba configuration file. You should read the
# smb.conf(5) manual page in order to understand the options listed
# here. Samba has a huge number of configurable options most of which
# are not shown in this example
#
# Some options that are often worth tuning have been included as
# commented-out examples in this file.
#  - When such options are commented with ";", the proposed setting
#    differs from the default Samba behaviour
#  - When commented with "#", the proposed setting is the default
#    behaviour of Samba but the option is considered important
#    enough to be mentioned here
#
# NOTE: Whenever you modify this file you should run the command
# "testparm" to check that you have not made any basic syntactic
# errors.

#======================= Global Settings =======================

[global]

## Browsing/Identification ###

# Change this to the workgroup/NT-domain name your Samba server will part of
   workgroup = WORKGROUP

#### Networking ####

# The specific set of interfaces / networks to bind to
# This can be either the interface name or an IP address/netmask;
# interface names are normally preferred
;   interfaces = 127.0.0.0/8 eth0

# Only bind to the named interfaces and/or networks; you must use the
# 'interfaces' option above to use this.
# It is recommended that you enable this feature if your Samba machine is
# not protected by a firewall or is a firewall itself.  However, this
# option cannot handle dynamic or non-broadcast interfaces correctly.
;   bind interfaces only = yes

#### Debugging/Accounting ####

# This tells Samba to use a separate log file for each machine
# that connects
   log file = /var/log/samba/log.%m

# Cap the size of the individual log files (in KiB).
   max log size = 1000

# We want Samba to only log to /var/log/samba/log.{smbd,nmbd}.
# Append syslog@1 if you want important messages to be sent to syslog too.
   logging = file

# Do something sensible when Samba crashes: mail the admin a backtrace
   panic action = /usr/share/samba/panic-action %d

# The specific set of interfaces / networks to bind to
# This can be either the interface name or an IP address/netmask;
# interface names are normally preferred
;   interfaces = 127.0.0.0/8 eth0

# Only bind to the named interfaces and/or networks; you must use the
# 'interfaces' option above to use this.
# It is recommended that you enable this feature if your Samba machine is
[... 172 lines cut ...]
   path = /var/spool/samba

   printable = yes
   guest ok = no
   read only = yes
   create mask = 0700

# Windows clients look for this share name as a source of downloadable
# printer drivers
[print$]
   comment = Printer Drivers
   path = /var/lib/samba/printers
   browseable = yes
   read only = yes
   guest ok = no

# Uncomment to allow remote administration of Windows print drivers.
# You may need to replace 'lpadmin' with the name of the group your
# admin users are members of.
# Please note that you also need to set appropriate Unix permissions
# to the drivers directory for these users to have write rights in it
;   write list = root, @lpadmin
`

Even if I put a working share at the bottom of the file, it does not work.

MichaIng, 2021-05-29:
That is the wrong file indeed. The installer should download and install this one: https://github.com/MichaIng/DietPi/blob/dev/.conf/dps_96/conf

Could you reinstall it, verify that there are not errors shown, and check again?
```
dietpi-software reinstall 96
```

BillyCorgan1, 2021-05-29:
> 
> 
> That is the wrong file indeed. The installer should download and install this one: https://github.com/MichaIng/DietPi/blob/dev/.conf/dps_96/conf
> 
> Could you reinstall it, verify that there are not errors shown, and check again?
> 
> ```
> dietpi-software reinstall 96
> ```

This is the output of the command:

`root@DietPi:~# dietpi-software reinstall 96
[  OK  ] DietPi-Software | Initialised database
[  OK  ] DietPi-Software | Reading database

 DietPi-Software
─────────────────────────────────────────────────────
 Mode: Automated reinstall

[ INFO ] DietPi-Software | 96: Samba Server is not currently installed
[ INFO ] DietPi-Software | Use "dietpi-software install 96" to install Samba Server.
[  OK  ] DietPi-Software | No changes applied for: Samba Server
root@DietPi:~#
`

Joulinar, 2021-05-29:
As said, I did a test installation (even 2 now) and got the correct result. I'm not able to confirm your assumption.

Joulinar, 2021-05-29:
how did you installed samba???

`[ INFO ] DietPi-Software | 96: Samba Server is not currently installed`

MichaIng, 2021-05-29:
Ah, it seems the install never finished. Likely it cancelled after the package was installed but before our config was downloaded. Then:
```
dietpi-software install 96
```

BillyCorgan1, 2021-05-29:
I installed it with the browse software menu item.

Joulinar, 2021-05-29:
It is not installed. Pls run installation again

BillyCorgan1, 2021-05-29:
> dietpi-software install 96

```
root@DietPi:~# dietpi-software install 96
[  OK  ] DietPi-Software | Initialised database
[  OK  ] DietPi-Software | Reading database

 DietPi-Software
─────────────────────────────────────────────────────
 Mode: Automated install

[  OK  ] DietPi-Software | Installing Samba Server: Feature-rich SMB/CIFS server
[  OK  ] DietPi-Software | Free space check: path=/ | available=56602 MiB | required=500 MiB
[  OK  ] DietPi-Software | DietPi-Userdata validation: /mnt/dietpi_userdata
[  OK  ] DietPi-Software | Checking network connectivity
[  OK  ] DietPi-Software | Checking DNS resolver
[  OK  ] Network time sync | Completed
[ SUB1 ] DietPi-Services > unmask
[  OK  ] DietPi-Services | unmask : plexmediaserver
[  OK  ] DietPi-Services | unmask : cron
[ SUB1 ] DietPi-Services > stop
[  OK  ] DietPi-Services | stop : cron
[  OK  ] DietPi-Services | stop : plexmediaserver
[  OK  ] DietPi-Software | mkdir -p /mnt/dietpi_userdata/Music /mnt/dietpi_userdata/Pictures /mnt/dietpi_userdata/Video /mnt/dietpi_userdata/downloads /var/www /opt
[  OK  ] DietPi-Software | chown dietpi:dietpi /mnt/dietpi_userdata/Music /mnt/dietpi_userdata/Pictures /mnt/dietpi_userdata/Video /mnt/dietpi_userdata/downloads
[  OK  ] DietPi-Software | chmod 775 /mnt/dietpi_userdata/Music /mnt/dietpi_userdata/Pictures /mnt/dietpi_userdata/Video /mnt/dietpi_userdata/downloads
[ INFO ] DietPi-Software | APT update, please wait...
Hit:1 https://downloads.plex.tv/repo/deb public InRelease
Hit:2 https://archive.raspberrypi.org/debian buster InRelease
Hit:3 https://deb.debian.org/debian buster InRelease
Hit:4 https://deb.debian.org/debian buster-updates InRelease
Get:5 https://deb.debian.org/debian-security buster/updates InRelease [65.4 kB]
Hit:6 https://deb.debian.org/debian buster-backports InRelease
Fetched 65.4 kB in 2s (41.9 kB/s)
Reading package lists...
[  OK  ] DietPi-Software | APT update

 DietPi-Software
─────────────────────────────────────────────────────
 Mode: Checking for prerequisite software

 DietPi-Software
─────────────────────────────────────────────────────
 Mode: Installing Samba Server: Feature-rich SMB/CIFS server

[  OK  ] DietPi-Software | rm -Rf /var/cache/samba
[  OK  ] DietPi-Software | mkdir -p /run/samba-cache
[  OK  ] DietPi-Software | ln -s /run/samba-cache /var/cache/samba
[ INFO ] DietPi-Software | APT install for: samba, please wait...
debconf: delaying package configuration, since apt-utils is not installed
Selecting previously unselected package python-dnspython.
(Reading database ... 20047 files and directories currently installed.)
Preparing to unpack .../0-python-dnspython_1.16.0-1_all.deb ...
Unpacking python-dnspython (1.16.0-1) ...
Selecting previously unselected package python-crypto.
Preparing to unpack .../1-python-crypto_2.6.1-9+b1_arm64.deb ...
Unpacking python-crypto (2.6.1-9+b1) ...
Selecting previously unselected package python-ldb.
Preparing to unpack .../2-python-ldb_2%3a1.5.1+really1.4.6-3+deb10u1_arm64.deb ...
Unpacking python-ldb (2:1.5.1+really1.4.6-3+deb10u1) ...
Selecting previously unselected package python-tdb.
Preparing to unpack .../3-python-tdb_1.3.16-2+b1_arm64.deb ...
Unpacking python-tdb (1.3.16-2+b1) ...
Selecting previously unselected package python-samba.
Preparing to unpack .../4-python-samba_2%3a4.9.5+dfsg-5+deb10u1_arm64.deb ...
Unpacking python-samba (2:4.9.5+dfsg-5+deb10u1) ...
Selecting previously unselected package samba-common-bin.
Preparing to unpack .../5-samba-common-bin_2%3a4.9.5+dfsg-5+deb10u1_arm64.deb ...
Unpacking samba-common-bin (2:4.9.5+dfsg-5+deb10u1) ...
Selecting previously unselected package tdb-tools.
[... 46 lines cut ...]
[  OK  ] DietPi-Software | cd /tmp/DietPi-Software
[  OK  ] DietPi-Software | curl -sSfL https://raw.githubusercontent.com/MichaIng/DietPi/beta/.conf/dps_96/conf -o conf
[ INFO ] DietPi-Software | Updating file: /etc/samba/smb.conf
[  OK  ] DietPi-Software | mv conf /etc/samba/smb.conf
[  OK  ] DietPi-Software | Setting in /etc/samba/smb.conf adjusted: max connections = 8
[  OK  ] DietPi-Software | systemctl daemon-reload
[ SUB1 ] DietPi-Services > dietpi_controlled
[  OK  ] DietPi-Services | dietpi_controlled : nmbd
[  OK  ] DietPi-Services | dietpi_controlled : smbd
[  OK  ] DietPi-Services | dietpi_controlled : plexmediaserver
[  OK  ] DietPi-Services | dietpi_controlled : cron

 DietPi-Software
─────────────────────────────────────────────────────
 Mode: Installation completed

[  OK  ] DietPi-Survey | Sending survey data
[ SUB1 ] DietPi-Services > restart
[  OK  ] DietPi-Services | restart : nmbd
[  OK  ] DietPi-Services | restart : smbd
[  OK  ] DietPi-Services | restart : plexmediaserver
[  OK  ] DietPi-Services | restart : cron
root@DietPi:~#

```

Joulinar, 2021-05-29:
correct config was download as shown on your output

```
[  OK  ] DietPi-Software | /etc/samba/smb.conf: backup to /etc/samba/smb.conf.bak_290521_0
[ INFO ] DietPi-Software | For a full list of backup items, please see /var/tmp/dietpi/logs/G_BACKUP_FP.db
[  OK  ] DietPi-Software | Checking URL: https://raw.githubusercontent.com/MichaIng/DietPi/beta/.conf/dps_96/conf
[  OK  ] DietPi-Software | cd /tmp/DietPi-Software
[  OK  ] DietPi-Software | curl -sSfL https://raw.githubusercontent.com/MichaIng/DietPi/beta/.conf/dps_96/conf -o conf
[ INFO ] DietPi-Software | Updating file: /etc/samba/smb.conf
[  OK  ] DietPi-Software | mv conf /etc/samba/smb.conf
```

BillyCorgan1, 2021-05-29:
I have rebooted the system and the smb.conf is as expected:

```
[global]

        workgroup = WORKGROUP
        server string = %h server
        dns proxy = no
        log file = /var/log/samba/log.%m
        max log size = 1000
        syslog only = no
        syslog = 0

        panic action = /usr/share/samba/panic-action %d

        security = user
        encrypt passwords = true
        passdb backend = tdbsam
        obey pam restrictions = yes
        unix password sync = yes

        passwd program = /usr/bin/passwd %u
        passwd chat = *Enter\snew\s*\spassword:* %n\n *Retype\snew\s*\spassword:* %n\n *password\supdated\ssuccessfully* .
        pam password change = yes
        map to guest = bad user

        load printers = no
        printcap name = /dev/null
        disable spoolss = yes

[dietpi]
        comment = DietPi Share
        path = /mnt/dietpi_userdata
        browseable = yes
        create mask = 0664
        directory mask = 0775
        valid users = dietpi
        writeable = yes
max connections = 8

```

I can also now browse to the default share in Windows as I expected to. There is something causing the browse menu installation to fail (in my opinion). The file server way works as expected.

Joulinar, 2021-05-29:
> There is something causing the browse menu installation to fail (in my opinion).

This is working for me on 2 test installation. Nothing I can confirm. There was something on your side that cancelled the installation

BillyCorgan1, 2021-05-29:
Is it possible for you to be able to do a clean installation of dietpi and install samba just using the browse menu please. I would really appreciate it. I have tried many installations just using the menus and it fails. Thanks.

MichaIng, 2021-05-29:
Jep, regardless which way you install it, via browse menu, search option, file server menu of CLI (as now), the exactly same code runs. But APT package installs can fail at some point. Always good to have an eye at how the whole process ended (last outputs).

Joulinar, 2021-05-29:
@BillyCorgan1 
Even a 3rd run did not change the result. It's working same way without any difference.

BillyCorgan1, 2021-05-29:
I'm sorry I'm not very experienced at this, but I can't understand how it fails one specific way and not the other. Thanks for your time I appreciate it.

BillyCorgan1, 2021-05-29:
> 
> 
> @BillyCorgan1
> Even a 3rd run did not change the result. It's working same way without any difference.

Thank you for trying and helping me.

Joulinar, 2021-05-29:
there is just ONE way. There are no differences in how you select the package. It will always execute the same code. There seems to be an issue happen as you tried it initially as even the samba package did not got installed.

BillyCorgan1, 2021-05-29:
I will close this issue in 48 hours if no problems (with DietPi) come up. Please close this issue if you feel it is solved. Thanks for your time and help.

MichaIng, 2021-05-29:
Just tried is as well, successfully. Probably there is a very specific file or package state present at the moment, when setting up a fresh RPi 64-bit image, that produces the error, e.g. a system directory missing, created at a later point, or even by the first package install steps.

I have no 64-bit-capable RPi but will boot the image in a container. @BillyCorgan1 did you select Samba right on first login, when dietpi-software comes up?

Joulinar, 2021-05-29:
I tested on a RPi4B 64bit 2 times and another test on RPi3B+ 32bit

MichaIng, 2021-05-29:
Okay, with a fresh image as well? Then my test won't produce better results.

Joulinar, 2021-05-29:
did this now the 4th time. 

1) Flashed a new SD card using 64bit version
2) performed a minimal install
3) rebooted the system
4) install SAMBA via `Browse Software`
5) SAMBA got installed as expected

MichaIng, 2021-05-29:
I also did it once, installed it directly from first dietpi-software prompt (no reboot). Whatever caused the failure in OP case, we'd require the console output from that to check for the reason.

I mark this as closed. Feel free to reopen when required, especially when the same happens again, then as said please keep and paste the console output of the whole install process here.

closed 2021-05-29 by MichaIng