MichaIng/DietPi #4436: Unable to uninstall MariaDB due to service failing

psbankar, 2021-05-29:
#### Details:
- Date           | Sun May 30 01:51:31 IST 2021
- DietPi version | v7.1.2 (MichaIng/master)
- Image creator  | DietPi Core Team
- Pre-image      | Raspberry Pi OS Lite
- Hardware       | RPi 3 Model B (armv7l) (ID=3)
- Kernel version | Linux Dietpi 5.10.17-v7+ #1414 SMP Fri Apr 30 13:18:35 BST 2021 armv7l GNU/Linux
- Distro         | buster (ID=5,RASPBIAN=1)
- Command        | systemctl start mariadb
- Exit code      | 1
- Software title | DietPi-Software
#### Steps to reproduce:
Attempt to uninstall MariaDB when the service is failing to start. It tries to create a backup of the database but since the service could not start, the backup doesn't take place, and uninstall fails.
#### Expected behaviour:
An option should be provided to choose whether a backup should be created or not or another option could be provided after backup fails whether to continue or not
#### Actual behaviour:
No such option in place currently
#### Extra details:
```
Job for mariadb.service failed because the control process exited with error code.
See "systemctl status mariadb.service" and "journalctl -xe" for details.
```
#### Additional logs:
```
May 30 01:58:04 Dietpi systemd[1]: Starting MariaDB 10.3.27 database server...
May 30 01:58:04 Dietpi mysqld[28887]: 2021-05-30  1:58:04 0 [Note] /usr/sbin/mysqld (mysqld 10.3.27-MariaDB-0+deb10u1) starting as process 28887 ...
May 30 01:58:04 Dietpi mysqld[28887]: 2021-05-30  1:58:04 0 [Warning] Can't create test file /var/lib/mysql/Dietpi.lower-test
May 30 01:58:04 Dietpi mysqld[28887]: [97B blob data]
May 30 01:58:04 Dietpi mysqld[28887]: 2021-05-30  1:58:04 0 [ERROR] Aborting
May 30 01:58:04 Dietpi systemd[1]: mariadb.service: Main process exited, code=exited, status=1/FAILURE
May 30 01:58:04 Dietpi systemd[1]: mariadb.service: Failed with result 'exit-code'.
May 30 01:58:04 Dietpi systemd[1]: Failed to start MariaDB 10.3.27 database server.
```

MichaIng, 2021-05-29:
Many thanks for your report.

Indeed, at least when the `mysqldump` command is still available (hence the MariaDB server package was not removed yet), we create a backup. It's meant to protect against accidents/errors, as the database usually contains the very most important and sensitive software data.

But I see the point and I'm also regularly annoyed when a broken software package cannot be removed just because it is broken. As a workaround, you can use "Change command" and enter ":" or "true" to continue. The following two commands will fail as well but those are not error-handled. As we already show a dialog before the backup is created, we should indeed turn it into a yes-no choice. Also the service start can be error-handled a way so that one can choose "Ignore" instead of "Exit". So it will be still possible to retry and open a shell to check whats wrong, but the uninstall can be continued regardless. Additionally we could keep the raw database files then, as an alternative backup, not as good as an SQL dump, as it only works with the particular MariaDB version, but better than nothing.

What is actually the reason for MariaDB to fail?

psbankar, 2021-05-30:
Thanks for the workaround! It helped me uninstall MariaDB successfully. 

> What is actually the reason for MariaDB to fail?

No idea actually. My dietpi stopped working suddenly so I flashed it again and restored the backup but everything else worked fine except MariaDB. Tried a lot to recover the DB but nothing worked. Thankfully the DB didn't contain anything important so I can start from scratch.

Joulinar, 2021-05-30:
we could troubleshoot the database if you like. Following would be needed

```
systemctl restart mariadb.service
journalctl -u mariadb
cat /var/log/mysql/error.log
readlink /var/lib/mysql
readlink -f /var/lib/mysql
```

MichaIng, 2021-05-30:
Probably we can get it working, however, MariaDB version and raw database files usually need to match in a certain degree. So for important databases I recommend to regularly do an SQL dump:
```
mysqldump -A > /path/to/backup.sql
```
to backup all databases, or
```
mysqldump specific_database_name > /path/to/backup.sql
```
to backup a specific database.

These dumps contain the SQL instructions to recreate the databases on different SQL-compatible database systems and definitely multiple MariaDB versions.

psbankar, 2021-05-31:
@Joulinar Thanks but I already moved on from recovering it 😁. It was just Home Assistant database so wasn't that important.
@MichaIng yes thanks for the suggestion. I will take backups of the database regularly from now on.