MichaIng/DietPi #4443: Cannot install Bazarr on MiniPC

PrzemekSkw, 2021-05-30:
Hello, I cannot install Bazarr on Dietpi on my MiniPC (64bit). I see ffmpeg error.

#### Details:
- Date           | Sun May 30 14:08:35 CEST 2021
- DietPi version | v7.2.3 (MichaIng/master)
- Image creator  | DietPi Core Team
- Pre-image      | Debian mini.iso
- Hardware       | Native PC (x86_64) (ID=21)
- Kernel version | Linux DietPi 4.19.0-16-amd64 #1 SMP Debian 4.19.181-1 (2021-03-19) x86_64 GNU/Linux
- Distro         | buster (ID=5)
- Command        | apt-get -qq --allow-change-held-packages install ffmpeg
- Exit code      | 100
- Software title | DietPi-Software
#### Steps to reproduce:

1. I try to install Bazarr trough dietpi-software
#### Expected behaviour:

- Bazarr should install.
#### Actual behaviour:

- Installation stops with error.
#### Additional logs:
```
E: Unable to correct problems, you have held broken packages.
```
```
│ APT install for: ffmpeg                                                                                              │ 
             │  - Command: apt-get -qq --allow-change-held-packages install ffmpeg                                                  │ 
             │  - Exit code: 100                                                                                                    │ 
             │  - DietPi version: v7.2.3 (MichaIng/master) | HW_MODEL: 21 | HW_ARCH: 10 | DISTRO: 5                                 │ 
             │  - Image creator: DietPi Core Team                                                                                   │ 
             │  - Pre-image: Debian mini.iso                                                                                        │ 
             │  - Error log:                                                                                                        │ 
             │ E: Unable to correct problems, you have held broken packages.
```
regards.

Joulinar, 2021-05-30:
Hi,

Pls can you run `apt update && apt upgrade` and post full output

PrzemekSkw, 2021-05-30:
Hi @Joulinar, thanks for reply.
```
root@DietPi:/# apt update && apt upgrade
Stary:1 https://deb.debian.org/debian buster InRelease
Stary:2 https://deb.debian.org/debian buster-updates InRelease
Stary:3 https://deb.debian.org/debian-security buster/updates InRelease
Stary:4 https://www.collaboraoffice.com/repos/CollaboraOnline/CODE-debian10 ./ InRelease
Czytanie list pakietów... Gotowe
Budowanie drzewa zależności       
Odczyt informacji o stanie... Gotowe
All packages are up to date.
Czytanie list pakietów... Gotowe
Budowanie drzewa zależności       
Odczyt informacji o stanie... Gotowe
Obliczanie aktualizacji... Gotowe
0 aktualizowanych, 0 nowo instalowanych, 0 usuwanych i 0 nieaktualizowanych.
```
regards.

Joulinar, 2021-05-30:
Can you try `apt install ffmpeg`

PrzemekSkw, 2021-05-30:
I try:
```
dietpi@DietPi:~$ sudo apt install ffmpeg
Czytanie list pakietów... Gotowe
Budowanie drzewa zależności       
Odczyt informacji o stanie... Gotowe
Nie udało się zainstalować niektórych pakietów. Może to oznaczać,
że zażądano niemożliwej sytuacji lub użyto dystrybucji niestabilnej,
w której niektóre pakiety nie zostały jeszcze utworzone lub przeniesione
z katalogu Incoming ("Przychodzące").
Następujące informacje mogą pomóc rozwiązać sytuację:

Następujące pakiety mają niespełnione zależności:
 ffmpeg : Wymaga: libavcodec58 (= 7:4.1.6-1~deb10u1)
          Wymaga: libavdevice58 (= 7:4.1.6-1~deb10u1) ale nie zostanie zainstalowany
          Wymaga: libavfilter7 (= 7:4.1.6-1~deb10u1)
          Wymaga: libavformat58 (= 7:4.1.6-1~deb10u1) ale nie zostanie zainstalowany
          Wymaga: libavresample4 (= 7:4.1.6-1~deb10u1) ale nie zostanie zainstalowany
          Wymaga: libavutil56 (= 7:4.1.6-1~deb10u1) ale nie zostanie zainstalowany
          Wymaga: libpostproc55 (= 7:4.1.6-1~deb10u1) ale nie zostanie zainstalowany
          Wymaga: libswresample3 (= 7:4.1.6-1~deb10u1) ale nie zostanie zainstalowany
          Wymaga: libswscale5 (= 7:4.1.6-1~deb10u1) ale nie zostanie zainstalowany
```
dependiences issues.
regards.

PrzemekSkw, 2021-05-30:
Maybe I have that issues because I install Kodi nightly build some time ago.

MichaIng, 2021-05-30:
Is an `libavformat58` package installed already? If so what happens if you try to purge it? Else what happens if you explicitly try to install the required version? You can copy&paste the following code as a block into your console/SSH. Do not confirm what APT offers, if you're unsure:
```
if dpkg-query -s libavformat58
then apt purge libavformat58
else apt install libavformat58=7:4.1.6-1~deb10u1
fi
```

PrzemekSkw, 2021-05-30:
@MichaIng I don't have that package:
```
dietpi@DietPi:~$ sudo apt install libavformat58
Czytanie list pakietów... Gotowe
Budowanie drzewa zależności       
Odczyt informacji o stanie... Gotowe
Nie udało się zainstalować niektórych pakietów. Może to oznaczać,
że zażądano niemożliwej sytuacji lub użyto dystrybucji niestabilnej,
w której niektóre pakiety nie zostały jeszcze utworzone lub przeniesione
z katalogu Incoming ("Przychodzące").
Następujące informacje mogą pomóc rozwiązać sytuację:

Następujące pakiety mają niespełnione zależności:
 libavformat58 : Wymaga: libavcodec58 (= 7:4.1.6-1~deb10u1)
                 Wymaga: libavutil56 (= 7:4.1.6-1~deb10u1) ale nie zostanie zainstalowany
                 Wymaga: libchromaprint1 (>= 1.3.2) ale nie zostanie zainstalowany
E: Nie udało się naprawić problemów, zatrzymano uszkodzone pakiety.
```
```
root@DietPi:/# if dpkg-query -s libavformat58
> then apt purge libavformat58
> else apt install libavformat58=7:4.1.6-1~deb10u1
> fi
dpkg-query: pakiet "libavformat58" nie jest zainstalowany i informacje o nim nie są dostępne
Użyj dpkg --info (= dpkg-deb --info), aby zobaczyć informacje o pakiecie.
Czytanie list pakietów... Gotowe
Budowanie drzewa zależności       
Odczyt informacji o stanie... Gotowe
Nie udało się zainstalować niektórych pakietów. Może to oznaczać,
że zażądano niemożliwej sytuacji lub użyto dystrybucji niestabilnej,
w której niektóre pakiety nie zostały jeszcze utworzone lub przeniesione
z katalogu Incoming ("Przychodzące").
Następujące informacje mogą pomóc rozwiązać sytuację:

Następujące pakiety mają niespełnione zależności:
 libavformat58 : Wymaga: libavcodec58 (= 7:4.1.6-1~deb10u1)
                 Wymaga: libavutil56 (= 7:4.1.6-1~deb10u1) ale nie zostanie zainstalowany
                 Wymaga: libchromaprint1 (>= 1.3.2) ale nie zostanie zainstalowany
E: Nie udało się naprawić problemów, zatrzymano uszkodzone pakiety.
```
regards.

Joulinar, 2021-05-30:
It would be helpful if you could switch to English language for the time being

MichaIng, 2021-05-30:
> It would be helpful if you could switch to English language for the time being

Before you use `dietpi-config`, to temporarily change it:
```
export LC_ALL='UTF-8'
```

MichaIng, 2021-05-30:
Please try:
```
apt install libchromaprint1
```

PrzemekSkw, 2021-05-30:
> > It would be helpful if you could switch to English language for the time being
> 
> Before you use `dietpi-config`, to temporarily change it:
> 
> ```
> export LC_ALL='UTF-8'
> ```
> 
> that should be enough.

```
bash: warning: setlocale: LC_ALL: cannot change locale (UTF-8)
```

> Please try:
> 
> ```
> apt install libchromaprint1
> ```
```
root@DietPi:/# LC_ALL=C apt install libchromaprint1
bash: warning: setlocale: LC_ALL: cannot change locale (UTF-8)
Reading package lists... Done
Building dependency tree       
Reading state information... Done
Some packages could not be installed. This may mean that you have
requested an impossible situation or if you are using the unstable
distribution that some required packages have not yet been created
or been moved out of Incoming.
The following information may help to resolve the situation:

The following packages have unmet dependencies:
 libchromaprint1 : Depends: libavcodec58 (>= 7:4.0)
                   Depends: libavutil56 (>= 7:4.0) but it is not going to be installed
E: Unable to correct problems, you have held broken packages.
```

MichaIng, 2021-05-30:
Ah sorry, the C was missing:
```
export LC_ALL='C.UTF-8'
```

PrzemekSkw, 2021-05-30:
```
root@DietPi:/# apt install libavutil56 
Reading package lists... Done
Building dependency tree       
Reading state information... Done
Some packages could not be installed. This may mean that you have
requested an impossible situation or if you are using the unstable
distribution that some required packages have not yet been created
or been moved out of Incoming.
The following information may help to resolve the situation:

The following packages have unmet dependencies:
 libavutil56 : Depends: libva-drm2 (>= 1.3) but it is not going to be installed
               Depends: libva-x11-2 (>= 1.3) but it is not going to be installed
E: Unable to correct problems, you have held broken packages.
```

MichaIng, 2021-05-30:
```
apt install libva-drm2
```
You get the idea. Sadly APT does not show the very last package that actually fails to be installed, so need to try them one after another until a meaningful error is shown.

PrzemekSkw, 2021-05-30:
```
root@DietPi:/# apt install libva-drm2
Reading package lists... Done
Building dependency tree       
Reading state information... Done
Some packages could not be installed. This may mean that you have
requested an impossible situation or if you are using the unstable
distribution that some required packages have not yet been created
or been moved out of Incoming.
The following information may help to resolve the situation:

The following packages have unmet dependencies:
 libva-drm2 : Depends: libva2 (< 2.4.0.1) but 2.10.0-1~bpo10+1 is to be installed
E: Unable to correct problems, you have held broken packages.
root@DietPi:/# apt install libva2
Reading package lists... Done
Building dependency tree       
Reading state information... Done
libva2 is already the newest version (2.10.0-1~bpo10+1).
libva2 set to manually installed.
0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.
```

MichaIng, 2021-05-30:
Okay, you installed `libva2` (probably as dependency of Kodi), from a 3rd party repository, where a backported v2.10 is available, while the Debian repository does not have such backports. All Debian packages which depend on the libva2 version shipped by the Debian repo, are hence incompatible with this, which includes ffmpeg.

You said you purged Kodi already? Then you should be able to purge this library as well to have the correct one pulled from the Debian repo:
```
apt purge --autoremove libva2
dpkg -l | grep 'bpo10'
```
The second command is to check for further backported packages that might still be installed.

PrzemekSkw, 2021-05-30:
```
root@DietPi:/# apt purge --autoremove libva2
Reading package lists... Done
Building dependency tree       
Reading state information... Done
The following packages will be REMOVED:
  i965-va-driver* libva2*
0 upgraded, 0 newly installed, 2 to remove and 0 not upgraded.
After this operation, 2040 kB disk space will be freed.
Do you want to continue? [Y/n] y
(Reading database ... 48806 files and directories currently installed.)
Removing i965-va-driver:amd64 (2.3.0+dfsg1-1) ...

Progress: [  0%] [............................................................................................................................] 
Progress: [ 11%] [#############...............................................................................................................] 
Progress: [ 22%] [###########################.................................................................................................] 
Progress: [ 33%] [#########################################...................................................................................] 
Removing libva2:amd64 (2.10.0-1~bpo10+1) ...#############################.....................................................................] 

Progress: [ 56%] [####################################################################........................................................] 
Progress: [ 67%] [##################################################################################..........................................] 
Progress: [ 78%] [################################################################################################............................] 
Processing triggers for libc-bin (2.28-10) ...##################################################################################..............] 
root@DietPi:/# dpkg -l | grep 'bpo10'
rc  libjs-jquery                  3.5.1+dfsg-4~bpo10+1         all          JavaScript library for dynamic web applications
rc  libpulse0:amd64               13.0-3~bpo10+1               amd64        PulseAudio client libraries
```
do I need to purge some other packages?

PrzemekSkw, 2021-05-30:
OK, that helps. I install ffmpeg and then Bazarr. Thanks guys very much.
regards.

closed 2021-05-30 by PrzemekSkw

MichaIng, 2021-05-30:
Great 🙂