MichaIng/DietPi #4444: DietPi Error installing software as root user

v1rot-8, 2021-05-30:
#### Details:
- Date           | Mon May 31 10:33:23 NZST 2021
- DietPi version | v7.2.3 (MichaIng/master)
- Image creator  | DietPi Core Team
- Pre-image      | Raspberry Pi OS Lite (64-bit)
- Hardware       | RPi 3 Model B+ (aarch64) (ID=3)
- Kernel version | Linux SupportRpi-1 5.10.17-v8+ #1414 SMP PREEMPT Fri Apr 30 1                                           3:23:25 BST 2021 aarch64 GNU/Linux
- Distro         | buster (ID=5,RASPBIAN=0)
- Command        | chown dietpi:dietpi /mnt/dietpi_userdata/Music /mnt/dietpi_userdata/Pictures /mnt/dietpi_userdata/Video /mnt/dietpi_userdata/downloads
- Exit code      | 1
- Software title | DietPi-Software
#### Steps to reproduce:

1. ...
2. ...
#### Expected behaviour:

- ...
#### Actual behaviour:

Previous console lines:

root@SupportRpi-1:~# dietpi-software
[  OK  ] DietPi-Software | Initialised database
[  OK  ] DietPi-Software | Reading database
[  OK  ] DietPi-Software | Free space check: path=/ | available=54735 MiB | required=500 MiB
[  OK  ] DietPi-Software | DietPi-Userdata validation: /mnt/dietpi_userdata
[  OK  ] DietPi-Software | Checking network connectivity
[  OK  ] DietPi-Software | Checking DNS resolver
[  OK  ] Network time sync | Completed
[ SUB1 ] DietPi-Services > unmask
[  OK  ] DietPi-Services | unmask : avahi-daemon
[  OK  ] DietPi-Services | unmask : isc-dhcp-server
[  OK  ] DietPi-Services | unmask : rpimonitor
[  OK  ] DietPi-Services | unmask : webmin
[  OK  ] DietPi-Services | unmask : cron
[ SUB1 ] DietPi-Services > stop
[  OK  ] DietPi-Services | stop : cron
[  OK  ] DietPi-Services | stop : webmin
[  OK  ] DietPi-Services | stop : rpimonitor
[  OK  ] DietPi-Services | stop : isc-dhcp-server
[  OK  ] DietPi-Services | stop : avahi-daemon
[  OK  ] DietPi-Software | mkdir -p /mnt/dietpi_userdata/Music /mnt/dietpi_userdata/Pictures /mnt/dietpi_userdata/Video /mnt/dietpi_userdata/downloads /var/www/opt
[FAILED] DietPi-Software | chown dietpi:dietpi /mnt/dietpi_userdata/Music /mnt/d                                           ietpi_userdata/Pictures /mnt/dietpi_userdata/Video /mnt/dietpi_userdata/downloads
[FAILED] DietPi-Software | chown dietpi:dietpi /mnt/dietpi_userdata/Music /mnt/d                                           ietpi_userdata/Pictures /mnt/dietpi_userdata/Video /mnt/dietpi_userdata/downloads

- ...
#### Extra details:

- ...
#### Additional logs:
```
chown: invalid user: ‘dietpi:dietpi’
```

Joulinar, 2021-05-30:
Hi,

did you deleted user `dietpi` in the past? Because it should be available by default.

v1rot-8, 2021-05-30:
Finger trouble when cloning dietpi user last week Had renamed dietpi as another name. All fixed.

closed 2021-05-30 by v1rot-8