MichaIng/DietPi #4445: SSL certificate problem

cocoflan, 2021-05-31:
Get this error after upgrading to latest version on boot:

- WAN IP : curl: (60) SSL certificate problem: unable to get local issuer certificate 
More details here: https://curl.haxx.se/docs/sslcerts.html 
 
curl failed to verify the legitimacy of the server and therefore could not 
establish a secure connection to it. To learn more about this situation and 
how to fix it

Joulinar, 2021-05-31:
Hi,

i did a quick check and it is working. Can you ran it manually as follow

`curl -sSfLm 3 https://freegeoip.app/csv/ 2>&1 | mawk -F, '($5){r=$5" "}{print $1" "r$3}'`

cocoflan, 2021-05-31:
i get this on running that command

curl: (60) SSL certificate problem: unable to get local issuer certificate 
More details here: https://curl.haxx.se/docs/sslcerts.html 
 
curl failed to verify the legitimacy of the server and therefore could not 
establish a secure connection to it. To learn more about this situation and 
how to fix it

Joulinar, 2021-05-31:
Did you checked if your system time is up to date?

cocoflan, 2021-05-31:
Yes time is up to date.

MichaIng, 2021-05-31:
Please try to reinstall the `ca-certificates` package:
```
apt install --reinstall ca-certificates
```

cocoflan, 2021-05-31:
stays the same, get again the error.

MichaIng, 2021-05-31:
Does wget have the same issue?
```
wget -q https://freegeoip.app/csv/ -O -
```

If so, as I remember an old bug, can you try:
```
c_rehash
```

and show:
```
ls -l /etc/ssl/certs
```

cocoflan, 2021-06-01:
it stays the same. 

output of

ls -l /etc/ssl/certs
total 772
lrwxrwxrwx 1 root root     26 Jun  1 09:05  00673b5b.0 -> thawte_Primary_Root_CA.pem
lrwxrwxrwx 1 root root     45 Jun  1 09:05  02265526.0 -> Entrust_Root_Certification_Authority_-_G2.pem
lrwxrwxrwx 1 root root     36 Jun  1 09:05  03179a64.0 -> Staat_der_Nederlanden_EV_Root_CA.pem
lrwxrwxrwx 1 root root     41 Jun  1 09:05  04f60c28.0 -> USERTrust_ECC_Certification_Authority.pem
lrwxrwxrwx 1 root root     27 Jun  1 09:05  062cdee6.0 -> GlobalSign_Root_CA_-_R3.pem
lrwxrwxrwx 1 root root     25 Jun  1 09:05  064e0aa9.0 -> QuoVadis_Root_CA_2_G3.pem
lrwxrwxrwx 1 root root     50 Jun  1 09:05  06dc52d5.0 -> SSL.com_EV_Root_Certification_Authority_RSA_R2.pem
lrwxrwxrwx 1 root root     20 Jun  1 09:05  080911ac.0 -> QuoVadis_Root_CA.pem
lrwxrwxrwx 1 root root     54 Jun  1 09:05  09789157.0 -> Starfield_Services_Root_Certificate_Authority_-_G2.pem
lrwxrwxrwx 1 root root     15 Jun  1 09:05  0a775a30.0 -> GTS_Root_R3.pem
lrwxrwxrwx 1 root root     16 Jun  1 09:05  0b1b94ef.0 -> CFCA_EV_ROOT.pem
lrwxrwxrwx 1 root root     44 Jun  1 09:05  0bf05006.0 -> SSL.com_Root_Certification_Authority_ECC.pem
lrwxrwxrwx 1 root root     34 Jun  1 09:05  0c4c9b6c.0 -> Global_Chambersign_Root_-_2008.pem
lrwxrwxrwx 1 root root     31 Jun  1 09:05  0d69c7e1.0 -> GlobalSign_ECC_Root_CA_-_R4.pem
lrwxrwxrwx 1 root root     32 Jun  1 09:05  0f5dc4f3.0 -> UCA_Extended_Validation_Root.pem
lrwxrwxrwx 1 root root     26 Jun  1 09:05  0f6fa695.0 -> GDCA_TrustAUTH_R5_ROOT.pem
lrwxrwxrwx 1 root root     15 Jun  1 09:05  1001acf7.0 -> GTS_Root_R1.pem
lrwxrwxrwx 1 root root     54 Jun  1 09:05  10531352.0 -> Starfield_Services_Root_Certificate_Authority_-_G2.pem
lrwxrwxrwx 1 root root     46 Jun  1 09:05  106f3e4d.0 -> Entrust_Root_Certification_Authority_-_EC1.pem
lrwxrwxrwx 1 root root     27 Jun  1 09:05  111e6273.0 -> GlobalSign_Root_CA_-_R2.pem
lrwxrwxrwx 1 root root     49 Jun  1 09:05  116bf586.0 -> GeoTrust_Primary_Certification_Authority_-_G2.pem
lrwxrwxrwx 1 root root     35 Jun  1 09:05  128805a3.0 -> EE_Certification_Centre_Root_CA.pem
lrwxrwxrwx 1 root root     18 Jun  1 09:05  12d55845.0 -> DST_Root_CA_X3.pem
lrwxrwxrwx 1 root root     27 Jun  1 09:05  14bc7599.0 -> emSign_ECC_Root_CA_-_G3.pem
lrwxrwxrwx 1 root root     26 Jun  1 09:05  157753a5.0 -> AddTrust_External_Root.pem
lrwxrwxrwx 1 root root     59 Jun  1 09:05  1636090b.0 -> Hellenic_Academic_and_Research_Institutions_RootCA_2011.pem
lrwxrwxrwx 1 root root     31 Jun  1 09:05  17b51fe6.0 -> Certplus_Class_2_Primary_CA.pem
lrwxrwxrwx 1 root root     23 Jun  1 09:05  18856ac4.0 -> SecureSign_RootCA11.pem
lrwxrwxrwx 1 root root     31 Jun  1 09:05  1d3472b9.0 -> GlobalSign_ECC_Root_CA_-_R5.pem
lrwxrwxrwx 1 root root     15 Jun  1 09:05  1dcd6f4c.0 -> Taiwan_GRCA.pem
lrwxrwxrwx 1 root root     34 Jun  1 09:05  1df5a75f.0 -> D-TRUST_Root_Class_3_CA_2_2009.pem
lrwxrwxrwx 1 root root     37 Jun  1 09:05  1e08bfd1.0 -> IdenTrust_Public_Sector_Root_CA_1.pem
lrwxrwxrwx 1 root root     32 Jun  1 09:05  1e09d511.0 -> T-TeleSec_GlobalRoot_Class_2.pem
lrwxrwxrwx 1 root root     32 Jun  1 09:05  1e1eab7c.0 -> T-TeleSec_GlobalRoot_Class_3.pem
lrwxrwxrwx 1 root root     27 Jun  1 09:05  1e8e7201.0 -> GlobalSign_Root_CA_-_R3.pem
lrwxrwxrwx 1 root root     36 Jun  1 09:05  1eb37bdf.0 -> Chambers_of_Commerce_Root_-_2008.pem
lrwxrwxrwx 1 root root     25 Jun  1 09:05  1f58a078.0 -> QuoVadis_Root_CA_2_G3.pem
lrwxrwxrwx 1 root root     23 Jun  1 09:05  219d9499.0 -> Go_Daddy_Class_2_CA.pem
lrwxrwxrwx 1 root root     24 Jun  1 09:05  23f4c490.0 -> Starfield_Class_2_CA.pem
lrwxrwxrwx 1 root root     38 Jun  1 09:05  244b5494.0 -> DigiCert_High_Assurance_EV_Root_CA.pem
lrwxrwxrwx 1 root root     49 Jun  1 09:05  27af790d.0 -> GeoTrust_Primary_Certification_Authority_-_G2.pem
lrwxrwxrwx 1 root root     23 Jun  1 09:05  2923b3f9.0 -> emSign_Root_CA_-_G1.pem
lrwxrwxrwx 1 root root     31 Jun  1 09:05  2add47b6.0 -> GlobalSign_ECC_Root_CA_-_R5.pem
lrwxrwxrwx 1 root root     20 Jun  1 09:05  2ae6433e.0 -> CA_Disig_Root_R2.pem
lrwxrwxrwx 1 root root     26 Jun  1 09:05  2b349938.0 -> AffirmTrust_Commercial.pem
lrwxrwxrwx 1 root root     22 Jun  1 09:05  2c543cd1.0 -> GeoTrust_Global_CA.pem
lrwxrwxrwx 1 root root     27 Jun  1 09:05  2d9dafe4.0 -> Buypass_Class_3_Root_CA.pem
lrwxrwxrwx 1 root root     26 Jun  1 09:05  2e4eed3c.0 -> thawte_Primary_Root_CA.pem
lrwxrwxrwx 1 root root     18 Jun  1 09:05  2e5ac55d.0 -> DST_Root_CA_X3.pem
lrwxrwxrwx 1 root root     43 Jun  1 09:05  2fa87019.0 -> Network_Solutions_Certificate_Authority.pem
lrwxrwxrwx 1 root root     20 Jun  1 09:05  302904dd.0 -> Certigna_Root_CA.pem
lrwxrwxrwx 1 root root     22 Jun  1 09:05  304d27c3.0 -> UCA_Global_G2_Root.pem
lrwxrwxrwx 1 root root     49 Jun  1 09:05  31188b5e.0 -> TUBITAK_Kamu_SM_SSL_Kok_Sertifikasi_-_Surum_1.pem
lrwxrwxrwx 1 root root     59 Jun  1 09:05  32888f65.0 -> Hellenic_Academic_and_Research_Institutions_RootCA_2015.pem
lrwxrwxrwx 1 root root     44 Jun  1 09:05  33ee480d.0 -> SSL.com_Root_Certification_Authority_RSA.pem
lrwxrwxrwx 1 root root     26 Jun  1 09:05  343eb6cb.0 -> Cybertrust_Global_Root.pem
lrwxrwxrwx 1 root root     10 Jun  1 09:05  349f2832.0 -> EC-ACC.pem
lrwxrwxrwx 1 root root     41 Jun  1 09:05  35105088.0 -> USERTrust_RSA_Certification_Authority.pem
lrwxrwxrwx 1 root root     27 Jun  1 09:05  3513523f.0 -> DigiCert_Global_Root_CA.pem
lrwxrwxrwx 1 root root     27 Jun  1 09:05  399e7759.0 -> DigiCert_Global_Root_CA.pem
[... cut ...]
lrwxrwxrwx 1 root root     49 Jun  1 09:05  ff34af3f.0 -> TUBITAK_Kamu_SM_SSL_Kok_Sertifikasi_-_Surum_1.pem
drwxr-xr-x 2 root root   4096 May  4 15:39  java
lrwxrwxrwx 1 root root     59 May 31 03:48  thawte_Primary_Root_CA.pem -> /usr/local/share/ca-certificates/thawte_Primary_Root_CA.crt
lrwxrwxrwx 1 root root     64 May 31 03:48  thawte_Primary_Root_CA_-_G2.pem -> /usr/local/share/ca-certificates/thawte_Primary_Root_CA_-_G2.crt
lrwxrwxrwx 1 root root     64 May 31 03:48  thawte_Primary_Root_CA_-_G3.pem -> /usr/local/share/ca-certificates/thawte_Primary_Root_CA_-_G3.crt

MichaIng, 2021-06-01:
```
ACCVRAIZ1.pem -> /usr/local/share/ca-certificates/ACCVRAIZ1.crt
```
Hmm, the linked certs are located in the local admin managed directory instead of in the system managed directory. To compare on all my systems it's:
```
ACCVRAIZ1.pem -> /usr/share/ca-certificates/mozilla/ACCVRAIZ1.crt
```

Are these links to each others or indeed different files?
```
readlink -f /usr/local/share/ca-certificates/ACCVRAIZ1.crt /usr/share/ca-certificates/mozilla/ACCVRAIZ1.crt
```

Do you otherwise have an idea where this is coming from? A package should usually not ship it:
```
dpkg -S /usr/local/share/ca-certificates/ACCVRAIZ1.crt
dpkg -S "$(readlink -f /usr/local/share/ca-certificates/ACCVRAIZ1.crt)"
```

cocoflan, 2021-06-01:
readlink -f /usr/local/share/ca-certificates/ACCVRAIZ1.crt /usr/share/ca-certificates/mozilla/ACCVRAIZ1.crt
/usr/local/share/ca-certificates/ACCVRAIZ1.crt
/usr/share/ca-certificates/mozilla/ACCVRAIZ1.crt

cocoflan, 2021-06-01:
dpkg -S /usr/local/share/ca-certificates/ACCVRAIZ1.crt
dpkg-query: no path found matching pattern /usr/local/share/ca-certificates/ACCVRAIZ1.crt

dpkg -S "$(readlink -f /usr/local/share/ca-certificates/ACCVRAIZ1.crt)"
dpkg-query: no path found matching pattern /usr/local/share/ca-certificates/ACCVRAIZ1.crt

I have no idea where it is comming from. could it be netdata, i have installed it from source code in the past, and on one of my devices there is nothing more installed than netdata.

MichaIng, 2021-06-01:
Netdata is a monitoring tool and won't install any CA certificates on your system.

Let's try to (re)move those and re-configure the package to update the symlinks:
```
mv /usr/local/share/ca-certificates{,_bak}
dpkg-reconfigure -f noninteractive ca-certificates
ls -lL /etc/ssl/certs/ACCVRAIZ1.crt
```

cocoflan, 2021-06-01:
mv /usr/local/share/ca-certificates{,_bak}

dpkg-reconfigure -f noninteractive ca-certificates
Updating certificates in /etc/ssl/certs...
0 added, 0 removed; done.
Processing triggers for ca-certificates (20200601~deb10u2) ...
Updating certificates in /etc/ssl/certs...

ls -lL /etc/ssl/certs/ACCVRAIZ1.crt
ls: cannot access '/etc/ssl/certs/ACCVRAIZ1.crt': No such file or directory

cocoflan, 2021-06-01:
Get same error after reboot 
 
curl: (60) SSL certificate problem: unable to get local issuer certificate 
More details here: https://curl.haxx.se/docs/sslcerts.html 
 
curl failed to verify the legitimacy of the server and therefore could not 
establish a secure connection to it. To learn more about this situation and 
how to fix it

MichaIng, 2021-06-01:
Ah sorry, wrong filename:
```
ls -l /etc/ssl/certs/ACCVRAIZ1.pem
```

cocoflan, 2021-06-01:
ls -l /etc/ssl/certs/ACCVRAIZ1.pem
lrwxrwxrwx 1 root root 48 Jun  1 11:54 /etc/ssl/certs/ACCVRAIZ1.pem -> /usr/share/ca-certificates/mozilla/ACCVRAIZ1.crt

MichaIng, 2021-06-01:
Okay, the symlink is as expected now.

As edited in above, please reinstall the package one more time and check all symlinks, just to be sure:
```
apt install --reinstall ca-certificates
ls -l /etc/ssl/certs
```

cocoflan, 2021-06-01:
reinstalled it.

ls -l /etc/ssl/certs
total 564
lrwxrwxrwx 1 root root     45 Jun  1 09:08  02265526.0 -> Entrust_Root_Certification_Authority_-_G2.pem
lrwxrwxrwx 1 root root     36 Jun  1 09:08  03179a64.0 -> Staat_der_Nederlanden_EV_Root_CA.pem
lrwxrwxrwx 1 root root     27 Jun  1 09:08  062cdee6.0 -> GlobalSign_Root_CA_-_R3.pem
lrwxrwxrwx 1 root root     25 Jun  1 09:08  064e0aa9.0 -> QuoVadis_Root_CA_2_G3.pem
lrwxrwxrwx 1 root root     50 Jun  1 09:08  06dc52d5.0 -> SSL.com_EV_Root_Certification_Authority_RSA_R2.pem
lrwxrwxrwx 1 root root     20 Jun  1 09:08  080911ac.0 -> QuoVadis_Root_CA.pem
lrwxrwxrwx 1 root root     54 Jun  1 09:08  09789157.0 -> Starfield_Services_Root_Certificate_Authority_-_G2.pem
lrwxrwxrwx 1 root root     15 Jun  1 09:08  0a775a30.0 -> GTS_Root_R3.pem
lrwxrwxrwx 1 root root     16 Jun  1 09:08  0b1b94ef.0 -> CFCA_EV_ROOT.pem
lrwxrwxrwx 1 root root     44 Jun  1 09:08  0bf05006.0 -> SSL.com_Root_Certification_Authority_ECC.pem
lrwxrwxrwx 1 root root     34 Jun  1 09:08  0c4c9b6c.0 -> Global_Chambersign_Root_-_2008.pem
lrwxrwxrwx 1 root root     32 Jun  1 09:08  0f5dc4f3.0 -> UCA_Extended_Validation_Root.pem
lrwxrwxrwx 1 root root     26 Jun  1 09:08  0f6fa695.0 -> GDCA_TrustAUTH_R5_ROOT.pem
lrwxrwxrwx 1 root root     15 Jun  1 09:08  1001acf7.0 -> GTS_Root_R1.pem
lrwxrwxrwx 1 root root     46 Jun  1 09:08  106f3e4d.0 -> Entrust_Root_Certification_Authority_-_EC1.pem
lrwxrwxrwx 1 root root     35 Jun  1 09:08  128805a3.0 -> EE_Certification_Centre_Root_CA.pem
lrwxrwxrwx 1 root root     27 Jun  1 09:08  14bc7599.0 -> emSign_ECC_Root_CA_-_G3.pem
lrwxrwxrwx 1 root root     59 Jun  1 09:08  1636090b.0 -> Hellenic_Academic_and_Research_Institutions_RootCA_2011.pem
lrwxrwxrwx 1 root root     23 Jun  1 09:08  18856ac4.0 -> SecureSign_RootCA11.pem
lrwxrwxrwx 1 root root     31 Jun  1 09:08  1d3472b9.0 -> GlobalSign_ECC_Root_CA_-_R5.pem
lrwxrwxrwx 1 root root     37 Jun  1 09:08  1e08bfd1.0 -> IdenTrust_Public_Sector_Root_CA_1.pem
lrwxrwxrwx 1 root root     32 Jun  1 09:08  1e09d511.0 -> T-TeleSec_GlobalRoot_Class_2.pem
lrwxrwxrwx 1 root root     38 Jun  1 09:08  244b5494.0 -> DigiCert_High_Assurance_EV_Root_CA.pem
lrwxrwxrwx 1 root root     23 Jun  1 09:08  2923b3f9.0 -> emSign_Root_CA_-_G1.pem
lrwxrwxrwx 1 root root     20 Jun  1 09:08  2ae6433e.0 -> CA_Disig_Root_R2.pem
lrwxrwxrwx 1 root root     26 Jun  1 09:08  2b349938.0 -> AffirmTrust_Commercial.pem
lrwxrwxrwx 1 root root     18 Jun  1 09:08  2e5ac55d.0 -> DST_Root_CA_X3.pem
lrwxrwxrwx 1 root root     59 Jun  1 09:08  32888f65.0 -> Hellenic_Academic_and_Research_Institutions_RootCA_2015.pem
lrwxrwxrwx 1 root root     10 Jun  1 09:08  349f2832.0 -> EC-ACC.pem
lrwxrwxrwx 1 root root     27 Jun  1 09:08  3513523f.0 -> DigiCert_Global_Root_CA.pem
lrwxrwxrwx 1 root root     61 Jun  1 09:08  3bde41ac.0 -> Autoridad_de_Certificacion_Firmaprofesional_CIF_A62634068.pem
lrwxrwxrwx 1 root root     26 Jun  1 09:08  3e44d2f7.0 -> TrustCor_RootCert_CA-2.pem
lrwxrwxrwx 1 root root     27 Jun  1 09:08  3e45d192.0 -> Hongkong_Post_Root_CA_1.pem
lrwxrwxrwx 1 root root     31 Jun  1 09:08  40193066.0 -> Certum_Trusted_Network_CA_2.pem
lrwxrwxrwx 1 root root     16 Jun  1 09:08  4042bcee.0 -> ISRG_Root_X1.pem
lrwxrwxrwx 1 root root     34 Jun  1 09:08  40547a79.0 -> COMODO_Certification_Authority.pem
lrwxrwxrwx 1 root root     23 Jun  1 09:08  406c9bb1.0 -> emSign_Root_CA_-_C1.pem
lrwxrwxrwx 1 root root     43 Jun  1 09:08  4304c5e5.0 -> Network_Solutions_Certificate_Authority.pem
lrwxrwxrwx 1 root root     29 Jun  1 09:08  48bec511.0 -> Certum_Trusted_Network_CA.pem
lrwxrwxrwx 1 root root     27 Jun  1 09:08  4a6481c9.0 -> GlobalSign_Root_CA_-_R2.pem
lrwxrwxrwx 1 root root     27 Jun  1 09:08  4b718d9b.0 -> emSign_ECC_Root_CA_-_C3.pem
lrwxrwxrwx 1 root root     45 Jun  1 09:08  4bfab552.0 -> Starfield_Root_Certificate_Authority_-_G2.pem
lrwxrwxrwx 1 root root     26 Jun  1 09:08  4f316efb.0 -> SwissSign_Gold_CA_-_G2.pem
lrwxrwxrwx 1 root root     35 Jun  1 09:08  5273a94c.0 -> E-Tugra_Certification_Authority.pem
lrwxrwxrwx 1 root root     32 Jun  1 09:08  5443e9e3.0 -> T-TeleSec_GlobalRoot_Class_3.pem
lrwxrwxrwx 1 root root     27 Jun  1 09:08  54657681.0 -> Buypass_Class_2_Root_CA.pem
lrwxrwxrwx 1 root root     28 Jun  1 09:08  57bcb2da.0 -> SwissSign_Silver_CA_-_G2.pem
lrwxrwxrwx 1 root root     38 Jun  1 09:08  5a4d6896.0 -> Staat_der_Nederlanden_Root_CA_-_G3.pem
lrwxrwxrwx 1 root root     22 Jun  1 09:08  5ad8a5d6.0 -> GlobalSign_Root_CA.pem
lrwxrwxrwx 1 root root     38 Jun  1 09:08  5c44d531.0 -> Staat_der_Nederlanden_Root_CA_-_G2.pem
lrwxrwxrwx 1 root root     26 Jun  1 09:08  5cd81ad7.0 -> TeliaSonera_Root_CA_v1.pem
lrwxrwxrwx 1 root root     26 Jun  1 09:08  5d3033c5.0 -> TrustCor_RootCert_CA-1.pem
lrwxrwxrwx 1 root root     45 Jun  1 09:08  5e98733a.0 -> Entrust_Root_Certification_Authority_-_G4.pem
lrwxrwxrwx 1 root root     23 Jun  1 09:08  5f15c80c.0 -> TWCA_Global_Root_CA.pem
lrwxrwxrwx 1 root root     27 Jun  1 09:08  607986c7.0 -> DigiCert_Global_Root_G2.pem
lrwxrwxrwx 1 root root     15 Jun  1 09:08  626dceaf.0 -> GTS_Root_R2.pem
lrwxrwxrwx 1 root root     15 Jun  1 09:08  6410666e.0 -> Taiwan_GRCA.pem
lrwxrwxrwx 1 root root     29 Jun  1 09:08  653b494a.0 -> Baltimore_CyberTrust_Root.pem
lrwxrwxrwx 1 root root     27 Jun  1 09:08  68dd7389.0 -> Hongkong_Post_Root_CA_3.pem
lrwxrwxrwx 1 root root     40 Jun  1 09:08  6b99d060.0 -> Entrust_Root_Certification_Authority.pem
[... cut ...]
lrwxrwxrwx 1 root root     24 Jun  1 09:08  f387163d.0 -> Starfield_Class_2_CA.pem
lrwxrwxrwx 1 root root     18 Jun  1 09:08  f39fc864.0 -> SecureTrust_CA.pem
lrwxrwxrwx 1 root root     20 Jun  1 09:08  f51bb24c.0 -> Certigna_Root_CA.pem
lrwxrwxrwx 1 root root     41 Jun  1 09:08  fc5a8f99.0 -> USERTrust_RSA_Certification_Authority.pem
lrwxrwxrwx 1 root root     19 Jun  1 09:08  fe8a2cd8.0 -> SZAFIR_ROOT_CA2.pem
lrwxrwxrwx 1 root root     49 Jun  1 09:08  ff34af3f.0 -> TUBITAK_Kamu_SM_SSL_Kok_Sertifikasi_-_Surum_1.pem
drwxr-xr-x 2 root root   4096 May  4 15:39  java

cocoflan, 2021-06-01:
Get still same error after reboot

curl: (60) SSL certificate problem: unable to get local issuer certificate
More details here: https://curl.haxx.se/docs/sslcerts.html

curl failed to verify the legitimacy of the server and therefore could not
establish a secure connection to it. To learn more about this situation and
how to fix it

MichaIng, 2021-06-01:
Okay, all certs are now as expected.

Let's check the certificate you get:
```
openssl s_client -showcerts -servername freegeoip.app -connect freegeoip.app:443
```

cocoflan, 2021-06-01:
openssl s_client -showcerts -servername freegeoip.app -connect freegeoip.app:443
CONNECTED(00000003)
depth=2 O = NextDNS Inc., CN = NextDNS Blockpage CA
verify error:num=20:unable to get local issuer certificate
verify return:1
depth=1 O = NextDNS Inc., CN = NextDNS Blockpage Edge CA
verify return:1
depth=0 O = NextDNS Inc., CN = blockpage.nextdns.io
verify return:1
---
Certificate chain
 0 s:O = NextDNS Inc., CN = blockpage.nextdns.io
   i:O = NextDNS Inc., CN = NextDNS Blockpage Edge CA
-----BEGIN CERTIFICATE-----
MIIBtzCCAV2gAwIBAgIJAOUr/KgS13yJMAoGCCqGSM49BAMCMDsxFTATBgNVBAoT
DE5leHRETlMgSW5jLjEiMCAGA1UEAxMZTmV4dEROUyBCbG9ja3BhZ2UgRWRnZSBD
QTAeFw0yMTA1MzEwMDAwMDFaFw0yMTA2MDUwMDAwMDFaMDYxFTATBgNVBAoTDE5l
eHRETlMgSW5jLjEdMBsGA1UEAxMUYmxvY2twYWdlLm5leHRkbnMuaW8wWTATBgcq
hkjOPQIBBggqhkjOPQMBBwNCAAR36LMPs7rS9VNpgYfZk7vPkPhfAdxZIZ7aJ+nM
t7hmDoJtbOFoyiwhC7mFda3JwTSfi3yF3PmNqnXeaBy87+Hdo08wTTAOBgNVHQ8B
Af8EBAMCBaAwEwYDVR0lBAwwCgYIKwYBBQUHAwEwDAYDVR0TAQH/BAIwADAYBgNV
HREEETAPgg1mcmVlZ2VvaXAuYXBwMAoGCCqGSM49BAMCA0gAMEUCIQDPTZXQ8zeT
aXv3kmboXkxxE1n4On3mvVFfKK78WdUx6wIgbvjUVq5iST33MfFnFts/16bHgaTz
usNxdf8f2hA2pNQ=
-----END CERTIFICATE-----
 1 s:O = NextDNS Inc., CN = NextDNS Blockpage Edge CA
   i:O = NextDNS Inc., CN = NextDNS Blockpage CA
-----BEGIN CERTIFICATE-----
MIIB0zCCAXmgAwIBAgIRANnnYyf/xllFao2aWYUSGlkwCgYIKoZIzj0EAwIwNjEV
MBMGA1UEChMMTmV4dEROUyBJbmMuMR0wGwYDVQQDExROZXh0RE5TIEJsb2NrcGFn
ZSBDQTAeFw0yMTA1MzEwMDAwMDFaFw0yMTA2MDUwMDAwMDFaMDsxFTATBgNVBAoT
DE5leHRETlMgSW5jLjEiMCAGA1UEAxMZTmV4dEROUyBCbG9ja3BhZ2UgRWRnZSBD
QTBZMBMGByqGSM49AgEGCCqGSM49AwEHA0IABHfosw+zutL1U2mBh9mTu8+Q+F8B
3Fkhnton6cy3uGYOgm1s4WjKLCELuYV1rcnBNJ+LfIXc+Y2qdd5oHLzv4d2jYzBh
MA4GA1UdDwEB/wQEAwICBDAPBgNVHRMBAf8EBTADAQH/MB0GA1UdDgQWBBR7QId+
pYe5FLpDyvx1sLBbj8Wb8DAfBgNVHSMEGDAWgBTQLJ4rTtsqz+S22OVpYIZjBN9f
1DAKBggqhkjOPQQDAgNIADBFAiBG1YtEDc3bVDGqeREwSOTHOI4yhyAH1DET3ZHD
wom9LgIhAIP7FU2wYpeaK1CyyusCSKvtXxOjR9lWZq4RDux8uR3l
-----END CERTIFICATE-----
 2 s:O = NextDNS Inc., CN = NextDNS Blockpage CA
   i:O = NextDNS Inc., CN = NextDNS Root CA
-----BEGIN CERTIFICATE-----
MIIDjTCCAXWgAwIBAgIRAIxV6Ouw4eDaOPpBqxkTr10wDQYJKoZIhvcNAQELBQAw
MTEVMBMGA1UEChMMTmV4dEROUyBJbmMuMRgwFgYDVQQDEw9OZXh0RE5TIFJvb3Qg
Q0EwHhcNMjAwNjEyMDMwMzQyWhcNMjUwNjEyMDMwMzQyWjA2MRUwEwYDVQQKEwxO
ZXh0RE5TIEluYy4xHTAbBgNVBAMTFE5leHRETlMgQmxvY2twYWdlIENBMFkwEwYH
KoZIzj0CAQYIKoZIzj0DAQcDQgAEcTI8DMT9znsbhsj+mq4ydx1ziMp1avGy8cgd
pJeYz9Ui8kfO7xZkGe5Sv/mpaFmTRVeI5KUXFx1SHtUNJbY53qNmMGQwDgYDVR0P
AQH/BAQDAgIEMBIGA1UdEwEB/wQIMAYBAf8CAQEwHQYDVR0OBBYEFNAsnitO2yrP
5LbY5WlghmME31/UMB8GA1UdIwQYMBaAFHX0wb8dexh2ldvQgETQerb8fQCvMA0G
CSqGSIb3DQEBCwUAA4ICAQCzmy2tjMMgtkPfiFPsVGY6u5xWUT0w7aCNdPQgplN5
G2dJhRt1GazasH1Kv3RL/GBIwJI2r+5NT4Xct0yd2GQpXoL3MSlK2gs8ZeB7l+y3
I+Va5URxX20zW4TY2T+CGwuaHLTuZC1nCrSfZ8ngarm79QDB2eC68X34+fRiD7mU
7abAVSmOrUGQTkrOu0dqdIjuRLX1WsoPZfgC4lujKmYRjxqg1m769PO6qFsgpeuR
dupVw1rFWsrYlAPMHSTKD09O//nrLJv70XZVtfdT02J+t0Se785UQ0HkAvAQg4Cj
po9pXbj7Wd6H3OqChBzMM9GoeYJALzvg3lVy1prvXdFtLP/8uBHuttdTMoxPR+Hj
yg310VtdAY8o9TItqi5xoyFjNT7qsC0uIsJOT7GWiUiJlltmrsOH6hdN5x7CuuLJ
nTIdL4GJ3RR17U0zLK7PfTTCr5L0ZpFbc4SYxw3jPxmnn4uroLhhUURQP1Qft9hk
JmVJ0xnjwCOlAq5LNWhXTW2fYAVF6Uf3FY5xMIyMXY6U7QWg5CVjKAr41xvKFS09
pXSnJ+YgwprjAK2tA5JawenxpskkjoGe5/txGxurrwqcOKlyAYNMYf1Dt9PTZqgY
r+1/5sTUlxFuei1TteEGA/25LICLETQ4wG8dK9whpnWt4W63N3+VP5ns/6bkO2SN
og==
-----END CERTIFICATE-----
---
Server certificate
subject=O = NextDNS Inc., CN = blockpage.nextdns.io

issuer=O = NextDNS Inc., CN = NextDNS Blockpage Edge CA

---
No client certificate CA names sent
Peer signing digest: SHA256
Peer signature type: ECDSA
Server Temp Key: X25519, 253 bits
---
SSL handshake has read 2191 bytes and written 369 bytes
Verification error: unable to get local issuer certificate
---
New, TLSv1.3, Cipher is TLS_CHACHA20_POLY1305_SHA256
Server public key is 256 bit
Secure Renegotiation IS NOT supported
Compression: NONE
Expansion: NONE
No ALPN negotiated
Early data was not sent
Verify return code: 20 (unable to get local issuer certificate)
---
---
Post-Handshake New Session Ticket arrived:
SSL-Session:
    Protocol  : TLSv1.3
    Cipher    : TLS_CHACHA20_POLY1305_SHA256
    Session-ID: CB944A3B37CAA820DA445FA4B8B705028CDF06369CCE0B7F0B12BDDD3E604B4D
    Session-ID-ctx: 
    Resumption PSK: 837D305552FB6554DC48FC932D00CEDA98A9E06B0F24BFAE817B688993CF4471
    PSK identity: None
    PSK identity hint: None
    SRP username: None
    TLS session ticket lifetime hint: 604800 (seconds)
    TLS session ticket:
    0000 - d4 34 c6 be 0f 22 58 dc-f7 27 29 fd b4 a1 b6 e6   .4..."X..').....
    0010 - 8d 8a d1 33 8f 22 a0 91-17 09 7a d2 e6 aa 77 9b   ...3."....z...w.
    0020 - 01 c0 4d d1 30 d4 cf 55-bd 4e 75 86 a7 2b 6a e6   ..M.0..U.Nu..+j.
    0030 - 11 42 7c 77 1d 5d 23 43-1c 3d 86 f5 fd 94 49 da   .B|w.]#C.=....I.
    0040 - 95 6b e1 39 19 1f b3 20-e2 37 6f 12 43 30 f8 dd   .k.9... .7o.C0..
    0050 - 86 63 ba 4c 32 90 59 72-96 09 1e 1a 6e d1 cb 10   .c.L2.Yr....n...
    0060 - f5 76 b8 96 36 90 2d 96-40 6d ef 1b 30 e0 9e 55   .v..6.-.@m..0..U
    0070 - 36                                                6

    Start Time: 1622542865
    Timeout   : 7200 (sec)
    Verify return code: 20 (unable to get local issuer certificate)
    Extended master secret: no
    Max Early Data: 0
---
read R BLOCK
closed

cocoflan, 2021-06-01:
So must be nextdns, i use a client on al my devices.

Joulinar, 2021-06-01:
looks like nextdns is rewriting the certificate?

Usually it looks like this

```
root@DietPi3:~# openssl s_client -showcerts -servername freegeoip.app -connect freegeoip.app:443
CONNECTED(00000003)
depth=2 C = IE, O = Baltimore, OU = CyberTrust, CN = Baltimore CyberTrust Root
verify return:1
depth=1 C = US, O = "Cloudflare, Inc.", CN = Cloudflare Inc ECC CA-3
verify return:1
depth=0 C = US, ST = CA, L = San Francisco, O = "Cloudflare, Inc.", CN = sni.cloudflaressl.com
verify return:1
---
Certificate chain
 0 s:C = US, ST = CA, L = San Francisco, O = "Cloudflare, Inc.", CN = sni.cloudflaressl.com
   i:C = US, O = "Cloudflare, Inc.", CN = Cloudflare Inc ECC CA-3
```

MichaIng, 2021-06-01:
NextDNS is blocking this domain. That's bad... We had a longer check (and me discussion with a blocklist maintainer) about whether it is reasonable to have this blocked or not:
- https://github.com/MichaIng/DietPi/pull/4398#issuecomment-845480973
- https://github.com/lightswitch05/hosts/issues/281

To me it still does not make sense, when seeing the whole consequent picture of this. Everyone that you connect to, has your IP, and can use any API or a self-hosted service to derive your location from this. Blocking a GEO IP service breaks it for you and not for anyone else, so first of all you are unable to see your own IP or location as a result. While locally running evil JavaScripts can of course use this and send result to anywhere, the recipient could do this by itself as well, so blocking it on your system does not really protect your privacy. But many blocklist maintainers see some of those GEO IP APIs as tracking tools, as they are called in combination with tracking scripts. Sadly the reliable and fast providers are of course also more often used, hence land on blocklists, while small unreliable providers (like the one we used before, which had server downtimes often and strict rate limits) are not blocked. Funnily Google Analytics is then not blocked in all cases, one of the most intrusive tracking scripts that sends all data to Google and is used by ~50% of all popular websites...

Sorry, it makes me a little angry because to me this is all not logic, but we're unable to do anything about it 😄.

cocoflan, 2021-06-01:
Fixed!! added geoip.app to the whitelist on nextdns account!

Many thanks to help me to fix the problem!

MichaIng, 2021-06-01:
Okay, great that this works 👍. I tried to find the responsible blocklist entry, but it seems NextDNS is pulling in a large list of third party blocklists, time-consuming to check them all: https://github.com/nextdns/metadata/tree/master/privacy/blocklists
I guess lists can be added/enabled/disabled then?

cocoflan, 2021-06-01:
[image]

Yes you can add a selection of lists and also deny and allow manually.

MichaIng, 2021-06-01:
Btw, you may want to remove the obsolete certificates directory:
```
rm -R /usr/local/share/ca-certificates_bak
```
While it was not the issue here, as these files are not updated with the `ca-certificates`, they would have caused issues earlier or later, when contained CA certs were expired. It's definitely better to use the trusted list of certs, shipped with the trusted APT package 🙂.

As of `freegeoip.app`, I think best is to handle curl error 60 specifically and show a crafted message instead that the domain seems to be blocked and hence either locally running DNS sinkholes or blocklists of the public provider (if it applies any) should be checked.

MichaIng, 2021-08-14:
Done via: #4638

closed 2021-08-20 by MichaIng