MichaIng/DietPi #4447: Can't use SSH after enabling dietpi-vpn killswitch setting

yslupdates, 2021-05-31:
#### Required Information
- DietPi version | 7.2.3
- Distro version | buster
- Kernel version | Linux DietPi 5.10.17-v8+ #1421 SMP PREEMPT Thu May 27 14:01:37 BST 2021 aarch64 GNU/Linux
- SBC model | Raspberry Pi 4
- Power supply used | USB C rpi official charger
- SDcard used | samsung

#### Additional Information

Unable to remote in via SSH after enabling killswitch setting in new dietpi-vpn app

#### Steps to reproduce

1. Setup dietpi-vpn with surfshark custom .ovpn file successfully
2. Enable killswitch, iptables installed automatically, works fine while still logged on remotely via SSH.
3. Do a reboot, unable to connect to SSH but Pi connects to network just fine. I can see it on my router

#### Expected behaviour

- Should be able to SSH

#### Actual behaviour

- Killswitch is blocking SSH by default I think? No changes made on my end to block port 22, nor is anything changed

#### Extra details

- Posted about this issue in dietpi forums also https://dietpi.com/phpbb/viewtopic.php?f=11&t=9082

Joulinar, 2021-05-31:
There is a possible solution provided on the forum as well

MichaIng, 2021-05-31:
Many thanks for your report.

@Joulinar 
Thanks, I was just thinking about such, but trendy already mentions those.

I'm just wondering if this implies weakening the killswitch, hence if this is bullet prove to reliably prevent the system to start connections (however initiated) outside of the tunnel.

MichaIng, 2021-06-26:
Done with: https://github.com/MichaIng/DietPi/commit/cbad2b84702c0a18c68eb5b5da591dee92988080

closed 2021-06-26 by MichaIng