MichaIng/DietPi #4450: Jacket update

Psych85, 2021-06-01:
Updating Jackett via the program breaks the program

Joulinar, 2021-06-01:
Can you be a little bit more specific on your issue? Did you try to post it on Jackett specific board as well? https://github.com/Jackett/Jackett

Psych85, 2021-06-01:
Apologies for the brief information. This is the first time I'm raising an issue on GitHub. 

I wonder if this might be a dependency issue. Reading being that following an update of dietpi Jackett stopped responding. I restored from a backup and all was well. 

I'm more than happy to provide more information but may need some guidance and direction to do so. 

I have not raised this on the Jackett GitHub page.

Joulinar, 2021-06-01:
pls describe the steps you have done before Jackett stopped

MichaIng, 2021-06-01:
There was one intermediate Jackett version which removed the configuration on update. Could that be the case? That was resolved with the latest version but I guess when you're currently running the affected one, the "old" updater still does this. Workaround would be to copy the .Jackett dir inside of the program dir outside, perform the update (which is always the latest version) and copy .Jackett back inside.

Psych85, 2021-06-01:
Many thanks. I will give this a go and report back. 

In terms of reproducing the problem from the previous question. There is a "check for updates" button within Jackett which once clicked downloads the latest version. Following a brief moment Jackett becomes u reachable.

Joulinar, 2021-06-01:
If the issue persists, it might be good to report at Jackett GitHub

Psych85, 2021-06-01:
I can confirm that copying the Jackett folder seems to have done the trick. 

Thanks for all the help and I will endeavour to better at reporting issues.

MichaIng, 2021-06-01:
> I will endeavour to better at reporting issues.

No problem. It would be great, if for some reason the next Jackett update breaks it again, if you could report back 🙂.

Psych85, 2021-06-01:
> > I will endeavour to better at reporting issues.
> 
> 
> 
> No problem. It would be great, if for some reason the next Jackett update breaks it again, if you could report back 🙂.

Sure no problem.

Joulinar, 2021-06-03:
I guess this one can be closed as we have a workaround. Feel free to reopen if needed.

closed 2021-06-03 by Joulinar