MichaIng/DietPi #4451: DietPi-Software | Jupyter: Web-based IDE

MichaIng, 2021-06-01:
https://jupyter.org/

fpetru, 2021-06-02:
## Interactive development using JupyterHub with DietPi

With it, you could start a notebook from any device in your network (at home or office). Jupyter notebooks could be quite useful, especially for Python users. It's easy to setup them locally, but it's always easier to have run via a server (through JupyterHub). Everything becomes at handy, and the notebook could be started from any device (tablet, phone, computer etc.). 

Site: https://jupyter.org/

JupyterHub consists of three components:
- A single-user notebook server that is started for every user on the system when they log in
- The hub that manages the user accounts, authentication and coordinates the single-user notebook servers.
- A proxy that routes the user requests to the hub and the notebook servers

Installing the hub could be done using `pip` command:

`pip3 install notebook jupyterhub`

This uses a configuration file: `jupyterhub_config.py` and here we could set also the binding port (and choose a different port than default `8000`).

` jupyterhub -f /path/jupyterhub_config.py`

We could have it together installed with jupyterlab  

`pip3 install jupyterlab` 

`jupyter serverextension enable --py jupyterlab --system`

And we could have `numpy` and `seaborn` packages as ones of the most used packages:

`pip3 install numpy`
`pip3 install seaborn`

With them any user could start running graphs, and process numbers.

Reference: <https://towardsdatascience.com/setup-your-home-jupyterhub-on-a-raspberry-pi-7ad32e20eed>

MichaIng, 2026-10-27:
This is a simple Python project with end user executable, nothing we could do any reasonable ready-to-run setup: https://jupyter.org/install
```sh
python3 -m venv jupyter
jupyter/bin/pip install -U jupyterlab
jupyter/bin/jupyter
```