MichaIng/DietPi #4457: Unable to update Amiberry to version 4.1.2

corasaniti, 2021-06-02:
After updating Amiberry to version 4.1.2 and starting it doesn't work.

dietpi-software reinstall 108

The mouse gets stuck in the upper left corner of the screen.
I downgraded to previous version 3.3
Please fix the problem
Thank you

MichaIng, 2021-06-02:
Many thanks for your report. Which SBC model and Debian version do you run? Basically it would be great if you could fill the bug report template, that's why it exists 😉.

corasaniti, 2021-06-02:
#### Required Information
G_DIETPI_VERSION_CORE=7
G_DIETPI_VERSION_SUB=2
G_DIETPI_VERSION_RC=3
G_GITBRANCH='master'
G_GITOWNER='MichaIng'

Distro version 10.9
Kernel version Linux DietPi 5.10.17-v7l+ #1421 SMP Thu May 27 14:00:13 BST 2021 armv7l GNU/Linux
SBC model RPi4-8GB
Power supply 5V 1A 
SDcard used  SanDisk 

#### Steps to reproduce
update Amiberry with command
dietpi-software reinstall 108

#### Expected behaviour
Correct operation of the mouse in Amiberry

#### Actual behaviour
The mouse freezes at the top left of the screen

MichaIng, 2021-06-02:
Seems to be a bug in LibSDL2. I'll rebuild it with current nightly/development version.

MichaIng, 2021-06-02:
I compiled an uploaded a test Amiberry build: https://dietpi.com/downloads/binaries/buster/amiberry_rpi4_test.tar.bz2
Based on SDL v2.0.15, so the development version. Let's see if that has the issue resolved already. I'll also run some tests tomorrow. Otherwise we need to downgrade to SDL v2.0.12.

corasaniti, 2021-06-03:
Hi @MichaIng 
Many thanks,  I prefer to wait for it to be released along with the new version of Diet-pi
In the meantime, I continue to use version 3.3
Thanks again

MichaIng, 2021-06-08:
Same here: https://dietpi.com/phpbb/viewtopic.php?t=9056

MichaIng, 2021-07-24:
All Amyberry builds have been updated in the meantime.

closed 2021-07-24 by MichaIng