MichaIng/DietPi #4458: Update issue with Allo UsBridge

BelakScientific, 2021-06-02:
login as: root
root@192.168.1.7's password:

The programs included with the Debian GNU/Linux system are free software;
the exact distribution terms for each program are described in the
individual files in /usr/share/doc/*/copyright.

Debian GNU/Linux comes with ABSOLUTELY NO WARRANTY, to the extent
permitted by applicable law.
 ─────────────────────────────────────────────────────
 DietPi v6.26.3 : Update available
 ─────────────────────────────────────────────────────
 - Device model : RPi CM 3+ (armv7l)
 - CPU temp : 50'C : 122'F (Running warm, but safe)
 - LAN IP : 192.168.1.7 (eth0)
 - MOTD : DietPi v7.2 has been released, please update now.
 ─────────────────────────────────────────────────────

 DietPi Team     : MichaIng (lead), Daniel Knight (founder)
 Image           : DietPi Core Team (pre-image: Raspbian Lite)
 Web             : https://DietPi.com | https://twitter.com/dietpi_
 Patreon Legends : PINE64 community | oct8l
 Donate          : https://DietPi.com/#donate
 DietPi Hosting  : Powered by https://MyVirtualServer.com

 dietpi-update   : Run now to update DietPi from v6.26.3 to v6.35.0.

 dietpi-launcher : All the DietPi programs in one place.
 dietpi-config   : Feature rich configuration tool for your device.
 dietpi-software : Select optimized software for installation.
 htop            : Resource monitor.
 cpu             : Shows CPU information and stats.

root@DietPi:~# uname -a
Linux DietPi 4.19.73-v7+ #1266 SMP Tue Sep 17 20:52:18 BST 2019 armv7l GNU/Linux
root@DietPi:~# Linux DietPi 4.19.73-v7+ #1266 SMP Tue Sep 17 20:52:18 BST 2019 a                                    rmv7l GNU/Linux
-bash: Linux: command not found
root@DietPi:~# root@DietPi:~#
-bash: root@DietPi:~#: command not found
root@DietPi:~# ^C
root@DietPi:~# dietpi-update

 DietPi-Update
─────────────────────────────────────────────────────
 Mode: Checking for DietPi updates

[ INFO ] DietPi-Update | Checking mirror: https://raw.githubusercontent.com/Mich                                    aIng/DietPi/master/dietpi/server_version-6
[  OK  ] DietPi-Update | Using update server: https://raw.githubusercontent.com/                                    MichaIng/DietPi/master/dietpi/server_version-6
[  OK  ] DietPi-Update | Update available:
[ INFO ] DietPi-Update | Current version : v6.26.3
[ INFO ] DietPi-Update | Latest version  : v6.35.0
[  OK  ] DietPi-Update | DietPi-Userdata validation: /mnt/dietpi_userdata
[  OK  ] DietPi-Update | Free space check: path=/ | available=12399 MiB | requir                                    ed=100 MiB
[ INFO ] DietPi-Update | Lines required for Whiptail box and its contents (y=21,                                     z=0), exceeds screen dimensions (y=20).

 DietPi-Update
─────────────────────────────────────────────────────
 Mode: Applying DietPi update

[ SUB1 ] DietPi-Services > stop
[  OK  ] DietPi-Services | stop : cron
[  OK  ] DietPi-Services | stop : netdata
[  OK  ] DietPi-Services | stop : logitechmediaserver
[  OK  ] DietPi-Services | stop : roonbridge
[  OK  ] DietPi-Services | stop : networkaudiod
[  OK  ] DietPi-Services | stop : gmrender
[  OK  ] DietPi-Services | stop : squeezelite
[  OK  ] DietPi-Services | stop : shairport-sync
[  OK  ] DietPi-Services | stop : mpd
[  OK  ] DietPi-Services | stop : lighttpd
[... 725 lines cut ...]

1. ...
2. ...
#### Expected behaviour:

- ...
#### Actual behaviour:

- ...
#### Extra details:

- ...
#### Additional logs:
```
E: Unable to correct problems, you have held broken packages.
```
---------------------------------------------------------------------
[FAILED] DietPi-Software | Unable to continue, DietPi-Software will now terminate.
[FAILED] DietPi-Update | An error occured during incremental patching. Please check the above log or /var/tmp/dietpi/logs/dietpi-update.log for errors, and rerun "dietpi-update" after the cause has been solved.
[  OK  ] DietPi-Patch | Everything done! Terminating the obsolete DietPi-Update parent instance...

Terminated
root@DietPi:~# ^C
root@DietPi:~# ^C
root@DietPi:~#

ravenclaw900, 2021-06-02:
Hello,
Could you post the full output of:
```
apt update && apt upgrade
```

BelakScientific, 2021-06-02:
Here it is:

login as: root
root@192.168.1.7's password:

The programs included with the Debian GNU/Linux system are free software;
the exact distribution terms for each program are described in the
individual files in /usr/share/doc/*/copyright.

Debian GNU/Linux comes with ABSOLUTELY NO WARRANTY, to the extent
permitted by applicable law.
 ─────────────────────────────────────────────────────
 DietPi v6.26.3 : Update available
 ─────────────────────────────────────────────────────
 - Device model : RPi CM 3+ (armv7l)
 - CPU temp : 50'C : 122'F (Running warm, but safe)
 - LAN IP : 192.168.1.7 (eth0)
 - MOTD : DietPi v7.2 has been released, please update now.
 ─────────────────────────────────────────────────────

 DietPi Team     : MichaIng (lead), Daniel Knight (founder)
 Image           : DietPi Core Team (pre-image: Raspbian Lite)
 Web             : https://DietPi.com | https://twitter.com/dietpi_
 Patreon Legends : PINE64 community | oct8l
 Donate          : https://DietPi.com/#donate
 DietPi Hosting  : Powered by https://MyVirtualServer.com

 dietpi-update   : Run now to update DietPi from v6.26.3 to v6.35.0.

 dietpi-launcher : All the DietPi programs in one place.
 dietpi-config   : Feature rich configuration tool for your device.
 dietpi-software : Select optimized software for installation.
 htop            : Resource monitor.
 cpu             : Shows CPU information and stats.

root@DietPi:~# uname -a
Linux DietPi 4.19.73-v7+ #1266 SMP Tue Sep 17 20:52:18 BST 2019 armv7l GNU/Linux
root@DietPi:~# Linux DietPi 4.19.73-v7+ #1266 SMP Tue Sep 17 20:52:18 BST 2019 a                                                                                                                                                                                                                                                                                                                                                           rmv7l GNU/Linux
-bash: Linux: command not found
root@DietPi:~# root@DietPi:~#
-bash: root@DietPi:~#: command not found
root@DietPi:~# ^C
root@DietPi:~# dietpi-update

 DietPi-Update
─────────────────────────────────────────────────────
 Mode: Checking for DietPi updates

[ INFO ] DietPi-Update | Checking mirror: https://raw.githubusercontent.com/Mich                                                                                                                                                                                                                                                                                                                                                           aIng/DietPi/master/dietpi/server_version-6
[  OK  ] DietPi-Update | Using update server: https://raw.githubusercontent.com/                                                                                                                                                                                                                                                                                                                                                           MichaIng/DietPi/master/dietpi/server_version-6
[  OK  ] DietPi-Update | Update available:
[ INFO ] DietPi-Update | Current version : v6.26.3
[ INFO ] DietPi-Update | Latest version  : v6.35.0
[  OK  ] DietPi-Update | DietPi-Userdata validation: /mnt/dietpi_userdata
[  OK  ] DietPi-Update | Free space check: path=/ | available=12399 MiB | requir                                                                                                                                                                                                                                                                                                                                                           ed=100 MiB
[ INFO ] DietPi-Update | Lines required for Whiptail box and its contents (y=21,                                                                                                                                                                                                                                                                                                                                                            z=0), exceeds screen dimensions (y=20).

 DietPi-Update
─────────────────────────────────────────────────────
 Mode: Applying DietPi update

[ SUB1 ] DietPi-Services > stop
[  OK  ] DietPi-Services | stop : cron
[  OK  ] DietPi-Services | stop : netdata
[  OK  ] DietPi-Services | stop : logitechmediaserver
[  OK  ] DietPi-Services | stop : roonbridge
[  OK  ] DietPi-Services | stop : networkaudiod
[  OK  ] DietPi-Services | stop : gmrender
[  OK  ] DietPi-Services | stop : squeezelite
[  OK  ] DietPi-Services | stop : shairport-sync
[... 762 lines cut ...]
W: Some index files failed to download. They have been ignored, or old ones used instead.
Reading package lists... Done
Building dependency tree
Reading state information... Done
Calculating upgrade... Done
The following package was automatically installed and is no longer required:
  libbluray2
Use 'apt autoremove' to remove it.
The following NEW packages will be installed:
  libbluray1 libicu64
The following packages will be upgraded:
  libavformat57 libxml2
2 upgraded, 2 newly installed, 0 to remove and 0 not upgraded.
Need to get 9,723 kB of archives.
After this operation, 30.7 MB of additional disk space will be used.
Do you want to continue? [Y/n] y
[... cut ...]                                                                     4 [8,102 kB]
Err:4 https://packages.sury.org/php stretch/main armhf libxml2 armhf 2.9.9+dfsg-1+0~20190914.4+debian9~1.gbp3b6674
  404  Not Found
Fetched 9,108 kB in 27s (327 kB/s)
E: Failed to fetch https://packages.sury.org/php/pool/main/libx/libxml2/libxml2_2.9.9+dfsg-1+0~20190914.4+debian9~1.gbp3b6674_armhf.deb  404  Not Found
E: Unable to fetch some archives, maybe run apt-get update or try with --fix-missing?
root@DietPi:~#

BelakScientific, 2021-06-02:
Here it is:

root@DietPi:~# apt update && apt upgrade
Hit:1 http://raspbian.raspberrypi.org/raspbian stretch InRelease
Hit:3 https://archive.raspberrypi.org/debian stretch InRelease
Err:2 https://packages.sury.org/php stretch InRelease
  The following signatures were invalid: EXPKEYSIG B188E2B695BD4743 DEB.SURY.ORG Automatic Signing Key <deb@sury.org                                                                                                                                                                                                                                                                                                                       >
Reading package lists... Done
Building dependency tree
Reading state information... Done
2 packages can be upgraded. Run 'apt list --upgradable' to see them.
W: An error occurred during the signature verification. The repository is not updated and the previous index files w                                                                                                                                                                                                                                                                                                                       ill be used. GPG error: https://packages.sury.org/php stretch InRelease: The following signatures were invalid: EXPK                                                                                                                                                                                                                                                                                                                       EYSIG B188E2B695BD4743 DEB.SURY.ORG Automatic Signing Key <deb@sury.org>
W: Failed to fetch https://packages.sury.org/php/dists/stretch/InRelease  The following signatures were invalid: EXP                                                                                                                                                                                                                                                                                                                       KEYSIG B188E2B695BD4743 DEB.SURY.ORG Automatic Signing Key <deb@sury.org>
W: Some index files failed to download. They have been ignored, or old ones used instead.
Reading package lists... Done
Building dependency tree
Reading state information... Done
Calculating upgrade... Done
The following package was automatically installed and is no longer required:
  libbluray2
Use 'apt autoremove' to remove it.
The following NEW packages will be installed:
  libbluray1 libicu64
The following packages will be upgraded:
  libavformat57 libxml2
2 upgraded, 2 newly installed, 0 to remove and 0 not upgraded.
Need to get 9,723 kB of archives.
After this operation, 30.7 MB of additional disk space will be used.
Do you want to continue? [Y/n] y
Get:1 http://mirror.sjc02.svwh.net/raspbian/raspbian stretch/main armhf libavformat57 armhf 7:3.2.15-0+deb9u2 [882 k                                                                                                                                                                                                                                                                                                                       B]
Get:2 https://archive.raspberrypi.org/debian stretch/main armhf libbluray1 armhf 1:0.9.3-4 [124 kB]
Get:3 https://packages.sury.org/php stretch/main armhf libicu64 armhf 64.1-0.1+0~20190410090943.5+stretch~1.gbp38f69                                                                                                                                                                                                                                                                                                                       4 [8,102 kB]
Err:4 https://packages.sury.org/php stretch/main armhf libxml2 armhf 2.9.9+dfsg-1+0~20190914.4+debian9~1.gbp3b6674
  404  Not Found
Fetched 9,108 kB in 27s (327 kB/s)
E: Failed to fetch https://packages.sury.org/php/pool/main/libx/libxml2/libxml2_2.9.9+dfsg-1+0~20190914.4+debian9~1.gbp3b6674_armhf.deb  404  Not Found
E: Unable to fetch some archives, maybe run apt-get update or try with --fix-missing?
root@DietPi:~#

BelakScientific, 2021-06-02:
Sorry, I kept pasting too much text. Here it is again:

root@DietPi:~# apt update && apt upgrade
Hit:1 http://raspbian.raspberrypi.org/raspbian stretch InRelease
Hit:3 https://archive.raspberrypi.org/debian stretch InRelease
Err:2 https://packages.sury.org/php stretch InRelease
  The following signatures were invalid: EXPKEYSIG B188E2B695BD4743 DEB.SURY.ORG Automatic Signing Key <deb@sury.org                                                                                                                                                                                                                                                                                                                       >
Reading package lists... Done
Building dependency tree
Reading state information... Done
2 packages can be upgraded. Run 'apt list --upgradable' to see them.
W: An error occurred during the signature verification. The repository is not updated and the previous index files w                                                                                                                                                                                                                                                                                                                       ill be used. GPG error: https://packages.sury.org/php stretch InRelease: The following signatures were invalid: EXPK                                                                                                                                                                                                                                                                                                                       EYSIG B188E2B695BD4743 DEB.SURY.ORG Automatic Signing Key <deb@sury.org>
W: Failed to fetch https://packages.sury.org/php/dists/stretch/InRelease  The following signatures were invalid: EXP                                                                                                                                                                                                                                                                                                                       KEYSIG B188E2B695BD4743 DEB.SURY.ORG Automatic Signing Key <deb@sury.org>
W: Some index files failed to download. They have been ignored, or old ones used instead.
Reading package lists... Done
Building dependency tree
Reading state information... Done
Calculating upgrade... Done
The following package was automatically installed and is no longer required:
  libbluray2
Use 'apt autoremove' to remove it.
The following NEW packages will be installed:
  libbluray1 libicu64
The following packages will be upgraded:
  libavformat57 libxml2
2 upgraded, 2 newly installed, 0 to remove and 0 not upgraded.
Need to get 9,723 kB of archives.
After this operation, 30.7 MB of additional disk space will be used.
Do you want to continue? [Y/n] y
Get:1 http://mirror.sjc02.svwh.net/raspbian/raspbian stretch/main armhf libavformat57 armhf 7:3.2.15-0+deb9u2 [882 k                                                                                                                                                                                                                                                                                                                       B]
Get:2 https://archive.raspberrypi.org/debian stretch/main armhf libbluray1 armhf 1:0.9.3-4 [124 kB]
Get:3 https://packages.sury.org/php stretch/main armhf libicu64 armhf 64.1-0.1+0~20190410090943.5+stretch~1.gbp38f69                                                                                                                                                                                                                                                                                                                       4 [8,102 kB]
Err:4 https://packages.sury.org/php stretch/main armhf libxml2 armhf 2.9.9+dfsg-1+0~20190914.4+debian9~1.gbp3b6674
  404  Not Found
Fetched 9,108 kB in 27s (327 kB/s)
E: Failed to fetch https://packages.sury.org/php/pool/main/libx/libxml2/libxml2_2.9.9+dfsg-1+0~20190914.4+debian9~1.gbp3b6674_armhf.deb  404  Not Found
E: Unable to fetch some archives, maybe run apt-get update or try with --fix-missing?
root@DietPi:~#

ravenclaw900, 2021-06-02:
It looks like you might have 2 different problems, but I'm not sure.  Let's start by fixing the first one, the expired repo key.  Could you please run:
```
apt-key del '95BD4743' # delete the old key
curl -sSfL 'https://packages.sury.org/php/apt.gpg' -o /etc/apt/trusted.gpg.d/dietpi-php.gpg # download the new key
/boot/dietpi/func/dietpi-set_software apt-cache clean # clean the apt cache
```
And then post the output of `apt update && apt upgrade` again.

BelakScientific, 2021-06-02:
After running the final command again I got a lot of text (which I did not copy and now cannot access) and then this at the end

[image]

MichaIng, 2021-06-02:
Select "Keep the local version".

BelakScientific, 2021-06-02:
Here is the result--please confirm it's fixed

DietPi-Set_software
-----------------------------------------------------
 Mode: apt-cache (clean)

[  OK  ] apt-cache clean | Completed
root@DietPi:~# apt update && apt upgrade
Get:1 http://raspbian.raspberrypi.org/raspbian stretch InRelease [15.0 kB]
Get:2 https://packages.sury.org/php stretch InRelease [6,824 B]
Get:3 https://archive.raspberrypi.org/debian stretch InRelease [25.3 kB]
Get:4 http://raspbian.raspberrypi.org/raspbian stretch/main armhf Packages [11.7 MB]
Get:5 https://archive.raspberrypi.org/debian stretch/main armhf Packages [192 kB]
Get:6 https://packages.sury.org/php stretch/main armhf Packages [320 kB]
Get:7 https://archive.raspberrypi.org/debian stretch/ui armhf Packages [44.6 kB]
Get:8 http://raspbian.raspberrypi.org/raspbian stretch/contrib armhf Packages [56.9 kB]
Get:9 http://raspbian.raspberrypi.org/raspbian stretch/rpi armhf Packages [1,360 B]
Get:10 http://raspbian.raspberrypi.org/raspbian stretch/non-free armhf Packages [98.9 kB]
Fetched 12.4 MB in 12s (995 kB/s)
Reading package lists... Done
Building dependency tree
Reading state information... Done
27 packages can be upgraded. Run 'apt list --upgradable' to see them.
Reading package lists... Done
Building dependency tree
Reading state information... Done
Calculating upgrade... Done
The following package was automatically installed and is no longer required:
  libbluray2
Use 'apt autoremove' to remove it.
The following NEW packages will be installed:
  libbluray1 libicu65 libonig4 php8.0-apcu php8.0-cli php8.0-common php8.0-curl php8.0-fpm php8.0-gd php8.0-mbstring php8.0-mysql php8.0-opcache php8.0-readline php8.0-xml php8.0-zip
The following packages will be upgraded:
  libavformat57 libidn2-0 libpcre2-8-0 libpcre3 libxml2 libzip4 php-apcu php-common php-curl php-fpm php-gd php-mbstring php-mysql php-xml php-zip php7.3-cli php7.3-common php7.3-curl php7.3-fpm php7.3-gd php7.3-json php7.3-mbstring php7.3-mysql php7.3-opcache php7.3-readline php7.3-xml php7.3-zip
27 upgraded, 15 newly installed, 0 to remove and 0 not upgraded.
Need to get 18.5 MB of archives.
After this operation, 46.9 MB of additional disk space will be used.
Do you want to continue? [Y/n] y
Get:1 http://mirror.sjc02.svwh.net/raspbian/raspbian stretch/main armhf libavformat57 armhf 7:3.2.15-0+deb9u2 [882 kB]
Get:2 http://mirror.sjc02.svwh.net/raspbian/raspbian stretch/main armhf libonig4 armhf 6.1.3-2+deb9u2 [128 kB]
Get:3 https://archive.raspberrypi.org/debian stretch/main armhf libbluray1 armhf 1:0.9.3-4 [124 kB]
Get:4 https://packages.sury.org/php stretch/main armhf php-common all 2:82+0~20210313.35+debian9~1.gbpa8195f [17.6 kB]
Get:5 https://packages.sury.org/php stretch/main armhf php8.0-common armhf 8.0.5-2+0~20210503.19+debian9~1.gbp3e4ad0 [601 kB]
Get:6 https://packages.sury.org/php stretch/main armhf php-apcu armhf 5.1.20+4.0.11-1+0~20210304.24+debian9~1.gbp1c3d4d [6,814 B]
Get:7 https://packages.sury.org/php stretch/main armhf php8.0-apcu armhf 5.1.20+4.0.11-1+0~20210304.24+debian9~1.gbp1c3d4d [40.1 kB]
Get:8 https://packages.sury.org/php stretch/main armhf libpcre3 armhf 2:8.44-2+0~20210301.9+debian9~1.gbpa278ad [324 kB]
Get:9 https://packages.sury.org/php stretch/main armhf libicu65 armhf 65.1-1+0~20200223.8+debian9~1.gbp519cf3 [8,208 kB]
Get:10 https://packages.sury.org/php stretch/main armhf libxml2 armhf 2.9.9+dfsg-1+0~20200226.5+debian9~1.gbp3b6674 [615 kB]
Get:11 https://packages.sury.org/php stretch/main armhf libidn2-0 armhf 2.2.0-2+0~20200302.4+debian9~1.gbpf85c2e [123 kB]
Get:12 https://packages.sury.org/php stretch/main armhf libpcre2-8-0 armhf 10.36-2+0~20210212.6+debian9~1.gbp6138a4 [216 kB]
Get:13 https://packages.sury.org/php stretch/main armhf libzip4 armhf 1.7.3-1+0~20210114.10+debian9~1.gbp4c125d [49.6 kB]
Get:14 https://packages.sury.org/php stretch/main armhf php8.0-curl armhf 8.0.5-2+0~20210503.19+debian9~1.gbp3e4ad0 [28.7 kB]
Get:15 https://packages.sury.org/php stretch/main armhf php-curl all 2:8.0+82+0~20210313.35+debian9~1.gbpa8195f [6,946 B]
Get:16 https://packages.sury.org/php stretch/main armhf php8.0-opcache armhf 8.0.5-2+0~20210503.19+debian9~1.gbp3e4ad0 [178 kB]
Get:17 https://packages.sury.org/php stretch/main armhf php8.0-readline armhf 8.0.5-2+0~20210503.19+debian9~1.gbp3e4ad0 [10.8 kB]
Get:18 https://packages.sury.org/php stretch/main armhf php8.0-cli armhf 8.0.5-2+0~20210503.19+debian9~1.gbp3e4ad0 [1,221 kB]
Get:19 https://packages.sury.org/php stretch/main armhf php8.0-fpm armhf 8.0.5-2+0~20210503.19+debian9~1.gbp3e4ad0 [1,230 kB]
Get:20 https://packages.sury.org/php stretch/main armhf php-fpm all 2:8.0+82+0~20210313.35+debian9~1.gbpa8195f [7,004 B]
Get:21 https://packages.sury.org/php stretch/main armhf php8.0-gd armhf 8.0.5-2+0~20210503.19+debian9~1.gbp3e4ad0 [22.7 kB]
Get:22 https://packages.sury.org/php stretch/main armhf php-gd all 2:8.0+82+0~20210313.35+debian9~1.gbpa8195f [6,942 B]
Get:23 https://packages.sury.org/php stretch/main armhf php8.0-mbstring armhf 8.0.5-2+0~20210503.19+debian9~1.gbp3e4ad0 [374 kB]
Get:24 https://packages.sury.org/php stretch/main armhf php-mbstring all 2:8.0+82+0~20210313.35+debian9~1.gbpa8195f [6,956 B]
Get:25 https://packages.sury.org/php stretch/main armhf php8.0-mysql armhf 8.0.5-2+0~20210503.19+debian9~1.gbp3e4ad0 [100 kB]
Get:26 https://packages.sury.org/php stretch/main armhf php-mysql all 2:8.0+82+0~20210313.35+debian9~1.gbpa8195f [6,948 B]
[... cut ...]
Setting up php-fpm (2:8.0+82+0~20210313.35+debian9~1.gbpa8195f) ...
Processing triggers for libc-bin (2.24-11+deb9u4) ...
Processing triggers for php8.0-cli (8.0.5-2+0~20210503.19+debian9~1.gbp3e4ad0) ...
Processing triggers for php7.3-cli (7.3.28-1+0~20210503.84+debian9~1.gbp6819da) ...
Processing triggers for systemd (232-25+deb9u12) ...
Processing triggers for php7.3-fpm (7.3.28-1+0~20210503.84+debian9~1.gbp6819da) ...
Processing triggers for php8.0-fpm (8.0.5-2+0~20210503.19+debian9~1.gbp3e4ad0) ...
root@DietPi:~#

BelakScientific, 2021-06-02:
Are there any other steps needed?

MichaIng, 2021-06-02:
You have now two different PHP versions installed. Do you remember when you or another installer installed these PHP meta packages? **php-apcu php-curl php-fpm php-gd php-mbstring php-mysql php-xml php-zip**

As you are on Stretch with Ondrej's PHP repo installed, those meta packages will pull the current PHP version whenever a new one is released, which is not intended as long as you know that all your software does support it and how to configure it to use the new PHP version and how to remove the old one. If you didn't intended to install PHP8, please do the following:
```
apt purge --autoremove php-apcu php-curl php-fpm php-gd php-mbstring php-mysql php-xml php-zip
apt install php7.3-apcu
```

And then Squeezelite failed to install previously. Retry it manually:
```
cd /tmp
curl -sSfLO https://dietpi.com/downloads/binaries/stretch/squeezelite_armv7l.deb
apt install ./squeezelite_armv7l.deb
rm squeezelite_armv7l.deb
```

BelakScientific, 2021-06-02:
These came pre-installed with my Allo USBridge Signature so I can't tell you when that happened exactly but it was about a year ago. When I tried to uninstall PHP8 there were a few errors toward the end due to directories not being empty and so not being removed.

login as: root
root@192.168.1.7's password:

The programs included with the Debian GNU/Linux system are free software;
the exact distribution terms for each program are described in the
individual files in /usr/share/doc/*/copyright.

Debian GNU/Linux comes with ABSOLUTELY NO WARRANTY, to the extent
permitted by applicable law.
 ─────────────────────────────────────────────────────
 DietPi v6.26.3 : Update available
 ─────────────────────────────────────────────────────
 - Device model : RPi CM 3+ (armv7l)
 - CPU temp : 50'C : 122'F (Running warm, but safe)
 - LAN IP : 192.168.1.7 (eth0)
 - MOTD : DietPi v7.2 has been released, please update now.
 ─────────────────────────────────────────────────────

 DietPi Team     : MichaIng (lead), Daniel Knight (founder)
 Image           : DietPi Core Team (pre-image: Raspbian Lite)
 Web             : https://DietPi.com | https://twitter.com/dietpi_
 Patreon Legends : PINE64 community | oct8l
 Donate          : https://DietPi.com/#donate
 DietPi Hosting  : Powered by https://MyVirtualServer.com

 dietpi-update   : Run now to update DietPi from v6.26.3 to v6.35.0.

 dietpi-launcher : All the DietPi programs in one place.
 dietpi-config   : Feature rich configuration tool for your device.
 dietpi-software : Select optimized software for installation.
 htop            : Resource monitor.
 cpu             : Shows CPU information and stats.

root@DietPi:~# uname -a
Linux DietPi 4.19.73-v7+ #1266 SMP Tue Sep 17 20:52:18 BST 2019 armv7l GNU/Linux
root@DietPi:~# Linux DietPi 4.19.73-v7+ #1266 SMP Tue Sep 17 20:52:18 BST 2019 a                                    rmv7l GNU/Linux
-bash: Linux: command not found
root@DietPi:~# root@DietPi:~#
-bash: root@DietPi:~#: command not found
root@DietPi:~# ^C
root@DietPi:~# dietpi-update

 DietPi-Update
─────────────────────────────────────────────────────
 Mode: Checking for DietPi updates

[ INFO ] DietPi-Update | Checking mirror: https://raw.githubusercontent.com/Mich                                    aIng/DietPi/master/dietpi/server_version-6
[  OK  ] DietPi-Update | Using update server: https://raw.githubusercontent.com/                                    MichaIng/DietPi/master/dietpi/server_version-6
[  OK  ] DietPi-Update | Update available:
[ INFO ] DietPi-Update | Current version : v6.26.3
[ INFO ] DietPi-Update | Latest version  : v6.35.0
[  OK  ] DietPi-Update | DietPi-Userdata validation: /mnt/dietpi_userdata
[  OK  ] DietPi-Update | Free space check: path=/ | available=12399 MiB | requir                                    ed=100 MiB
[ INFO ] DietPi-Update | Lines required for Whiptail box and its contents (y=21,                                     z=0), exceeds screen dimensions (y=20).

 DietPi-Update
─────────────────────────────────────────────────────
 Mode: Applying DietPi update

[ SUB1 ] DietPi-Services > stop
[  OK  ] DietPi-Services | stop : cron
[  OK  ] DietPi-Services | stop : netdata
[  OK  ] DietPi-Services | stop : logitechmediaserver
[  OK  ] DietPi-Services | stop : roonbridge
[  OK  ] DietPi-Services | stop : networkaudiod
[  OK  ] DietPi-Services | stop : gmrender
[  OK  ] DietPi-Services | stop : squeezelite
[  OK  ] DietPi-Services | stop : shairport-sync
[... 1156 lines cut ...]
Removing php8.0-curl (8.0.5-2+0~20210503.19+debian9~1.gbp3e4ad0) ...
Removing php8.0-gd (8.0.5-2+0~20210503.19+debian9~1.gbp3e4ad0) ...
Removing php8.0-mysql (8.0.5-2+0~20210503.19+debian9~1.gbp3e4ad0) ...
Removing php8.0-opcache (8.0.5-2+0~20210503.19+debian9~1.gbp3e4ad0) ...
Removing php8.0-readline (8.0.5-2+0~20210503.19+debian9~1.gbp3e4ad0) ...
Removing php8.0-common (8.0.5-2+0~20210503.19+debian9~1.gbp3e4ad0) ...
Processing triggers for libc-bin (2.24-11+deb9u4) ...
(Reading database ... 40841 files and directories currently installed.)
Purging configuration files for php8.0-curl (8.0.5-2+0~20210503.19+debian9~1.gbp3e4ad0) ...
Purging configuration files for php8.0-fpm (8.0.5-2+0~20210503.19+debian9~1.gbp3e4ad0) ...
Purging configuration files for php-apcu (5.1.20+4.0.11-1+0~20210304.24+debian9~1.gbp1c3d4d) ...
Purging configuration files for php8.0-opcache (8.0.5-2+0~20210503.19+debian9~1.gbp3e4ad0) ...
Purging configuration files for php8.0-apcu (5.1.20+4.0.11-1+0~20210304.24+debian9~1.gbp1c3d4d) ...
dpkg: warning: while removing php8.0-apcu, directory '/etc/php/8.0/mods-available' not empty so not removed
Purging configuration files for php8.0-xml (8.0.5-2+0~20210503.19+debian9~1.gbp3e4ad0) ...
Purging configuration files for php8.0-mysql (8.0.5-2+0~20210503.19+debian9~1.gbp3e4ad0) ...
Purging configuration files for php8.0-mbstring (8.0.5-2+0~20210503.19+debian9~1.gbp3e4ad0) ...
Purging configuration files for php8.0-zip (8.0.5-2+0~20210503.19+debian9~1.gbp3e4ad0) ...
Purging configuration files for php8.0-cli (8.0.5-2+0~20210503.19+debian9~1.gbp3e4ad0) ...
dpkg: warning: while removing php8.0-cli, directory '/etc/php/8.0' not empty so not removed
Purging configuration files for php8.0-gd (8.0.5-2+0~20210503.19+debian9~1.gbp3e4ad0) ...
Purging configuration files for php8.0-common (8.0.5-2+0~20210503.19+debian9~1.gbp3e4ad0) ...
Purging configuration files for php8.0-readline (8.0.5-2+0~20210503.19+debian9~1.gbp3e4ad0) ...
Processing triggers for systemd (232-25+deb9u12) ...
root@DietPi:~#

BelakScientific, 2021-06-02:
These came pre-installed with my Allo USBridge Signature so I can't tell you when that happened exactly but it was about a year ago. When I tried to uninstall PHP8 there were a few errors toward the end due to directories not being empty and so not being removed

root@DietPi:~# apt purge --autoremove php-apcu php-curl php-fpm php-gd php-mbstring php-mysql php-xml php-zip
Reading package lists... Done
Building dependency tree
Reading state information... Done
The following packages will be REMOVED:
  libbluray2* libonig4* php-apcu* php-curl* php-fpm* php-gd* php-mbstring* php-mysql* php-xml* php-zip*
  php8.0-apcu* php8.0-cli* php8.0-common* php8.0-curl* php8.0-fpm* php8.0-gd* php8.0-mbstring* php8.0-mysql*
  php8.0-opcache* php8.0-readline* php8.0-xml* php8.0-zip*
0 upgraded, 0 newly installed, 22 to remove and 0 not upgraded.
After this operation, 16.8 MB disk space will be freed.
Do you want to continue? [Y/n] Y
(Reading database ... 41071 files and directories currently installed.)
Removing libbluray2:armhf (1:1.0.2-1) ...
Removing php-mbstring (2:8.0+82+0~20210313.35+debian9~1.gbpa8195f) ...
Removing php8.0-mbstring (8.0.5-2+0~20210503.19+debian9~1.gbp3e4ad0) ...
Removing libonig4:armhf (6.1.3-2+deb9u2) ...
Removing php-apcu (5.1.20+4.0.11-1+0~20210304.24+debian9~1.gbp1c3d4d) ...
Removing php-curl (2:8.0+82+0~20210313.35+debian9~1.gbpa8195f) ...
Removing php-fpm (2:8.0+82+0~20210313.35+debian9~1.gbpa8195f) ...
Removing php-gd (2:8.0+82+0~20210313.35+debian9~1.gbpa8195f) ...
Removing php-mysql (2:8.0+82+0~20210313.35+debian9~1.gbpa8195f) ...
Removing php-xml (2:8.0+82+0~20210313.35+debian9~1.gbpa8195f) ...
Removing php-zip (2:8.0+82+0~20210313.35+debian9~1.gbpa8195f) ...
Removing php8.0-apcu (5.1.20+4.0.11-1+0~20210304.24+debian9~1.gbp1c3d4d) ...
Removing php8.0-fpm (8.0.5-2+0~20210503.19+debian9~1.gbp3e4ad0) ...
Removing php8.0-cli (8.0.5-2+0~20210503.19+debian9~1.gbp3e4ad0) ...
update-alternatives: using /usr/bin/php7.3 to provide /usr/bin/php (php) in auto mode
update-alternatives: using /usr/bin/phar7.3 to provide /usr/bin/phar (phar) in auto mode
update-alternatives: using /usr/bin/phar.phar7.3 to provide /usr/bin/phar.phar (phar.phar) in auto mode
Removing php8.0-zip (8.0.5-2+0~20210503.19+debian9~1.gbp3e4ad0) ...
Removing php8.0-xml (8.0.5-2+0~20210503.19+debian9~1.gbp3e4ad0) ...
Removing php8.0-curl (8.0.5-2+0~20210503.19+debian9~1.gbp3e4ad0) ...
Removing php8.0-gd (8.0.5-2+0~20210503.19+debian9~1.gbp3e4ad0) ...
Removing php8.0-mysql (8.0.5-2+0~20210503.19+debian9~1.gbp3e4ad0) ...
Removing php8.0-opcache (8.0.5-2+0~20210503.19+debian9~1.gbp3e4ad0) ...
Removing php8.0-readline (8.0.5-2+0~20210503.19+debian9~1.gbp3e4ad0) ...
Removing php8.0-common (8.0.5-2+0~20210503.19+debian9~1.gbp3e4ad0) ...
Processing triggers for libc-bin (2.24-11+deb9u4) ...
(Reading database ... 40841 files and directories currently installed.)
Purging configuration files for php8.0-curl (8.0.5-2+0~20210503.19+debian9~1.gbp3e4ad0) ...
Purging configuration files for php8.0-fpm (8.0.5-2+0~20210503.19+debian9~1.gbp3e4ad0) ...
Purging configuration files for php-apcu (5.1.20+4.0.11-1+0~20210304.24+debian9~1.gbp1c3d4d) ...
Purging configuration files for php8.0-opcache (8.0.5-2+0~20210503.19+debian9~1.gbp3e4ad0) ...
Purging configuration files for php8.0-apcu (5.1.20+4.0.11-1+0~20210304.24+debian9~1.gbp1c3d4d) ...
dpkg: warning: while removing php8.0-apcu, directory '/etc/php/8.0/mods-available' not empty so not removed
Purging configuration files for php8.0-xml (8.0.5-2+0~20210503.19+debian9~1.gbp3e4ad0) ...
Purging configuration files for php8.0-mysql (8.0.5-2+0~20210503.19+debian9~1.gbp3e4ad0) ...
Purging configuration files for php8.0-mbstring (8.0.5-2+0~20210503.19+debian9~1.gbp3e4ad0) ...
Purging configuration files for php8.0-zip (8.0.5-2+0~20210503.19+debian9~1.gbp3e4ad0) ...
Purging configuration files for php8.0-cli (8.0.5-2+0~20210503.19+debian9~1.gbp3e4ad0) ...
dpkg: warning: while removing php8.0-cli, directory '/etc/php/8.0' not empty so not removed
Purging configuration files for php8.0-gd (8.0.5-2+0~20210503.19+debian9~1.gbp3e4ad0) ...
Purging configuration files for php8.0-common (8.0.5-2+0~20210503.19+debian9~1.gbp3e4ad0) ...
Purging configuration files for php8.0-readline (8.0.5-2+0~20210503.19+debian9~1.gbp3e4ad0) ...
Processing triggers for systemd (232-25+deb9u12) ...
root@DietPi:~#

BelakScientific, 2021-06-02:
Here's what happened when I tried to unistall Squeezelite

root@DietPi:~# cd /tmp
root@DietPi:/tmp# curl -sSfLO https://dietpi.com/downloads/binaries/stretch/squeezelite_armv7l.deb
root@DietPi:/tmp# apt install ./squeezelite_armv7l.deb
Reading package lists... Done
Building dependency tree
Reading state information... Done
Note, selecting 'squeezelite' instead of './squeezelite_armv7l.deb'
Some packages could not be installed. This may mean that you have
requested an impossible situation or if you are using the unstable
distribution that some required packages have not yet been created
or been moved out of Incoming.
The following information may help to resolve the situation:

The following packages have unmet dependencies:
 squeezelite : Depends: libmpg123-0 (>= 1.23.8-1+b1) but 1.23.8-1 is to be installed
E: Unable to correct problems, you have held broken packages.
root@DietPi:/tmp# rm squeezelite_armv7l.deb
root@DietPi:/tmp#

MichaIng, 2021-06-02:
The warnings about directories that are not empty and hence not removed on package uninstalls are normal, especially when you have multiple versions of the same program installed. Nothing to worry about.

The issue with Squeezelite is due to a mismatch of a library version between Raspbian Stretch and Debian Stretch. I'll repack that package tomorrow, then you're able to install it.

MichaIng, 2021-06-03:
Okay, repacket Squeezelite is ready:
```
cd /tmp
curl -sSfLO https://dietpi.com/downloads/binaries/stretch/squeezelite_armv7l.deb
apt install ./squeezelite_armv7l.deb
rm squeezelite_armv7l.deb
```

BelakScientific, 2021-06-03:
Thanks again, here is my output. Note that it's now prompting me on login to update from v6.34.0 to v7.2.3. I'll ignore that unless this requires action (?)
login as: root
root@192.168.1.7's password:

root@DietPi:~# cd /tmp
root@DietPi:/tmp# curl -sSfLO https://dietpi.com/downloads/binaries/stretch/sque                                                                                                             ezelite_armv7l.deb
root@DietPi:/tmp# apt install ./squeezelite_armv7l.deb
Reading package lists... Done
Building dependency tree
Reading state information... Done
Note, selecting 'squeezelite' instead of './squeezelite_armv7l.deb'
The following packages will be upgraded:
  squeezelite
1 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.
Need to get 0 B/70.5 kB of archives.
After this operation, 12.3 kB disk space will be freed.
Get:1 /tmp/squeezelite_armv7l.deb squeezelite armhf 1.9.9-1386-dietpi1 [70.5 kB]
(Reading database ... 40825 files and directories currently installed.)
Preparing to unpack /tmp/squeezelite_armv7l.deb ...
Unpacking squeezelite (1.9.9-1386-dietpi1) over (1.8-4) ...
Setting up squeezelite (1.9.9-1386-dietpi1) ...
Creating Squeezelite service user ...
Configuring Squeezelite systemd service ...
Created symlink /etc/systemd/system/multi-user.target.wants/squeezelite.service → /etc/systemd/system/squeezelite.service.
root@DietPi:/tmp#

Joulinar, 2021-06-03:
Looks fine. No errors showing up

BelakScientific, 2021-06-03:
Thanks for your help!

MichaIng, 2021-06-03:
> Note that it's now prompting me on login to update from v6.34.0 to v7.2.3. I'll ignore that unless this requires action (?)

Since the update aborted at that point because of the APT errors, this is correct. `dietpi-update 1` should now run through without issues.

BelakScientific, 2021-06-03:
No errors,. Thanks again

MichaIng, 2021-06-03:
Okay, I mark the issue as closed then. Feel free to reopen if required.

closed 2021-06-03 by MichaIng