MichaIng/DietPi #4459: Plex program installation error

bsoener, 2021-06-02:
Hi. I want to install plex. unfortunately I get this error message. Update and upgrade have already been done.

#### Additional logs:
```
E: Failed to fetch https://downloads.plex.tv/repo/deb/pool/plexmediaserver_1.22.3.4523-d0ce30438_armhf.deb  404  Not Found [IP: 104.18.156.41 443]
E: Unable to fetch some archives, maybe run apt-get update or try with --fix-missing?
```

ravenclaw900, 2021-06-02:
Hello,
This is a known error, and we are waiting from a fix from Plex.  See https://dietpi.com/phpbb/viewtopic.php?f=11&t=9095.

Joulinar, 2021-06-03:
adding the Plex forum post https://forums.plex.tv/t/404-when-fetching-deb-from-repository/719611/9

MichaIng, 2021-06-03:
They seem build two sets of ARMv7 packages now, one with hardware + software floating point support. But APT only knows one armhf, so not sure how this shall be rolled out through the currently one repository 😄. Actually ARMv6 hardware floating point would make much more sense than ARMv7 software floating point builds, as then Raspberry Pi 1 and Zero would be supported. But maybe there exists another set of ARM devices that are outside of my view 🤔.

ravenclaw900, 2021-06-03:
Could the one with software support be `armel`?

Joulinar, 2021-06-03:
whatever they did, they broke the `apt` repository. Via `apt` you will get `Version: 1.22.3`. But on the download page you already have `1.23.2`

MichaIng, 2021-06-03:
> Could the one with software support be `armel`?

In theory yes, but it doesn't make sense to keep ARMv7 + hardware floating point support then, as `armel` is actually ARMv6 + earlier and software floating point only. For APT/DPkg mixing soft-float and hard-float is not foreseen, while mixing ARM versions theoretically is, which is why many repos ship armhf for ARMv6 + ARMv7, like Raspbian, to support all Raspberry Pi models. We'll see.

They said the HF+SF builds are without NEON, respectively the others are with NEON. It's a command set introduced with ARMv7, so I wonder whether without it, binaries would run on Raspberry Pi 1/Zero, or if there is another left difference between v6 and v7 that breaks it 🤔.

MichaIng, 2021-11-09:
Not an issue anymore, tested on ARMv6 RPi image (ARMv7 capable RPi).

closed 2021-11-09 by MichaIng