MichaIng/DietPi #4460: Upgrade from v7.1.2 to v7.2.3 fails on ROCK64

robinrehbein, 2021-06-02:
### Creating a bug report/issue

#### Required Information
- DietPi version | `cat /boot/dietpi/.version`
- Distro version | `echo $G_DISTRO_NAME` or `cat /etc/debian_version`
- Kernel version | `uname -a`
- SBC model | `echo $G_HW_MODEL_NAME` or (EG: RPi3)
- Power supply used | (EG: 5V 1A RAVpower)
- SDcard used | (EG: SanDisk ultra)

#### Additional Information (if applicable)
- Software title | (EG: Nextcloud)
- Was the software title installed freshly or updated/migrated?
- Can this issue be replicated on a fresh installation of DietPi?

- Bug report ID | `echo $G_HW_UUID`

#### Steps to reproduce

1. dietpi-update
2. run update

#### Expected behaviour
- finishes update properly

#### Actual behaviour
- fails install and executes with an Error

#### Extra details
- when i now try to run apt upgrade it failes too
```
root@ROCK64:~# apt-get upgrade
Reading package lists... Done
Building dependency tree
Reading state information... Done
Calculating upgrade... Done
The following packages have been kept back:
  linux-buster-root-current-rock64
The following packages will be upgraded:
  armbian-firmware liblz4-1 libx11-6 libx11-data linux-dtb-current-rockchip64 linux-image-current-rockchip64
  linux-u-boot-rock64-current
7 upgraded, 0 newly installed, 0 to remove and 1 not upgraded.
Need to get 0 B/48.9 MB of archives.
After this operation, 207 kB of additional disk space will be used.
Do you want to continue? [Y/n] Y
debconf: delaying package configuration, since apt-utils is not installed
dpkg: unrecoverable fatal error, aborting:
 files list file for package 'libzip4:arm64' is missing final newline
E: Sub-process /usr/bin/dpkg returned an error code (2)
root@ROCK64:~#
```
#### Additional logs:
```
debconf: delaying package configuration, since apt-utils is not installed
dpkg: unrecoverable fatal error, aborting:
 files list file for package 'libzip4:arm64' is missing final newline
E: Sub-process /usr/bin/dpkg returned an error code (2)
```

ravenclaw900, 2021-06-02:
Hello,
Could you try running:
```
rm /var/lib/dpkg/info/libzip4.list
apt install libzip4 --reinstall
```
The last command will probably show a warning, it's okay to ignore it.  See if `apt upgrade` works after that.

robinrehbein, 2021-06-02:
Steps i tried to fix the error with the libzip4 package:
- sudo dpkg --configure -a
- apt-get upgrade
- sudo apt-get -f install
- sudo apt-get (auto)clean
- apt-get install --reinstall libzip4

robinrehbein, 2021-06-02:
actually i didn't remove the list. try that now

robinrehbein, 2021-06-02:
> Hello,
> Could you try running:
> 
> ```
> rm /var/lib/dpkg/info/libzip4.list
> apt install libzip4 --reinstall
> ```
> 
> The last command will probably show a warning, it's okay to ignore it. See if `apt upgrade` works after that.

unfortunately the list does not exist

robinrehbein, 2021-06-02:
> > Hello,
> > Could you try running:
> > ```
> > rm /var/lib/dpkg/info/libzip4.list
> > apt install libzip4 --reinstall
> > ```
> > 
> > 
> >     
> >       
> >     
> > 
> >       
> >     
> > 
> >     
> >   
> > The last command will probably show a warning, it's okay to ignore it. See if `apt upgrade` works after that.
> 
> unfortunately the list does not exist
> 

Well got it!
Thang You @ravenclaw900 
Had to run it with :arm64 added at the end of the package name...

```
root@ROCK64:/var/lib/dpkg/info# rm /var/lib/dpkg/info/libzip4:arm64.list
root@ROCK64:/var/lib/dpkg/info# apt install libzip4 --reinstall
Reading package lists... Done
Building dependency tree
Reading state information... Done
0 upgraded, 0 newly installed, 1 reinstalled, 0 to remove and 8 not upgraded.
Need to get 0 B/48.0 kB of archives.
After this operation, 0 B of additional disk space will be used.
debconf: delaying package configuration, since apt-utils is not installed
dpkg: warning: files list file for package 'libzip4:arm64' missing; assuming package has no files currently installed
(Reading database ... 21530 files and directories currently installed.)
Preparing to unpack .../libzip4_1.5.1-4_arm64.deb ...
Unpacking libzip4:arm64 (1.5.1-4) over (1.5.1-4) ...
Setting up libzip4:arm64 (1.5.1-4) ...
Processing triggers for libc-bin (2.28-10) ...
root@ROCK64:/var/lib/dpkg/info#
```

closed 2021-06-02 by robinrehbein