MichaIng/DietPi #4461: .NET interactive

fpetru, 2021-06-02:
### Enable .NET live editor in your browser using dotnet/interactive

#### Is your feature request related to a problem? Please describe:

- ...

#### Describe the solution you'd like:

- ...

#### Describe alternatives you've considered:

- ...

#### Additional context

- ...

#### Vote for this feature on FeatHub: https://feathub.com/MichaIng/DietPi/

MichaIng, 2021-06-02:
Just to not forget about it: First we should implement .NET as install option, before we implement an IDE for it 😉.

MichaIng, 2026-10-27:
.NET interactive has been deprecated and archived: https://github.com/dotnet/interactive/issues/4163

Jupyter supports other backends in the meantime.

closed 2026-10-27 by MichaIng