MichaIng/DietPi #4462: Roon extension fails to install

Esad-np, 2021-06-03:
#### Details:
- Date           | Thu Jun  3 19:21:11 BST 2021
- Bug report     | ba5e27a9-ab3c-44c8-b801-1590fc929440
- DietPi version | v7.2.3 (MichaIng/master)
- Image creator  | DietPi Core Team
- Pre-image      | Raspberry Pi OS Lite
- Hardware       | RPi 4 Model B (armv7l) (ID=4)
- Kernel version | Linux DietPi 5.10.17-v7l+ #1414 SMP Fri Apr 30 13:20:47 BST 2021 armv7l GNU/Linux
- Distro         | buster (ID=5,RASPBIAN=1)
- Command        | ./rem-setup.sh
- Exit code      | 1
- Software title | DietPi-Software
#### Steps to reproduce:
Selected Roon extension to be installed
#### Expected behaviour:
Roon extension installed
- ...
#### Actual behaviour:
Installation failed- ...
#### Extra details:

#### Additional logs:
```
Roon Extension Manager setup script - version 1.0.0

Checking for root privileges...		[ OK ]
Checking for Docker socket...		[ OK ]
Downloading shell script...		[FAIL]

Joulinar, 2021-06-03:
Hi,

I did a short test on my RPi3B+ without issues. Can you retry please

Esad-np, 2021-06-03:
Hi,

Retried : same issue still present    . 

Esad

Esad-np, 2021-06-03:
I managed to install it successfully  by running:
1. sudo su -
2. dietpi-software

In my previous attempts I run :
1. sudo dietpi-software

Not sure what is the difference ...

Thanks,
E

MichaIng, 2021-06-03:
Instead of `sudo su -`, you can use `sudo -s` (interactive shell) or `sudo -i` (interactive login shell), so no need to combine two commands 😉.

But yeah, `sudo dietpi-software` should just do it. I'll check if I can replicate it.

Joulinar, 2021-06-03:
yep, using user `dietpi` and installation will fail. Means as workaround, user `root` to be used

MichaIng, 2021-06-03:
It uses the `$SUDO_USER` variable. I have seen this in other cases. The idea is that a regular user uses sudo, not because we want to have software installed "for" the root user, but because some install steps require root permissions, while in this case the container shall start as originating user.

The following should work, right before calling the installer:
```
unset -v SUDO_USER
```
Since we do not use this variable in our scripts, we could also do this as part of `G_INIT`.

Joulinar, 2021-07-25:
@MichaIng 
I guess something we can close?

MichaIng, 2021-07-26:
Fixed in `dietpi-software` for now: https://github.com/MichaIng/DietPi/commit/e8aca2c21dc715fc75e5e985d8552e523728ee42
I still think we should unset it in G_INIT already as I cannot imagine a situation where this is wanted but more leading to unexpected situations with anything we do in our scripts 🤔.

closed 2021-08-20 by MichaIng