MichaIng/DietPi #4464: NanoPi R1 + DietPi 7.2.3 -> only one eth detected (eth0 = wan) instead of two

robyscar, 2021-06-04:
### Creating a bug report/issue
latest DiePi 7.2.3 from official website, recognize only 1 of 2 eth.
wifi is recognized.

latest armbian recognized both eth and can install on eMMC

#### Required Information
- DietPi version | `cat /boot/dietpi/.version`
- Distro version | `echo $G_DISTRO_NAME` or `cat /etc/debian_version`
- Kernel version | `uname -a`
- SBC model | `echo $G_HW_MODEL_NAME` or (EG: RPi3)
- Power supply used | (EG: 5V 1A RAVpower)
- SDcard used | (EG: SanDisk ultra)

#### Additional Information (if applicable)
- Software title | (EG: Nextcloud)
- Was the software title installed freshly or updated/migrated?
- Can this issue be replicated on a fresh installation of DietPi?

- Bug report ID | `echo $G_HW_UUID`

#### Steps to reproduce

1. ...
2. ...

#### Expected behaviour

- ...

#### Actual behaviour

- ...

#### Extra details

- ...

MichaIng, 2021-06-04:
Many thanks for you report.

Did you flash the new NanoPi R1 image from our website, which was just uploaded a few days ago? Because that has been verified to detect both Ethernet devices, even on NanoPi R1S-H3, and it is based on the Armbian-kernel, device tree and bootloader, so should behave the same, regarding hardware detection.

But I see now the major difference between R1 and R1S: The R1S has the second Ethernet internally attached via USB2.0 bus, hence the single USB port only, while R1 has it as 100 MiB, via PCI bus attached, I guess. Can you paste the output of:
```
ip l
dmesg
```

robyscar, 2021-06-07:
Ciao MichaIng,

yes, of course i did: just official and downloaded 30 minutes before this report
the SBC is the R1 as i have told when opened the Image request.

i will try another run and upload the info you need.

Roby

robyscar, 2021-06-07:
dear @MichaIng, good evening

i have uploaded in a separate Repo (i don't want to mess up things in other's houses) some info you should find useful for the debug

repo is here : 

https://github.com/robyscar/DietPi-NanoPi-R1

files are here
1. [](https://github.com/robyscar/DietPi-NanoPi-R1/blob/main/debug-logs/20210607%20BLKID%20%2B%20MOUNT.txt)

2. [](https://github.com/robyscar/DietPi-NanoPi-R1/blob/main/debug-logs/20210607%20IP%20L%20%2B%20DMESG%20after%201st%20round%20of%20automatic%20setup.txt)

3. [](https://github.com/robyscar/DietPi-NanoPi-R1/blob/main/debug-logs/20210607%20LSUSB%20%2B%20LSUSB%20-V.txt)

- today WLAN worked 
(i have already read around that this SBC is quite floppy, and usually give problems on the TTYS0 and TTYS1 (you disconnect power, set apart for some hours... then magic happens and work again)

- as you can see, kernel recognize the eth1, but is missing the part of the DietPi script to set it up.
- probably is missing all the part related to the routing/iptables/docker

p.s.: i have tried to dig inside your magic scripts ..... but i got lost after few parsecs of the script (+/- at row 2.365E62) :)

MichaIng, 2021-06-07:
Let me see the logs:
- eMMC is detected.
- All network adapters are detected as such. Interesting is that the second Ethernet is as well internally attached via USB. I though that would be different on R1 vs R1S-H3 and that this would be the reason that R1S-H3 has a single USB port only while R1 has two.
- Okay so the serial consoles now work? Boot messages and should be done to UART1 (ttyS1, most likely) as well as a login console. The debug UART0 (ttyS0, most likely) should be enabled in a hardcoded way (I do not see it in the boot configs). As well by default one USB port is used as gadget mode serial console `ttyGS0`, with a login console active on it. That USB port cannot be used in other way then, I guess. That is default we might want to change (away from how Armbian ships it). Not sure if `systemctl disable --now serial-getty@ttyGS0` alone solves this, as I cannot find a boot setting to control it device-wise.

Btw, both Ethernet devices can generally used the same way. And, if you used the declared LAN port (`eth1`), I guess I know why it is not auto-configured:
- On first boot, it basically enables the first Ethernet adapter `eth0` as none has been configured yet (that is what the firstboot setup script aims to do).
- Best would be to configure the first Ethernet adapter WHICH has a cable connected, but as long as the network has not been configured (e.g. `ifup eth1`), it is impossible to detect this. `ethtool eth1`, `/sys/class/net/eth1/carrier`, `/sys/class/net/eth0/operstate` and `ip a` will all produce the exact same output, with connected and disconnected cable. This basically means that Linux is powering down a de-configured network interface completely and then no signal or such can be send/received to detect a physical connection state.

Since we just recently started to ship images for dual-Ethernet SBCs, this has never really been an issue. Not sure how other distros handle it. Naked Debian does not configure the network at all, I guess high level tools like NetworkManager will then loop through detected network interfaces and bring them all up to detect connection state.

What we could do in the firstboot script, before detecting used network interfaces, to bring up all of them: https://github.com/MichaIng/DietPi/blob/dev/rootfs/var/lib/dietpi/services/dietpi-firstboot.bash#L216
```
for i in /sys/class/net/eth*
do
	[[ -e $i ]] && ip l set up "${i#*net/}"
done
```
Then `ip -br a` will show `UP` for all physically connected adapters and even an IPv6 address, when enabled, due to SLAAC auto-configuration. To prefer physically connected interfaces as well when no IPv6 is enabled, hence no IPv6 address present, a tiny change to the network detection script needs to be done as well.

As workaround, for now, you can initially connect to the WAN port, then, after initial setup, change in `/etc/network/interfaces` `eth0` to `eth1`, to have the LAN port brought up at boot. In case of DHCP, this means a new lease due to change MAC address.

robyscar, 2021-06-14:
Hi dear @MichaIng,
sorry for taking long time to reply and try your solution.

1. tty part: let's skipy it for now. is a minor problem compared to the rest
2. the FOR loop, worked of course, and all the NIC are pulled UP
3. fast NANO of the /etc/network/interfaces, is ok, but this SBC seems to have another problem embedded

it simply... "explodes" with all the 3NIC (2 eth + 1 wlan) pulled up. Is a problem that also IPFIRE has.
i am debugging in parallel both DIETPI and IPFIRE, and with the latter, the system crash (BADLY! with FS corruption) after setting up the WLAN and.... HOSTAPD

I will dig into the problem, trying to go step by step, but i have 3 problems:
a) logging the crash of the kernel and the loop can be a problem (both skillset and hw instability)
b) also from a fresh start, the DIETPI first run script is quite invasive and could be difficult to intercept it and/or stop it, or set it in debug mode (again skillset)
c) the access to UBOOT and the EMMC is quite messy on this SBC. Now i have ARMBIAN installed on it, and before booting from the SD card, from the SerialTTY you can see that ARMIAN try to boot once or twice, then boot DIETPI (or whatever)

one NTH (nice to have) feature on DietPi, could be also the management and unpdate/upgrade of UBOOT, as also the DiskWizard for the EMMC.

Il'' keep you posted.

MichaIng, 2021-06-15:
> it simply... "explodes" with all the 3NIC

What do you mean by this? It cashes/does not continue?

For setting up hostapd on first boot, best is to keep WiFi disabled in dietpi.txt, so it won't be configured as primary interface. Selecting WiFi Hotspot will then enable WiFi. But this does not work reliably with automated install yet, as the DHCP server fails to be installed when no WiFi adapter is available yet (e.g. the module blacklisted), and when not selecting WiFi Hotspot interactively from the menu, WiFi is re-enabled too late. We just found this as of a user report in the forum and will be fixed with next release.

> logging the crash of the kernel and the loop can be a problem

First step would be to disable DietPi-RAMlog and enable persistent system logs:
```
dietpi-software uninstall 103
mkdir -p /var/log/journal
# then after crash, the following shows logs from before the crash as well
journalctl
```

MichaIng, 2021-09-01:
I'll mark this as closed. Feel free to reopen if the issue persists.

closed 2021-09-01 by MichaIng