MichaIng/DietPi #4466: LXDE non-root autologin without userdata drive

C0D3-M4513R, 2021-06-04:
### Creating a bug report/issue

#### Required Information
- DietPi version | `cat /boot/dietpi/.version`
``` 
G_DIETPI_VERSION_CORE=7
G_DIETPI_VERSION_SUB=2
G_DIETPI_VERSION_RC=3
G_GITBRANCH='master'
G_GITOWNER='MichaIng'
```
- Distro version | `echo $G_DISTRO_NAME` or `cat /etc/debian_version`
`buster`
- Kernel version | `uname -a`
`Linux DietPi 5.10.17-v8+ #1414 SMP PREEMPT Fri Apr 30 13:23:25 BST 2021 aarch64 GNU/Linux`
- SBC model | `echo $G_HW_MODEL_NAME` or (EG: RPi3)
`RPi 4 Model B (aarch64)`
- Power supply used | (EG: 5V 1A RAVpower)
`5V3A `
- SDcard used | SanDisk?
- echo $G_HW_MODEL
`4`

#### Additional Information (if applicable)
- Software title | dietpi-software
- Was the software title installed freshly or updated/migrated? fresh install
- Can this issue be replicated on a fresh installation of DietPi? yes

- Bug report ID | `echo $G_HW_UUID`

#### Steps to reproduce

1. Install mariadb
2. transfer user-dir to external drive
3. Configure Desktop non-root autologin
4. shutdown
5. disconnect drive
6. restart
7. Watch being unable forced to wait, and then watch the autologin fail/not proceed, due to lightdm not starting

#### Expected behaviour

The Autologin should still succeed, and the login should not block.

#### Actual behaviour

The Login is aborted, after being forced to wait a little bit.
The last thing being executed is this:

https://github.com/MichaIng/DietPi/blob/4ffa0e5ef0d8fd6fa241535ad96eae5482135797/dietpi/postboot#L47

Side effect:
- dietpi-drive_manager doesn't start. It just does nothing! Maybe it fails to find userdata partition?
- Unable to remove bootwait on external drive (even when using nobootwait on all fstab entries for that drive)

#### Extra details

lightdm debug logs:
```
Jun 04 16:19:47 DietPi lightdm[2512]: [+0.00s] DEBUG: Registered seat module xremote
Jun 04 16:19:47 DietPi lightdm[2512]: [+0.00s] DEBUG: Registered seat module unity
Jun 04 16:19:47 DietPi lightdm[2512]: [+0.00s] DEBUG: Using D-Bus name org.freedesktop.DisplayManager
Jun 04 16:19:47 DietPi lightdm[2512]: [+0.03s] DEBUG: Monitoring logind for seats
Jun 04 16:19:47 DietPi lightdm[2512]: [+0.03s] DEBUG: New seat added from logind: seat0
Jun 04 16:19:47 DietPi lightdm[2512]: [+0.03s] DEBUG: Seat seat0: Loading properties from config section Seat:*
Jun 04 16:19:47 DietPi lightdm[2512]: [+0.03s] DEBUG: Seat seat0: Starting
Jun 04 16:19:47 DietPi lightdm[2512]: [+0.03s] DEBUG: Seat seat0: Creating user session
Jun 04 16:19:47 DietPi lightdm[2512]: [+0.04s] WARNING: Error getting user list from org.freedesktop.Accounts: GDBus.Error:org.freedesktop.DBus.Error.ServiceUnknown: The name org.freedesktop.Accounts was not provided by an
Jun 04 16:19:47 DietPi lightdm[2512]: [+0.04s] DEBUG: Loading user config from /etc/lightdm/users.conf
```

C0D3-M4513R, 2021-06-04:
If this is a known issue, is there some hack, I can do, to get it working temporarily?

MichaIng, 2021-06-04:
Many thanks for your report.

This is currently a known limitation:
- DietPi-Drive_Manager adds mounted drives to `/etc/fstab`, hence on boot it looks those drives, to create the automount unit, respectively tries to mount it as fast as anything is trying to access the mount point.
- It should not wait forever, but 90 seconds is the default systemd timeout, AFAIK.

Generally for quick temporary mounts:
```
# Attach drive
# Check for drive partition device path
lsblk -p
# Mount it to existing mount point
mount /dev/sda1 /mnt/<mountpoint>
```
I have a `/mnt/external` for such cases, also `/media` could be used, which is used (subdirectories though) by desktop auto-mount tools, to mount drives as they are attached.

When using `dietpi-drive_manager` for temporary mounts, you also need to unmount it afterwards, to have the fstab entry removed.

C0D3-M4513R, 2021-06-04:
My concern is lxde not starting. I need the pi in a couple of days, to start without the external disk to desktop.
If all fails, I will prepare a usb, and spoof part-uuids and the disk uuid, to get the pi to "recognise" the disk, and start lxde.

MichaIng, 2021-06-04:
It it the dietpi_userdata drive or why don't you want to simply "unmount" it via `dietpi-drive_manager`? I just played around with this, and actually systemd itself has no issue with the automount as long as nothing is accessing the mount point. And also DietPi itself does not access even dietpi_userdata, so it all works fine without delay.

But the first process that accesses the mount point, is then delayed, as then systemd tries to mount the drive which is not available, and it gives up after precisely 90 seconds indeed. There is no setting to control this timeout for automount units explicitly: https://manpages.debian.org/systemd.automount
So I guess it is the regular service start timeout. I changed in `/etc/systemd/system.conf` `DefaultTimeoutStartSec=20s`, and indeed then the accessing command is precisely delayed for 20 seconds. But this is then true for every service start as well.

So in your case, something seems to access the mount point of that drive during boot. If it's the dietpi_userdata drive, then most likely a service that has it's config or data stored there, like MariaDB or such. Instead of having those services running into a failure, I would exclude them from being started up when the userdata drive cannot be attached, so nothing is trying to access it and hence nothing is delayed. You can do that via `dietpi-services` > service name > `Include/Exclude`.

C0D3-M4513R, 2021-06-04:
## Idea of a driveless boot
I could do it manually for the time being (which will be my choice for a dirty fix), but isn't there a way, to add a dependency to services, that are being stored on the dietpi_userdata mount, to block them from starting, if the drive isn't present, to allow a proper boot, without having a drive attached?

Too complicated? 

Is userdata Drive attached?
 - Services start normally

Else
 - All services on the userdata drive don't start
 - Hope for the best!

### Question awnsers:

>Is it the dietpi_userdata drive or why don't you want to simply "unmount" it via `dietpi-drive_manager` ?

See initial post:

>Steps to reproduce
>1. Install mariadb
>2. **transfer user-dir to external drive**

YES! it is my dietpi_userdata drive. But fun thing: Dietpi-drive_manager just WIPED all my manually added mounts. so I have to go to the pain of adding them all. 

## Fix idea?

OR I could (probably) have this whole issue solved, by just transitioning to systemd mounts directly, because there I can specify the mount as a required dependent of another service. 
Technically it is advised against having systemd-mounts, because everything can also be done in fstab, but since dietpi-drive_manager and the fstab to systemd-mount converter don't like me, I will transition to systemd-mounts, and hope for the best.

C0D3-M4513R, 2021-06-04:
Edit: the services starting conditionally works!
Here is an example unit below.
Note: /mnt/dietpi_userdata is a symlink to /mnt/external for my case!
mnt-external-mysql.mount
```
[Unit]
Description=/mnt/external/mysql mount
RequiresMountsFor=/mnt/external
PartOf=mnt-external.mount

BindsTo=mariadb.service
Before=mariadb.service

RefuseManualStart=true
RefuseManualStop=true

[Mount]

Type=btrfs
What=UUID=59fe09af-dbac-4fb1-988c-c2800a8e02c3
Where=/mnt/external/mysql
Options=subvol=@mysql,rw,autodefrag,noatime,lazytime,nofail

[Install]
RequiredBy=mariadb.service
```
mnt-external.mount
```
[Unit]
Description=/mnt/external mount

[Mount]
Type=btrfs
What=UUID=59fe09af-dbac-4fb1-988c-c2800a8e02c3
Where=/mnt/external

ReadWriteOnly=false

Options=subvol=@,rw,autodefrag,noatime,lazytime,nofail

[Install]
WantedBy=local-fs.target
```

C0D3-M4513R, 2021-06-04:
I did the above for all the services, and the desktop login now works!

MichaIng, 2021-06-05:
Good idea with the `RequiresMountsFor`. Does it work as well when using `RequiresMountsFor=/mnt/dietpi_userdata/mysql`, so using a path that contains the symlink? Otherwise it's difficult to implement, e.g. MariaDB is installed, then the userdata drive changed, that setting needs to be changed as well.

https://www.freedesktop.org/software/systemd/man/systemd.unit.html#RequiresMountsFor=
> Mount points marked with noauto are not mounted automatically through `local-fs.target`, but are still honored for the purposes of this option, i.e. they will be pulled in by this unit.

How does the service actually fail when the mount is not present? `journalctl -u mariadb`
And, if it is enabled via `systemctl enable mariadb`, does it wait for the mount, starting up as fast as it is there, and again probably delaying the boot/login prompt, or does it deny to start immediately? There is a difference `Requires=` (added here) and `Requisite=` (the way to assure that a service is not even tried to be started, when such a requisite is missing or failed).

C0D3-M4513R, 2021-06-05:
### `RequiresMountsFor=/mnt/external`
adds a Requires and after dependency, for the mnt-external.mount. Dunno, if it works with symlinks. Normally `/mnt/dietpi_userdata` should be a mount though, right? So this shouldn't be a problem on vanilla things.

### `PartOf=mnt-external.mount`
Makes this service also remount, if `mnt-external.mount` is restarted and stop if `mnt-external.mount` is stopped. The reverse is NOT the case.

### `BindsTo=mariadb.service` 
If mariadb is stopped, this service will be stopped too.
In conjunction with `Before=mariadb.service` it requires the `mnt-external-mysql.mount` to be strictly active, when Starting `mariadb.service`

### `RequiredBy=mariadb.service`
Actually start this service, when `mariadb.service` starts. 

I also added a `TimeoutSec=10s`, to make the mount fail after 10 seconds. mariadb doesn't start at all, because for the service to start, it needs the `mnt-external-mysql.mount` to be started, because of the `Before` property (considering, that the mount is enabled of course... Had a fun time, trying to figure that out!).
 
`Requisite` would not work here, since:
> Similar to Requires=. However, if the units listed here are not started already, they will not be started and the starting of this unit will fail immediately.
>RequisiteOf= dependency cannot be specified directly.

One would need to modify `mariadb.service`.
If you wanted to only modify `mariadb.service`, and have the mounts still via fstab, it could be done. (next post)

MichaIng, 2021-06-05:
> One would need to modify mariadb.service.

That was what I had in mind, so modify services or create drop-in configs to assure that the userdata mount is present, for all services which require access to it to start.

`RequiresMountsFor=/mnt/dietpi_userdata/mysql` will as well look for the `mnt-dietpi_userdata.mount`, basically for all mounts from root on up to the last sub directory. But when `/mnt/dietpi_userdata` is a symlink to `/mnt/external`, I'm not sure if it looks for `mnt-external-mysql.mount` and `mnt-external.mount` as well. I'll run some tests later.

C0D3-M4513R, 2021-06-08:
Sadly it seems, that `RequiresMountsFor` doesn't respect symlinks. This should though be a non-issue, since the user should just mount the drive, where it is intended. If the user doesn't, he should have enough knowledge, to know what is happening.
```
root@DietPi:/etc/systemd/system# ls -ahl /mnt/
total 32K
drwxr-xr-x 10 root   root   4.0K Jun  4 20:37 .
drwxr-xr-x 21 root   root   4.0K Apr 21 23:58 ..
drwxr-xr-x  1 root   root    238 Jun  3 15:51 data
drwxr-xr-x  2 root   root      0 Feb 14  2019 dietpibackup
lrwxrwxrwx  1 ftp    ftp      14 May 11 08:07 dietpi_userdata -> /mnt/external/
drwxr-xr-x  2 root   root   4.0K May 11 08:07 external
drwxrwxr-x  2 dietpi dietpi 4.0K Jan 21 22:53 ftp_client
drwxr-xr-x  5 root   root   4.0K May 12 08:05 nas
drwxrwxr-x  2 dietpi dietpi 4.0K Jan 21 22:57 nfs_client
drwxrwxr-x  2 dietpi dietpi 4.0K Jan 21 22:57 samba
drwxr-xr-x  2 ftp    ftp    4.0K May 11 08:05 tmp
root@DietPi:/etc/systemd/system# cat test-service.service
[Unit]
Description=testunit for testing systemd stuff
RequiresMountsFor=/mnt/dietpi_userdata/

[Service]
Type=simple
ExecStart=/bin/sleep 10
root@DietPi:/etc/systemd/system# systemctl start test-service &&  systemctl status test-service && systemctl status mnt-external.mount && systemctl status test-service
● test-service.service - testunit for testing systemd stuff
   Loaded: loaded (/etc/systemd/system/test-service.service; static; vendor preset: enabled)
   Active: active (running) since Tue 2021-06-08 14:50:19 CEST; 101ms ago
 Main PID: 3077 (sleep)
    Tasks: 1 (limit: 4411)
   CGroup: /system.slice/test-service.service
           └─3077 /bin/sleep 10

Jun 08 14:50:19 DietPi systemd[1]: Started testunit for testing systemd stuff.
● mnt-external.mount - /mnt/external mount
   Loaded: loaded (/etc/systemd/system/mnt-external.mount; disabled; vendor preset: enabled)
   Active: inactive (dead)
    Where: /mnt/external
     What: UUID=59fe09af-dbac-4fb1-988c-c2800a8e02c3
```

MichaIng, 2021-06-08:
All fstab entries are translated to systemd mount units automatically at boot. The problem is not the mount units, but the automount units. The mount itself fails immediately if the drive is not available, but the automount, when accessing the mount point, either kernel-wise waits forever and/or systemd-wise waits for `TimeoutStartSec` to be reached.

C0D3-M4513R, 2021-06-08:
> The problem is not the mount units, but the automount units.

then don't add x-systemd.automount as an fstab-option

>       x-systemd.automount
>           An automount unit will be created for the file system. See
>           systemd.automount(5) for details.

MichaIng, 2021-06-08:
We use that purposefully. It is often required for external mounts, which are not fully available yet when regular mount calls based on fstab are done (without `noauto` and `x-systemd.automount`, like root and boot partitions). Drives would then never be mounted and it might hang as well (not tested). Also for network drives (NFS, CIFS) it is nearly always required. Also it is quite a standard nowadays, like `allow-hotplug eth0` for network interfaces. So removing it causes more real issues. Just note that in your case you keep instructing the system to mount external drives which are not attached. We can enhance the behaviour for this case but since you could simply ordinary unmount the drive, it is a minor issue IMO.